greearb/ath10k-ct #1: qca9887 ath10k-ct TX bitrate forever only 6 Mbps

brealin, 2017-01-06:
1. AP(qca9887)--->PC:
root@lede:~# **iperf -c 192.168.1.106 -i 1**
------------------------------------------------------------
Client connecting to 192.168.1.106, TCP port 5001
TCP window size: 43.8 KByte (default)
------------------------------------------------------------
[  3] local 192.168.1.1 port 35710 connected with 192.168.1.106 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0- 1.0 sec   768 KBytes  6.29 Mbits/sec
[  3]  1.0- 2.0 sec   512 KBytes  4.19 Mbits/sec
[  3]  2.0- 3.0 sec   512 KBytes  4.19 Mbits/sec
[  3]  3.0- 4.0 sec   640 KBytes  5.24 Mbits/sec
[  3]  4.0- 5.0 sec   512 KBytes  4.19 Mbits/sec
[  3]  5.0- 6.0 sec   512 KBytes  4.19 Mbits/sec
[  3]  6.0- 7.0 sec   640 KBytes  5.24 Mbits/sec
[  3]  7.0- 8.0 sec   512 KBytes  4.19 Mbits/sec
[  3]  8.0- 9.0 sec   640 KBytes  5.24 Mbits/sec
[  3]  9.0-10.0 sec   512 KBytes  4.19 Mbits/sec
[  3]  0.0-10.1 sec  5.62 MBytes  4.69 Mbits/sec

2.PC-->AP(qca9887)
root@lede:~# **iperf -s -i1**
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size: 85.3 KByte (default)
------------------------------------------------------------
[  4] local 192.168.1.1 port 5001 connected with 192.168.1.10 port 63171
[ ID] Interval       Transfer     Bandwidth
[  4]  0.0- 1.0 sec  11.3 MBytes  94.7 Mbits/sec
[  4]  1.0- 2.0 sec  11.3 MBytes  94.7 Mbits/sec
[  4]  2.0- 3.0 sec  11.3 MBytes  94.7 Mbits/sec
[  4]  3.0- 4.0 sec  11.3 MBytes  94.7 Mbits/sec
[  4]  4.0- 5.0 sec  11.3 MBytes  94.7 Mbits/sec
[  4]  5.0- 6.0 sec  11.3 MBytes  94.7 Mbits/sec
[  4]  6.0- 7.0 sec  11.3 MBytes  94.7 Mbits/sec
[  4]  7.0- 8.0 sec  11.3 MBytes  94.7 Mbits/sec
[  4]  8.0- 9.0 sec  11.3 MBytes  94.7 Mbits/sec
[  4]  9.0-10.0 sec  11.3 MBytes  94.7 Mbits/sec
[  4]  0.0-10.0 sec   113 MBytes  94.7 Mbits/sec

`root@lede:~# iw wlan0 station dump
Station 5c:c5:d4:53:ad:c7 (on wlan0)
	inactive time:	0 ms
	rx bytes:	545107
	rx packets:	6103
	tx bytes:	14817641
	tx packets:	10075
	tx retries:	0
	tx failed:	0
	rx drop misc:	1
	signal:  	-57 dBm
	signal avg:	-57 dBm
	**tx bitrate:	6.0 MBit/s**
	rx bitrate:	433.3 MBit/s VHT-MCS 9 80MHz short GI VHT-NSS 1
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	short
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	DTIM period:	2
	beacon interval:100
	short preamble:	yes
	short slot time:yes
	connected time:	119 seconds
`

greearb, 2017-06-21:
I think this has been fixed, please try latest firmware.

greearb, 2017-06-21:
Please re-open if you see it with recent firmware/driver.

closed 2017-06-21 by greearb

jhaws1982, 2018-01-11:
I see this same problem on a Dell Precision 5520 with ath10k_pci,  Xubuntu 16.04.3.

```
  *-network               
       description: Wireless interface
       product: QCA6174 802.11ac Wireless Network Adapter
       vendor: Qualcomm Atheros
       physical id: 0
       bus info: pci@0000:02:00.0
       logical name: wlan0
       version: 32
       serial: b0:52:16:8c:4f:6b
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress bus_master cap_list ethernet physical wireless
       configuration: broadcast=yes driver=ath10k_pci driverversion=4.13.0-26-generic firmware=WLAN.RM.4.4.1-00079-QCARMSWPZ-1 ip=10.0.0.131 latency=0 link=yes multicast=yes wireless=IEEE 802.11
       resources: irq:140 memory:ed200000-ed3fffff
```
```
Linux C01547 4.13.0-26-generic #29~16.04.2-Ubuntu SMP Tue Jan 9 22:00:44 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

Thoughts on how to fix this?

brealin, 2018-01-11:
Hi Greearb:
         I didn't do anything else at the time. I just reconfigured it in your way, and it should have been usable and I didn't study it anymore.

 http://www.candelatech.com/downloads/ath10k-9887/ath10k-fw-beta/firmware-2-ct-htt-mgt-nrcc-community.bin
root@LEDE:/# cat /lib/firmware/ath10k/fwcfg-pci-0000\:00\:00.0.txt
vdevs = 4
peers = 80 

li-xiao-xi@163.com
 
From: jhaws1982
Date: 2018-01-11 12:56
To: greearb/ath10k-ct
CC: brealin; Manual
Subject: Re: [greearb/ath10k-ct] qca9887 ath10k-ct TX bitrate forever only 6 Mbps (#1)
I see this same problem on a Dell Precision 5520 with ath10k_pci, Xubuntu 16.04.3.
  *-network               
       description: Wireless interface
       product: QCA6174 802.11ac Wireless Network Adapter
       vendor: Qualcomm Atheros
       physical id: 0
       bus info: pci@0000:02:00.0
       logical name: wlan0
       version: 32
       serial: b0:52:16:8c:4f:6b
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress bus_master cap_list ethernet physical wireless
       configuration: broadcast=yes driver=ath10k_pci driverversion=4.13.0-26-generic firmware=WLAN.RM.4.4.1-00079-QCARMSWPZ-1 ip=10.0.0.131 latency=0 link=yes multicast=yes wireless=IEEE 802.11
       resources: irq:140 memory:ed200000-ed3fffff
Linux C01547 4.13.0-26-generic #29~16.04.2-Ubuntu SMP Tue Jan 9 22:00:44 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

Thoughts on how to fix this?

greearb, 2018-01-11:
Please show me full 'dmesg' of bootup.

Thanks,
Ben