greearb/ath10k-ct #4: Problems with wifi tests failing

slthomason, 2017-06-21:
Driver commit: 956e2609b7e42c8c710bba10ef925a5be1be5137

Normal firmware, we got this:
Mon Jun 19 10:42:51 2017 kern.warn kernel: [  635.132430] ath10k_pci 0000:01:00.0: htt peer map vdev 0 peer ec:08:6b:f9:74:7d id 419
Mon Jun 19 10:42:52 2017 daemon.info hostapd: wlan1: STA 20:16:d8:bd:f5:83 WPA: group key handshake completed (RSN)
Mon Jun 19 10:42:52 2017 daemon.info hostapd: wlan1: STA b0:c5:54:19:fb:df WPA: group key handshake completed (RSN)
Mon Jun 19 10:42:52 2017 daemon.info hostapd: wlan1: STA e0:66:78:27:4d:72 WPA: group key handshake completed (RSN)
Mon Jun 19 10:42:52 2017 daemon.info hostapd: wlan1: STA 4c:66:41:9c:0f:8e WPA: group key handshake completed (RSN)
Mon Jun 19 10:42:55 2017 daemon.info hostapd: wlan1: STA e0:66:78:19:31:5d IEEE 802.11: authenticated
Mon Jun 19 10:42:55 2017 daemon.info hostapd: wlan1: STA e0:66:78:19:31:5d IEEE 802.11: associated (aid 5)
Mon Jun 19 10:42:55 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED e0:66:78:19:31:5d
Mon Jun 19 10:42:55 2017 daemon.info hostapd: wlan1: STA e0:66:78:19:31:5d RADIUS: starting accounting session 81674F3F9239404A
Mon Jun 19 10:42:55 2017 daemon.info hostapd: wlan1: STA e0:66:78:19:31:5d WPA: pairwise key handshake completed (RSN)
Mon Jun 19 10:42:55 2017 daemon.info dnsmasq-dhcp[1672]: DHCPREQUEST(br-lan) 192.168.2.5 e0:66:78:19:31:5d
Mon Jun 19 10:42:55 2017 daemon.info dnsmasq-dhcp[1672]: DHCPACK(br-lan) 192.168.2.5 e0:66:78:19:31:5d Caryns-iPad
Mon Jun 19 10:47:07 2017 kern.warn kernel: [  887.408156] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:32 2017 kern.warn kernel: [  894.016329] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:52 2017 kern.warn kernel: [  894.023804] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:52 2017 kern.warn kernel: [  894.555731] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:52 2017 kern.warn kernel: [  894.563205] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:52 2017 kern.warn kernel: [  894.570644] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:52 2017 kern.warn kernel: [  896.039231] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:52 2017 kern.warn kernel: [  902.210334] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:52 2017 kern.warn kernel: [  903.411095] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:52 2017 kern.warn kernel: [  911.037291] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:52 2017 kern.warn kernel: [  911.436323] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:52 2017 kern.warn kernel: [  911.443829] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:52 2017 kern.warn kernel: [  916.143585] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:52 2017 kern.warn kernel: [  916.151054] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:52 2017 kern.warn kernel: [  916.513572] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:52 2017 kern.warn kernel: [  916.521042] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:52 2017 kern.warn kernel: [  917.058136] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:52 2017 kern.warn kernel: [  917.065572] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:52 2017 kern.warn kernel: [  917.365014] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:52 2017 kern.warn kernel: [  918.554659] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:52 2017 kern.warn kernel: [  928.775235] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:52 2017 kern.warn kernel: [  929.137677] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:52 2017 kern.warn kernel: [  929.612764] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:53 2017 kern.warn kernel: [  935.994486] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:53 2017 kern.warn kernel: [  936.001966] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:53 2017 kern.warn kernel: [  936.397429] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:53 2017 kern.warn kernel: [  936.404895] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:47:53 2017 daemon.notice hostapd: nl80211: nl80211_recv_beacons->nl_recvmsgs failed: -5
Mon Jun 19 10:47:53 2017 daemon.notice hostapd: nl80211: wpa_driver_nl80211_event_receive->nl_recvmsgs failed: -5
Mon Jun 19 10:47:53 2017 daemon.notice hostapd: wlan1: STA 34:a3:95:00:a8:6e IEEE 802.11: did not acknowledge authentication response
Mon Jun 19 10:47:53 2017 daemon.notice hostapd: wlan1: STA 34:a3:95:00:a8:6e IEEE 802.11: did not acknowledge authentication response
Mon Jun 19 10:47:53 2017 daemon.notice hostapd: wlan1: STA 34:a3:95:00:a8:6e IEEE 802.11: did not acknowledge authentication response
Mon Jun 19 10:47:53 2017 daemon.notice hostapd: wlan1: STA 34:a3:95:00:a8:6e IEEE 802.11: did not acknowledge authentication response
Mon Jun 19 10:47:53 2017 daemon.notice hostapd: wlan1: STA 34:a3:95:00:a8:6e IEEE 802.11: did not acknowledge authentication response
Mon Jun 19 10:48:09 2017 kern.warn kernel: [  952.669842] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:48:09 2017 kern.warn kernel: [  952.677269] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:48:09 2017 kern.warn kernel: [  952.684740] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:48:10 2017 kern.warn kernel: [  953.255552] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:48:10 2017 kern.warn kernel: [  953.263029] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:48:10 2017 kern.warn kernel: [  953.651328] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:48:10 2017 kern.warn kernel: [  953.658802] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:48:51 2017 kern.warn kernel: [  963.463492] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:48:52 2017 kern.warn kernel: [  969.131534] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:48:52 2017 kern.warn kernel: [  971.165135] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:48:52 2017 kern.warn kernel: [  973.947326] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:48:52 2017 kern.warn kernel: [  977.036276] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:48:52 2017 kern.warn kernel: [  979.639102] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:48:52 2017 kern.warn kernel: [  979.646532] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:48:52 2017 kern.warn kernel: [  979.654000] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:48:52 2017 kern.warn kernel: [  979.661438] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:48:52 2017 kern.warn kernel: [  979.668873] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:48:52 2017 kern.warn kernel: [  980.226541] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:48:52 2017 kern.warn kernel: [  980.234021] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:48:52 2017 kern.warn kernel: [  980.803839] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jun 19 10:48:52 2017 kern.warn kernel: [  980.811314] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[... 68 lines cut ...]
[... cut ...]
[ 1675.034619] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[ 1720.782013] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[ 1721.438926] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[ 1721.809215] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[ 1767.992163] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[ 1812.076470] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[ 1812.435222] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon

greearb, 2017-07-10:
I believed this was fixed by the user upgrading to a newer LEDE release.  Please re-open if problem persists.

closed 2017-07-10 by greearb

slthomason, 2017-07-12:
We have spent the last week trying to figure out what reproduces this.  Today we were finally able to narrow it down.

We only get these errors when 2Ghz is under load (we replicate it doing an iperf test connected to the 2Ghz).  Running the CPU up to 100% does not cause this behavior, but when we run certain systems of ours that call heavily into the kernel - anytime the kworker starts to get even the least bit burdened (over 30% CPU), while doing an iperf test connected to the 2Ghz channel, we get the ath10k errors.

Further, the odd thing is that these errors are on the 5Ghz code (ath10k), however, we only replicate this when connected to the 2Ghz band (ath9k).  When we connect to the 5Ghz and put load from our systems and the iperf - it handles it fine.  However, when we go back to putting load from our system + iperf on the 2Ghz band, we get the below errors.

We have seen this with any system that puts load on kernel calls - not just our systems.  When the kworker gets at all burdened - this happens.

We have now confirmed this on the following build combinations:
LEDE 17.01 + ath10k (non-ct)
LEDE 17.01.2 + very latest ath10k-ct
LEDE 17.01.2 + ath10k (non-ct)
many, many other combinations

A couple of questions:
1) Where are these calls getting generated from?  
2) What is the actual problem happening here?
3) We are connected to the 2Ghz (ath9k).  Why is the code erroring in the the ath10k code?

kern.warn kernel: [ 1482.891420] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
kern.warn kernel: [ 1487.252602] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
kern.warn kernel: [ 1487.556625] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon

BrianLian, 2018-02-05:
I have a same problem.
Any one find out what happan?

blackyx81, 2018-05-11:
Device: D-Link DIR-869 (2,4GHz and 5GHz active)
LEDE 17.01.4

Same problem here:
[ 6585.271882] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[ 6586.587435] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
...

After kernel warning the device freezes for a while, after 10 seconds the device responds normal.

Can someone clarify why this happens?

Catfriend1, 2019-06-20:
Having those too:

```
[1193818.182719] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 2, skipped old beacon
[1193818.190325] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[1212675.045773] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 2, skipped old beacon
[1212675.053382] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[1227045.561476] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 1, skipped old beacon
[1227045.569089] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 2, skipped old beacon
```

Will also try the solution from #87

titiviking, 2022-01-12:
Same problem on my side:

`[314041.061508] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[314041.069122] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[314041.076715] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[314041.084296] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[314041.091896] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[314041.099477] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[314041.107067] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[314041.114650] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[314041.122233] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[314041.129781] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[314041.137353] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[314041.144942] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[314041.152526] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[314041.160100] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[314041.167669] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon`

Any working solution yet? 
I'm currently running OpenWRT 21.02

klukonin, 2022-01-12:
Well, it's not a good approach to use patches from #87.
Those patches are for old kernel and current kernel should already has all patched parameters.

Please try to use htt firmware
https://openwrt.org/packages/pkgdata_owrt18_6/ath10k-firmware-qca988x-ct-htt