greearb/ath10k-ct #5: Scanning aborted when there is an huge number of AP around

oamouroux, 2017-06-21:
HW: QCA9882

When there is 80 to 100 APs around, scanning aborted:
iw ap0 scan
scan aborted! 

Here is my setup:
iw dev
phy#0
    Interface ap0_0
        ifindex 19
        wdev 0x5
        addr a2:f0:21:31:54:c6
        ssid MyAPVLAN-WPA-RADIUS
        type AP
        channel 36 (5180 MHz), width: 40 MHz, center1: 5190 MHz
        txpower 19.00 dBm
    Interface ap0
        ifindex 18
        wdev 0x4
        addr 04:f0:21:31:54:c6
        ssid MyPublicAP-OPEN
        type AP
        channel 36 (5180 MHz), width: 40 MHz, center1: 5190 MHz
        txpower 19.00 dBm
    Interface adhoc0
        ifindex 17
        wdev 0x3
        addr 06:f0:21:31:54:c6
        ssid \x01\x02\x03\x04\x05
        type IBSS
        channel 36 (5180 MHz), width: 40 MHz, center1: 5190 MHz
        txpower 19.00 dBm
    Interface mon.ap0
        ifindex 16
        wdev 0x2
        addr 04:f0:21:31:54:c6
        type monitor
        txpower 0.00 dBm

greearb, 2017-06-21:
I don't have an easy way to reproduce this.  Do you see any hex in dmesg that looks like this?

[518070.334445] ath10k_pci 0000:07:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[518070.334476] ath10k: [0000]: 079DF922 10044C1D 00432BAC 00000001 00000000 00000140 079DF922 10044C1D
[518070.334478] ath10k: [0008]: 00432BAC 00000001 00000004 00000140
[518070.334480] ath10k_pci 0000:07:00.0: ATH10K_END

If so, please send me (or attach) 'dmesg' or similar right after this problem happens.  If not, then I guess you either are not running the ath10k-ct driver or you have disabled the firmware debuglog printing using debug_mask in ath10k?

oamouroux, 2017-06-22:
Here is the generated log when the scan is aborted:

[18064.230000] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[18064.230000] ath10k: [0000]: B8581A01 4F4C0814 BC000000 744A4200 14B94300 00000000 00007100 B8581A01
[18064.230000] ath10k: [0008]: 4F4C0814 AC000000 AC4A4200 14B94300 00000000 01007200 B8581A01 564C0814
[18064.230000] ath10k: [0016]: 14B94300 60119C00 11000000 00000000 203D0000 B8581A01 564C0814 14B94300
[18064.230000] ath10k: [0024]: BCCF4300 00000000 0C000000 001D0000 B8581A01 035C0814 00000000 06F02131
[18064.230000] ath10k: [0032]: 5B050000 74AD9B00 14B94300 D5581A01 524C0814 14B94300 00000000 01000000
[18064.230000] ath10k: [0040]: 02000000 08000000 D5581A01 0128FC17 070B1071 02000001 05000000 00000000
[18064.230000] ath10k: [0048]: 00000000 D5581A01 0128FC17 80881071 08000000 F4864400 08009600 F00F0000
[18064.230000] ath10k: [0056]: D5581A01 0128FC17 80881071 00000100 F4864400 08009600 00000000 D5581A01
[18064.230000] ath10k: [0064]: 0128FC17 80881071 00000200 F4864400 08009600 00000000 D5581A01 0128FC17
[18064.230000] ath10k: [0072]: 80881071 00000300 F4864400 08009600 00000000 D5581A01 0128FC17 80881071
[18064.230000] ath10k: [0080]: 00000400 F4864400 08009600 00000000 D5581A01 0128FC17 80881071 00000500
[18064.230000] ath10k: [0088]: F4864400 08009600 00000000
[18064.230000] ath10k_pci 0000:01:00.0: ATH10K_END
[18073.710000] ath10k_pci 0000:01:00.0: removing peer, htt peer unmap vdev 2 peer 06:f0:21:31:5b:05 id 188
[18073.720000] ath10k_pci 0000:01:00.0: removing peer, htt peer unmap vdev 2 peer 06:f0:21:31:5b:05 id 172
[18074.230000] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[18074.230000] ath10k: [0000]: CD801A01 025C0814 80F34300 74AD9B00 06F02131 5B050000 01000000 CD801A01
[18074.230000] ath10k: [0008]: 574C0814 14B94300 60119C00 11000000 00000000 A23D0000 CD801A01 574C0814
[18074.230000] ath10k: [0016]: 14B94300 BCCF4300 00000000 0C000000 821D0000 CD801A01 504C0810 BB001071
[18074.230000] ath10k: [0024]: 744A4200 14B94300 98000000 CD801A01 504C0810 BB001071 AC4A4200 14B94300
[18074.230000] ath10k: [0032]: 98000000
[18074.230000] ath10k_pci 0000:01:00.0: ATH10K_END
[18075.100000] ath10k_pci 0000:01:00.0: htt peer map vdev 2 peer 06:f0:21:31:5b:05 id 188
[18075.110000] ath10k_pci 0000:01:00.0: htt peer map vdev 2 peer 06:f0:21:31:5b:05 id 172
[18075.120000] ath10k_pci 0000:01:00.0: band: 1  ratemask: 0xffffffff  hw-nss: 3 dev-id: 0x3c
[18075.130000] ath10k_pci 0000:01:00.0: band: 1  ratemask: 0xffffffff  hw-nss: 3 dev-id: 0x3c
[18075.230000] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[18075.230000] ath10k: [0000]: 5F861A01 4F4C0814 BC000000 E44A4200 A4B94300 00000000 00007100 5F861A01
[18075.230000] ath10k: [0008]: 4F4C0814 AC000000 1C4B4200 A4B94300 00000000 01007200 5F861A01 564C0814
[18075.230000] ath10k: [0016]: A4B94300 A8119C00 11000000 00000000 203D0000 5F861A01 564C0814 A4B94300
[18075.230000] ath10k: [0024]: 04D04300 00000000 0C000000 001D0000 5F861A01 035C0814 00000000 06F02131
[18075.230000] ath10k: [0032]: 5B050000 74AD9B00 A4B94300 7A861A01 524C0814 A4B94300 00000000 01000000
[18075.230000] ath10k: [0040]: 02000000 08000000 7A861A01 0128FC17 070B1071 02000001 05000000 00000000
[18075.230000] ath10k: [0048]: 00000000 7A861A01 0128FC17 80881071 08000000 F4864400 08009600 F00F0000
[18075.230000] ath10k: [0056]: 7A861A01 0128FC17 80881071 00000100 F4864400 08009600 00000000 7A861A01
[18075.230000] ath10k: [0064]: 0128FC17 80881071 00000200 F4864400 08009600 00000000 7A861A01 0128FC17
[18075.230000] ath10k: [0072]: 80881071 00000300 F4864400 08009600 00000000 7A861A01 0128FC17 80881071
[18075.230000] ath10k: [0080]: 00000400 F4864400 08009600 00000000 7A861A01 0128FC17 80881071 00000500
[18075.230000] ath10k: [0088]: F4864400 08009600 00000000 85861A01 524C0814 A4B94300 C2000000 01000000
[18075.230000] ath10k: [0096]: 02000000 08000000 85861A01 0128FC17 070B1071 02000903 07B00102 FAFF0000
[18075.230000] ath10k: [0104]: FFFF0000 85861A01 0128FC17 80881071 08000000 F4864400 3E009600 F00F0000
[18075.230000] ath10k: [0112]: 85861A01 0128FC17 80881071 0A000100 F4864400 3E009600 1FF80000 85861A01
[18075.230000] ath10k: [0120]: 0128FC17 80881071 0A000200 F4864400 3E009600 1FF80000 85861A01 0128FC17
[18075.230000] ath10k: [0128]: 80881071 0B000300 F4864400 3E009600 1FE00700 85861A01 0128FC17 80881071
[18075.230000] ath10k: [0136]: 0B000400 F4864400 3E009600 1FE00700 85861A01 0128FC17 80881071 0C000500
[18075.230000] ath10k: [0144]: F4864400 3E009600 1FE00F00
[18075.230000] ath10k_pci 0000:01:00.0: ATH10K_END
[18085.110000] ath10k_pci 0000:01:00.0: removing peer, htt peer unmap vdev 2 peer 06:f0:21:31:5b:05 id 188
[18085.120000] ath10k_pci 0000:01:00.0: removing peer, htt peer unmap vdev 2 peer 06:f0:21:31:5b:05 id 172
[18085.230000] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[18085.230000] ath10k: [0000]: 67AE1A01 025C0814 80F34300 74AD9B00 06F02131 5B050000 01000000 67AE1A01
[18085.230000] ath10k: [0008]: 574C0814 A4B94300 A8119C00 11000000 00000000 A23D0000 67AE1A01 574C0814
[18085.230000] ath10k: [0016]: A4B94300 04D04300 00000000 0C000000 821D0000 67AE1A01 504C0810 BB001071
[18085.230000] ath10k: [0024]: E44A4200 A4B94300 98000000 67AE1A01 504C0810 BB001071 1C4B4200 A4B94300
[18085.230000] ath10k: [0032]: 98000000
[18085.230000] ath10k_pci 0000:01:00.0: ATH10K_END
[18089.950000] ath10k_pci 0000:01:00.0: htt peer map vdev 2 peer 06:f0:21:31:5b:05 id 188
[... cut ...]
[18120.230000] ath10k: [0112]: EB391B01 0128FC17 80881071 0A000100 F4864400 3E009600 1FF80000 EB391B01
[18120.230000] ath10k: [0120]: 0128FC17 80881071 0A000200 F4864400 3E009600 1FF80000 EB391B01 0128FC17
[18120.230000] ath10k: [0128]: 80881071 0B000300 F4864400 3E009600 1FE00700 EB391B01 0128FC17 80881071
[18120.230000] ath10k: [0136]: 0B000400 F4864400 3E009600 1FE00700 EB391B01 0128FC17 80881071 0C000500
[18120.230000] ath10k: [0144]: F4864400 3E009600 1FE00F00
[18120.230000] ath10k_pci 0000:01:00.0: ATH10K_END

greearb, 2017-10-25:
I do not see any obvious problems in that debug messsage dump, nor do I see any info at all about scanning.  Are you still able to reproduce this problem?

oamouroux, 2018-03-15:
The issue is still reproducible on OpenWRT Chaos Calmer, Kernel 3.18.45, mac80211 backported from v4.4-rc5-1913-gc8fdf68, driver ath10k-ct 4.7 2f3171d, fW-020-64bb715

I am also able to reproduce it often.

You can find attached some logs
[scan.txt](https://github.com/greearb/ath10k-ct/files/1816980/scan.txt)

greearb, 2018-03-16:
I do not see any useful info.  The only message is telling me that scan was canceled.  I see no logs about it being started.  Please enable ath10k driver debugging and re-run your test.

echo 0xc0000032 > /sys/kernel/debug/ieee80211/phy0/ath10k/debug_level

And if possible, put this in modprobe.d so that it boots up this way.  Maybe the original
error happened before you tweaked the debug_level to enable firmware logging:

$ cat /etc/modprobe.d/ath10k.conf
# Boot & MAC
#options ath10k_core debug_mask=0xc0000030

# Boot, MAC, and WMI
options ath10k_core debug_mask=0xc0000032

greearb, 2018-03-26:
FYI, new firmware and driver patches were accepted into LEDE/OpenWRT recently.  Please re-test with these applied and see if that helps.

oamouroux, 2018-04-06:
Here is the log after enabling the debug traces and trying the scan:
[20521.330000] ath10k_pci 0000:01:00.0: Setting debug-mask to: 0xc0000032  old: 0x10000000
[20525.100000] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[20525.110000] ath10k: [0000]: 2CCD4001 025C0414 5CF34300 44B09B00 065E0C11 EED30000 01000000 2CCD4001
[20525.120000] ath10k: [0008]: 574C0414 F0AB4300 58D59B00 11000000 00000000 A23D0000 2CCD4001 574C0414
[20525.120000] ath10k: [0016]: F0AB4300 38C84300 00000000 0C000000 821D0000 2DCD4001 504C0410 BB001071
[20525.130000] ath10k: [0024]: 18F34100 F0AB4300 98000000 2DCD4001 504C0410 BB001071 50F34100 F0AB4300
[20525.140000] ath10k: [0032]: 98000000 4ACD4001 4F4C0414 7A010000 68F44100 909F4300 00000000 00007100
[20525.150000] ath10k: [0040]: 4ACD4001 4F4C0414 6A010000 A0F44100 909F4300 00000000 01007200 4ACD4001
[20525.160000] ath10k: [0048]: 564C0414 909F4300 08D79B00 11000000 00000000 203D0000 4ACD4001 564C0414
[20525.170000] ath10k: [0056]: 909F4300 10C94300 00000000 0C000000 001D0000 4ACD4001 035C0414 00000000
[20525.180000] ath10k: [0064]: 065E0C11 EED30000 44B09B00 909F4300 4ACD4001 524C0414 909F4300 00000000
[20525.190000] ath10k: [0072]: 01000000 02000000 08000000 4ACD4001 0128FC17 070B1071 02000001 05000000
[20525.200000] ath10k: [0080]: 00000000 00000000 4ACD4001 0128FC17 80881071 08000000 18854400 08009600
[20525.210000] ath10k: [0088]: F00F0000 4ACD4001 0128FC17 80881071 00000100 18854400 08009600 00000000
[20525.220000] ath10k: [0096]: 4ACD4001 0128FC17 80881071 00000200 18854400 08009600 00000000 4ACD4001
[20525.230000] ath10k: [0104]: 0128FC17 80881071 00000300 18854400 08009600 00000000 4ACD4001 0128FC17
[20525.230000] ath10k: [0112]: 80881071 00000400 18854400 08009600 00000000 4ACD4001 0128FC17 80881071
[20525.240000] ath10k: [0120]: 00000500 18854400 08009600 00000000 4BCD4001 524C0414 909F4300 C2000000
[20525.250000] ath10k: [0128]: 01000000 02000000 08000000 4BCD4001 0128FC17 070B1071 02000603 07B00000
[20525.260000] ath10k: [0136]: 00000000 FFFF0000 4BCD4001 0128FC17 80881071 08000000 18854400 1C009600
[20525.270000] ath10k: [0144]: F00F0000 4BCD4001 0128FC17 80881071 0A000100 18854400 1C009600 1FF80000
[20525.280000] ath10k: [0152]: 4BCD4001 0128FC17 80881071 0A000200 18854400 1C009600 1FF80000 4BCD4001
[20525.290000] ath10k: [0160]: 0128FC17 80881071 00000300 18854400 1C009600 00000000 4BCD4001 0128FC17
[20525.300000] ath10k: [0168]: 80881071 00000400 18854400 1C009600 00000000 4BCD4001 0128FC17 80881071
[20525.310000] ath10k: [0176]: 00000500 18854400 1C009600 00000000
[20525.310000] ath10k_pci 0000:01:00.0: ATH10K_END
[20527.100000] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[20527.110000] ath10k: [0000]: 6FD34001 025C0414 5CF34300 7CB09B00 065E0C11 E72B0000 01000000 6FD34001
[20527.120000] ath10k: [0008]: 574C0414 709E4300 A0D59B00 11000000 00000000 A23D0000 6FD34001 574C0414
[20527.120000] ath10k: [0016]: 709E4300 80C84300 00000000 0C000000 821D0000 6FD34001 504C0410 BB001071
[20527.130000] ath10k: [0024]: 88F34100 709E4300 98000000 6FD34001 504C0410 BB001071 C0F34100 709E4300
[20527.140000] ath10k: [0032]: 98000000 EED34001 4F4C0414 99000000 D8F44100 20A04300 00000000 00007100
[20527.150000] ath10k: [0040]: EED34001 4F4C0414 A9000000 10F54100 20A04300 00000000 01007200 EED34001
[20527.160000] ath10k: [0048]: 564C0414 20A04300 50D79B00 11000000 00000000 203D0000 EED34001 564C0414
[20527.170000] ath10k: [0056]: 20A04300 58C94300 00000000 0C000000 001D0000 EED34001 035C0414 00000000
[20527.180000] ath10k: [0064]: 065E0C11 E72B0000 7CB09B00 20A04300 EED34001 524C0414 20A04300 00000000
[20527.190000] ath10k: [0072]: 01000000 02000000 08000000 EED34001 0128FC17 070B1071 02000001 05000000
[20527.200000] ath10k: [0080]: 00000000 00000000 EED34001 0128FC17 80881071 08000000 18854400 08009600
[20527.210000] ath10k: [0088]: F00F0000 EFD34001 0128FC17 80881071 00000100 18854400 08009600 00000000
[20527.220000] ath10k: [0096]: EFD34001 0128FC17 80881071 00000200 18854400 08009600 00000000 EFD34001
[20527.230000] ath10k: [0104]: 0128FC17 80881071 00000300 18854400 08009600 00000000 EFD34001 0128FC17
[20527.230000] ath10k: [0112]: 80881071 00000400 18854400 08009600 00000000 EFD34001 0128FC17 80881071
[20527.240000] ath10k: [0120]: 00000500 18854400 08009600 00000000 F3D34001 524C0414 20A04300 C2000000
[20527.250000] ath10k: [0128]: 01000000 02000000 08000000 F3D34001 0128FC17 070B1071 02000603 07B00000
[20527.260000] ath10k: [0136]: 00000000 FFFF0000 F3D34001 0128FC17 80881071 08000000 18854400 1C009600
[20527.270000] ath10k: [0144]: F00F0000 F3D34001 0128FC17 80881071 0A000100 18854400 1C009600 1FF80000
[20527.280000] ath10k: [0152]: F3D34001 0128FC17 80881071 0A000200 18854400 1C009600 1FF80000 F3D34001
[20527.290000] ath10k: [0160]: 0128FC17 80881071 00000300 18854400 1C009600 00000000 F3D34001 0128FC17
[20527.300000] ath10k: [0168]: 80881071 00000400 18854400 1C009600 00000000 F3D34001 0128FC17 80881071
[20527.310000] ath10k: [0176]: 00000500 18854400 1C009600 00000000
[20527.310000] ath10k_pci 0000:01:00.0: ATH10K_END
[20530.100000] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[20530.110000] ath10k: [0000]: 3EE14001 025C0414 5CF34300 D4AF9B00 065E0C11 EECC0000 01000000 3EE14001
[20530.120000] ath10k: [0008]: 574C0414 009F4300 C0D69B00 11000000 00000000 A23D0000 3FE14001 574C0414
[20530.120000] ath10k: [0016]: 009F4300 C8C84300 00000000 0C000000 821D0000 3FE14001 504C0410 BB001071
[... cut ...]
[20550.300000] ath10k: [0168]: 80881071 00000400 18854400 1C009600 00000000 A7314101 0128FC17 80881071
[20550.310000] ath10k: [0176]: 00000500 18854400 1C009600 00000000
[20550.320000] ath10k_pci 0000:01:00.0: ATH10K_END
[20552.100000] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[20552.110000] ath10k: [0000]: 38384101 0424FC13 B1001107 00000000 04000000 DC634400
[20552.120000] ath10k_pci 0000:01:00.0: ATH10K_END

Unfortantely I did not get the time for now to try new driver & firmware

greearb, 2018-09-28:
I do not see any errors in the logs, please test with latest OpenWRT, which includes recent ath10k-ct firmware and driver.  If problem persists, please update this bug...I can do some custom firmware builds if needed.  Also, if you test in this same environment on some other non-ath10k driver/firmware, (like ath9k), will it work there?  Possibly the issue is not with the firmware, but something higher in the mac80211 stack?