greearb/ath10k-ct #6: Mesh mode is not able to authenticate peers

Jojosep, 2017-06-22:
The problem is that two QCA9888 cards can't connect each other on mesh mode. We need to connect a non QCA988X and ping both QCA9888 to successfully ping from one QCA9888 to the other.

Our hardware is QCA9888 as for the software we use an ubuntu 14.04  with the kernel provided by you (4.9.27) installed on a Gateworks Ventana.

This is the dmesg output when I ping (without any answer):

[   22.569560] ath10k_pci 0000:07:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   22.569573] ath10k_pci 0000:07:00.0: msdu-desc: 2500  skid: 32
[   22.598790] ath10k_pci 0000:07:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0'
[   22.599526] ath10k_pci 0000:07:00.0: wmi print 'free: 133180 iram: 18948 sram: 39944'
[   22.621522] ath10k_pci 0000:07:00.0: no channel configured; ignoring frame(s)!
[   22.723939] ath10k_pci 0000:07:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal otp max-sta 32 raw 0 hwcrypto 1
[   22.767001] ath10k_pci 0000:07:00.0: firmware crashed! (uuid 0b76e062-5b93-4152-851f-32c93f917f37)
[   22.775985] ath10k_pci 0000:07:00.0: qca9888 hw2.0 target 0x01000000 chip_id 0x00000000 sub 0000:0000
[   22.776000] ath10k_pci 0000:07:00.0: kconfig debug 1 debugfs 1 tracing 0 dfs 0 testmode 0
[   22.778850] ath10k_pci 0000:07:00.0: firmware ver 10.4-ct-9888-fW-010-98c2197 api 5 features peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT crc32 b1863e50
[   22.779207] ath10k_pci 0000:07:00.0: board_file api 2 bmi_id 0:17 crc32 74a04250
[   22.779220] ath10k_pci 0000:07:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal otp max-sta 32 raw 0 hwcrypto 1
[   22.791253] ath10k_pci 0000:07:00.0: failed to get memcpy hi address for firmware address 4: -16
[   22.791263] ath10k_pci 0000:07:00.0: failed to read firmware dump area: -16
[   22.798233] ath10k_pci 0000:07:00.0: in crash-regs-harder
[   25.198886] ath10k_pci 0000:07:00.0: in crash-regs-harder, firmware did not provide indicator: 0x2
[   25.198905] ath10k_pci 0000:07:00.0: failed to get memcpy hi address for firmware address 8: -28
[   25.198920] ath10k_pci 0000:07:00.0: failed to dump debug log area: -28
[   25.205551] ath10k_pci 0000:07:00.0: failed to get memcpy hi address for firmware address 72: -28
[   25.205565] ath10k_pci 0000:07:00.0: failed to get memcpy hi address for firmware address 72: -28
[   25.205572] ath10k_pci 0000:07:00.0: failed to get memcpy hi address for firmware address 76: -28
[   25.205579] ath10k_pci 0000:07:00.0: failed to get memcpy hi address for firmware address 76: -28
[   25.205591] ath10k_pci 0000:07:00.0: failed to read firmware RAM BSS memory from 4284224 (22168 B): -28
[   25.205605] ath10k_pci 0000:07:00.0: failed to read firmware ROM BSS memory from 4197376 (11312 B): -28
[   25.213510] ath: EEPROM regdomain: 0x0
[   25.213516] ath: EEPROM indicates default country code should be used
[   25.213518] ath: doing EEPROM country->regdmn map search
[   25.213524] ath: country maps to regdmn code: 0x3a
[   25.213528] ath: Country alpha2 being used: US
[   25.213530] ath: Regpair used: 0x3a
[   25.216360] cfg80211: Setting DFS Master region in update_regulatory, was: US unset, new: US FCC  lr: bc7c5440  regdom: bc8eac00
[   25.277412] ath10k_pci 0000:07:00.0: cannot restart a device that hasn't been started
[   33.115472] init: failsafe main process (496) killed by TERM signal
[   33.551088] init: plymouth-upstart-bridge main process ended, respawning
[   74.727387] random: crng init done
[  118.582287] ath10k_pci 0000:07:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[  118.582299] ath10k_pci 0000:07:00.0: msdu-desc: 2500  skid: 32
[  118.611498] ath10k_pci 0000:07:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0'
[  118.612260] ath10k_pci 0000:07:00.0: wmi print 'free: 133180 iram: 18948 sram: 39944'
[  118.842984] ath10k_pci 0000:07:00.0: htt peer map vdev 0 peer 02:00:00:00:05:00 id 0
[  118.843022] ath10k_pci 0000:07:00.0: htt peer map vdev 0 peer 02:00:00:00:05:00 id 0
[  118.846478] IPv6: ADDRCONF(NETDEV_UP): mesh0: link is not ready
[  119.577252] ath10k_pci 0000:07:00.0: dropping dbg buffer due to crash since read
[  119.577270] ath10k_pci 0000:07:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[  119.577292] ath10k: [0000]: 00000DFB 13FC0007 60002070 00000010 00000030 00060001 00000E18 17FC582E
[  119.577340] ath10k: [0008]: 00000003 0000001E 0000094C 0000094C 00000000 00000E18 17FC582E 00000012
[  119.577356] ath10k: [0016]: 0000001E 0000094C 0000094C 00000000 00000E18 17FC582E 00000045 0000001E
[  119.577392] ath10k: [0024]: 0000094C 0000094C 00000000 00000E18 17FC582E 00000067 0000001E 0000094C
[  119.577428] ath10k: [0032]: 0000094C 00000000 00000E18 0BFC582E 00000100 11E1A300 00000E1A 14005853
[  119.577455] ath10k: [0040]: 51100001 000F0CB0 000003FC 00000001 00449FC8 00000E1A 14005856 71109990
[  119.577469] ath10k: [0048]: 00426E00 00000000 0044FA24 00001200 00000E1A 14005856 71109990 00426E00
[  119.577484] ath10k: [0056]: 00000000 0044FA10 00098000 00000E1A 10005851 00426E00 00449FC8 00000001
[  119.577495] ath10k: [0064]: 00000000 00000E1A 14006403 00000000 00000100 00000000 00000000 00000000
[  119.577509] ath10k: [0072]: 00000E1A 14005856 71109990 00426E00 00000000 0044F9FC 01000000 00000E1B
[  119.577521] ath10k: [0080]: 14006402 71103331 00000000 00000100 00000001 00000000 00000E1B 14006402
[  119.577534] ath10k: [0088]: 71103332 00000000 00000100 0042486C 00000000 00000E1B 14005854 51100002
[  119.577547] ath10k: [0096]: 000F0CB0 00000002 00000000 00449FC8 00000E1B 10005852 00426E00 00449FC8
[  119.577560] ath10k: [0104]: 00000000 00000000 00000E1B 10005857 71109991 00426E00 00000000 0044FA24
[  119.577574] ath10k: [0112]: 00000E1B 10005857 71109991 00426E00 00000000 0044FA10 00000E1B 10005857
[  119.577587] ath10k: [0120]: 71109991 00426E00 00000000 0044F9FC 00000E30 13FC5822 9110B000 00000001
[  119.577598] ath10k: [0128]: 00000001 00000001 00000E46 13FC5822 9110B000 00000001 00000001 00000001
[  119.577610] ath10k: [0136]: 00000E5B 13FC5822 9110B000 00000001 00000001 00000001 00000E70 13FC5822
[  119.577623] ath10k: [0144]: 9110B000 00000001 00000001 00000001 00000E85 13FC5822 9110B000 00000001
[  119.577653] ath10k: [0152]: 00000001 00000001 00000E9A 13FC5822 9110B000 00000001 00000001 00000001
[... 17 lines cut ...]
[  121.577436] ath10k_pci 0000:07:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[  121.577459] ath10k: [0000]: 0000171F 08003812 000F12E8 00426E00 0000171F 10003809 0000143C 00000002
[  121.577476] ath10k: [0008]: 00000006 00000000 00001720 0BFC5805 0000143C 00000000 00001814 04003814
[  121.577492] ath10k: [0016]: 00000000 00001817 14045853 51100001 000F0C20 000003FC 00000003 00449ED8
[  121.577508] ath10k: [0024]: 00001817 14045853 51100001 000F0BD8 000003FE 00000004 00449ED8 00001817
[  121.577522] ath10k: [0032]: 14045856 71109990 00426D28 00000000 0044F9D4 00001200 00001817 14045856
[  121.577537] ath10k: [0040]: 71109990 00426D28 00000000 0044F9C0 00098000 00001817 10045851 00426D28
[  121.577553] ath10k: [0048]: 00449ED8 00000000 00000001 00001817 14046403 3021F004 0000F246 00000000
[  121.577566] ath10k: [0056]: 00000001 00000001 00001817 14045856 71109990 00426D28 00000000 0044F9AC
[  121.577582] ath10k: [0064]: 01000000 00001817 08043812 000F13AC 00426D28 00001817 10043809 0000143C
[  121.577593] ath10k: [0072]: 00000003 00000000 00000000
[  121.577604] ath10k_pci 0000:07:00.0: ATH10K_END
[  121.868732] IPv6: ADDRCONF(NETDEV_CHANGE): mesh0: link becomes ready
[  122.577711] ath10k_pci 0000:07:00.0: dropping dbg buffer due to crash since read
[  122.577730] ath10k_pci 0000:07:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[  122.577752] ath10k: [0000]: 00001B20 04043814 00000000 00001B20 13FC5808 711A0A01 00000000 00000002
[  122.577767] ath10k: [0008]: 00000003 00001B20 14045808 00000003 00000000 00000000 00000000 00000001
[  122.577784] ath10k: [0016]: 00001B20 04045806 00000003 00001B20 13FC5808 711A0A01 00000002 00000002
[  122.577798] ath10k: [0024]: 00000000 00001B20 14005808 00000000 00000001 00000000 00000000 00000001
[  122.577813] ath10k: [0032]: 00001B20 04005806 00000000 00001B20 1400380F 61100101 0000143C 0000143C
[  122.577827] ath10k: [0040]: 00000000 00000000 00001B20 0BFC3023 00000001 00000000 00001B20 13FC3807
[  122.577838] ath10k: [0048]: 00000000 0000143C 00000000 00000000
[  122.577849] ath10k_pci 0000:07:00.0: ATH10K_END

Note that on the dmesg output we can see that the peer gets mapped before the "link is ready" message; this may be the problem.

greearb, 2017-10-25:
I think this is fixed now?

greearb, 2018-03-26:
Assuming issue is resolved, please re-open if not.

closed 2018-03-26 by greearb