greearb/ath10k-ct #7: Networking hangs when destroying interface of type MP (Mesh Point)

Jojosep, 2017-06-22:
This issue is present only with the latest firmware version of the CT firmware. Whenever we destroy a previously created MP virtual interface the networking systems stops responding any further commands (iw info, iw list,...) and we're not able to interact with it until system restart.

To reproduce this issue we execute:
- iw wlan0 interface add mesh0 type mp
- iw mesh0 set channel 36
- ifconfig mesh0 up 192.168.50.5
- iw mesh0 mesh join newMesh
- (We do now some tests  that involves ping, arping and iperf)
- iw mesh0 del <-- crash!

We use QCA9888 NICs on ubuntu 14.04 with your modified kernel (4.9.27).

Here's the dmesg output for this whole sequence of commands:

[    7.176277] ath10k_pci 0000:07:00.0: board_file api 2 bmi_id 0:17 crc32 74a04250
[   10.630584] ath10k_pci 0000:07:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   10.630596] ath10k_pci 0000:07:00.0: msdu-desc: 2500  skid: 32
[   10.659748] ath10k_pci 0000:07:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0'
[   10.660467] ath10k_pci 0000:07:00.0: wmi print 'free: 133180 iram: 18948 sram: 39944'
[   10.785184] ath10k_pci 0000:07:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal otp max-sta 32 raw 0 hwcrypto 1
[   10.792477] ath10k_pci 0000:07:00.0: no channel configured; ignoring frame(s)!
[   10.826473] ath10k_pci 0000:07:00.0: firmware crashed! (uuid 31eaaae3-73c2-4243-b0bb-9c5ff4babf79)
[   10.835465] ath10k_pci 0000:07:00.0: qca9888 hw2.0 target 0x01000000 chip_id 0x00000000 sub 0000:0000
[   10.835474] ath10k_pci 0000:07:00.0: kconfig debug 1 debugfs 1 tracing 0 dfs 0 testmode 0
[   10.838334] ath10k_pci 0000:07:00.0: firmware ver 10.4-ct-9888-fW-010-98c2197 api 5 features peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT crc32 b1863e50
[   10.838684] ath10k_pci 0000:07:00.0: board_file api 2 bmi_id 0:17 crc32 74a04250
[   10.838693] ath10k_pci 0000:07:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal otp max-sta 32 raw 0 hwcrypto 1
[   10.850731] ath10k_pci 0000:07:00.0: failed to get memcpy hi address for firmware address 4: -16
[   10.850738] ath10k_pci 0000:07:00.0: failed to read firmware dump area: -16
[   10.857717] ath10k_pci 0000:07:00.0: in crash-regs-harder
[   12.567754] fec 2188000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   12.567778] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   13.259504] ath10k_pci 0000:07:00.0: in crash-regs-harder, firmware did not provide indicator: 0x2
[   13.259518] ath10k_pci 0000:07:00.0: failed to get memcpy hi address for firmware address 8: -28
[   13.259524] ath10k_pci 0000:07:00.0: failed to dump debug log area: -28
[   13.266148] ath10k_pci 0000:07:00.0: failed to get memcpy hi address for firmware address 72: -28
[   13.266156] ath10k_pci 0000:07:00.0: failed to get memcpy hi address for firmware address 72: -28
[   13.266164] ath10k_pci 0000:07:00.0: failed to get memcpy hi address for firmware address 76: -28
[   13.266172] ath10k_pci 0000:07:00.0: failed to get memcpy hi address for firmware address 76: -28
[   13.266183] ath10k_pci 0000:07:00.0: failed to read firmware RAM BSS memory from 4284224 (22168 B): -28
[   13.266193] ath10k_pci 0000:07:00.0: failed to read firmware ROM BSS memory from 4197376 (11312 B): -28
[   13.273576] ath: EEPROM regdomain: 0x0
[   13.273581] ath: EEPROM indicates default country code should be used
[   13.273584] ath: doing EEPROM country->regdmn map search
[   13.273589] ath: country maps to regdmn code: 0x3a
[   13.273593] ath: Country alpha2 being used: US
[   13.273595] ath: Regpair used: 0x3a
[   13.276310] cfg80211: Setting DFS Master region in update_regulatory, was: US unset, new: US FCC  lr: bcab0040  regdom: bcaaff00
[   13.367323] ath10k_pci 0000:07:00.0: cannot restart a device that hasn't been started
[   22.682004] init: failsafe main process (420) killed by TERM signal
[   24.278143] init: plymouth-upstart-bridge main process ended, respawning
[   77.047352] random: crng init done
[  210.156050] ath10k_pci 0000:07:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[  210.156069] ath10k_pci 0000:07:00.0: msdu-desc: 2500  skid: 32
[  210.185354] ath10k_pci 0000:07:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0'
[  210.186089] ath10k_pci 0000:07:00.0: wmi print 'free: 133180 iram: 18948 sram: 39944'
[  210.424385] ath10k_pci 0000:07:00.0: htt peer map vdev 0 peer 02:00:00:00:05:00 id 0
[  210.424426] ath10k_pci 0000:07:00.0: htt peer map vdev 0 peer 02:00:00:00:05:00 id 0
[  210.427810] IPv6: ADDRCONF(NETDEV_UP): mesh0: link is not ready
[  211.151809] ath10k_pci 0000:07:00.0: dropping dbg buffer due to crash since read
[  211.151836] ath10k_pci 0000:07:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[  211.151857] ath10k: [0000]: 00000E0A 13FC0007 60002070 00000010 00000030 00060001 00000E26 17FC582E
[  211.151872] ath10k: [0008]: 00000003 0000001E 0000094C 0000094C 00000000 00000E26 17FC582E 00000012
[  211.151886] ath10k: [0016]: 0000001E 0000094C 0000094C 00000000 00000E26 17FC582E 00000045 0000001E
[  211.151900] ath10k: [0024]: 0000094C 0000094C 00000000 00000E26 17FC582E 00000067 0000001E 0000094C
[  211.151934] ath10k: [0032]: 0000094C 00000000 00000E26 0BFC582E 00000100 11E1A300 00000E29 14005853
[  211.151970] ath10k: [0040]: 51100001 000F0CB0 000003FC 00000001 00449FC8 00000E29 14005856 71109990
[  211.151996] ath10k: [0048]: 00426E00 00000000 0044FA24 00001200 00000E29 14005856 71109990 00426E00
[  211.152022] ath10k: [0056]: 00000000 0044FA10 00098000 00000E29 10005851 00426E00 00449FC8 00000001
[  211.152035] ath10k: [0064]: 00000000 00000E29 14006403 00000000 00000100 00000000 00000000 00000000
[... 72 lines cut ...]
[  248.723365] ath10k_pci 0000:07:00.0: [56]: 0x0A00AFB0 0x000001F4 0x00000FFF 0x00000000
[  248.733335] ath10k_pci 0000:07:00.0: debug log header, dbuf: 0x41b538  dropped: 0
[  248.734359] ath10k_pci 0000:07:00.0: [0] next: 0x41b520 buf: 0x417950 sz: 1500 len: 44 count: 2 free: 0
[  248.735378] ath10k_pci 0000:07:00.0: dropping dbg buffer due to crash since read
[  248.735385] ath10k_pci 0000:07:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[  248.741593] ath10k: [0000]: 0000A7C2 08003813 000F12E8 00000001 0000A7C2 17FC0001 009C2F31 00000000
[  248.750665] ath10k: [0008]: 00000000 00405B3C 91104569
[  248.755811] ath10k_pci 0000:07:00.0: ATH10K_END
[  248.761402] ath10k_pci 0000:07:00.0: [1] next: 0x41b538 buf: 0x417360 sz: 1500 len: 0 count: 0 free: 0
[  248.786891] ath10k_pci 0000:07:00.0: failed to to request monitor vdev 1 stop: -108
[  248.786898] ath10k_pci 0000:07:00.0: failed to synchronize monitor vdev 1 stop: -108
[  248.786904] ath10k_pci 0000:07:00.0: failed to stop monitor vdev: -108
[  248.786910] ath10k_pci 0000:07:00.0: failed to recalc monitor: -108
[  248.786928] ath10k_pci 0000:07:00.0: failed to put down monitor vdev 1: -108
[  248.786937] ath10k_pci 0000:07:00.0: failed to to request monitor vdev 1 stop: -108
[  248.786943] ath10k_pci 0000:07:00.0: failed to synchronize monitor vdev 1 stop: -108
[  248.786948] ath10k_pci 0000:07:00.0: failed to stop monitor vdev: -108
[  248.786957] ath10k_pci 0000:07:00.0: removing peer, cleanup-all, deleting: peer bc897c00 vdev: 1 addr: 04:f0:21:30:46:f2 
[  248.786965] ath10k_pci 0000:07:00.0: removing peer, cleanup-all, deleting: peer bc896200 vdev: 0 addr: 02:00:00:00:05:00 
[  248.890117] ieee80211 phy0: Hardware restart was requested
[  248.890225] ath10k_pci 0000:07:00.0: failed to stop WMI vdev 0: -108
[  248.890245] ath10k_pci 0000:07:00.0: failed to stop vdev 0: -108
[  248.890457] ath10k_pci 0000:07:00.0: failed to submit AP/IBSS self-peer removal on vdev 0: -108
[  248.890481] ath10k_pci 0000:07:00.0: failed to delete WMI vdev 0: -108
[  248.902703] ath10k_pci 0000:07:00.0: failed to read hi_board_data address: -16

greearb, 2017-10-25:
I have since done some mesh testing successfully...do you know if this problem still exists in the latest ath10k-ct driver/firmware?

Jojosep, 2017-10-26:
With the latests firmware versions I don't experience this error. Thank you.

closed 2017-10-26 by Jojosep