greearb/ath10k-ct #8: Survey dump is not working

oamouroux, 2017-07-10:
There is no useful information when using the survey dump command:
iw dev ap0 survey dump
Survey data from ap0
	frequency:			5180 MHz [in use]
Survey data from ap0
	frequency:			5200 MHz
Survey data from ap0
	frequency:			5220 MHz
Survey data from ap0
	frequency:			5240 MHz
Survey data from ap0
	frequency:			5260 MHz
Survey data from ap0
	frequency:			5280 MHz
Survey data from ap0
	frequency:			5300 MHz
Survey data from ap0
	frequency:			5320 MHz
Survey data from ap0
	frequency:			5500 MHz
Survey data from ap0
	frequency:			5520 MHz
Survey data from ap0
	frequency:			5540 MHz
Survey data from ap0
	frequency:			5560 MHz
Survey data from ap0
	frequency:			5580 MHz
Survey data from ap0
	frequency:			5600 MHz
Survey data from ap0
	frequency:			5620 MHz
Survey data from ap0
	frequency:			5640 MHz
Survey data from ap0
	frequency:			5660 MHz
Survey data from ap0
	frequency:			5680 MHz
Survey data from ap0
	frequency:			5700 MHz

Here are the logs:
[ 1277.080000] ath10k: [0000]: A6191400 025C0414 00F34300 04AD9B00 06CA6D10 02B60000 01000000 A6191400
[ 1277.080000] ath10k: [0008]: 574C0414 B4AC4300 78029C00 11000000 00000000 A23D0000 A6191400 574C0414
[ 1277.080000] ath10k: [0016]: B4AC4300 64CA4300 00000000 0C000000 821D0000 A6191400 574C0414 B4AC4300
[ 1277.080000] ath10k: [0024]: 08039C00 06000000 0C000000 023D0000 A6191400 504C0410 BB001071 240A4200
[ 1277.080000] ath10k: [0032]: B4AC4300 98000000 A6191400 504C0410 BB001071 5C0A4200 B4AC4300 98000000
[ 1277.080000] ath10k: [0040]: BF191400 4F4C0414 69010000 940A4200 44AD4300 00000000 00007100 BF191400
[ 1277.080000] ath10k: [0048]: 4F4C0414 79010000 CC0A4200 44AD4300 00000000 01007200 BF191400 564C0414
[ 1277.080000] ath10k: [0056]: 44AD4300 50039C00 11000000 00000000 203D0000 BF191400 564C0414 44AD4300
[ 1277.080000] ath10k: [0064]: ACCA4300 00000000 0C000000 001D0000 BF191400 035C0414 00000000 06CA6D10
[ 1277.080000] ath10k: [0072]: 02B60000 04AD9B00 44AD4300 BF191400 524C0414 44AD4300 00000000 01000000
[ 1277.080000] ath10k: [0080]: 02000000 08000000 BF191400 0128FC17 070B1071 02000001 05000000 00000000
[ 1277.080000] ath10k: [0088]: 00000000 BF191400 0128FC17 80881071 08000000 F4864400 08009600 F00F0000
[ 1277.080000] ath10k: [0096]: BF191400 0128FC17 80881071 00000100 F4864400 08009600 00000000 BF191400
[ 1277.080000] ath10k: [0104]: 0128FC17 80881071 00000200 F4864400 08009600 00000000 BF191400 0128FC17
[ 1277.080000] ath10k: [0112]: 80881071 00000300 F4864400 08009600 00000000 BF191400 0128FC17 80881071
[ 1277.080000] ath10k: [0120]: 00000400 F4864400 08009600 00000000 BF191400 0128FC17 80881071 00000500
[ 1277.080000] ath10k: [0128]: F4864400 08009600 00000000 C0191400 524C0414 44AD4300 C2000000 01000000
[ 1277.080000] ath10k: [0136]: 02000000 08000000 C0191400 0128FC17 070B1071 02000603 07B00000 00000000
[ 1277.080000] ath10k: [0144]: FFFF0000 C0191400 0128FC17 80881071 08000000 F4864400 1C009600 F00F0000
[ 1277.080000] ath10k: [0152]: C0191400 0128FC17 80881071 0A000100 F4864400 1C009600 1FF80000 C0191400
[ 1277.080000] ath10k: [0160]: 0128FC17 80881071 0A000200 F4864400 1C009600 1FF80000 C0191400 0128FC17
[ 1277.080000] ath10k: [0168]: 80881071 00000300 F4864400 1C009600 00000000 C0191400 0128FC17 80881071
[ 1277.080000] ath10k: [0176]: 00000400 F4864400 1C009600 00000000 C0191400 0128FC17 80881071 00000500
[ 1277.080000] ath10k: [0184]: F4864400 1C009600 00000000 C3191400 564C0414 44AD4300 98039C00 06000000
[ 1277.080000] ath10k: [0192]: 0C000000 003D0000

greearb, 2017-07-10:
Please provide useful debug info like what NIC/chipset, firmware version, OS, etc you are using.

And, does survey-dump work with stock QCA firmware on your system?

oamouroux, 2017-07-11:
QCA9882
Firmware version:  10.1-ct-8x-__fW-019-6375cb7
OS: Openwrt Chaos Calmer (Kernel 3.18.45, mac80211 backported from v4.4-rc5-1913-gc8fdf68)
This is working using official ath10k from v4.4-rc5-1913-gc8fdf68 + official firmware 10.2.4.70.52

greearb, 2017-07-11:
If you use ath10k-ct driver and QCA firmware 10.2.4.70.52, does it work then?  I am trying to understand if it is an ath10k-ct driver or a firmware limitation.

oamouroux, 2017-07-11:
This is working using ath10k-ct driver and QCA firmware 10.2.4.70.52

greearb, 2017-10-25:
Maybe this is some new feature in 10.2 firmware that has not yet been backported to 10.1 ath10k-ct firmware.

greearb, 2018-09-28:
I believe I have this working in latest driver/firmware.  Please re-open if you still see trouble.

closed 2018-09-28 by greearb