greearb/ath10k-ct #12: QCA9888 (Compex WLE650V5-18A) not reporting tx_rate with beta9 firmware

syoder89, 2017-08-08:
driver: ath10k-ct
version: 4.4.74
firmware-version: 10.4-ct-9888-fW-009-a611947
LEDE-version: 254f0da6d2d9a7dac5e1281de8dcf450aff7bacd    (Wed Jul 12 10:34:00 2017)
Card: QCA9888 (Compex WLE650V5-18A)

Tx bitrate from iw .. station dump shows 6.0 Mbps.

The ath10k-ct driver shows 0 for resp->data_tx_completion.flag_ack_rssi_filled and resp->data_tx_completion.flag_tx_rate_filled for all transmitted frames (I put a printk in to test).

I've tried 2 beta9 firmwares:
http://www.candelatech.com/downloads/ath10k-9888-10-4/ath10k-fw-beta/firmware-5-ct-full-htt-mgt-community.bin
http://www.candelatech.com/downloads/ath10k-9888-10-4/firmware-5-ct-full-community-9.bin-lede.003

Both firmwares are not reporting the connected station tx rate.

greearb, 2017-08-08:
I don't test much with 9888, but TX reporting works with 9984 (which is built from the same firmware source).  Can you attach 'dmesg' output from a boot, as well as text dump of how you are getting the rx rate report?
And, in case you can test with a 9984 or 9880 NIC on the same software build, it would be good to know if the problem is only with 9888 or not.

syoder89, 2017-08-08:
Sure I can pop a 9984 card in (WLE1216) and test.

syoder89, 2017-08-09:
Ok so I wasn't able to boot my system (WPJ563) with the WLE1216V5-23 even with the 5V mPCIe power jumper on, but I was able to get it to work in an older WPJ344, just not with the QCA9888 firmware.

I tried most of the QCA9984 firmwares in the ath10k-9984-10-4 directory and even the latest beta but see the same 6.0 Mbps transmit rate. The 9984 should have no problem with TX reporting right?

Here is a station dump and dmesg from the latest beta 9884 firmware:
http://www.candelatech.com/downloads/ath10k-9984-10-4/ath10k-fw-beta/firmware-5-ct-full-community.bin
I'm using ath10k-ct-2017-05-16-c048ef0c. Is there something in the latest commits that could solve this?

<pre>
iw wlan2 station dump
Station a4:08:ea:19:a8:ee (on wlan2)
	inactive time:	20 ms
	rx bytes:	57132
	rx packets:	438
	tx bytes:	64712
	tx packets:	171
	tx retries:	0
	tx failed:	6
	rx drop misc:	0
	signal:  	-60 dBm
	signal avg:	-59 dBm
	tx bitrate:	6.0 MBit/s
	rx bitrate:	433.3 MBit/s VHT-MCS 9 80MHz short GI VHT-NSS 1
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	DTIM period:	2
	beacon interval:100
	short slot time:yes
	connected time:	34 seconds

</pre>
This doesn't have those extra printks...
<pre>
[  117.595769] ath10k driver, optimized for CT firmware, probing pci device: 0x46.
[  117.604718] ath10k_pci 0000:00:00.0: pci irq legacy oper_irq_mode 1 irq_mode 0 reset_mode 0
[  117.755745] ath10k_pci 0000:00:00.0: Direct firmware load for ath10k/pre-cal-pci-0000:00:00.0.bin failed with error -2
[  117.766623] ath10k_pci 0000:00:00.0: Falling back to user helper
[  117.856036] firmware ath10k!pre-cal-pci-0000:00:00.0.bin: firmware_loading_store: map pages failed
[  117.865355] ath10k_pci 0000:00:00.0: Direct firmware load for ath10k/cal-pci-0000:00:00.0.bin failed with error -2
[  117.875885] ath10k_pci 0000:00:00.0: Falling back to user helper
[  118.048073] firmware ath10k!cal-pci-0000:00:00.0.bin: firmware_loading_store: map pages failed
[  118.057096] ath10k_pci 0000:00:00.0: Direct firmware load for ath10k/fwcfg-pci-0000:00:00.0.txt failed with error -2
[  118.067799] ath10k_pci 0000:00:00.0: Falling back to user helper
[  118.243787] firmware ath10k!fwcfg-pci-0000:00:00.0.txt: firmware_loading_store: map pages failed
[  118.258315] ath10k_pci 0000:00:00.0: too long firmware RAM BSS length: 50144
[  118.265540] ath10k_pci 0000:00:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[  118.275620] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[  118.295418] ath10k_pci 0000:00:00.0: firmware ver 10.4-ct-9984-fW-009-f493c92 api 5 features peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT crc32 7b264a44
[  119.836754] ath10k_pci 0000:00:00.0: board_file api 2 bmi_id 0:1 crc32 751efba1
[  122.478953] ath10k_pci 0000:00:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[  122.486613] ath10k_pci 0000:00:00.0: msdu-desc: 2500  skid: 32
[  122.567382] ath10k_pci 0000:00:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0'
[  122.577612] ath10k_pci 0000:00:00.0: wmi print 'free: 90144 iram: 17252 sram: 24080'
[  122.735799] ath10k_pci 0000:00:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal otp max-sta 32 raw 0 hwcrypto 1
[  122.856852] ath: EEPROM regdomain: 0x0
[  122.860667] ath: EEPROM indicates default country code should be used
[  122.867234] ath: doing EEPROM country->regdmn map search
[  122.872624] ath: country maps to regdmn code: 0x3a
[  122.877496] ath: Country alpha2 being used: US
[  122.881995] ath: Regpair used: 0x3a
[  128.152401] ath10k_pci 0000:00:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[  128.160065] ath10k_pci 0000:00:00.0: msdu-desc: 2500  skid: 32
[  128.240914] ath10k_pci 0000:00:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0'
[  128.251144] ath10k_pci 0000:00:00.0: wmi print 'free: 90144 iram: 17252 sram: 24080'
[  128.534886] ath10k_pci 0000:00:00.0: htt peer map vdev 0 peer 04:f0:21:18:e3:ab id 0
[  128.542755] ath10k_pci 0000:00:00.0: htt peer map vdev 0 peer 04:f0:21:18:e3:ab id 0
[  128.554086] IPv6: ADDRCONF(NETDEV_UP): wlan2: link is not ready
[  128.569145] ath10k_pci 0000:00:00.0: removing peer, htt peer unmap vdev 0 peer 04:f0:21:18:e3:ab id 0
[  128.579056] ath10k_pci 0000:00:00.0: peer-unmap-event: unknown peer id 0
[  131.578993] ath10k_pci 0000:00:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[  131.586645] ath10k_pci 0000:00:00.0: msdu-desc: 2500  skid: 32
[  131.667450] ath10k_pci 0000:00:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0'
[  131.677685] ath10k_pci 0000:00:00.0: wmi print 'free: 90144 iram: 17252 sram: 24080'
[  131.961430] ath10k_pci 0000:00:00.0: htt peer map vdev 0 peer 04:f0:21:18:e3:ab id 0
[  131.969320] ath10k_pci 0000:00:00.0: htt peer map vdev 0 peer 04:f0:21:18:e3:ab id 0
[  131.980683] IPv6: ADDRCONF(NETDEV_UP): wlan2: link is not ready
[  131.987539] device wlan2 entered promiscuous mode
[  133.088942] IPv6: ADDRCONF(NETDEV_CHANGE): wlan2: link becomes ready
[  134.185558] ath10k_pci 0000:00:00.0: htt peer map vdev 0 peer 04:f0:21:18:e3:ab id 0
[  149.093845] ath10k_pci 0000:00:00.0: htt peer map vdev 0 peer a4:08:ea:19:a8:ee id 1
[  149.111337] ath10k_pci 0000:00:00.0: band: 1  ratemask: 0xffffffff  hw-nss: 4 dev-id: 0x46
</pre>

greearb, 2017-08-12:
I don't have time right now to poke at LEDE, but in case you can boot up
our 4.9 or 4.7 kernel (linked from ath10k pages), it might be worth testing.

TX/RX rate reporting is working for us on 4.9 kernels, at least, and very likely 4.7 unless
I somehow broke it recently.

We normally run on Fedora 24 Linux currently, but that shouldn't matter much.

Thanks,
Ben

greearb, 2017-09-01:
I noticed while porting patches to the 4.13 kernel that the mac80211 internals w/regard to reporting rates have changed, so if you are using my older driver on a new stack, that might be the cause.  I'm starting testing 4.13 soon, and will update the ath10k-ct LEDE driver to be based on that when it is stable.

greearb, 2017-10-25:
I am pretty sure this is fixed now, so closing the bug.  Please re-open if you see the issue again.

closed 2017-10-25 by greearb