greearb/ath10k-ct #14: endless looping message "dropping dbg buffer due to crash since read "

oamouroux, 2017-09-25:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).
Messages caught from the boot. It seems to be linked to a RF environment. Changing the channel did not solve the issue.

Software (OS, Firmware version, kernel, driver, etc)
OpenWRT Chaos, Calmer Kernel 3.18, driver 1df6b26afd1766aa476f41b74605b56a79b5ec64, fw 18

Hardware (NIC chipset, platform, etc)
QCA9882

Logs (dmesg, maybe supplicant and/or hostap)

[ 2769.600000] ath10k_pci 0000:01:00.0: no channel configured; ignoring frame(s)! 
[ 2769.610000] ath10k_pci 0000:01:00.0: no channel configured; ignoring frame(s)! 
[ 2769.620000] ath10k_pci 0000:01:00.0: no channel configured; ignoring frame(s)! 
[ 2769.630000] ath10k_pci 0000:01:00.0: firmware crashed! (uuid 038fb12b-6655-4f54-85ea-2d80b9785765) 
[ 2769.640000] ath10k_pci 0000:01:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043222ff sub 0000:0000 
[ 2769.650000] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 0 testmode 0 
[ 2769.660000] ath10k_pci 0000:01:00.0: firmware ver 10.1-ct-8x-_fW-019-ddf2a35 api 2 features wmi-10.x,has-wmi-mgmt-tx,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-C
T crc32 01d901a9 
[ 2769.690000] ath10k_pci 0000:01:00.0: board_file api 1 bmi_id N/A crc32 bebc7c08 
[ 2769.690000] ath10k_pci 0000:01:00.0: htt-ver 2.1 wmi-op 2 htt-op 2 cal otp max-sta 128 raw 0 hwcrypto 1 
[ 2769.710000] ath10k_pci 0000:01:00.0: firmware register dump: 
[ 2769.710000] ath10k_pci 0000:01:00.0: [00]: 0x4100016C 0x000015B3 0x009A9B51 0x00955B31 
[ 2769.720000] ath10k_pci 0000:01:00.0: [04]: 0x009A9B51 0x00060730 0x00000005 0x00000032 
[ 2769.730000] ath10k_pci 0000:01:00.0: [08]: 0x0040EC80 0x00411080 0x002B5D29 0x00400000 
[ 2769.730000] ath10k_pci 0000:01:00.0: [12]: 0x00000009 0x00000000 0x00958360 0x0095836B 
[ 2769.740000] ath10k_pci 0000:01:00.0: [16]: 0x00958080 0x0094085D 0x00000000 0x00000000 
[ 2769.750000] ath10k_pci 0000:01:00.0: [20]: 0x409A9B51 0x0040AE44 0x00000005 0x0040CFB0 
[ 2769.760000] ath10k_pci 0000:01:00.0: [24]: 0x80944C31 0x0040AEA4 0xF573FFFF 0xC09A9B51 
[ 2769.770000] ath10k_pci 0000:01:00.0: [28]: 0x80942BE7 0x0040AED4 0x004112E4 0x00000000 
[ 2769.780000] ath10k_pci 0000:01:00.0: [32]: 0x809B6209 0x0040AEF4 0x004090A0 0x00409110 
[ 2769.780000] ath10k_pci 0000:01:00.0: [36]: 0x80940F18 0x0040AF14 0x00000008 0x00403A20 
[ 2769.790000] ath10k_pci 0000:01:00.0: [40]: 0x80940EEA 0x0040AF44 0x00400000 0x00000000 
[ 2769.800000] ath10k_pci 0000:01:00.0: [44]: 0x80940F31 0x0040AF64 0x00401C10 0x00400600 
[ 2769.810000] ath10k_pci 0000:01:00.0: [48]: 0x40940024 0x0040AF84 0x004068E8 0x004068E8 
[ 2769.820000] ath10k_pci 0000:01:00.0: [52]: 0x00000000 0x0040AFA4 0x009BB001 0x00040020 
[ 2769.820000] ath10k_pci 0000:01:00.0: [56]: 0x00403994 0x00000000 0x00000001 0x00400600 
[ 2769.830000] ath10k_pci 0000:01:00.0: debug log header, dbuf: 0x412598  dropped: 0 
[ 2769.840000] ath10k_pci 0000:01:00.0: [0] next: 0x4125b0 buf: 0x4103fc sz: 1500 len: 448 count: 17 free: 0 
[ 2769.850000] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER: 
[ 2769.860000] ath10k: [0000]: 295D2B00 3204FC17 00000000 4B080000 05000000 03000000 00000000 295D2B00 
[ 2769.870000] ath10k: [0008]: 3204FC17 03771071 00000000 00000000 FFFFFFFF FFFFFFFF 295D2B00 3204FC0F 
[ 2769.880000] ath10k: [0016]: 13771071 03000000 03000000 295D2B00 3204FC17 03771071 23000100 22000000 
[ 2769.890000] ath10k: [0024]: 03000000 03000000 295D2B00 3204FC17 03771071 07000200 06000000 07000000 
[ 2769.900000] ath10k: [0032]: 07000000 295D2B00 3204FC17 03771071 00000300 00000200 FFFFFFFF FFFFFFFF 
[ 2769.900000] ath10k: [0040]: 295D2B00 3204FC0F 13771071 05000000 05000000 295D2B00 3204FC17 03771071 
[ 2769.910000] ath10k: [0048]: 00000400 00000100 FFFFFFFF FFFFFFFF 295D2B00 3204FC0F 13771071 03000000 
[ 2769.920000] ath10k: [0056]: 03000000 295D2B00 3204FC17 03771071 21080500 20080000 01000000 01000000 
[ 2769.930000] ath10k: [0064]: 295D2B00 3204FC17 00000000 00000000 00000000 00000000 00000000 295D2B00 
[ 2769.940000] ath10k: [0072]: 3204FC17 01000000 00000000 00000000 00000000 00000000 295D2B00 3204FC17 
[ 2769.950000] ath10k: [0080]: 02000000 00000000 00000000 00000000 00000000 295D2B00 3204FC17 03000000 
[ 2769.960000] ath10k: [0088]: 00000000 00000000 00000000 00000000 295D2B00 3204FC17 04000000 00000000 
[ 2769.970000] ath10k: [0096]: 00000000 00000000 00000000 295D2B00 3204FC13 45331091 B1292B00 295D2B00 
[ 2769.980000] ath10k: [0104]: 401F0000 295D2B00 0100FC17 B3150000 B3150000 34AD4000 6C010041 FFFFFFFF 
[ 2769.990000] ath10k_pci 0000:01:00.0: ATH10K_END 
[ 2769.990000] ath10k_pci 0000:01:00.0: [1] next: 0x412598 buf: 0x4109ec sz: 1500 len: 0 count: 0 free: 0 
[ 2770.150000] device ap0 entered promiscuous mode 
[ 2770.290000] ieee80211 phy0: Hardware restart was requested 
[ 2770.300000] ath10k_pci 0000:01:00.0: failed to synchronize setup for vdev 0 restart 0: -143 
[ 2770.310000] ath10k_pci 0000:01:00.0: failed to start vdev 0 addr 04:f0:21:31:61:25 on freq 5660: -143 
[ 2770.320000] ath10k_pci 0000:01:00.0: failed to flush transmit queue (skip 1 ar-state 2): 500 
[ 2770.330000] device ap0 left promiscuous mode 
[ 2770.340000] br-lan: port 1(ap0) entered disabled state 
[ 2770.350000] ath10k_pci 0000:01:00.0: failed to submit AP/IBSS self-peer removal on vdev 0: -143 
[ 2770.350000] ath10k_pci 0000:01:00.0: failed to delete WMI vdev 0: -143 
[ 2770.360000] ath10k_pci 0000:01:00.0: found vif peer 04:f0:21:31:61:25 id: 136 entry on vdev 0 after it was supposedly removed 
[ 2770.370000] ath10k_pci 0000:01:00.0: removing stale peer 04:f0:21:31:61:25 from vdev_id 0 
[ 2770.380000] ath10k_pci 0000:01:00.0: removing peer mapping 04:f0:21:31:61:25 peer-id: 136 
[ 2770.390000] ath10k_pci 0000:01:00.0: removing peer mapping 04:f0:21:31:61:25 peer-id: 152 
[ 2780.720000] batman_adv: bat0: Removing interface: adhoc0 
[ 2781.420000] batman_adv: bat0: Adding interface: adhoc0 
[ 2781.430000] batman_adv: bat0: The MTU of interface adhoc0 is too small (1500) to handle the transport of batman-adv packets. Packets going over this interface will be fragmented on layer2 which could impact the performance. Setting t
he MTU to 1532 would solve the problem. 
[ 2781.450000] batman_adv: bat0: Not using interface adhoc0 (retrying later): interface not active 
[ 2782.630000] ath10k_pci 0000:01:00.0: 10.1 wmi init: vdevs: 16  peers: 127  tid: 256 
[ 2782.650000] ath10k_pci 0000:01:00.0: wmi print 'P 128 V 8 T 410' 
[ 2782.650000] ath10k_pci 0000:01:00.0: wmi print 'msdu-desc: 1424  sw-crypt: 0' 
[ 2782.660000] ath10k_pci 0000:01:00.0: wmi print 'alloc rem: 25908 iram: 25760' 
[ 2782.750000] device ap0 entered promiscuous mode 
[ 2782.750000] ath10k_pci 0000:01:00.0: no channel configured; ignoring frame(s)! 
[ 2782.750000] ath10k_pci 0000:01:00.0: no channel configured; ignoring frame(s)! 
[ 2782.770000] ath10k_pci 0000:01:00.0: no channel configured; ignoring frame(s)! 
[ 2782.780000] ath10k_pci 0000:01:00.0: no channel configured; ignoring frame(s)! 
[ 2782.810000] br-lan: port 1(ap0) entered forwarding state 
[ 2782.810000] br-lan: port 1(ap0) entered forwarding state 
[ 2782.970000] batman_adv: bat0: Interface activated: adhoc0 
[ 2783.080000] batman_adv: bat0: Interface deactivated: adhoc0 
[ 2783.080000] ath10k_pci 0000:01:00.0: dropping dbg buffer due to crash since read 
[ 2783.090000] batman_adv: bat0: Interface activated: adhoc0 
[ 2783.210000] batman_adv: bat0: Interface deactivated: adhoc0 
[ 2783.210000] batman_adv: bat0: Interface activated: adhoc0 
[ 2783.220000] adhoc0: Created IBSS using preconfigured BSSID ba:f2:be:e9:36:ab 
[ 2783.230000] adhoc0: Creating new IBSS network, BSSID ba:f2:be:e9:36:ab 
[ 2783.250000] ath10k_pci 0000:01:00.0: dropping dbg buffer due to crash since read 
[ 2783.300000] ath10k_pci 0000:01:00.0: dropping dbg buffer due to crash since read 
[ 2783.630000] ath10k_pci 0000:01:00.0: dropping dbg buffer due to crash since read 
[ 2784.810000] br-lan: port 1(ap0) entered forwarding state 
[ 2793.330000] ath10k_pci 0000:01:00.0: dropping dbg buffer due to crash since read 
[ 2793.630000] ath10k_pci 0000:01:00.0: dropping dbg buffer due to crash since read

greearb, 2017-09-25:
On 09/25/2017 09:38 AM, oamouroux wrote:
> Please provide this info. See this link for more info on how to gather debug
> info: http://www.candelatech.com/ath10k-bugs.php
>
> Description of the problem (how to configure, how to reproduce, how often it happens).
> Messages caught from the boot. It seems to be linked to a RF environment. Changing the channel did not solve the issue.
>
> Software (OS, Firmware version, kernel, driver, etc)
> OpenWRT Chaos, Calmer Kernel 3.18, driver 1df6b26 <https://github.com/greearb/ath10k-ct/commit/1df6b26afd1766aa476f41b74605b56a79b5ec64>, fw 18
>
> Hardware (NIC chipset, platform, etc)
> QCA9882
>
> Logs (dmesg, maybe supplicant and/or hostap)

I'll decode the crash tomorrow when I'm back in the office, but to make the 'dropping dbg buffer'
messages go away, you need to read the binary crash dump file (it automatically clears on
read).

There is info on reading the binary dump here:
http://www.candelatech.com/ath10k-bugs.php

Thanks,
Ben

-- 
Ben Greear <greearb@candelatech.com>
Candela Technologies Inc  http://www.candelatech.com

greearb, 2017-09-29:
This crash is due to a WMI keep-alive timeout in the firmware.  It did not receive a keep-alive message from the driver within about 5 seconds.  Usually this means the firmware and/or driver is in a bad state,
and crashing and restarting firmware is the best option.

Is this crash reproducible?

oamouroux, 2017-10-24:
Unfortunately not for now. If I can reproduce it, I will send the crash report

greearb, 2018-03-26:
Closing since it does not seem easy to reproduce and I do not have any idea of the root cause.

closed 2018-03-26 by greearb