greearb/ath10k-ct #16: Can not make WLE1216V5-20 working under ath10k-ct

master8282, 2018-01-19:
Dear Sir,
Recently I have bought WLE1216V5-20 and I already became sad cause was involved in such strong troubleshooting.

wifi module: WLE1216V5-20 with QCA9984 after October, 2017

firmware-5.bin was taken here:
https://www.candelatech.com/downloads/ath10k-9984-10-4/firmware-5-ct-non-commercial-htt-mgt-9.bin

boad-2.bin was taken here:
https://www.candelatech.com/downloads/ath10k-9984-10-4/board-2-ct.bin

trying on Lede:

root@OpenWrt:~# uname -a
Linux OpenWrt 4.9.76 #0 SMP Wed Jan 17 21:00:43 2018 x86_64 GNU/Linux

root@OpenWrt:~# cat load_m 
modprobe cfg80211
modprobe ath
modprobe mac80211
modprobe ath10k_core
modprobe ath10k_pci

root@OpenWrt:~# sh load_m 
root@OpenWrt:~# lsmod | grep ath
ath                    16211  1 ath10k_core
ath10k_core           254566  1 ath10k_pci
ath10k_pci             29467  0 
cfg80211              199024  3 ath10k_core,mac80211,ath
mac80211              374887  1 ath10k_core

dmesg:

...
[ 1594.261893] ath10k 4.13 driver, optimized for CT firmware, probing pci device: 0x46.
[ 1594.263594] ath10k_pci 0000:02:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[ 1594.433632] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/fwcfg-pci-0000:02:00.0.txt failed with error -2
[ 1594.434303] ath10k_pci 0000:02:00.0: Falling back to user helper
[ 1594.438423] firmware ath10k!fwcfg-pci-0000:02:00.0.txt: firmware_loading_store: map pages failed
[ 1594.439108] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/pre-cal-pci-0000:02:00.0.bin failed with error -2
[ 1594.439784] ath10k_pci 0000:02:00.0: Falling back to user helper
[ 1594.443757] firmware ath10k!pre-cal-pci-0000:02:00.0.bin: firmware_loading_store: map pages failed
[ 1594.444489] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/cal-pci-0000:02:00.0.bin failed with error -2
[ 1594.445151] ath10k_pci 0000:02:00.0: Falling back to user helper
[ 1594.448908] firmware ath10k!cal-pci-0000:02:00.0.bin: firmware_loading_store: map pages failed
[ 1594.449618] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/QCA9984/hw1.0/firmware-6.bin failed with error -2
[ 1594.450292] ath10k_pci 0000:02:00.0: Falling back to user helper
[ 1594.454115] firmware ath10k!QCA9984!hw1.0!firmware-6.bin: firmware_loading_store: map pages failed
[ 1594.455560] ath10k_pci 0000:02:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[ 1594.456213] ath10k_pci 0000:02:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[ 1594.458720] ath10k_pci 0000:02:00.0: firmware ver 10.4-ct-9984-xtH-009-dfa0083 api 5 features peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,rxswcrypt-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,no-bmiss-CT crc32 50d1a82f
[ 1595.901052] ath10k_pci 0000:02:00.0: board_file api 2 bmi_id 0:31 crc32 e807b522
[ 1598.490391] ath10k_pci 0000:02:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[ 1598.490920] ath10k_pci 0000:02:00.0: msdu-desc: 2500  skid: 32
[ 1598.575386] ath10k_pci 0000:02:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0'
[ 1598.575988] ath10k_pci 0000:02:00.0: wmi print 'free: 123180 iram: 13444 sram: 42240'
[ 1598.740229] ath10k_pci 0000:02:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal otp max-sta 32 raw 0 hwcrypto 1
[ 1598.768195] ath10k_pci 0000:02:00.0: no channel configured; ignoring frame(s)!
[ 1598.876732] ath: EEPROM regdomain: 0x0
[ 1598.876737] ath: EEPROM indicates default country code should be used
[ 1598.876738] ath: doing EEPROM country->regdmn map search
[ 1598.876741] ath: country maps to regdmn code: 0x3a
[ 1598.876743] ath: Country alpha2 being used: US
[ 1598.876744] ath: Regpair used: 0x3a
[ 1601.551841] ath10k_pci 0000:02:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[ 1601.552356] ath10k_pci 0000:02:00.0: msdu-desc: 2500  skid: 32
[ 1601.636663] ath10k_pci 0000:02:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0'
[ 1601.637257] ath10k_pci 0000:02:00.0: wmi print 'free: 123180 iram: 13444 sram: 42240'
[ 1601.942882] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 1602.551038] ath10k_pci 0000:02:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[... 688 lines cut ...]
[ 2318.735302] device wlp2s0 entered promiscuous mode
[ 2318.807239] IPv6: ADDRCONF(NETDEV_UP): br0: link is not ready
[ 2321.540414] IPv6: ADDRCONF(NETDEV_CHANGE): wlp2s0: link becomes ready

I can see network, it appears and dissappers in the loop and I can not connect to it.

If I take firmware for regular QCA9984 from:

https://github.com/kvalo/ath10k-firmware/tree/master/QCA9984/hw1.0

I have this tace:

[ 2857.205737] ath10k_pci 0000:02:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[ 2857.325714] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/pre-cal-pci-0000:02:00.0.bin failed with error -2
[ 2857.325734] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/cal-pci-0000:02:00.0.bin failed with error -2
[ 2857.325750] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/QCA9984/hw1.0/firmware-6.bin failed with error -2
[ 2857.325886] ath10k_pci 0000:02:00.0: invalid firmware magic
[ 2857.326005] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/QCA9984/hw1.0/firmware-4.bin failed with error -2
[ 2857.327401] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/QCA9984/hw1.0/firmware-3.bin failed with error -2
[ 2857.327426] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/QCA9984/hw1.0/firmware-2.bin failed with error -2
[ 2857.327430] ath10k_pci 0000:02:00.0: Failed to find firmware-N.bin (N between 2 and 6) from ath10k/QCA9984/hw1.0: -2
[ 2857.327550] ath10k_pci 0000:02:00.0: could not fetch firmware files (-2)
[ 2857.327624] ath10k_pci 0000:02:00.0: could not probe fw (-2)

Could you please help me in trubleshuting of existing firmware, or advice a workable firmware for WLE1216V5-20.

greearb, 2018-01-19:
> [ 1603.538566] [] ? dump_stack+0x5d/0x79
> [ 1603.538985] [] ? __warn+0xb2/0xcb
> [ 1603.539313] [] ? ath10k_htt_t2h_msg_handler+0xb80/0x1861 [ath10k_core]
> [ 1603.539798] [] ? ath10k_htt_t2h_msg_handler+0xcfe/0x1861 [ath10k_core]
> [ 1603.540282] [] ? ath10k_htt_txrx_compl_task+0x8b4/0xfde [ath10k_core]
> [ 1603.540760] [] ? ath10k_pci_hif_exchange_bmi_msg+0x10ee/0x110c [ath10k_pci]
> [ 1603.541255] [] ? ath10k_ce_per_engine_service+0xba/0xcc [ath10k_pci]
> [ 1603.541730] [] ? ath10k_pci_enable_legacy_irq+0x8e/0x107 [ath10k_pci]
> [ 1603.542210] [] ? net_rx_action+0xeb/0x25a
> [ 1603.542560] [] ? __do_softirq+0xb8/0x1a1
> [ 1603.542906] [] ? irq_exit+0x4f/0x5b
> [ 1603.543238] [] ? do_IRQ+0xa4/0xba
> [ 1603.543566] [] ? common_interrupt+0x93/0x93
> [ 1603.544026] [] ? native_safe_halt+0x2/0x3
> [ 1603.544451] [] ? default_idle+0x5/0x8
> [ 1603.544790] [] ? cpu_startup_entry+0x11c/0x1ce
> [ 1603.545154] [] ? start_secondary+0x12b/0x12e

Hello,

First, the warning splat from LEDE might not be a real problem.  I do not see any
obvious errors in the dbglog output from the firmware.

And, on DFS channels, it will not work because your radio detects
RADAR, which will then properly disable DFS channels.

Does LEDE work on non-DFS channels?

THanks,
Ben

master8282, 2018-01-19:
Hello Ben,

"Does LEDE work on non-DFS channels?" On lede unfortunately I'm not able to start the device as DFS as without DFS, before I start hostapd daemon, I see the traces and when load ath10k_pci module.

[ 1603.538238] Call Trace:
[ 1603.538453] 
[ 1603.538566] [] ? dump_stack+0x5d/0x79
[ 1603.538985] [] ? __warn+0xb2/0xcb
[ 1603.539313] [] ? ath10k_htt_t2h_msg_handler+0xb80/0x1861 [ath10k_core]
[ 1603.539798] [] ? ath10k_htt_t2h_msg_handler+0xcfe/0x1861 [ath10k_core]
[ 1603.540282] [] ? ath10k_htt_txrx_compl_task+0x8b4/0xfde [ath10k_core]
[ 1603.540760] [] ? ath10k_pci_hif_exchange_bmi_msg+0x10ee/0x110c [ath10k_pci]
[ 1603.541255] [] ? ath10k_ce_per_engine_service+0xba/0xcc [ath10k_pci]
[ 1603.541730] [] ? ath10k_pci_enable_legacy_irq+0x8e/0x107 [ath10k_pci]
[ 1603.542210] [] ? net_rx_action+0xeb/0x25a
[ 1603.542560] [] ? __do_softirq+0xb8/0x1a1
[ 1603.542906] [] ? irq_exit+0x4f/0x5b
[ 1603.543238] [] ? do_IRQ+0xa4/0xba
[ 1603.543566] [] ? common_interrupt+0x93/0x93
[ 1603.543917] 
[ 1603.544026] [] ? native_safe_halt+0x2/0x3
[ 1603.544451] [] ? default_idle+0x5/0x8
[ 1603.544790] [] ? cpu_startup_entry+0x11c/0x1ce
[ 1603.545154] [] ? start_secondary+0x12b/0x12e
[ 1603.545448] ---[ end trace 8d4d257a180fa625 ]---

And after that hostapd can not connect with wlan interface.

About Ubuntu:
"And, on DFS channels, it will not work because your radio detects
RADAR, which will then properly disable DFS channels."
I would imagine if I could have wrong hostaptd config or requested frequencies are occupied.
But I have one more wifi card -   Compex WLE900V5-27 and I use almost same configs (only without 160 MHz setting), and no any problem with working the second wifi card. And wifi spectrum analyzer shows many vacant  frequencies.  

I already thinking about I got factory defective wle1216v5-20 and maybe send it back to seller for checking. Unfortunate I don't know about any another firmwares except candelatech  to check the card with alternative firmware and be 100% sure about hardware issue.

Thank you.

greearb, 2018-03-26:
I have recently posted patches that were accepted into LEDE.  These enable my latest driver and firmware.  Maybe re-test with those and see if the problem is improved?  Please post more dmesg logs if errors are seen with the latest...

master8282, 2018-06-14:
Dear Ben,

After much suffering and tries I could build working configuration.
I found out that mainboard makes sense.
Now I use the intel NUC mainboard with UEFI and embedded core i3 inside.

I made some tests using different drivers, here is my report:

I didn't get any issues even using standard linux drivers to make it working under 80 Mhz, only boars-2.bin I had to take at compex drivers. 

But for 160MHz native linux ath10k drivers don't work for 160Mhz I see the following errors:
"
nl80211: Start radar detection (CAC) 5500 MHz (ht_enabled=1, vht_enabled=1, bandwidth=160 MHz, cf1=5570 MHz, cf2=0 MHz)
 * freq=5500
 * vht_enabled=1
 * ht_enabled=1
 * bandwidth=160
 * channel_width=5
 * center_freq1=5570
 * center_freq2=0
nl80211: Failed to start radar detection: -16 (Device or resource busy)
DFS start_dfs_cac() failed, -1
Interface initialization failed
"
I used kernel (stable:  4.16.15) and built your ath10k-ct, got four ko modules, and just changed them with native from linux.
And the wonderful has happened, using new modules and candella tech firmware (only firmware-5-ct-non-commercial.bin - it is important), I made working configuration.
If I use any another CT firmware, it works fine for 80MHz, but for 160MHz I see the following error:

"
nl80211: Start radar detection (CAC) 5500 MHz (ht_enabled=1, vht_enabled=1, bandwidth=160 MHz, cf1=5570 MHz, cf2=0 MHz)
 * freq=5500
 * vht_enabled=1
 * ht_enabled=1
 * bandwidth=160
 * channel_width=5
 * center_freq1=5570
 * center_freq2=0
nl80211: Failed to start radar detection: -110 (Connection timed out)
DFS start_dfs_cac() failed, -1
Interface initialization failed
"
Here is my workable hostapd.conf if someone needs:
"
 interface=<your_wlan>
driver=nl80211

### IEEE 802.11
ssid=<your_ssid>
hw_mode=a
channel=100
auth_algs=1

### DFS
ieee80211h=1
ieee80211d=1
country_code=AU

### IEEE 802.11n
ieee80211n=1
ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40]

### IEEE 802.11ac
ieee80211ac=1
vht_oper_chwidth=2
vht_oper_centr_freq_seg0_idx=114
vht_capab=[RXLDPC][SHORT-GI-80][SHORT-GI-160][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][VHT160-80PLUS80][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7]

### WPA/IEEE 802.11i
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_passphrase=<your_password>
wpa_pairwise=CCMP
"
Channels can be set in 36 and 50, then also works fine.

Ben I have only one question - why 160 Mhz mode work only under firmware-5-ct-non-commercial.bin ?

And Ben thank you so much for you work and your drivers, I very appreciate it, it helped me much.
I will continue testing my configuration under load and let you know if get any new issues.

Thank you,
Alex.

greearb, 2018-06-15:
I am not sure why just the non-commercial one works.  The special features that it enables will not even be turned on unless you have special driver configuration enabled.

Can you send me 'dmesg' output of the successful 160Mhz and of the failed attempt with one of the other firmware flavors?

master8282, 2018-06-15:
Hi Ben,

The requested dmesgs here:

1. for "http://www.candelatech.com/downloads/ath10k-9984-10-4/firmware-5-ct-non-commercial.bin"
[dmesg_non_com.txt](https://github.com/greearb/ath10k-ct/files/2106716/dmesg_non_com.txt)

2. for "http://www.candelatech.com/downloads/ath10k-9984-10-4/firmware-5-ct-full-community.bin"
[dmesg_com.txt](https://github.com/greearb/ath10k-ct/files/2106717/dmesg_com.txt)

master8282, 2018-06-16:
Hello Ben,

Next issue is very high ping to clients side (ping of the gateway from clients is ok).

root@ubuntu1804:~# ping 192.168.111.61
PING 192.168.111.61 (192.168.111.61) 56(84) bytes of data.
64 bytes from 192.168.111.61: icmp_seq=1 ttl=64 time=1.52 ms
64 bytes from 192.168.111.61: icmp_seq=9 ttl=64 time=988 ms
64 bytes from 192.168.111.61: icmp_seq=10 ttl=64 time=1.84 ms
64 bytes from 192.168.111.61: icmp_seq=11 ttl=64 time=4003 ms
64 bytes from 192.168.111.61: icmp_seq=12 ttl=64 time=2996 ms
64 bytes from 192.168.111.61: icmp_seq=13 ttl=64 time=1972 ms
64 bytes from 192.168.111.61: icmp_seq=14 ttl=64 time=948 ms
64 bytes from 192.168.111.61: icmp_seq=15 ttl=64 time=1.58 ms
64 bytes from 192.168.111.61: icmp_seq=16 ttl=64 time=3052 ms
64 bytes from 192.168.111.61: icmp_seq=17 ttl=64 time=2046 ms
64 bytes from 192.168.111.61: icmp_seq=18 ttl=64 time=1022 ms
64 bytes from 192.168.111.61: icmp_seq=19 ttl=64 time=0.960 ms

root@ubuntu1804:~# ping 192.168.111.54
PING 192.168.111.54 (192.168.111.54) 56(84) bytes of data.
64 bytes from 192.168.111.54: icmp_seq=1 ttl=64 time=3999 ms
64 bytes from 192.168.111.54: icmp_seq=2 ttl=64 time=2968 ms
64 bytes from 192.168.111.54: icmp_seq=3 ttl=64 time=1944 ms
64 bytes from 192.168.111.54: icmp_seq=4 ttl=64 time=920 ms
64 bytes from 192.168.111.54: icmp_seq=5 ttl=64 time=2.24 ms
64 bytes from 192.168.111.54: icmp_seq=30 ttl=64 time=1.94 ms
64 bytes from 192.168.111.54: icmp_seq=31 ttl=64 time=1229 ms
64 bytes from 192.168.111.54: icmp_seq=32 ttl=64 time=214 ms
^C
--- 192.168.111.54 ping statistics ---
34 packets transmitted, 8 received, 76% packet loss, time 33671ms

root@ubuntu1804:~# tracepath 192.168.111.61
 1?: [LOCALHOST]                      pmtu 1500
 1:  192.168.111.61                                        1.858ms reached
 1:  192.168.111.61                                        3.771ms reached
     Resume: pmtu 1500 hops 1 back 1

root@ubuntu1804:~# tcpdump -lei wlp2s0 icmp
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wlp2s0, link-type EN10MB (Ethernet), capture size 262144 bytes
13:03:29.704776 04:f0:21:38:98:ae (oui Unknown) > f4:5c:89:c4:7e:2b (oui Unknown), ethertype IPv4 (0x0800), length 98: 192.168.111.1 > 192.168.111.61: ICMP echo request, id 1991, seq 159, length 64
13:03:30.719582 04:f0:21:38:98:ae (oui Unknown) > f4:5c:89:c4:7e:2b (oui Unknown), ethertype IPv4 (0x0800), length 98: 192.168.111.1 > 192.168.111.61: ICMP echo request, id 1991, seq 160, length 64
13:03:31.743561 04:f0:21:38:98:ae (oui Unknown) > f4:5c:89:c4:7e:2b (oui Unknown), ethertype IPv4 (0x0800), length 98: 192.168.111.1 > 192.168.111.61: ICMP echo request, id 1991, seq 161, length 64
13:03:32.140034 f4:5c:89:c4:7e:2b (oui Unknown) > 04:f0:21:38:98:ae (oui Unknown), ethertype IPv4 (0x0800), length 98: 192.168.111.61 > 192.168.111.1: ICMP echo reply, id 1991, seq 159, length 64
13:03:32.140315 f4:5c:89:c4:7e:2b (oui Unknown) > 04:f0:21:38:98:ae (oui Unknown), ethertype IPv4 (0x0800), length 98: 192.168.111.61 > 192.168.111.1: ICMP echo reply, id 1991, seq 160, length 64
13:03:32.140438 f4:5c:89:c4:7e:2b (oui Unknown) > 04:f0:21:38:98:ae (oui Unknown), ethertype IPv4 (0x0800), length 98: 192.168.111.61 > 192.168.111.1: ICMP echo reply, id 1991, seq 161, length 64
13:03:32.744242 04:f0:21:38:98:ae (oui Unknown) > f4:5c:89:c4:7e:2b (oui Unknown), ethertype IPv4 (0x0800), length 98: 192.168.111.1 > 192.168.111.61: ICMP echo request, id 1991, seq 162, length 64
13:03:33.759559 04:f0:21:38:98:ae (oui Unknown) > f4:5c:89:c4:7e:2b (oui Unknown), ethertype IPv4 (0x0800), length 98: 192.168.111.1 > 192.168.111.61: ICMP echo request, id 1991, seq 163, length 64
13:03:33.759566 f4:5c:89:c4:7e:2b (oui Unknown) > 04:f0:21:38:98:ae (oui Unknown), ethertype IPv4 (0x0800), length 98: 192.168.111.61 > 192.168.111.1: ICMP echo reply, id 1991, seq 162, length 64
13:03:33.761992 f4:5c:89:c4:7e:2b (oui Unknown) > 04:f0:21:38:98:ae (oui Unknown), ethertype IPv4 (0x0800), length 98: 192.168.111.61 > 192.168.111.1: ICMP echo reply, id 1991, seq 163, length 64
13:03:34.761298 04:f0:21:38:98:ae (oui Unknown) > f4:5c:89:c4:7e:2b (oui Unknown), ethertype IPv4 (0x0800), length 98: 192.168.111.1 > 192.168.111.61: ICMP echo request, id 1991, seq 164, length 64
13:03:35.775549 04:f0:21:38:98:ae (oui Unknown) > f4:5c:89:c4:7e:2b (oui Unknown), ethertype IPv4 (0x0800), length 98: 192.168.111.1 > 192.168.111.61: ICMP echo request, id 1991, seq 165, length 64

root@ubuntu1804:~# ip r
default via 172.16.0.1 dev eno1
172.16.0.0/24 dev eno1 proto kernel scope link src 172.16.0.18
192.168.111.0/24 dev br0 proto kernel scope link src 192.168.111.1

root@ubuntu1804:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eno1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether c0:3f:d5:6f:aa:5f brd ff:ff:ff:ff:ff:ff
    inet 172.16.0.18/24 brd 172.16.0.255 scope global eno1
       valid_lft forever preferred_lft forever
    inet6 fe80::c23f:d5ff:fe6f:aa5f/64 scope link
       valid_lft forever preferred_lft forever
3: wlp2s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br0 state UP group default qlen 1000
[... cut ...]
    inet 192.168.111.1/24 brd 192.168.111.254 scope global br0
       valid_lft forever preferred_lft forever
    inet6 fe80::ccd0:9aff:fe0f:7e4/64 scope link
       valid_lft forever preferred_lft forever

root@ubuntu1804:~# brctl show
bridge name    	bridge id      		STP enabled    	interfaces
br0    		8000.04f0213898ae      	no     		wlp2s0

the outputs were taken when hostapd.conf has only necessary params for starting the daemon, no any additional tuning.
Also I tried many tunings but unfortunately no result and I still have the issue.

master8282, 2018-06-17:
Ben,
Today I tried default firmware-5.bin on 80Mhz and the worst ping was about 550ms,
ant@ubuntu1804:~$ ping 192.168.111.54
PING 192.168.111.54 (192.168.111.54) 56(84) bytes of data.
64 bytes from 192.168.111.54: icmp_seq=1 ttl=64 time=164 ms
64 bytes from 192.168.111.54: icmp_seq=2 ttl=64 time=186 ms
64 bytes from 192.168.111.54: icmp_seq=3 ttl=64 time=3.60 ms
64 bytes from 192.168.111.54: icmp_seq=4 ttl=64 time=27.5 ms
64 bytes from 192.168.111.54: icmp_seq=5 ttl=64 time=47.3 ms
64 bytes from 192.168.111.54: icmp_seq=6 ttl=64 time=72.3 ms
64 bytes from 192.168.111.54: icmp_seq=7 ttl=64 time=93.2 ms
64 bytes from 192.168.111.54: icmp_seq=8 ttl=64 time=114 ms
64 bytes from 192.168.111.54: icmp_seq=9 ttl=64 time=138 ms
64 bytes from 192.168.111.54: icmp_seq=10 ttl=64 time=161 ms
64 bytes from 192.168.111.54: icmp_seq=11 ttl=64 time=184 ms
64 bytes from 192.168.111.54: icmp_seq=12 ttl=64 time=207 ms
64 bytes from 192.168.111.54: icmp_seq=13 ttl=64 time=25.1 ms
64 bytes from 192.168.111.54: icmp_seq=14 ttl=64 time=46.8 ms
64 bytes from 192.168.111.54: icmp_seq=15 ttl=64 time=116 ms
64 bytes from 192.168.111.54: icmp_seq=16 ttl=64 time=2.25 ms
64 bytes from 192.168.111.54: icmp_seq=17 ttl=64 time=114 ms
64 bytes from 192.168.111.54: icmp_seq=18 ttl=64 time=341 ms
64 bytes from 192.168.111.54: icmp_seq=19 ttl=64 time=364 ms
64 bytes from 192.168.111.54: icmp_seq=20 ttl=64 time=386 ms
64 bytes from 192.168.111.54: icmp_seq=21 ttl=64 time=518 ms
64 bytes from 192.168.111.54: icmp_seq=22 ttl=64 time=25.1 ms
64 bytes from 192.168.111.54: icmp_seq=23 ttl=64 time=359 ms
64 bytes from 192.168.111.54: icmp_seq=24 ttl=64 time=378 ms
64 bytes from 192.168.111.54: icmp_seq=25 ttl=64 time=94.1 ms
64 bytes from 192.168.111.54: icmp_seq=26 ttl=64 time=420 ms
64 bytes from 192.168.111.54: icmp_seq=27 ttl=64 time=447 ms
64 bytes from 192.168.111.54: icmp_seq=28 ttl=64 time=469 ms
64 bytes from 192.168.111.54: icmp_seq=29 ttl=64 time=495 ms
64 bytes from 192.168.111.54: icmp_seq=30 ttl=64 time=516 ms
64 bytes from 192.168.111.54: icmp_seq=31 ttl=64 time=24.5 ms
64 bytes from 192.168.111.54: icmp_seq=32 ttl=64 time=355 ms
^C
--- 192.168.111.54 ping statistics ---
81 packets transmitted, 81 received, 0% packet loss, time 80098ms

After switched back to http://www.candelatech.com/downloads/ath10k-9984-10-4/firmware-5-ct-non-commercial.bin" and got up to 4500ms and lost

ant@ubuntu1804:~$ ping 192.168.111.54
64 bytes from 192.168.111.54: icmp_seq=6 ttl=64 time=62.5 ms
64 bytes from 192.168.111.54: icmp_seq=7 ttl=64 time=1.20 ms
64 bytes from 192.168.111.54: icmp_seq=8 ttl=64 time=2.03 ms
64 bytes from 192.168.111.54: icmp_seq=9 ttl=64 time=30.3 ms
64 bytes from 192.168.111.54: icmp_seq=10 ttl=64 time=1456 ms
64 bytes from 192.168.111.54: icmp_seq=11 ttl=64 time=437 ms
64 bytes from 192.168.111.54: icmp_seq=12 ttl=64 time=182 ms
64 bytes from 192.168.111.54: icmp_seq=13 ttl=64 time=2449 ms
64 bytes from 192.168.111.54: icmp_seq=14 ttl=64 time=1441 ms
64 bytes from 192.168.111.54: icmp_seq=15 ttl=64 time=418 ms
64 bytes from 192.168.111.54: icmp_seq=22 ttl=64 time=3460 ms
64 bytes from 192.168.111.54: icmp_seq=23 ttl=64 time=2437 ms
64 bytes from 192.168.111.54: icmp_seq=24 ttl=64 time=1413 ms
64 bytes from 192.168.111.54: icmp_seq=25 ttl=64 time=389 ms
64 bytes from 192.168.111.54: icmp_seq=26 ttl=64 time=22.9 ms
64 bytes from 192.168.111.54: icmp_seq=27 ttl=64 time=53.9 ms
64 bytes from 192.168.111.54: icmp_seq=34 ttl=64 time=2804 ms
64 bytes from 192.168.111.54: icmp_seq=35 ttl=64 time=1780 ms
64 bytes from 192.168.111.54: icmp_seq=36 ttl=64 time=756 ms
64 bytes from 192.168.111.54: icmp_seq=37 ttl=64 time=3.29 ms
64 bytes from 192.168.111.54: icmp_seq=41 ttl=64 time=4389 ms
64 bytes from 192.168.111.54: icmp_seq=42 ttl=64 time=3365 ms
64 bytes from 192.168.111.54: icmp_seq=43 ttl=64 time=2342 ms
64 bytes from 192.168.111.54: icmp_seq=44 ttl=64 time=1322 ms
64 bytes from 192.168.111.54: icmp_seq=45 ttl=64 time=298 ms
^C
--- 192.168.111.54 ping statistics ---
50 packets transmitted, 25 received, +3 errors, 50% packet loss, time 49822ms
rtt min/avg/max/mdev = 1.206/1252.958/4389.891/1279.999 ms, pipe 5

greearb, 2018-06-18:
Maybe try disabling power-save?

And, it sounds like this is different from the original bug you opened, so maybe you should open a new bug so the issue can be tracked properly?

master8282, 2018-06-18:
Hello Ben,
Yes, I agree about creating the separated bug regarding high ping. But tell me please, where, in which place you develop CT firmware, if I understand correct, the current place for discussing ath10k-ct driver but not firmware, I suppose here is wrong place.
Btw "disabling power-save" doesn't help, I tried all CT non-commercial firmwares since the earliest ver supported QCA9984 hardware, and all of then have same issue for me, stock firmware diesn't.

greearb, 2018-06-19:
The ath10k-ct github project tracks bugs for both the ath10k-ct driver and firmware.  Please open a new bug to track the ping time issue.

greearb, 2018-06-19:
With regard to the 'commercial' firmware having issues, the dmesg you attached has a kernel splat, but it seems to be missing a few lines that would tell why and where it crashed:

[   87.947073] IPv6: ADDRCONF(NETDEV_UP): wlp2s0: link is not ready

### Normally right here there would be some BUG or WARNING text.

 intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel pcbc ath10k_pci(OE) ath10k_core(OE) snd_hda_codec_realtek snd_hda_codec_hdmi snd_hda_codec_generic aesni_intel snd_hda_intel ath aes_x86_64 crypto_simd glue_helper cryptd mac80211 snd_hda_codec cfg80211 snd_hda_core intel_cstate snd_hwdep snd_pcm joydev nuvoton_cir intel_rapl_perf input_leds snd_timer rc_core mei_me mei shpchp lpc_ich mac_hid snd soundcore sch_fq_codel ip_tables x_tables autofs4 hid_logitech_hidpp i915 i2c_algo_bit drm_kms_helper hid_logitech_dj hid_generic syscopyarea sysfillrect sysimgblt ahci
[   88.276460]  usbhid libahci fb_sys_fops e1000e hid drm video
[   88.276476] CPU: 3 PID: 0 Comm: swapper/3 Tainted: G           OE    4.16.15 #1
[   88.276479] Hardware name:  /D34010WYB, BIOS WYLPT10H.86A.0026.2014.0514.1714 05/14/2014
[   88.276500] RIP: 0010:ath10k_wmi_event_vdev_start_resp+0x89/0xa0 [ath10k_core]
[   88.276502] RSP: 0018:ffff974c1fb83cd0 EFLAGS: 00010202
....

Can you reproduce this and see if you can capture the BUG or WARNING part of this splat?
Maybe your kernel needs to be compiled with verbose BUG support, or something like that?

master8282, 2018-06-19:
Sure, will try to do it tomorrow, on Fri

master8282, 2018-06-20:
Hi Ben,

I added "debug=8" in the kernel boot params and made new dumps, looks like they are more detailed now and will help you Ben.

[dmesg_dbg_full-com.txt](https://github.com/greearb/ath10k-ct/files/2120478/dmesg_dbg_full-com.txt)
[dmesg_dbg_noncom.txt](https://github.com/greearb/ath10k-ct/files/2120480/dmesg_dbg_noncom.txt)

Thank you,
Alex

greearb, 2018-09-28:
Sorry for the late response.  That dmesg_c0000032.txt has good debugging.  The kernel splat is because of a warning, vdev create didn't work properly for some reason.  The firmware logs appear to be missing some vdev create logs I'd expect to see.  I also fixed the driver to print out some better info in case the vdev create fails like it did in your case.  Can you re-test this with recent firmware and driver and let me know if the problem still exists?  Please generate logs how you did for the dmesg_c0000032.txt file.

greearb, 2018-12-18:
Closing this bug, OpenWRT has newer firmware and driver in it now, please re-open a new bug if problems persist.

closed 2018-12-18 by greearb