greearb/ath10k-ct #17: Compex WLE650V5-18A (QCA9888) not working - Firmware crashes

gojpdchx, 2018-02-07:
**PROBLEM DESCRIPTION**
My Compex WLE650V5-18A WiFi module does not work using Candela Technologies (CT) firmware.
From the log, it appears that the firmware crashes.
The stock firmware from https://github.com/kvalo/ath10k-firmware also crashes, too. [1]

**SOFTWARE**
**Operating System**: LEDE (Version: OpenWrt SNAPSHOT r5997-ecc347d [2])
**Kernel**: 4.9.77 (LEDE stock kernel)
**Firmware / Driver**: 
`board-2.bin`: Obtained from https://github.com/kvalo/ath10k-firmware/blob/master/QCA9888/hw2.0/board-2.bin . 
`firmware-5.bin`: Version 10.4-ct-9888-fW-009-2559479`(http://www.candelatech.com/downloads/ath10k-9888-10-4/firmware-5-ct-full-community.bin)
Also tested (However it still does not solve the problem):
ath10k CT 10.4.3 Beta 10: http://www.candelatech.com/downloads/ath10k-9888-10-4/ath10k-fw-beta/firmware-5-ct-full-community.bin

**HARDWARE**
**Board**: UniElec U7621-06
**WiFi Card 1 (Problematic with CT firmware)**: WLE650V5-18A ("Model: WLE650V5-18A 7A" is printed on the card, as shown in https://www.compex.com.sg/wp-content/uploads/2016/08/WLE650V5-18A_HEATSINK_TRANS_1-1500.png)
**WiFi Card 2 (Works fine with CT firmware 10.1-ct-8x-__xtW-019-395e9f1)**: Brand unknown, 168c:003c, lspci: `Qualcomm Atheros QCA986x/988x 802.11ac Wireless Network Adapter`

**LOGS**
`dmesg`:
```
[   13.382385] ath10k_pci 0000:02:00.0: pci irq legacy oper_irq_mode 1 irq_mode 0 reset_mode 0
[   13.580675] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/pre-cal-pci-0000:02:00.0.bin failed with error -2
[   13.591373] ath10k_pci 0000:02:00.0: Falling back to user helper
[   13.634860] firmware ath10k!pre-cal-pci-0000:02:00.0.bin: firmware_loading_store: map pages failed
[   13.644388] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/cal-pci-0000:02:00.0.bin failed with error -2
[   13.654749] ath10k_pci 0000:02:00.0: Falling back to user helper
[   13.725709] firmware ath10k!cal-pci-0000:02:00.0.bin: firmware_loading_store: map pages failed
[   13.736189] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/QCA9888/hw2.0/firmware-6.bin failed with error -2
[   13.746941] ath10k_pci 0000:02:00.0: Falling back to user helper
[   13.812923] firmware ath10k!QCA9888!hw2.0!firmware-6.bin: firmware_loading_store: map pages failed
[   14.480678] ath10k_pci 0000:02:00.0: Unknown FW IE: 30
[   14.485847] ath10k_pci 0000:02:00.0: qca9888 hw2.0 target 0x01000000 chip_id 0x00000000 sub 0000:0000
[   14.495066] ath10k_pci 0000:02:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 1
[   14.515364] ath10k_pci 0000:02:00.0: firmware ver 10.4-ct-9888-fW-009-2559479 api 5 features peer-flow-ctrl crc32 331b488d
[   16.605446] ath10k_pci 0000:02:00.0: board_file api 2 bmi_id 0:17 crc32 74a04250
[   20.056984] ath10k_pci 0000:02:00.0: firmware crashed! (guid 4c26a6ee-bac8-40f4-8c0f-fac426ec2c3d)
[   20.065976] ath10k_pci 0000:02:00.0: qca9888 hw2.0 target 0x01000000 chip_id 0x00000000 sub 0000:0000
[   20.075166] ath10k_pci 0000:02:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 1
[   20.095432] ath10k_pci 0000:02:00.0: firmware ver 10.4-ct-9888-fW-009-2559479 api 5 features peer-flow-ctrl crc32 331b488d
[   20.107940] ath10k_pci 0000:02:00.0: board_file api 2 bmi_id 0:17 crc32 74a04250
[   20.115321] ath10k_pci 0000:02:00.0: htt-ver 0.0 wmi-op 6 htt-op 4 cal otp max-sta 32 raw 0 hwcrypto 1
[   20.126664] ath10k_pci 0000:02:00.0: firmware register dump:
[   20.132307] ath10k_pci 0000:02:00.0: [00]: 0x01000000 0x000015B3 0x009B2221 0x00975B31
[   20.140205] ath10k_pci 0000:02:00.0: [04]: 0x009B2221 0x00060730 0x0000001E 0x0004A000
[   20.148102] ath10k_pci 0000:02:00.0: [08]: 0x00000000 0x00000050 0x0000000D 0x40000060
[   20.156001] ath10k_pci 0000:02:00.0: [12]: 0x00000009 0x00000000 0x009735D1 0x009735DF
[   20.163882] ath10k_pci 0000:02:00.0: [16]: 0x00973820 0x009606B8 0x00000000 0x00000000
[   20.171779] ath10k_pci 0000:02:00.0: [20]: 0x409B2221 0x0040604C 0xFFFEFFFF 0x0000000D
[   20.179678] ath10k_pci 0000:02:00.0: [24]: 0x809B226A 0x004060AC 0x00402130 0xC09B2221
[   20.187579] ath10k_pci 0000:02:00.0: [28]: 0x8098002B 0x004060CC 0x00448001 0x0000000D
[   20.195478] ath10k_pci 0000:02:00.0: [32]: 0x809607C1 0x004060EC 0xD8400100 0x004060FC
[   20.203358] ath10k_pci 0000:02:00.0: [36]: 0x80960892 0x0040610C 0x00000000 0x00000008
[   20.211255] ath10k_pci 0000:02:00.0: [40]: 0x80960E81 0x0040612C 0x00400000 0x00400600
[   20.219153] ath10k_pci 0000:02:00.0: [44]: 0x80960EE5 0x0040614C 0x00000000 0x00400600
[   20.227052] ath10k_pci 0000:02:00.0: [48]: 0x40960024 0x0040616C 0x00403830 0x00403830
[   20.234933] ath10k_pci 0000:02:00.0: [52]: 0x00000000 0x0040618C 0x00448001 0x00040020
[   20.242830] ath10k_pci 0000:02:00.0: [56]: 0x80960EE5 0x0040614C 0x00000000 0x00400600
[   20.250731] ath10k_pci 0000:02:00.0: Copy Engine register dump:
[   20.256650] ath10k_pci 0000:02:00.0: [00]: 0x0004a000  12  12  29  28
[   20.263066] ath10k_pci 0000:02:00.0: [01]: 0x0004a400   6   6   6   6
[   20.269498] ath10k_pci 0000:02:00.0: [02]: 0x0004a800   0   0   0   0
[   20.275933] ath10k_pci 0000:02:00.0: [03]: 0x0004ac00   0   0   0   0
[   20.282350] ath10k_pci 0000:02:00.0: [04]: 0x0004b000   0   0   0   0
[   20.288783] ath10k_pci 0000:02:00.0: [05]: 0x0004b400   0   0   0   0
[   20.295212] ath10k_pci 0000:02:00.0: [06]: 0x0004b800   0   0   0   0
[   20.301629] ath10k_pci 0000:02:00.0: [07]: 0x0004bc00   1   1   1   1
[   20.308061] ath10k_pci 0000:02:00.0: [08]: 0x0004c000   0   0   0   0
[   20.314478] ath10k_pci 0000:02:00.0: [09]: 0x0004c400   0   0   0   0
[... 27 lines cut ...]
[   23.553178] ath10k_pci 0000:03:00.0: Unknown FW IE: 30
[   23.558388] ath10k_pci 0000:03:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000
[   23.567610] ath10k_pci 0000:03:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 1
[   23.579825] ath10k_pci 0000:03:00.0: firmware ver 10.1-ct-8x-__xtW-019-395e9f1 api 2 features wmi-10.x,has-wmi-mgmt-tx crc32 7d9c7869
[   23.619610] ath10k_pci 0000:03:00.0: Direct firmware load for ath10k/QCA988X/hw2.0/board-2.bin failed with error -2
[   23.630034] ath10k_pci 0000:03:00.0: Falling back to user helper
[   23.670228] firmware ath10k!QCA988X!hw2.0!board-2.bin: firmware_loading_store: map pages failed
[   23.681505] ath10k_pci 0000:03:00.0: board_file api 1 bmi_id N/A crc32 bebc7c08
[   24.553454] ath10k_pci 0000:03:00.0: htt-ver 2.1 wmi-op 2 htt-op 2 cal otp max-sta 128 raw 0 hwcrypto 1
[   24.696850] ath: EEPROM regdomain: 0x6a
[   24.696861] ath: EEPROM indicates we should expect a direct regpair map
[   24.696882] ath: Country alpha2 being used: 00
```
`lspci`:
```
02:00.0 Network controller: Qualcomm Atheros Device 0056
03:00.0 Network controller: Qualcomm Atheros QCA986x/988x 802.11ac Wireless Network Adapter
```
```
02:00.0 0280: 168c:0056
03:00.0 0280: 168c:003c
```

[1] I have tested versions `10.4-3.5.1-00014`, `10.4-3.5.3-00053`, `10.4-3.2-00072`, `10.4-3.4-00104` from kvalo's repository, but none of them works. `dmesg` output is the same except the firmware version.
[2] Downloaded from https://downloads.lede-project.org/snapshots/targets/ramips/mt7621/openwrt-ramips-mt7621-u7621-06-256M-16M-squashfs-sysupgrade.bin . This links to the latest snapshot, which is usually updated daily. I downloaded on 2018/02/05, which corresponds to r5997-ecc347d.

greearb, 2018-02-07:
Please try the attached firmware, and send me the new crash trace if it still crashes.

Do you have a second NIC of the same kind that you could test against to see if it
is specific to one particular piece of hardware?

Can you use the 'ath10k-ct' driver in LEDE as well?  That might provide more
debugging output for me to diagnose.

Thanks,
Ben

greearb, 2018-02-07:
Here is the file I tried to attach to my last comment:

[firmware-5-full-community.bin.gz](https://github.com/greearb/ath10k-ct/files/1704321/firmware-5-full-community.bin.gz)

gojpdchx, 2018-02-08:
@greearb 
> Please try the attached firmware

It still crashes. Please see Attachment 2 for the new crash trace.

> Can you use the 'ath10k-ct' driver in LEDE as well?

I removed `kmod-ath10k` and installed `kmod-ath10k-ct`, but then "unknown symbol" errors appeared and the module could not be loaded. (Please refer to Attachment 3.) Am I supposed to install both of them?

> Do you have a second NIC of the same kind that you could test against to see if it is specific to one particular piece of hardware?

Unfortunately I only have this card. All other cards I have are of different chipsets and vendors.

Attachment 1:
`20180208T013000.tar.xz`: Contains the `board-2.bin` and `firmware-5.bin` in use after you have attached the firmware for debugging.
[20180208T015000.tar.gz](https://github.com/greearb/ath10k-ct/files/1705410/20180208T015000.tar.gz)

Attachment 2:
`dmesg_kmod-ath10k_crash_trace.txt`: The new crash trace.
[dmesg_kmod-ath10k_crash_trace.txt](https://github.com/greearb/ath10k-ct/files/1705411/dmesg_kmod-ath10k_crash_trace.txt)

Attachment 3:
`dmesg_kmod-ath10k-ct.txt`: "Unknown symbol" errors as described above.
[dmesg_kmod-ath10k-ct.txt](https://github.com/greearb/ath10k-ct/files/1705412/dmesg_kmod-ath10k-ct.txt)

greearb, 2018-02-08:
It appears that the cpu core on your NIC is running at some un-supported rate.  Please try the attached firmware, but you must be running ath10k-ct driver in order for the extra debug to print out.  To load the driver and get around the module-symbols issue, manually load mac80211 first:

modprobe mac80211
modprobe ath10k_pci
[firmware-5-full-community.bin.gz](https://github.com/greearb/ath10k-ct/files/1708338/firmware-5-full-community.bin.gz)

Where did you get this NIC from?  Perhaps it just has busted hardware somehow?  Can you get another one to test with?

gojpdchx, 2018-02-09:
@greearb 
> To load the driver and get around the module-symbols issue, manually load mac80211 first

Unfortunately I still could not get the ath10k-ct driver to work. I tried loading mac80211 first then ath10k_pci, but I still have the "Unknown symbol" errors.

I have asked for help on this issue by posting on the LEDE forum. Once I successfully load the ath10k-ct driver, I will post the updated debug information here.

> Where did you get this NIC from?

I bought the NIC on a Far East online marketplace (for nearly USD 40). The seller has been in excellent rating and the NIC I received appeared to be new. There was no sign of use / wear on the mini PCI Express pins.

> Can you get another one to test with?

Currently I have no plan to buy another one because it can be difficult to return the product. I am not living in that country (where that marketplace is based), so the (return) shipping fee is high (Shipping to my country is cheap, shipping *from* my country is expensive) and I have to go through the customs procedures.

Using kmod-ath10k and the latest firmware you provided (I will comment again if I can get ath10k-ct to work):
```
[   13.391780] ath10k_pci 0000:02:00.0: pci irq legacy oper_irq_mode 1 irq_mode 0 reset_mode 0
[   13.587699] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/pre-cal-pci-0000:02:00.0.bin failed with error -2
[   13.598403] ath10k_pci 0000:02:00.0: Falling back to user helper
[   13.640890] firmware ath10k!pre-cal-pci-0000:02:00.0.bin: firmware_loading_store: map pages failed
[   13.650322] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/cal-pci-0000:02:00.0.bin failed with error -2
[   13.660680] ath10k_pci 0000:02:00.0: Falling back to user helper
[   13.731398] firmware ath10k!cal-pci-0000:02:00.0.bin: firmware_loading_store: map pages failed
[   13.741450] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/QCA9888/hw2.0/firmware-6.bin failed with error -2
[   13.752210] ath10k_pci 0000:02:00.0: Falling back to user helper
[   13.818400] firmware ath10k!QCA9888!hw2.0!firmware-6.bin: firmware_loading_store: map pages failed
[   14.568363] ath10k_pci 0000:02:00.0: Unknown FW IE: 30
[   14.573512] ath10k_pci 0000:02:00.0: qca9888 hw2.0 target 0x01000000 chip_id 0x00000000 sub 0000:0000
[   14.582708] ath10k_pci 0000:02:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 1
[   14.603022] ath10k_pci 0000:02:00.0: firmware ver 10.4-ct-9888-fW-010-1d078e0 api 5 features peer-flow-ctrl crc32 3ff7686c
[   16.766138] ath10k_pci 0000:02:00.0: board_file api 2 bmi_id 0:17 crc32 74a04250
[   20.220828] ath10k_pci 0000:02:00.0: firmware crashed! (guid abe7b6a8-d411-40b4-8e54-3e16cf699788)
[   20.229825] ath10k_pci 0000:02:00.0: qca9888 hw2.0 target 0x01000000 chip_id 0x00000000 sub 0000:0000
[   20.239029] ath10k_pci 0000:02:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 1
[   20.259360] ath10k_pci 0000:02:00.0: firmware ver 10.4-ct-9888-fW-010-1d078e0 api 5 features peer-flow-ctrl crc32 3ff7686c
[   20.271903] ath10k_pci 0000:02:00.0: board_file api 2 bmi_id 0:17 crc32 74a04250
[   20.279289] ath10k_pci 0000:02:00.0: htt-ver 0.0 wmi-op 6 htt-op 4 cal otp max-sta 32 raw 0 hwcrypto 1
[   20.290617] ath10k_pci 0000:02:00.0: firmware register dump:
[   20.296284] ath10k_pci 0000:02:00.0: [00]: 0x01000000 0x000015B3 0x009B24A8 0x00975B31
[   20.304166] ath10k_pci 0000:02:00.0: [04]: 0x009B24A8 0x00060730 0x0000001E 0x00000004
[   20.312063] ath10k_pci 0000:02:00.0: [08]: 0x00000000 0x00080200 0x0000000D 0x0000000D
[   20.319957] ath10k_pci 0000:02:00.0: [12]: 0x00000009 0x00000000 0x009735D1 0x009735DF
[   20.327862] ath10k_pci 0000:02:00.0: [16]: 0x00973820 0x009606CA 0x00000000 0x00000000
[   20.335762] ath10k_pci 0000:02:00.0: [20]: 0x409B24A8 0x0040603C 0xFFFEFFFF 0x0000000D
[   20.343644] ath10k_pci 0000:02:00.0: [24]: 0x809B24F2 0x0040609C 0x00000001 0xC09B24A8
[   20.351541] ath10k_pci 0000:02:00.0: [28]: 0x8098002B 0x004060CC 0x00448001 0x0000000D
[   20.359442] ath10k_pci 0000:02:00.0: [32]: 0x809607C1 0x004060EC 0xD8400100 0x004060FC
[   20.367342] ath10k_pci 0000:02:00.0: [36]: 0x80960892 0x0040610C 0x00000000 0x00000008
[   20.375224] ath10k_pci 0000:02:00.0: [40]: 0x80960E81 0x0040612C 0x00400000 0x00400600
[   20.383121] ath10k_pci 0000:02:00.0: [44]: 0x80960EE5 0x0040614C 0x00000000 0x00400600
[   20.391027] ath10k_pci 0000:02:00.0: [48]: 0x40960024 0x0040616C 0x00403830 0x00403830
[   20.398922] ath10k_pci 0000:02:00.0: [52]: 0x00000000 0x0040618C 0x00448001 0x00040020
[   20.406821] ath10k_pci 0000:02:00.0: [56]: 0x00404FB8 0x00400000 0x000015B8 0x000013B8
[   20.414699] ath10k_pci 0000:02:00.0: Copy Engine register dump:
[   20.420622] ath10k_pci 0000:02:00.0: [00]: 0x0004a000   1   1   2   1
[   20.427053] ath10k_pci 0000:02:00.0: [01]: 0x0004a400   6   6   5   6
[   20.433469] ath10k_pci 0000:02:00.0: [02]: 0x0004a800   0   0  63   0
[   20.439902] ath10k_pci 0000:02:00.0: [03]: 0x0004ac00   0   0   0   0
[   20.446344] ath10k_pci 0000:02:00.0: [04]: 0x0004b000   0   0   0   0
[   20.452761] ath10k_pci 0000:02:00.0: [05]: 0x0004b400   0   0   0   0
[   20.459195] ath10k_pci 0000:02:00.0: [06]: 0x0004b800   0   0   0   0
[   20.465630] ath10k_pci 0000:02:00.0: [07]: 0x0004bc00   1   1   1   1
[   20.472047] ath10k_pci 0000:02:00.0: [08]: 0x0004c000   0   0 127   0
[   20.478488] ath10k_pci 0000:02:00.0: [09]: 0x0004c400   0   0   0   0
[   20.484905] ath10k_pci 0000:02:00.0: [10]: 0x0004c800   0   0   0   0
[... cut ...]
[   23.477086] ath10k_pci 0000:03:00.0: Falling back to user helper
[   23.517550] firmware ath10k!QCA988X!hw2.0!board-2.bin: firmware_loading_store: map pages failed
[   23.528620] ath10k_pci 0000:03:00.0: board_file api 1 bmi_id N/A crc32 bebc7c08
[   24.436214] ath10k_pci 0000:03:00.0: htt-ver 2.1 wmi-op 2 htt-op 2 cal otp max-sta 128 raw 0 hwcrypto 1
[   24.576977] ath: EEPROM regdomain: 0x6a
[   24.576987] ath: EEPROM indicates we should expect a direct regpair map
[   24.577008] ath: Country alpha2 being used: 00
```

gojpdchx, 2018-02-11:
@greearb 
It seems that I can now modprobe ath10k-ct (4.9.77+2017-06-13-bde..3-2, shipped with LEDE). 

Using kmod-ath10k:
```
[   76.933634] ath10k 4.13 driver, optimized for CT firmware, probing pci device: 0x56.
[   76.942785] ath10k_pci 0000:02:00.0: pci irq legacy oper_irq_mode 1 irq_mode 0 reset_mode 0
[   77.138685] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/fwcfg-pci-0000:02:00.0.txt failed with error -2
[   77.149192] ath10k_pci 0000:02:00.0: Falling back to user helper
[   77.190845] firmware ath10k!fwcfg-pci-0000:02:00.0.txt: firmware_loading_store: map pages failed
[   77.200151] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/pre-cal-pci-0000:02:00.0.bin failed with error -2
[   77.210870] ath10k_pci 0000:02:00.0: Falling back to user helper
[   77.281925] firmware ath10k!pre-cal-pci-0000:02:00.0.bin: firmware_loading_store: map pages failed
[   77.291486] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/cal-pci-0000:02:00.0.bin failed with error -2
[   77.301900] ath10k_pci 0000:02:00.0: Falling back to user helper
[   77.372659] firmware ath10k!cal-pci-0000:02:00.0.bin: firmware_loading_store: map pages failed
[   77.382819] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/QCA9888/hw2.0/firmware-6.bin failed with error -2
[   77.393561] ath10k_pci 0000:02:00.0: Falling back to user helper
[   77.459665] firmware ath10k!QCA9888!hw2.0!firmware-6.bin: firmware_loading_store: map pages failed
[   77.888343] ath10k_pci 0000:02:00.0: qca9888 hw2.0 target 0x01000000 chip_id 0x00000000 sub 0000:0000
[   77.897563] ath10k_pci 0000:02:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   77.917927] ath10k_pci 0000:02:00.0: firmware ver 10.4-ct-9888-fW-010-1d078e0 api 5 features peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT crc32 3ff7686c
[   80.015864] ath10k_pci 0000:02:00.0: board_file api 2 bmi_id 0:17 crc32 74a04250
[   83.471110] ath10k_pci 0000:02:00.0: firmware crashed! (uuid fd1399cc-4b0c-42d3-ab72-c6708f54810c)
[   83.480117] ath10k_pci 0000:02:00.0: qca9888 hw2.0 target 0x01000000 chip_id 0x00000000 sub 0000:0000
[   83.489390] ath10k_pci 0000:02:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   83.509881] ath10k_pci 0000:02:00.0: firmware ver 10.4-ct-9888-fW-010-1d078e0 api 5 features peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT crc32 3ff7686c
[   83.532618] ath10k_pci 0000:02:00.0: board_file api 2 bmi_id 0:17 crc32 74a04250
[   83.540088] ath10k_pci 0000:02:00.0: htt-ver 0.0 wmi-op 6 htt-op 4 cal otp max-sta 32 raw 0 hwcrypto 1
[   83.551478] ath10k_pci 0000:02:00.0: firmware register dump:
[   83.557123] ath10k_pci 0000:02:00.0: [00]: 0x01000000 0x000015B3 0x009B24A8 0x00975B31
[   83.565088] ath10k_pci 0000:02:00.0: [04]: 0x009B24A8 0x00060730 0x0000001E 0x00000004
[   83.573053] ath10k_pci 0000:02:00.0: [08]: 0x00000000 0x00080200 0x0000000D 0x0000000D
[   83.580993] ath10k_pci 0000:02:00.0: [12]: 0x00000009 0x00000000 0x009735D1 0x009735DF
[   83.588946] ath10k_pci 0000:02:00.0: [16]: 0x00973820 0x00960EA7 0x00000000 0x00000000
[   83.596831] ath10k_pci 0000:02:00.0: [20]: 0x409B24A8 0x0040603C 0xFFFEFFFF 0x0000000D
[   83.604783] ath10k_pci 0000:02:00.0: [24]: 0x809B24F2 0x0040609C 0x00000001 0xC09B24A8
[   83.612741] ath10k_pci 0000:02:00.0: [28]: 0x8098002B 0x004060CC 0x00448001 0x0000000D
[   83.620679] ath10k_pci 0000:02:00.0: [32]: 0x809607C1 0x004060EC 0xD8400100 0x004060FC
[   83.628640] ath10k_pci 0000:02:00.0: [36]: 0x80960892 0x0040610C 0x00000000 0x00000008
[   83.636530] ath10k_pci 0000:02:00.0: [40]: 0x80960E81 0x0040612C 0x00400000 0x00400600
[   83.644486] ath10k_pci 0000:02:00.0: [44]: 0x80960EE5 0x0040614C 0x00000000 0x00400600
[   83.652445] ath10k_pci 0000:02:00.0: [48]: 0x40960024 0x0040616C 0x00403830 0x00403830
[   83.660392] ath10k_pci 0000:02:00.0: [52]: 0x00000000 0x0040618C 0x00448001 0x00040020
[   83.668289] ath10k_pci 0000:02:00.0: [56]: 0x00404FB8 0x00400000 0x000015B8 0x000013B8
[   83.676246] ath10k_pci 0000:02:00.0: Copy Engine register dump:
[   83.682211] ath10k_pci 0000:02:00.0: [00]: 0x0004a000   2   2   3   2
[   83.688689] ath10k_pci 0000:02:00.0: [01]: 0x0004a400   6   6   6   6
[   83.695120] ath10k_pci 0000:02:00.0: [02]: 0x0004a800   0   0   0   0
[   83.701608] ath10k_pci 0000:02:00.0: [03]: 0x0004ac00   0   0   0   0
[   83.708048] ath10k_pci 0000:02:00.0: [04]: 0x0004b000   0   0   0   0
[   83.714532] ath10k_pci 0000:02:00.0: [05]: 0x0004b400   0   0   0   0
[   83.721017] ath10k_pci 0000:02:00.0: [06]: 0x0004b800   0   0   0   0
[   83.727445] ath10k_pci 0000:02:00.0: [07]: 0x0004bc00   1   1   1   1
[   83.733940] ath10k_pci 0000:02:00.0: [08]: 0x0004c000   0   0   0   0
[   83.740417] ath10k_pci 0000:02:00.0: [09]: 0x0004c400   0   0   0   0
[   83.746841] ath10k_pci 0000:02:00.0: [10]: 0x0004c800   0   0   0   0
[   83.753336] ath10k_pci 0000:02:00.0: [11]: 0x0004cc00   0   0   0   0
[   83.760855] ath10k_pci 0000:02:00.0: debug log header, dbuf: 0x96068c  dropped: 4197376
[   83.769938] ath10k_pci 0000:02:00.0: [0] next: 0xe0bbe59d buf: 0x7ec27190 sz: -2011895919 len: 172907295 count: -1208253521 free: -1865204219
[   83.782657] ath10k_pci 0000:02:00.0: debuglog buf length is out of bounds: 172907295
[... cut ...]
[   85.991860] ath10k_pci 0000:02:00.0: could not init core (-145)
[   85.997950] ath10k_pci 0000:02:00.0: could not probe fw (-145), releasing napi
```

This is all the driver produces in `dmesg`. Afterwards, it seems that the driver hangs. Renaming the QCA9888 firmware file from .bin to .bak will not solve the hang. My SparkLAN NIC card, having a larger slot number, is not probed by the ath10k-ct.
When switched back to non-ct driver, after failing to probe the Compex card, the SparkLAN card can still be probed normally.

greearb, 2018-02-11:
I guess the firmware crashes before it can set up the dbglog
buffer pointer, so the driver cannot read the crash log.

As for lockup, not sure.  I have fixed some bugs recently, but those
patches are not in LEDE currently.

I'll see if I can think of a way to get more debugging next week.

Thanks,
Ben

greearb, 2018-02-12:
Please try this firmware and send me the full dmesg log of it booting (or crashing), with the ath10k-ct driver.

[firmware-5-full-community.bin.gz](https://github.com/greearb/ath10k-ct/files/1717748/firmware-5-full-community.bin.gz)

Thanks,
Ben

gojpdchx, 2018-02-13:
`dmesg` after booting:
[dmesg_1.txt](https://github.com/greearb/ath10k-ct/files/1718540/dmesg_1.txt)

Sometimes the following appears in `dmesg` after doing some wireless operations, such as scanning networks or setting up an AP:
[dmesg_2.txt](https://github.com/greearb/ath10k-ct/files/1718542/dmesg_2.txt)

After switching to the latest firmware you provided, I rebooted the device several times for testing. I discovered that:
1. There is no change in `lspci` output - the Compex NIC is still being recognized as `Qualcomm Atheros Device 0056`.
2. `iw list` shows the correct information for the Compex NIC.
3. For SparkLAN NIC: occasionally the `ath10k-ct` driver will fail to probe it, but when probed successfully, it works fine. I will continue to observe the situation.
4. In LEDE LuCI web interface, the Compex NIC is recognized as `Generic MAC80211 802.11nac`.
5. Although I can configure an AP network (in LEDE) for the Compex NIC, it cannot be scanned (AP mode), nor can it scan any network (station mode). _(I remembered that I have connected the antennas firmly...)_

greearb, 2018-02-13:
Please try the attached firmware and send me new dmesg output.

[firmware-5-full-community.bin.gz](https://github.com/greearb/ath10k-ct/files/1720879/firmware-5-full-community.bin.gz)

erikarn, 2018-02-13:
Hi,

Would you please contact compex and see which firmware / driver they've tested it with?

It sounds like something in the board data / calibration data isn't lining up with what the firmware expects and it's handling it badly.

gojpdchx, 2018-02-14:
@greearb 
`dmesg` output, Compex NIC still not working, SparkLAN NIC works fine:
[dmesg.txt](https://github.com/greearb/ath10k-ct/files/1722366/dmesg.txt)

`dmesg` output, Compex NIC still not working, SparkLAN NIC only works when I changed the channel in the LEDE wireless configuration [1]:
[dmesg_2.txt](https://github.com/greearb/ath10k-ct/files/1722370/dmesg_2.txt)

[1] In this case, SparkLAN NIC is probed successfully but is not functioning as an AP (despite having proper configuration). The channel I selected actually works, but I have to first switch to another channel then switch back to make it work. So far, I only have this problem when using the ath10k-ct driver.

@erikarn 
I will try, but please note that I did not bought the device from Compex's store but from an online marketplace, so it can be possible that they will ignore my inquiry.

gojpdchx, 2018-02-14:
@erikarn 
I found https://www.compex.com.sg/forum/t/4 and tried the `board-2.bin` Compex provided in the thread. With kvalo's `10.4-3.5.1-00041` and `10.4-3.5.3-00053`, the same problem as described in the first post (firmware crash) still exists.

It is weird that with the original `board-2.bin` (and greearb's latest firmware for testing), at least `iw list` shows the correct information for the Compex NIC.

greearb, 2018-02-15:
I do not see any obvious bugs in the dmesg output for the compex NIC.  I am guessing that the CPU freq related assert is a symptom of some other problem, maybe with the calibration data for this particular NIC, or maybe the hardware is just broken.  If I hear any new info, I'll post it here, but I don't think there is much more I can do to help until I hear back from Compex.

gojpdchx, 2018-02-15:
@greearb 
May I know if you have contacted Compex regarding this matter?
If so, I am willing to email you my NIC's MAC address and S/N to facilitate their investigation.
_(According to the announcement on www.compex.com.sg , their office in Singapore is closed from 2018/02/15 to 2018/02/20.)_

P.S. I double-checked to make sure that the antennas are properly connected.

vuori, 2018-02-21:
I saw this issue on the Compex forum. I also have a recently purchased WLE650V5 that required the new  `board-2.bin` provided by Compex to even probe, since it uses `bmi-board-id` 30. I have essentially the same problems as described here with a vanilla 4.9.78 kernel on x86_64 Debian.

It seems likely that the innards of the WLE650V5 have been renewed while keeping the same model number, and that the calibration data provided by Compex is not the correct one. I'll try poking Compex a bit more later this week.

gojpdchx, 2018-02-22:
@vuori 
It turns out that my `bmi-board-id` is 17, which is an old one. Compex says that all modules meant for production now have the id 30.
I have provided my purchase details and MAC address via Private Message to the administrator on the Compex forum. He says that I will have to wait until next week because of the Lunar New Year.

greearb, 2018-02-27:
I also heard from Compex that the board-id 17 is a prototype batch.  Unless there is a functional board-data for it somehwere, then probably nothing I can do to make this NIC work, and possibly even then it would take more than a board-data file.

gojpdchx, 2018-03-01:
@greearb 
It seems that my purchase is regarded as legitimate and they are working on it, though I do not know how long it will take.
I will continue to pressure Compex to give a fix (or release the correct board data file). I will update you if I have any new information.

gojpdchx, 2018-03-02:
@greearb @erikarn 
Compex now has an updated board-2.bin at [https://downloads.compex.com.sg/?dir=uploads/boardData/For_Linux_PC_Unofficial](https://downloads.compex.com.sg/?dir=uploads/boardData/For_Linux_PC_Unofficial). They claim that this new boardData contains the support for board-id 17.

With that new board-2.bin and stock (i.e. non-CT) firmware, the result is the same: the firmware crashes.
With the latest debugging firmware provided by greearb, although the firmware can be probed successfully, there is no wireless signal (i.e. Client mode: unable to scan any network / AP mode: Clients cannot find the AP). In other words, the situation is the same.

I have attached two dmesg logs, the first one is ath10k + 10.4-3.5.1-00014, the second one is ath10k + 10.4-3.5.3-00053. If needed, I can also switch to ath10k-ct and provide more logs.

[dmesg_stock_10.4-3.5.1-00014.txt](https://github.com/greearb/ath10k-ct/files/1774855/dmesg_stock_10.4-3.5.1-00014.txt)
[dmesg_10.4-3.5.3-00053.txt](https://github.com/greearb/ath10k-ct/files/1774856/dmesg_10.4-3.5.3-00053.txt)

greearb, 2018-03-02:
Maybe see if they can suggest a particular driver/firmware that they know works with this NIC?

gojpdchx, 2018-03-06:
@greearb 
This is what I asked Compex:
> Could you please tell me which firmware-5.bin version has been tested with the board-id 17 version of WLE650V5-18A?

And this is how Compex replied:
> Our Engineering team had performed some testing.
We found that LEDE firmware does not support the WLE650V5-18A / WLE650V5-25A / QCA9888.

greearb, 2018-03-06:
Well, we know lots of things that *don't* work, ask them what firmware it *does* work with.  If they cannot provide any useful example, then maybe it doesn't actually work with anything.

gojpdchx, 2018-03-07:
@greearb 
According to Compex, they find the word working by using the Candela Tech firmware (which, according to the log, is of version 10.4-ct-9888-fW-009-2559479) on Ubuntu 16.04.1 with kernel 4.13.0.
They claim that kernel version 4.9.77 does not support this card. I will try compiling a LEDE 4.14 kernel if I can to see if it can solve the problem.

Meanwhile, I switched to the same version of CT firmware-5.bin (10.4-ct-9888-fW-009-2559479), using the latest board-2.bin provided by Compex, and ath10k (non-CT) driver. The firmware still crashes but I noticed that the register dumps seems to be different (than the ones with old boardData).

[dmesg_10.4-ct-9888-fW-009-2559479.txt](https://github.com/greearb/ath10k-ct/files/1788863/dmesg_10.4-ct-9888-fW-009-2559479.txt)

gojpdchx, 2018-03-09:
@greearb 
I compiled a LEDE firmware with kernel 4.14, using ath10k-ct driver and the latest board-2.bin by Compex. The firmware-5.bin file is the one your provided in [a previous comment](https://github.com/greearb/ath10k-ct/issues/17#issuecomment-365339168).

Still, it does not work. The situation is the same, i.e. if I use 10.4-ct-9888-fW-009-2559479, it crashes. If I use your firmware-5.bin, the firmware does not crash but it cannot scan any network / create an AP. The following appears in `dmesg` every time when I trigger a network scan using the Compex WLE650V5-18A card:

[4.14_dmesg_10.4-ct-9888-fW-010-73d5e8f.txt](https://github.com/greearb/ath10k-ct/files/1796092/4.14_dmesg_10.4-ct-9888-fW-010-73d5e8f.txt)

Could it be that my card has hardware problem?

greearb, 2018-03-26:
The dmesg logs from the firmware just show it changing channel, which is normal as part of scanning.  If it receives no responses, then I guess it must have some bug, possibly a hardware issue.  If you sniff, do you see probe requests go out?  Either way, not sure there is much I can do.

gojpdchx, 2018-04-10:
@greearb 
I cannot see any probe request going out when I sniff - I am guessing that there may be problems with the hardware.

closed 2018-04-10 by gojpdchx