greearb/ath10k-ct #18: In some rare RF conditions, no traffic on IBSS connections

oamouroux, 2018-03-14:
In specific field only RF conditions, some WPA2 IBSS links can be established but no traffic is received

OpenWRT Chaos Calmer, Kernel 3.18.45, mac80211 backported from v4.4-rc5-1913-gc8fdf68, driver ath10k-ct 4.7 2f3171d, fW-020-64bb715
This is also reproducible on ath10k-ct 4.7 1df6b26, fw 18

Hardware (NIC chipset, platform, etc)
QCA9882

Logs (dmesg, maybe supplicant and/or hostap)
[dmesg.txt](https://github.com/greearb/ath10k-ct/files/1812458/dmesg.txt)

greearb, 2018-03-16:
I think this might be a supplicant bug.  I saw this message in other testing scenarios:

[32m[349951.140828] [0m[33msta5002[0m: 04:f0:21:67:53:95 authenticate with f8:a0:97:44:94:55 at: 1520282299.350687
[32m[349951.141350] [0m[33msta5002[0m: failed to insert STA entry for the AP (error -17)

The supplicant does not know how to work around the EEXIST errno (-17).

Please manually hack in the check for EEXIST.  This patch will not apply to upstream supplicant, and I have not re-spun it quite yet, but I think it will be enough for you to make the change:

https://www.spinics.net/lists/hostap/msg04278.html

greearb, 2018-03-26:
Have you been able to test with supplicant patched as suggested?

oamouroux, 2018-05-24:
I added the suggested patch. I'm now using latest 4.9 driver, mac80211 backported wt-2017-01-31-0-ge882dff19e7 (same as LEDE-17.01), wpa_supplicant 2016-12-19 (same as LEDE-17.01) and it still does not work. Here is the log:

[   11.740000] ath10k 4.9 driver, optimized for CT firmware, probing pci device: 0x3c.
[   11.760000] ath10k_pci 0000:01:00.0: pci irq legacy oper_irq_mode 1 irq_mode 0 reset_mode 0
[   11.980000] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/pre-cal-pci-0000:01:00.0.bin failed with error -2
[   11.990000] ath10k_pci 0000:01:00.0: Falling back to user helper
[   12.050000] firmware ath10k!pre-cal-pci-0000:01:00.0.bin: firmware_loading_store: map pages failed
[   12.060000] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/cal-pci-0000:01:00.0.bin failed with error -2
[   12.070000] ath10k_pci 0000:01:00.0: Falling back to user helper
[   12.130000] firmware ath10k!cal-pci-0000:01:00.0.bin: firmware_loading_store: map pages failed
[   12.140000] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/fwcfg-pci-0000:01:00.0.txt failed with error -2
[   12.150000] ath10k_pci 0000:01:00.0: Falling back to user helper
[   12.220000] firmware ath10k!fwcfg-pci-0000:01:00.0.txt: firmware_loading_store: map pages failed
[   12.220000] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/QCA988X/hw2.0/ct-firmware-5.bin failed with error -2
[   12.240000] ath10k_pci 0000:01:00.0: Falling back to user helper
[   12.300000] firmware ath10k!QCA988X!hw2.0!ct-firmware-5.bin: firmware_loading_store: map pages failed
[   12.310000] ath10k_pci 0000:01:00.0: could not fetch firmware file 'ath10k/QCA988X/hw2.0/ct-firmware-5.bin': -11
[   12.320000] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/QCA988X/hw2.0/ct-firmware-2.bin failed with error -2
[   12.330000] ath10k_pci 0000:01:00.0: Falling back to user helper
[   12.390000] firmware ath10k!QCA988X!hw2.0!ct-firmware-2.bin: firmware_loading_store: map pages failed
[   12.400000] ath10k_pci 0000:01:00.0: could not fetch firmware file 'ath10k/QCA988X/hw2.0/ct-firmware-2.bin': -11
[   12.410000] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/QCA988X/hw2.0/firmware-5.bin failed with error -2
[   12.420000] ath10k_pci 0000:01:00.0: Falling back to user helper
[   12.490000] firmware ath10k!QCA988X!hw2.0!firmware-5.bin: firmware_loading_store: map pages failed
[   12.490000] ath10k_pci 0000:01:00.0: could not fetch firmware file 'ath10k/QCA988X/hw2.0/firmware-5.bin': -11
[   12.510000] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/QCA988X/hw2.0/firmware-4.bin failed with error -2
[   12.520000] ath10k_pci 0000:01:00.0: Falling back to user helper
[   12.580000] firmware ath10k!QCA988X!hw2.0!firmware-4.bin: firmware_loading_store: map pages failed
[   12.590000] ath10k_pci 0000:01:00.0: could not fetch firmware file 'ath10k/QCA988X/hw2.0/firmware-4.bin': -11
[   12.600000] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/QCA988X/hw2.0/firmware-3.bin failed with error -2
[   12.610000] ath10k_pci 0000:01:00.0: Falling back to user helper
[   12.670000] firmware ath10k!QCA988X!hw2.0!firmware-3.bin: firmware_loading_store: map pages failed
[   12.680000] ath10k_pci 0000:01:00.0: could not fetch firmware file 'ath10k/QCA988X/hw2.0/firmware-3.bin': -11
[   12.930000] ath10k_pci 0000:01:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043222ff sub 0000:0000
[   12.940000] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 0 testmode 0
[   12.950000] ath10k_pci 0000:01:00.0: firmware ver 10.1-ct-8x-__fW-021-75f739c api 2 features wmi-10.x,has-wmi-mgmt-tx,txstatus-noack,wmi-10.x-CT,ratemask-CT,get-temp-CT,tx-rc-CT,cust-stats-CT crc32 6c2a63ef
[   13.000000] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/QCA988X/hw2.0/board-2.bin failed with error -2
[   13.010000] ath10k_pci 0000:01:00.0: Falling back to user helper
[   13.080000] firmware ath10k!QCA988X!hw2.0!board-2.bin: firmware_loading_store: map pages failed
[   13.090000] ath10k_pci 0000:01:00.0: board_file api 1 bmi_id N/A crc32 bebc7c08
[   14.000000] ath10k_pci 0000:01:00.0: 10.1 wmi init: vdevs: 16  peers: 127  tid: 256
[   14.010000] ath10k_pci 0000:01:00.0: 10.1 features for rx-decap-mode: 0xc0
[   14.030000] ath10k_pci 0000:01:00.0: wmi print 'P 128 V 8 T 410'
[   14.030000] ath10k_pci 0000:01:00.0: wmi print 'msdu-desc: 1424  sw-crypt: 0'
[   14.040000] ath10k_pci 0000:01:00.0: wmi print 'alloc rem: 26172 iram: 24596'
[   14.100000] ath10k_pci 0000:01:00.0: htt-ver 2.1 wmi-op 2 htt-op 2 cal otp max-sta 128 raw 0 hwcrypto 1
[   32.620000] ath10k_pci 0000:01:00.0: 10.1 wmi init: vdevs: 16  peers: 127  tid: 256
[   32.630000] ath10k_pci 0000:01:00.0: 10.1 features for rx-decap-mode: 0xc0
[   32.640000] ath10k_pci 0000:01:00.0: wmi print 'P 128 V 8 T 410'
[   32.650000] ath10k_pci 0000:01:00.0: wmi print 'msdu-desc: 1424  sw-crypt: 0'
[   32.660000] ath10k_pci 0000:01:00.0: wmi print 'alloc rem: 26172 iram: 24596'
[   32.930000] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[   32.930000] ath10k: [0000]: F19C0000 0500FC17 70201031 08000000 80000000 58F84000 010006C0 FC9C0000
[   32.930000] ath10k: [0008]: 085CFC0F 30674400 08000000 80000000 FC9C0000 065CFC0F 58F84000 C8704400
[   32.930000] ath10k: [0016]: 19000000 FC9C0000 065CFC0F 09000000 C8704400 348D9C00 119D0000 0920000C
[   32.930000] ath10k: [0024]: B0F34300 01000000 00000000 119D0000 065C000C 08000000 C8704400 3C8D9C00
[... cut ...]
[   36.620000] ath10k: [0040]: 1FE00700 6FAA0000 0128FC17 80881071 0B000400 442A4400 3E009600 1FE00700
[   36.620000] ath10k: [0048]: 6FAA0000 0128FC17 80881071 0C000500 442A4400 3E009600 1FE00F00
[   36.620000] ath10k_pci 0000:01:00.0: ATH10K_END
[   43.620000] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[   43.620000] ath10k: [0000]: 13C70000 284C0408 40000000 03000000 1CC70000 234CFC0F 01000000 01000000
[   43.620000] ath10k: [0008]: 01000000
[   43.620000] ath10k_pci 0000:01:00.0: ATH10K_END

oamouroux, 2018-05-24:
Some additionnal information: I'm building the driver without DFS support (CONFIG_PACKAGE_ATH_DFS in OpenWrt)

greearb, 2018-05-24:
There are no obvious errors in the logs, though there is one failed-to-transmit message.
Normally one of those is OK and system recovers, but perhaps in your case it is important?

In this particular case that generated the logs, you are seeing no pkts on air from the ath10k system?

Have you tried testing the same setup without encryption to see if the problem is related to encryption?

In the case you just reported, does the problem always happen on this system, or is it transient?

And finally, please show me output of:
cat /sys/kernel/debug/ieee80211/phy0/ath10k/fw_regs

(Adjust for your phy name)

oamouroux, 2018-05-25:
It seems to be related to the encryption.  Broadcast traffic is exchanged unencrypted successfully but unicast traffic is sent & received encrypted and does not seem to be decrypted.

It always happens on these systems in this particular location (I guess there are some specific RF conditions in this area). I already changed all the HW several times (electronics, antenna, cables) and I always got the same result.

This is working fine using official ath10k + 802.11s +WPA2-SAE

cat /sys/kernel/debug/ieee80211/phy0/ath10k/fw_regs
ath10k Target Register Dump (extras-count: 0)
             =================

           MAC-FILTER-ADDR-L32 0x00000000
           MAC-FILTER-ADDR-U16 0x00000000
                 DCU-SLOT-TIME 0x00000000
               PHY-MODE-SELECT 0x00000000
           PHY-BB-GEN-CONTROLS 0x00000000
                       DMA-IMR 0x00000000
                  DMA-TXRX-IMR 0x00000000
                 PCU-BSSID-L32 0x00000000
                 PCU-BSSID-U16 0x00000000
                PCU-BSSID2-L32 0x00000000
                PCU-BSSID2-U16 0x00000000
              PCU-STA-ADDR-L32 0x00000000
              PCU-STA-ADDR-U16 0x00000000
                   MAC-DMA-CFG 0x00000000
                 MAC-DMA-TXCFG 0x00000000
                  SW-POWERMODE 0x00000000
               SW-CHAINMASK-TX 0x00000000
               SW-CHAINMASK-RX 0x00000000
                     SW-OPMODE 0x00000000
              MAC-PCU-RXFILTER 0x00000000
                   SW-RXFILTER 0x00000000
                 SHORT-RETRIES 0x00000000
                  LONG-RETRIES 0x00000000
                      ADC-TEMP 0x00000000
               NFCAL-PER-CHAIN 0x00000000

greearb, 2018-05-25:
Those register values must have been taken when the node was inactive?  I need to see them
in the case where the system is failing.

And, what you describe sounds like key setting issues, which I have previously seen, but in the cases I have debugged, it appears the stack and/or supplicant was doing the wrong thing.

I now suspect the previous 'dmesg' logs were also from a case that was not actively failing?  I need to see as complete of a boot log as possible for the case that is actually failing in order to debug this.

oamouroux, 2018-05-28:
[    0.000000] Linux version 3.18.92 (tpateloup@build) (gcc version 4.8.3 (OpenWrt/Linaro GCC 4.8-2014.04 5.3.0b1-10-gcfc70d75da) ) #209 Fri May 25 11:12:05 CEST 2018
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] SoC: Qualcomm Atheros QCA9558 ver 1 rev 0
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 08000000 @ 00000000 (usable)
[    0.000000] User-defined physical RAM map:
[    0.000000]  memory: 08000000 @ 00000000 (usable)
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x07ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000-0x07ffffff]
[    0.000000] Initmem setup node 0 [mem 0x00000000-0x07ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000] free_area_init_node: node 0, pgdat 80388f20, node_mem_map 81000000
[    0.000000]   Normal zone: 256 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 32512
[    0.000000] Kernel command line:  console=ttyS0,115200 root=/dev/mtdblock5 ro init=/etc/preinit panic=3 mtdparts=spi0.0:128k(u-boot),64k(u-boot-env),1280k(kernel1),6592k(rootfs1),1280k(kernel2),6592k(rootfs2),384k(config),64k(ART),7872k@0x30000(firmware1),7872k@0x7e0000(firmware2) board=JWAP230 mem=128M rootfstype=squashfs noinitrd
[    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 125860K/131072K available (2655K kernel code, 117K rwdata, 532K rodata, 144K init, 184K bss, 5212K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:51
[    0.000000] Clocks: CPU:720.000MHz, DDR:600.000MHz, AHB:200.000MHz, Ref:40.000MHz
[    0.000000] Calibrating delay loop... 358.80 BogoMIPS (lpj=1794048)
[    0.060000] pid_max: default: 32768 minimum: 301
[    0.060000] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.070000] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.080000] NET: Registered protocol family 16
[    0.080000] MIPS: machine is jjPlus JWAP230/232
[    0.090000] 
[    0.090000] WLAN firmware dump buffer allocation of 2097152 bytes @ address 0x87a00000- SUCCESS !!!
[    0.100000] ar724x-pci ar724x-pci.0: PCIe link is down
[    0.100000] registering PCI controller with io_map_base unset
[    0.110000] registering PCI controller with io_map_base unset
[    0.550000] PCI host bridge to bus 0000:00
[    0.560000] pci_bus 0000:00: root bus resource [mem 0x10000000-0x11ffffff]
[    0.560000] pci_bus 0000:00: root bus resource [io  0x0000]
[    0.570000] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
[    0.570000] pci_bus 0000:00: busn_res: [bus 00-ff] end is updated to 00
[    0.570000] PCI host bridge to bus 0000:01
[    0.580000] pci_bus 0000:01: root bus resource [mem 0x12000000-0x13ffffff]
[    0.580000] pci_bus 0000:01: root bus resource [io  0x0001]
[    0.590000] pci_bus 0000:01: No busn resource found for root bus, will use [bus 01-ff]
[    0.590000] pci 0000:01:00.0: [168c:003c] type 00 class 0x028000
[    0.590000] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x001fffff 64bit]
[    0.590000] pci 0000:01:00.0: reg 0x30: [mem 0x00000000-0x0000ffff pref]
[    0.590000] pci 0000:01:00.0: supports D1 D2
[    0.590000] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    0.590000] pci 0000:01:00.0: BAR 0: assigned [mem 0x12000000-0x121fffff 64bit]
[    0.600000] pci 0000:01:00.0: BAR 6: assigned [mem 0x12200000-0x1220ffff pref]
[    0.600000] pci 0000:01:00.0: using irq 41 for pin 1
[    0.610000] Switched to clocksource MIPS
[    0.610000] NET: Registered protocol family 2
[    0.620000] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
[    0.620000] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
[    0.630000] TCP: Hash tables configured (established 1024 bind 1024)
[    0.640000] TCP: reno registered
[    0.640000] UDP hash table entries: 256 (order: 0, 4096 bytes)
[    0.650000] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
[... 271 lines cut ...]
[   33.960000] ath10k: [0192]: 9EA10000 564C0414 54804300 C8DA9B00 10000000 0A000000 003D0000 BDA10000
[   33.960000] ath10k: [0200]: 4F4C0414 2C000000 ACE74100 54804300 01000000 01007300 BDA10000 1D4C0410
[   33.960000] ath10k: [0208]: 54804300 01000000 06000000 42000000 C1A10000 1C4C0410 54804300 00000000
[   33.960000] ath10k: [0216]: 06000000 42010000 07A20000 574C0414 54804300 C8DA9B00 10000000 1A000000
[   33.960000] ath10k: [0224]: 023D0000 07A20000 524C0414 54804300 E2010000 01000000 02000000 08400000
[   33.960000] ath10k: [0232]: 07A20000 0128FC17 070B1071 02000903 07B00102 FAFF0000 FFFF0000 07A20000
[... cut ...]
[   40.960000] ath10k: [0024]: 442A4400 3E009600 1FF80000 18BE0000 0128FC17 80881071 0A000200 442A4400
[   40.960000] ath10k: [0032]: 3E009600 1FF80000 18BE0000 0128FC17 80881071 0B000300 442A4400 3E009600
[   40.960000] ath10k: [0040]: 1FE00700 18BE0000 0128FC17 80881071 0B000400 442A4400 3E009600 1FE00700
[   40.960000] ath10k: [0048]: 18BE0000 0128FC17 80881071 0C000500 442A4400 3E009600 1FE00F00
[   40.960000] ath10k_pci 0000:01:00.0: ATH10K_END
[   66.410000] batman_adv: bat0: hop_penalty: Changing from: 20 to: 15

greearb, 2018-05-29:
Is that log from a working case, or a broken case?  If you restart that same node a few times, will it sometimes work and sometimes not?  If so, maybe it would be helpful if you could provide one log of broken and one log of the working case so I can compare.  Please clearly note any symptoms you see (or do not see) when posting logs so that I know what to look for.

I see no obvious errors in the log you posted above.

oamouroux, 2018-06-26:
It seems to happen when there is a mismatch on the NSS. For example:
side A:
iw adhoc0 station dump
Station 06:f0:21:31:5b:a1 (on adhoc0)
	inactive time:	0 ms
	rx bytes:	6569111
	rx packets:	40354
	tx bytes:	1604311
	tx packets:	5958
	tx retries:	0
	tx failed:	5953
	signal:  	-46 [-93, -46] dBm
	signal avg:	-46 [-92, -46] dBm
	tx bitrate:	81.0 MBit/s VHT-MCS 4 40MHz VHT-NSS 1
	rx bitrate:	162.0 MBit/s VHT-MCS 4 40MHz VHT-NSS 2
	authorized:	yes
	authenticated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	connected time:	321 seconds

side B:
Station 06:f0:21:3a:d7:9f (on adhoc0)
	inactive time:	0 ms
	rx bytes:	1344861
	rx packets:	8825
	tx bytes:	4838869
	tx packets:	14301
	tx retries:	0
	tx failed:	0
	signal:  	-49 [-68, -49] dBm
	signal avg:	-48 [-68, -48] dBm
	tx bitrate:	81.0 MBit/s VHT-MCS 4 40MHz VHT-NSS 1
	rx bitrate:	6.0 MBit/s
	authorized:	yes
	authenticated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	connected time:	122 seconds

klukonin, 2018-06-26:
Hello!

This looks quite like the same issue we faced and reported in dec 2016.
100% reproducible with 10.1 firmware and QCA988X.

@oamouroux 
Please can you check this issue again?
Right at the moment when tx begin to fail and  second device shows rx bitrate: 6.0 MBit/s (or sometimes rx bitrate: 48.0 MBit/s)
iw adhoc0 station dump should show 
"inactive time:"  WITH NEGATIVE VALUE.

After some time, "inactive time:" switches to a normal state, but no traffic can be sent.
In OPEN mode this issue with negative "inactive time:" value still exist, but when it switches to a positive state, traffic can go normally through the wireless IBSS link.

klukonin, 2018-07-19:
Hello, @oamouroux and @greearb .

We've jist triggered this issue in our lab again.
Look's like thr problem can be solved (or mitigated) with this parameters:
1) max_nss = 2 in fwcfg if we are using MIMO 2x2 device.
2) beacon_int = 10 in wpa_supplicant if we are using encrypted IBSS.

This issue look's like a beacon interval related, but I can't explain why. 

Best regards,
Lukonin Kirill.

klukonin, 2018-07-20:
Hello, Ben! @greearb 
Looks like a found a bug.
@oamouroux  Please check your issue too.

if we look into fwstats, there will be a strange behavior.
Look at the first peer and it's mac address. The first octet behave LIKE A COUNTER!
This behavior is reproducible on all 5 nodes in our lab at the same time.
I don't know why beacon_int =10 helps in this case. 
Looks like a bug, really.

This is fwstats PEES stats log below.

             ath10k PEER stats (3)
             =================

              Peer MAC address 65:01:00:00:1e:d4
                     Peer RSSI 55633
                  Peer TX rate 0
                  Peer RX rate 6000
              Peer RX duration 0

              Peer MAC address 0a:24:68:48:51:d9
                     Peer RSSI 0
                  Peer TX rate 6000
                  Peer RX rate 0
              Peer RX duration 0

              Peer MAC address 3e:9d:75:48:54:6a
                     Peer RSSI 9
                  Peer TX rate 58500
                  Peer RX rate 175500
              Peer RX duration 0
_________________________________________________________________

             ath10k PEER stats (3)
             =================

              Peer MAC address 67:01:00:00:1e:d4
                     Peer RSSI 55633
                  Peer TX rate 0
                  Peer RX rate 6000
              Peer RX duration 0

              Peer MAC address 0a:24:68:48:51:d9
                     Peer RSSI 0
                  Peer TX rate 6000
                  Peer RX rate 0
              Peer RX duration 0

              Peer MAC address 3e:9d:75:48:54:6a
                     Peer RSSI 9
                  Peer TX rate 58500
                  Peer RX rate 175500
              Peer RX duration 0
____________________________________________________________________

             ath10k PEER stats (3)
             =================

              Peer MAC address 69:01:00:00:1e:d4
                     Peer RSSI 55633
                  Peer TX rate 0
                  Peer RX rate 6000
              Peer RX duration 0

              Peer MAC address 0a:24:68:48:51:d9
                     Peer RSSI 0
                  Peer TX rate 6000
                  Peer RX rate 0
              Peer RX duration 0

              Peer MAC address 3e:9d:75:48:54:6a
                     Peer RSSI 9
                  Peer TX rate 58500
                  Peer RX rate 175500
              Peer RX duration 0
_________________________________________________________________

             ath10k PEER stats (3)
             =================

              Peer MAC address 6b:01:00:00:1e:d4
                     Peer RSSI 55633
                  Peer TX rate 0
                  Peer RX rate 6000
              Peer RX duration 0

              Peer MAC address 0a:24:68:48:51:d9
                     Peer RSSI 0
                  Peer TX rate 6000
                  Peer RX rate 0
              Peer RX duration 0

              Peer MAC address 3e:9d:75:48:54:6a
                     Peer RSSI 9
                  Peer TX rate 58500
                  Peer RX rate 175500
              Peer RX duration 0

klukonin, 2018-07-20:
@greearb 

We also noticed that the first peer should not exist at all.
there are no any mac address with this value BUT please look below.

             ath10k PEER stats (3)
             =================

              Peer MAC address 3d:00:00:00:3a:ed
                     Peer RSSI 53063
                  Peer TX rate 0
                  Peer RX rate 6000
              Peer RX duration 0

              Peer MAC address 06:0a:dc:48:44:e5
                     Peer RSSI 31
                  Peer TX rate 650000
                  Peer RX rate 650000
              Peer RX duration 0

              Peer MAC address 0e:16:25:48:54:6a
                     Peer RSSI 27
                  Peer TX rate 526500
                  Peer RX rate 526500
              Peer RX duration 0

__________________________________________________________________________

ip addr shows
ibss_5g_1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1564 qdisc mq master bat0 state UP group default qlen 1000
    link/ether 3a:ed:ad:48:47:cf brd ff:ff:ff:ff:ff:ff

3a:ed is the first two octets of the interface mac address and the last two octets of phantom peer.
Also there is a strange static RSSI value, hmmmm let's look closer....

printf %x\\n 53063
cf47

It's a reverse order of last two octets.

So may be there are some bugs with pointers inside the firmware. I don't know.
We use firmware ver 10.1-ct-8x-__fH-021-4fa9f30 api 2 features wmi-10.x,txstatus-noack,wmi-10.x-CT,ratemask-CT,get-temp-CT,tx-rc-CT,cust-stats-CT crc32 580c6146

greearb, 2018-07-20:
The stats parsing in ath10k is complex and bug prone.  The issue could easily be in the driver as well.  Do you see this weird first peer only in cases where you are having connectivity issues, or does it happen all the time?

And, the 'beacon int' that fixes this is sending around 100 beacons per second, or did I mis-understand that?

klukonin, 2018-07-20:
We see this phantom peer all the time, even with AP interface.

For example:

             ath10k PEER stats (1)
             =================

              Peer MAC address 2b:00:00:00:5a:be
                     Peer RSSI 45976
                  Peer TX rate 0
                  Peer RX rate 0
              Peer RX duration 0

ap_5g_1_1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 5a:be:7b:80:98:b3 brd ff:ff:ff:ff:ff:ff

root@OpenWrt:~# printf %x/n 45976
b398
The same pieces of mac and last two octets with reverse order.

> And, the 'beacon int' that fixes this is sending around 100 beacons per second, or did I mis-understand that?

You understand that 100% right.

Well, I don;'t want to say "Fixed", because we see a lot of overruns in dmesg.
 
[23211.095047] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[23356.278429] ath10k_pci 0000:00:00.0: wmi request nop (id 11199)
[23509.833606] ath10k_pci 0000:00:00.0: wmi vdev 0 install key peer 06:0a:dc:48:44:e5 idx 1 cipher 0 len 16 flags 0x1
[23509.833721] ath10k_pci 0000:00:00.0: wmi echo value 0x0ba991e9
[23509.833755] ath10k_pci 0000:00:00.0: wmi event echo value 0x0ba991e9
[23509.834109] ath10k_pci 0000:00:00.0: wmi vdev 0 install key peer 06:0a:dc:48:44:e5 idx 1 cipher 4 len 16 flags 0x1
[23509.834131] ath10k_pci 0000:00:00.0: wmi echo value 0x0ba991e9
[23509.834250] ath10k_pci 0000:00:00.0: wmi event echo value 0x0ba991e9
[23509.834319] ath10k_pci 0000:00:00.0: wmi vdev id 0x0 set param 31 value 1
[23510.071101] ath10k_pci 0000:00:00.0: wmi event debug mesg len 52
[23510.071115] ath10k_pci 0000:00:00.0: dropping dbg buffer due to crash since read
[23510.078805] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[23510.085194] ath10k: [0000]: 547D6F01 1D4C0010 E07E4300 01000000 00000000 E2010000 547D6F01 1D4C0010
[23510.094545] ath10k: [0008]: E07E4300 01000000 06000000 E2010000
[23510.100657] ath10k_pci 0000:00:00.0: ATH10K_END
[23510.110592] ath10k_pci 0000:00:00.0: wmi vdev 0 install key peer 3a:ed:ad:48:47:cf idx 1 cipher 0 len 16 flags 0x1
[23510.110613] ath10k_pci 0000:00:00.0: wmi echo value 0x0ba991e9
[23510.110737] ath10k_pci 0000:00:00.0: wmi event echo value 0x0ba991e9
[23510.110933] ath10k_pci 0000:00:00.0: wmi vdev 0 install key peer 3a:ed:ad:48:47:cf idx 1 cipher 4 len 16 flags 0x1
[23510.110951] ath10k_pci 0000:00:00.0: wmi echo value 0x0ba991e9
[23510.111087] ath10k_pci 0000:00:00.0: wmi event echo value 0x0ba991e9
[23510.111134] ath10k_pci 0000:00:00.0: wmi vdev id 0x0 set param 31 value 1
[23510.678282] ath10k_pci 0000:00:00.0: wmi vdev 0 install key peer 0e:16:25:48:54:6a idx 1 cipher 0 len 16 flags 0x1
[23510.678309] ath10k_pci 0000:00:00.0: wmi echo value 0x0ba991e9
[23510.678708] ath10k_pci 0000:00:00.0: wmi event echo value 0x0ba991e9
[23510.679101] ath10k_pci 0000:00:00.0: wmi vdev 0 install key peer 0e:16:25:48:54:6a idx 1 cipher 4 len 16 flags 0x1
[23510.679124] ath10k_pci 0000:00:00.0: wmi echo value 0x0ba991e9
[23510.679262] ath10k_pci 0000:00:00.0: wmi event echo value 0x0ba991e9
[23510.679319] ath10k_pci 0000:00:00.0: wmi vdev id 0x0 set param 31 value 1
[23511.071409] ath10k_pci 0000:00:00.0: wmi event debug mesg len 100
[23511.071423] ath10k_pci 0000:00:00.0: dropping dbg buffer due to crash since read

But traffic can pass normally through many IBSS hops without visible issues and throughput drops like in [this report](https://github.com/greearb/ath10k-ct/issues/19).
So let it be a "Mitigation", but not "Fix".
We tried beacon_int = 10 with reboots, interface restarts and didn't trigger this issue, so I confirm that this trick works all the time.

Also, as we can see, this bug can be reproduced with both 10.1 and 10.4 firmwares in IBSS mode.
We will check if beacon_int = 10 parameter helps with 10.4 firmware and let you know.

greearb, 2018-07-20:
So that peer thing is probably not related to the IBSS bug.  Please open a new bug for the peer stats problem.

For the IBSS problem, I cannot think of any reason why sending beacons more often would fix this particular bug report so I don't have any new ideas on how to fix this more properly in the firmware or driver.

@klukonin Earlier you mentioned you forced the system to 2x2 NSS as part of your fix.  Are both the 2x2 force and the beacon hack needed, or will one or the other by itself mitigate the problem?

klukonin, 2018-07-20:
I can confirm that max_nss = 2 with beacon_int = 10 helps.
max_nss = 2 doesn't help
We will try to test only beacon_int = 10 hack and I'll report about the result.
May be it will work without 2x2 NSS enforcement.

Ok. I will open a new bug report

klukonin, 2018-07-30:
Hello @oamouroux  and @greearb 

He have 100% working recipe how to trigger this issue and how to avoid it.

1) If we create ibss interface, do sleep 1 and start wpa_supplicant - every node will fail to do traffic exchange with others. Rarely one node can catch a state when it can transmit, but can't receive.
TCP acks and small packets can pass, but with very very low throughput, That's why sometimes traffic can pass only in one direction.

2) If we create ibss interface and run wpa_supplicant as fast as we can - this issue does not show itself anymore. It is important that beacon_int parameter should be 10ms.

So may be this issue is not beacon related, but TSF related. I remember a lot of weired things in ath10k and firmware about TSF.

klukonin, 2018-07-30:
Also,this can be interesting for you, With original 10.2.6 firmware we see how in mesh mode sometimes TSF offset can raise up to 34 days right after we boot up our lab. 
Mesh TSF logic can handle such situations normally, but I'm no sure about IBSS. 
Anyway, I don't know how to get current TSF offset value in ibss mode.