greearb/ath10k-ct #20: Compex WLE900VX - Fake crash due to SWBA overrun

gfduszynski, 2018-03-22:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).
I'm trying to setup AP for home usage in 5ghz band (hostapd.conf attached).

2 days ago, after experiencing many difficulties with stability/throughput I've decided to switch to LTS kernel and CT firmware(one specified below). All problems seemed resolved. Next day(after about 7h) I've checked everything and found out following:
- Laptop(win8.1) with Intel 7260 adapter could still connect and worked just fine.
- IPhone 6S (11.2.5) could not connect either throwing "wrong password" or that connection could not be established.

Unfortunately I didn't grab logs then.
I remember dmesg showed loads of "SWBA overrun" messages and firmware crash related to __sta_info_destroy_part2. 

I've decided to move to ath10k-ct. Firmware still crashed after night from the looks of it. However ath10k-ct seems a lot smarter and decided to force restart adapter. This morning wifi worked on both devices.

I'm not exactly sure whats causing instability .
I'm pretty sure each night IPhone went out of range for the night.
Laptop also went to sleep. There were likely no devices connected to AP at the time of crash.

I'm betting it has something to do with disconnection handling/going out of range.

Software (OS, Firmware version, kernel, driver, etc)
- Arch Linux x86-64
- 10.1-ct-8x-__fW-019-395e9f1
- 4.14.28-1-lts 
- ath10k-ct 

Hardware (NIC chipset, platform, etc)
- QCA988X (WLE900VX)
- https://www.asrock.com/mb/Intel/J4205-ITX/index.us.asp

Logs (dmesg, maybe supplicant and/or hostap)
NOTE: last 2 files have extra ".log" for me to be able to upload to github.
[journalctl.log](https://github.com/greearb/ath10k-ct/files/1837065/journalctl.log)
[dmesg.log](https://github.com/greearb/ath10k-ct/files/1837066/dmesg.log)
[crash-ath10k-ct.bin.log](https://github.com/greearb/ath10k-ct/files/1837070/crash-ath10k-ct.bin.log)
[hostapd.conf.log](https://github.com/greearb/ath10k-ct/files/1837073/hostapd.conf.log)

greearb, 2018-03-22:
Thank you for the detailed bug report.

The primary problem appears that sometimes we lose communication with the firmware.  As you notice, my driver will fake a crash so that the system can recover:

[ 8817.030140] ath10k_pci 0000:04:00.0: Cannot communicate with firmware, attempting to fake crash and restart firmware.
[ 8817.030166] ath10k_pci 0000:04:00.0: firmware crashed! (uuid 0893da43-3a96-4f84-bc71-29c9821b424d)

We have seen this same general problem for years, and I do not know how to make more progress on fixing the hang, so the work-around is the best that I can do at this time.

You might also try one of the 'htt-mgt' firmwares.  It can help stability in some cases, though it is unlikely to fix the firmware hang that you see.  The htt-mgt firmware requires use of the ath10k-ct driver as it will not work with the stock driver.

gfduszynski, 2018-03-22:
I will try to find a way to reliably recreate the problem.

Then I will try htt-mgt firmware.
Then Beta.

I'm quite determined to get this right, maybe we can finally fix this once and for all :D

gfduszynski, 2018-03-22:
Will it help you to find a firmware that does not experience this particular issue ?
If so, sharing the scenario to reproduce would help a lot.
I will literally go through all of them.

gfduszynski, 2018-03-23:
Yesterday I've switched to:
https://github.com/kvalo/ath10k-firmware/raw/master/QCA988X/hw2.0/10.2.4-1.0/firmware-5.bin_10.2.4-1.0-00037

Tortured it for couple of hours with iperf3, left for the night and it looks stable.
I will update in couple of days If it still does not crash.

It's very possible that the only reason it's working is because I can't yet reliably recreate the problem.

gfduszynski, 2018-03-27:
Mentioned firmware from kvalo repository is still running without problems.
It's unlikely that the problem that occured two days in a row with the old software is present in the mentioned firmware.

On the other hand maybe it was bad luck then.

Anyhow I would like to add that if anybody experiences a problem with Intel's 7260 AC it most likely to the crappy drivers, old ones like 17.15.0.5 can muster speeds between 45-50 MB/s up & down.