greearb/ath10k-ct #21: Warning: "received addba event for invalid peer_id: 2 tid: 0"

hauke, 2018-03-22:
I see these messages in my log file (dmesg)
```
[  146.642495] ath10k_pci 0000:02:00.0: received addba event for invalid peer_id: 2 tid: 0
[  146.658943] ath10k_pci 0000:02:00.0: received addba event for invalid peer_id: 2
[  146.664977] ath10k_pci 0000:02:00.0: received addba event for invalid peer_id: 2 tid: 0
[  166.486402] enter showtime
[  166.487901] IPv6: ADDRCONF(NETDEV_CHANGE): dsl0: link becomes ready
[  166.495351] IPv6: ADDRCONF(NETDEV_CHANGE): dsl0.7: link becomes ready
[  166.505167] enter showtime
[  172.055124] pppoe-wan: renamed from ppp0
[  219.424582] ath10k_pci 0000:02:00.0: received addba event for invalid peer_id: 1 tid: 0
[  616.441012] ath10k_pci 0000:02:00.0: received addba event for invalid peer_id: 1
[  619.252372] ath10k_pci 0000:02:00.0: received addba event for invalid peer_id: 1 tid: 0
```
I haven't noticed any real problem.

This was seen on a QCA9880 on a Lantiq VR9 MIPS BE SoC in a BT Home Hub 5A.
* With OpenWrt master from 22. March 2018, r6519
* ath10k-ct driver from 30827f7d5b9841905c4efe918da2d95fc518c921
* firmware-2-ct-full-htt-mgt-community-20.bin.lede.001

more logs:
```
[   12.433550] ath10k_pci 0000:02:00.0: pci irq legacy oper_irq_mode 1 irq_mode 0 reset_mode 0
[   12.665242] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/fwcfg-pci-0000:02:00.0.txt failed with error -2
[   12.674445] ath10k_pci 0000:02:00.0: Falling back to user helper
[   12.845717] firmware ath10k!fwcfg-pci-0000:02:00.0.txt: firmware_loading_store: map pages failed
[   12.853781] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/pre-cal-pci-0000:02:00.0.bin failed with error -2
[   12.863867] ath10k_pci 0000:02:00.0: Falling back to user helper
[   13.115287] firmware ath10k!pre-cal-pci-0000:02:00.0.bin: firmware_loading_store: map pages failed
[   13.124933] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/QCA988X/hw2.0/ct-firmware-5.bin failed with error -2
[   13.134604] ath10k_pci 0000:02:00.0: Falling back to user helper
[   13.381780] firmware ath10k!QCA988X!hw2.0!ct-firmware-5.bin: firmware_loading_store: map pages failed
[   13.394547] ath10k_pci 0000:02:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000
[   13.402477] ath10k_pci 0000:02:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   13.420116] ath10k_pci 0000:02:00.0: firmware ver 10.1-ct-8x-__fH-020-0e04baf api 2 features wmi-10.x,txstatus-noack,wmi-10.x-CT,ratemask-CT,get-temp-CT,tx-rc-CT,cust-stats-CT crc32 7b0564f1
[   13.463630] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/QCA988X/hw2.0/board-2.bin failed with error -2
[   13.472776] ath10k_pci 0000:02:00.0: Falling back to user helper
[   13.671499] firmware ath10k!QCA988X!hw2.0!board-2.bin: firmware_loading_store: map pages failed
[   13.717654] ath10k_pci 0000:02:00.0: board_file api 1 bmi_id N/A crc32 bebc7c08
[   14.581848] ath10k_pci 0000:02:00.0: 10.1 wmi init: vdevs: 16  peers: 127  tid: 256
[   14.597789] ath10k_pci 0000:02:00.0: wmi print 'P 128 V 8 T 410'
[   14.602401] ath10k_pci 0000:02:00.0: wmi print 'msdu-desc: 1424  sw-crypt: 0'
[   14.609897] ath10k_pci 0000:02:00.0: wmi print 'alloc rem: 26400 iram: 26980'
[   14.664088] ath10k_pci 0000:02:00.0: htt-ver 2.2 wmi-op 2 htt-op 2 cal file max-sta 128 raw 0 hwcrypto 1
```

I have a Samsung S4 with a Broadcom BCM4335 chip running LineageOS 14.1, two Debian systems with kernel 4.9, one using a Intel Corporation Wireless 7260 and the other an Intel Centrino Ultimate-N 6300. This is an urban environment with a lot of other wifi networks.

The vendor FW versions are very unstable and did not survive more than 3 days.

hauke, 2018-03-22:
This is my wireless configuration:
```
root@LEDE:~# cat /etc/config/wireless 

config wifi-device 'radio0'
	option type 'mac80211'
	option channel '124'
	option hwmode '11a'
	option path 'pci0000:01/0000:01:00.0/0000:02:00.0'
	option htmode 'VHT80'
	option country 'DE'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid '********'
	option key '********'
	option encryption 'psk2+ccmp'
	option ieee80211w '1'

config wifi-device 'radio1'
	option type 'mac80211'
	option channel '6'
	option hwmode '11g'
	option path 'pci0000:00/0000:00:0e.0'
	option htmode 'HT20'
	option country 'DE'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid ''********''
	option key ''********''
	option encryption 'psk2+ccmp'
	option ieee80211w '1'
```

greearb, 2018-03-26:
I think I see the problem.  It was a recent regression, bad backport from my 10.4 firmware.  Please try the attached firmware and see if it fixes the problem for you:
[firmware-2-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/1849445/firmware-2-full-htt-mgt-community.bin.gz)

greearb, 2018-03-26:
Bug appears fixed, and patch to update to latest FW has been posted to lede-dev, so closing this bug.

closed 2018-03-26 by greearb