greearb/ath10k-ct #22: [QCA9984] Wifi can't up when I use 160MHz bandwidth

ita93, 2018-04-05:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).

Software (OS, Firmware version, kernel, driver, etc)

Hardware (NIC chipset, platform, etc)

Logs (dmesg, maybe supplicant and/or hostap)
HI, I am trying the 160Mhz wifi on WLE1216(QCA9984) card with board WPQ864(IPQ8064), the board is running LEDE with kernel version 4.9.91.
Wifi never up if I use 160Mhz bandwidth, 80Mhz works well, however. 
This is hostapd configuration file:
`driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
hw_mode=a
beacon_int=100
channel=36

ieee80211n=1
ht_coex=0
ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40]
vht_oper_chwidth=2
vht_oper_centr_freq_seg0_idx=50
ieee80211ac=1
vht_capab=[RXLDPC][SHORT-GI-80][SHORT-GI-160][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][VHT160-80PLUS80][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7]

interface=wlan0
ctrl_interface=/var/run/hostapd
ap_isolate=1
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
auth_algs=1
wpa=0
ssid=OpenWrt
bridge=br-lan
bssid=04:f0:21:31:ee:ac
`

Could you help me check it. Thank you very much

greearb, 2018-04-05:
Please attach 'dmesg' output as well as hostapd logs.  Preferably verbose hostapd logs.

ita93, 2018-04-05:
Hi, 
**Hostapd log**:
`Sat Mar 31 17:32:47 2018 daemon.err hostapd: Configuration file: /var/run/hostapd-phy0.conf
Sat Mar 31 17:32:50 2018 kern.warn kernel: [   67.055050] ath10k_pci 0000:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
Sat Mar 31 17:32:50 2018 kern.warn kernel: [   67.055084] ath10k_pci 0000:01:00.0: msdu-desc: 2500  skid: 32
Sat Mar 31 17:32:50 2018 kern.info kernel: [   67.136139] ath10k_pci 0000:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0'
Sat Mar 31 17:32:50 2018 kern.info kernel: [   67.136813] ath10k_pci 0000:01:00.0: wmi print 'free: 89928 iram: 15476 sram: 23408'
Sat Mar 31 17:32:50 2018 daemon.info dhcpcd[549]: wlan0: waiting for carrier
Sat Mar 31 17:32:50 2018 daemon.info dhcpcd[549]: wlan0: carrier acquired
Sat Mar 31 17:32:50 2018 kern.info kernel: [   67.421431] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Sat Mar 31 17:32:50 2018 kern.warn kernel: [   67.424612] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 0
Sat Mar 31 17:32:50 2018 daemon.info dhcpcd[549]: wlan0: IAID 21:31:ee:ac
Sat Mar 31 17:32:50 2018 daemon.info dhcpcd[549]: wlan0: adding address fe80::e5e3:e65d:7d8c:8b7f
Sat Mar 31 17:32:53 2018 kern.warn kernel: [   70.292478] ath10k_pci 0000:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
Sat Mar 31 17:32:53 2018 kern.warn kernel: [   70.292507] ath10k_pci 0000:01:00.0: msdu-desc: 2500  skid: 32
Sat Mar 31 17:32:53 2018 kern.info kernel: [   70.373550] ath10k_pci 0000:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0'
Sat Mar 31 17:32:53 2018 kern.info kernel: [   70.374204] ath10k_pci 0000:01:00.0: wmi print 'free: 89928 iram: 15476 sram: 23408'
Sat Mar 31 17:32:53 2018 daemon.info dhcpcd[549]: wlan0: carrier lost
Sat Mar 31 17:32:53 2018 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->HT_SCAN
Sat Mar 31 17:32:53 2018 kern.info kernel: [   70.659129] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Sat Mar 31 17:32:53 2018 kern.info kernel: [   70.664732] br-lan: port 2(wlan0) entered blocking state
Sat Mar 31 17:32:53 2018 kern.info kernel: [   70.664779] br-lan: port 2(wlan0) entered disabled state
Sat Mar 31 17:32:53 2018 kern.info kernel: [   70.670126] device wlan0 entered promiscuous mode
Sat Mar 31 17:32:53 2018 daemon.info dhcpcd[549]: wlan0: deleting address fe80::e5e3:e65d:7d8c:8b7f
Sat Mar 31 17:32:53 2018 daemon.err hostapd: could not get valid channel
Sat Mar 31 17:32:53 2018 daemon.notice hostapd: wlan0: interface state HT_SCAN->DFS
Sat Mar 31 17:32:56 2018 kern.warn kernel: [   73.821028] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 0
Sat Mar 31 17:32:56 2018 daemon.notice hostapd: wlan0: INTERFACE-DISABLED
Sat Mar 31 17:32:56 2018 kern.info kernel: [   73.905570] br-lan: port 2(wlan0) entered disabled state
Sat Mar 31 17:32:56 2018 daemon.info dhcpcd[549]: wlan0: removing interface
Sat Mar 31 17:32:56 2018 kern.info kernel: [   73.979056] device wlan0 left promiscuous mode
Sat Mar 31 17:32:56 2018 kern.info kernel: [   73.979115] br-lan: port 2(wlan0) entered disabled state
Sat Mar 31 17:32:57 2018 daemon.notice netifd: radio0 (1421): command failed: Not supported (-95)
Sat Mar 31 17:32:57 2018 daemon.err hostapd: Configuration file: /var/run/hostapd-phy0.conf
Sat Mar 31 17:33:00 2018 kern.warn kernel: [   77.041013] ath10k_pci 0000:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
Sat Mar 31 17:33:00 2018 kern.warn kernel: [   77.041048] ath10k_pci 0000:01:00.0: msdu-desc: 2500  skid: 32
Sat Mar 31 17:33:00 2018 kern.info kernel: [   77.121944] ath10k_pci 0000:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0'
Sat Mar 31 17:33:00 2018 kern.info kernel: [   77.122621] ath10k_pci 0000:01:00.0: wmi print 'free: 89928 iram: 15476 sram: 23408'
Sat Mar 31 17:33:00 2018 daemon.info dhcpcd[549]: wlan0: waiting for carrier
Sat Mar 31 17:33:00 2018 daemon.info dhcpcd[549]: wlan0: carrier acquired
Sat Mar 31 17:33:00 2018 kern.info kernel: [   77.407078] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Sat Mar 31 17:33:00 2018 daemon.notice hostapd: wlan0: INTERFACE-ENABLED
Sat Mar 31 17:33:00 2018 kern.warn kernel: [   77.409443] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 0
Sat Mar 31 17:33:00 2018 daemon.notice hostapd: wlan0: INTERFACE-DISABLED
Sat Mar 31 17:33:00 2018 daemon.err hostapd: nl80211: Could not configure driver mode
Sat Mar 31 17:33:00 2018 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Sat Mar 31 17:33:00 2018 daemon.err hostapd: nl80211 driver initialization failed.
Sat Mar 31 17:33:00 2018 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->DISABLED
Sat Mar 31 17:33:00 2018 daemon.notice hostapd: wlan0: AP-DISABLED
Sat Mar 31 17:33:00 2018 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Sat Mar 31 17:33:00 2018 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Sat Mar 31 17:33:00 2018 daemon.notice netifd: radio0 (1421): cat: can't open '/var/run/wifi-phy0.pid': No such file or directory
Sat Mar 31 17:33:00 2018 daemon.info dhcpcd[549]: wlan0: IAID 21:31:ee:ac
Sat Mar 31 17:33:00 2018 daemon.info dhcpcd[549]: wlan0: adding address fe80::e5e3:e65d:7d8c:8b7f
Sat Mar 31 17:33:00 2018 daemon.info dhcpcd[549]: wlan0: carrier lost
Sat Mar 31 17:33:00 2018 daemon.notice netifd: radio0 (1421): WARNING (wireless_add_process): executable path /usr/sbin/hostapd does not match process  path ()
[... cut ...]
[   73.979115] br-lan: port 2(wlan0) entered disabled state
[   77.041013] ath10k_pci 0000:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   77.041048] ath10k_pci 0000:01:00.0: msdu-desc: 2500  skid: 32
[   77.121944] ath10k_pci 0000:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0'
[   77.122621] ath10k_pci 0000:01:00.0: wmi print 'free: 89928 iram: 15476 sram: 23408'
[   77.407078] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   77.409443] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 0`

greearb, 2018-04-05:
If you start hostapd with -dd option, you should get more logging information.

And, please paste:  iw phy phy0 info
output (for whatever phy you are using).

Will this 160Mhz work with stock ath10k driver/firmware on your system?

ita93, 2018-04-05:
Hi, 160Mhz can work with stock firmware .
Actually, when I run hostapd with -dd option, the log is still the same. The log I sent you in previous comment was token with following configuration:
wireless.default_radio0.log_level='0'
This is output of command iw phy phy0 info on my system:

> Wiphy phy0
        max # scan SSIDs: 16
        max scan IEs length: 199 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Available Antennas: TX 0xf RX 0xf
        Configured Antennas: TX 0xf RX 0xf
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
                 * P2P-client
                 * P2P-GO
                 * P2P-device
        Band 2:
                Capabilities: 0x19ef
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 8 usec (0x06)
                HT TX/RX MCS rate indexes supported: 0-31
                VHT Capabilities (0x339b79fa):
                        Max MPDU length: 11454
                        Supported Channel Width: 160 MHz, 80+80 MHz
                        RX LDPC
                        short GI (80 MHz)
                        short GI (160/80+80 MHz)
                        TX STBC
                        SU Beamformer
                        SU Beamformee
                        MU Beamformer
                        MU Beamformee
                        RX antenna pattern consistency
                        TX antenna pattern consistency
                VHT RX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: MCS 0-9
                        4 streams: MCS 0-9
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT RX highest supported: 1560 Mbps
                VHT TX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: MCS 0-9
                        4 streams: MCS 0-9
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT TX highest supported: 1560 Mbps
                Frequencies:
                        * 5180 MHz [36] (23.0 dBm)
                        * 5200 MHz [40] (23.0 dBm)
                        * 5220 MHz [44] (23.0 dBm)
                        * 5240 MHz [48] (23.0 dBm)
                        * 5260 MHz [52] (23.0 dBm) (radar detection)
                        * 5280 MHz [56] (23.0 dBm) (radar detection)
                        * 5300 MHz [60] (23.0 dBm) (radar detection)
                        * 5320 MHz [64] (23.0 dBm) (radar detection)
                        * 5500 MHz [100] (23.0 dBm) (radar detection)
                        * 5520 MHz [104] (23.0 dBm) (radar detection)
                        * 5540 MHz [108] (23.0 dBm) (radar detection)
                        * 5560 MHz [112] (23.0 dBm) (radar detection)
                        * 5580 MHz [116] (23.0 dBm) (radar detection)
                        * 5600 MHz [120] (23.0 dBm) (radar detection)
                        * 5620 MHz [124] (23.0 dBm) (radar detection)
                        * 5640 MHz [128] (23.0 dBm) (radar detection)
                        * 5660 MHz [132] (23.0 dBm) (radar detection)
                        * 5680 MHz [136] (23.0 dBm) (radar detection)
                        * 5700 MHz [140] (23.0 dBm) (radar detection)
                        * 5720 MHz [144] (23.0 dBm) (radar detection)
                        * 5745 MHz [149] (30.0 dBm)
                        * 5765 MHz [153] (30.0 dBm)
                        * 5785 MHz [157] (30.0 dBm)
                        * 5805 MHz [161] (30.0 dBm)
                        * 5825 MHz [165] (30.0 dBm)
                        * 5845 MHz [169] (disabled)
                        * 5865 MHz [173] (disabled)
        valid interface combinations:
                 * #{ managed } <= 16, #{ AP, mesh point } <= 16, #{ IBSS } <= 1,
                   total <= 16, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz, 80+80 MHz, 160 MHz }

        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
        Device supports VHT-IBSS.

greearb, 2018-04-05:
From the iw info command, it looks like 160Mhz should be supported, so I am not sure why it fails for you.  Maybe you can load the stock firmware that works and compare its iw phy info to the ath10k-ct firmware and see what is different.  Also, are you running the ath10k-ct driver as well as firmware?  Maybe try different combinations of ath10k-ct firmware and driver and stock firmware and driver to see if it is firmware or driver that makes 160Mhz not work?

ita93, 2018-04-05:
Hi, thanks for your very fast support. I can not do combine like your suggestion because the ath10k firmware/driver and qca-wifi structure are very different. 
Btw, Did you successfully run 160Mhz with any Radio Card and Board? Because, when I google, it seems no one can make 160Mhz works with ath10k or ath10k-ct.
And my LEDE run kernel 4.9.9, by default, it will install ath10k-ct 4.13 when I choose ath10k-ct module in make menuconfig. Does It have any problem?

(Sorry because of my bad English).

greearb, 2018-04-05:
In LEDE, you can select the 'ath10k-ct' firmware and driver, or you can use the built-in ath10k firmware and driver (the ones without -ct in the name).

We have successfully tested 160Mhz mode with 9984 in our 4.13 kernel running on Fedora, but not specifically tested it in LEDE.  Please note that 160Mhz mode means the 9984 NIC can only run at 2x2 speeds, so unless you are doing long-range communication where you expect 4x4 to not work, then you should probably just stay with 4x4 80Mhz mode.

naoufal51, 2018-04-10:
Hi Ben,
Did you test the WLE1216(QCA9984) card with  4.13 kernel on a linux pc?
Thanks in advance for your reply

greearb, 2018-04-10:
Yes, we test it as part of our LANforge product line, which runs on Fedora-24 (or higher), with our modified 4.13 kernel, Compex 9984 NICs on x86-64 PCs, our firmware, the hostapd from my github page, etc.  Next time I find time to re-run this test, I'll post the hostapd config file we use if this issue is not resolved, but I am busy with lots of other things currently so it might be a while.

ita93, 2018-04-11:
Hi, I have a question, which board data file did you use?

master8282, 2018-06-20:
Hi all!

I have  WLE1216V5-20 and was able to run it in 160Mhz
1) I used 4.16 kernel (built with supporting DFS)
2) Built four ko modules of 10ath-ct and changed with original.
3) Used this firmware-5.bin (https://www.candelatech.com/downloads/ath10k-9984-10-4/firmware-5-ct-non-commercial-htt-mgt-9.bin), another firmware don't work in 160MHz for me unfortunately.
My hostapd.conf is attached here in another bug branch.

Good luck.

ssdnvv, 2018-09-14:
@master8282 I'm using same module and would like to understand how exactly you made it.
1. Which distro are you running? Did you try with other kernel versions? What did you change in kernel to support DFS?
With 3. you mean [this thread](https://github.com/greearb/ath10k-ct/issues/24)?

greearb, 2018-09-28:
The kernels on my github site have a configs/ directory, and the configs in there enable DFS mode.
We use Fedora Linux.  OpenWRT has recent ath10k-ct drivers and firmware, so it will probably work with 160Mhz, but I have not tried it.

greearb, 2018-12-18:
160Mhz works for me, so closing this bug.

closed 2018-12-18 by greearb