greearb/ath10k-ct #25: Compex WLE1216V5-20 (QCA9984) channel setting problem on FCC3_FCCA regulatory domain

jhaakana, 2018-07-13:
- Firmware: firmware ver 10.4-ct-9984-fW-010-868495e (from https://www.candelatech.com/downloads/ath10k-9984-10-4/firmware-5-ct-full-community.bin)
- board-2.bin: stock board-2.bin or board-2-ct.bin (from https://www.candelatech.com/downloads/ath10k-9984-10-4/board-2-ct.bin), does not matter
- OS: LEDE latest, mvebu/cortexa9, arm_cortex-a9_vfpv3
- Kernel: 4.14.51
- Driver: ath10k-ct 4.13 (ath10k-ct-2018-05-30-127f9818)

**Problem description:**

When using default country code (US) and trying to switch monitor mode interface channel to certain widths, operation
times out and the following message is seen in dmesg:
 
 \# iw dev ath3 set channel 40 HT40+
command failed: Connection timed out (-110)

\# dmesg
[  447.448592] ------------[ cut here ]------------
[  447.453290] WARNING: CPU: 0 PID: 0 at /home/jhaakana/git-trees/dev/Sapphire/Eye/swdev/ecp/Platform/compile/Lapidary/build_dir/target-arm_cortex-a9+vfpv3_glibc_eabi/linux-mvebu_cortexa9/ath10k-ct-2018-05-30-127f9818/ath10k-4.13/wmi.c:3419 ath10k_wmi_10_4_op_rx+0x4d4/0x69c [ath10k_core]
[  447.478642] Modules linked in: pppoe ppp_async pppox ppp_generic iptable_nat iptable_mangle iptable_filter ipt_REJECT ipt_MASQUERADE ip_tables xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD x_tables slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_log_common nf_flow_table_hw nf_flow_table nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack crc_ccitt ath10k_pci ath10k_core mwlwifi mac80211 nfs lockd sunrpc grace ath cfg80211 compat gpio_button_hotplug
[  447.531989] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W       4.14.51 #0
[  447.539244] Hardware name: Marvell Armada 375 (Device Tree)
[  447.544861] [<c0115328>] (unwind_backtrace) from [<c01105ac>] (show_stack+0x10/0x14)
[  447.552647] [<c01105ac>] (show_stack) from [<c07ce290>] (dump_stack+0x88/0x9c)
[  447.559907] [<c07ce290>] (dump_stack) from [<c012cc90>] (warn_slowpath_null+0xbc/0x114)
[  447.567985] [<c012cc90>] (warn_slowpath_null) from [<bf1e70cc>] (ath10k_wmi_10_4_op_rx+0x4d4/0x69c [ath10k_core])
[  447.578353] [<bf1e70cc>] (ath10k_wmi_10_4_op_rx [ath10k_core]) from [<bf1cfc18>] (ath10k_htc_rx_completion_handler+0x1a0/0x1b0 [ath10k_core])
[  447.591142] [<bf1cfc18>] (ath10k_htc_rx_completion_handler [ath10k_core]) from [<bf222b44>] (ath10k_pci_hif_fetch_cal_eeprom+0x744/0x758 [ath10k_pci])
[  447.604696] [<bf222b44>] (ath10k_pci_hif_fetch_cal_eeprom [ath10k_pci]) from [<bf227a24>] (ath10k_ce_per_engine_service_any+0x94/0x110 [ath10k_pci])
[  447.618074] [<bf227a24>] (ath10k_ce_per_engine_service_any [ath10k_pci]) from [<bf224a44>] (ath10k_pci_napi_poll+0x90/0x1ac [ath10k_pci])
[  447.630493] [<bf224a44>] (ath10k_pci_napi_poll [ath10k_pci]) from [<c06e7470>] (net_rx_action+0x134/0x374)
[  447.640196] [<c06e7470>] (net_rx_action) from [<c01018a0>] (__do_softirq+0xe0/0x2a4)
[  447.647979] [<c01018a0>] (__do_softirq) from [<c0133a18>] (irq_exit+0xb0/0xf8)
[  447.655240] [<c0133a18>] (irq_exit) from [<c01908d0>] (__handle_domain_irq+0xac/0xbc)
[  447.663109] [<c01908d0>] (__handle_domain_irq) from [<c010161c>] (gic_handle_irq+0x5c/0x90)
[  447.671498] [<c010161c>] (gic_handle_irq) from [<c011148c>] (__irq_svc+0x6c/0x90)
[  447.679013] Exception stack(0xc0b01f48 to 0xc0b01f90)
[  447.684091] 1f40:                   00000001 00000000 00000000 c011c2c0 ffffe000 c0b03c74
[  447.692308] 1f60: c0b03c24 00000000 00000000 414fc091 00000001 00000000 c0b01f84 c0b01f98
[  447.700523] 1f80: c010d0fc c010d100 600e0013 ffffffff
[  447.705603] [<c011148c>] (__irq_svc) from [<c010d100>] (arch_cpu_idle+0x34/0x38)
[  447.713040] [<c010d100>] (arch_cpu_idle) from [<c0182b24>] (do_idle+0xf0/0x1f4)
[  447.720387] [<c0182b24>] (do_idle) from [<c0182e84>] (cpu_startup_entry+0x18/0x1c)
[  447.727999] [<c0182e84>] (cpu_startup_entry) from [<c0a00d74>] (start_kernel+0x494/0x4a0)
[  447.736233] ---[ end trace 103d016a5a79645b ]---
[  452.473007] ath10k_pci 0000:02:00.0: failed to synchronize setup for vdev 0 restart 0: -110
[  452.481409] ath10k_pci 0000:02:00.0: failed to start vdev 0 addr 00:00:00:00:00:00 on freq 5200: -110
 
The channel widths and the channels having this problem are:
* VHT160: all channels (36-128)
* HT40-: 44, 52, 60, 108, 116, 124, 132, 140, 157, 165
* HT40+: 40, 48, 56, 104, 112, 120, 128, 136, 153, 161
 
HT20/VHT20/VHT40/VHT80 seem to be fine.
 
If we add a fwcfg file and set regdom to 840 (US) we see the same error. But if we change the country codein fwcfg to any country that does not belong to FCC3_FCCA regulatory domain (and use iw reg set to set corresponding country code), setting channel with all available / supported bandwidths works without issues.
 
On the other hand, if we set any country belonging to FCC3_FCCA regulatory domain (f.e. Canada) in fwcfg file (and set the country code with iw reg set), we are getting the same error.
 
If we switch to stock firmware (e.g. firmware-5.bin_10.4-3.2-00072), but still use the same ath10k-ct
driver, the problem does not occur. All (supported) channel widths on FCC3_FCCA can be set without issues. So it looks like the problem is specific to Candelach firmware-5?

greearb, 2018-07-13:
I am not able to reproduce this on an x86-64 machine running our 4.13 kernel.  I am using a different
firmware variant, but probably that is not the issue since it mostly just compiles out some un-needed code.

Maybe you could enable the firmware debugging and also instrument the driver to print out
the status code that it is complaining about and send me that dmesg output?

[root@lf0313-6477 lanforge]# iw phy wiphy1 interface add moni7a type monitor
[root@lf0313-6477 lanforge]# ip link set dev moni7a up
[root@lf0313-6477 lanforge]#  iw dev moni7a set channel 40 HT40+
command failed: Device or resource busy (-16)
[root@lf0313-6477 lanforge]# ifconfig wlan1 down
[root@lf0313-6477 lanforge]#  iw dev moni7a set channel 40 HT40+
[root@lf0313-6477 lanforge]# iw reg get
global
country 00: DFS-UNSET
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, NO-IR
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, NO-IR
	(4910 - 5170 @ 20), (N/A, 23), (N/A)
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, NO-IR
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, NO-IR
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, NO-IR
	(5735 - 5835 @ 80), (N/A, 20), (N/A), NO-IR
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#1
country US: DFS-FCC
	(2402 - 2472 @ 40), (N/A, 30), (N/A)
	(4910 - 5170 @ 20), (N/A, 23), (N/A)
	(5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
	(5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
	(5735 - 5835 @ 80), (N/A, 30), (N/A)
	(57240 - 63720 @ 2160), (N/A, 40), (N/A)

phy#0
country US: DFS-FCC
	(2402 - 2472 @ 40), (N/A, 30), (N/A)
	(4910 - 5170 @ 20), (N/A, 23), (N/A)
	(5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
	(5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
	(5735 - 5835 @ 80), (N/A, 30), (N/A)
	(57240 - 63720 @ 2160), (N/A, 40), (N/A)

[root@lf0313-6477 lanforge]# ethtool -i moni7a
driver: ath10k_pci
version: 4.13.16+
firmware-version: 10.4-ct-9984-xtH-010-868495e
expansion-rom-version: 
bus-info: 0000:04:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
[root@lf0313-6477 lanforge]# 

[root@lf0313-6477 lanforge]# cat /lib/firmware/ath10k/fwcfg-pci-0000:04:00.0.txt
# Created by LANforge.  LANforge will over-write this file
# unless you add the string LEAVE-ME-Be  (with last E also capitalized)

vdevs = 64
peers = 128
active_peers = 128
stations = 128
rate_ctrl_objs = 7
regdom = 840
fwname = firmware-5-htt-mgt.bin
fwver = 5
nohwcrypt = 1
tx_desc = 2200
#max_nss = 3
tids = 256
skid_limit = 360
max_amsdus = 3
bname = WLE1216V5-2-board.bin

[root@lf0313-6477 lanforge]# iw dev moni7a info
Interface moni7a
	ifindex 19
	wdev 0x100000002
	addr 04:f0:21:38:98:f3
	type monitor
	wiphy 1
	channel 40 (5200 MHz), width: 40 MHz, center1: 5210 MHz
	txpower 0.00 dBm
[root@lf0313-6477 lanforge]#

jhaakana, 2018-07-16:
Did that. Enabling FW debugging does not provide extra information. Status code returned is 1:

\# iw reg get
global
country US: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
        (5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)
        (57240 - 63720 @ 2160), (N/A, 40), (N/A)

phy#1
country US: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
        (5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)
        (57240 - 63720 @ 2160), (N/A, 40), (N/A)

phy#0
country US: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
        (5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)
        (57240 - 63720 @ 2160), (N/A, 40), (N/A)

\# ethtool -i ath3
driver: ath10k_pci
version: 4.14.51
firmware-version: 10.4-ct-9984-fW-010-868495e
expansion-rom-version: 
bus-info: 0000:02:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
\# echo 0 > /sys/kernel/debug/ieee80211/phy1/ath10k/debug_level 
\# iw dev ath3 set channel 40 HT40+
command failed: Connection timed out (-110)
\# dmesg 
[  781.934059] ath10k_pci 0000:02:00.0: Setting debug-mask to: 0x0  old: 0x0
[  792.368189] ------------[ cut here ]------------
[  792.372910] WARNING: CPU: 0 PID: 0 at /home/jhaakana/git-trees/dev/Sapphire/Eye/swdev/ecp/Platform/compile/Lapidary/build_dir/target-arm_cortex-a9+vfpv3_glibc_eabi/linux-mvebu_cortexa9/ath10k-ct-2018-05-30-127f9818/ath10k-4.13/wmi.c:3419 ath10k_wmi_10_4_op_rx+0x4d4/0x6b0 [ath10k_core]
[  792.398234] Modules linked in: pppoe ppp_async pppox ppp_generic iptable_nat iptable_mangle iptable_filter ipt_REJECT ipt_MASQUERADE ip_tables xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD x_tables slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_log_common nf_flow_table_hw nf_flow_table nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack crc_ccitt ath10k_pci ath10k_core mwlwifi mac80211 nfs lockd sunrpc grace ath cfg80211 compat gpio_button_hotplug
[  792.451597] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W       4.14.51 #0
[  792.458853] Hardware name: Marvell Armada 375 (Device Tree)
[  792.464470] [<c0115328>] (unwind_backtrace) from [<c01105ac>] (show_stack+0x10/0x14)
[  792.472256] [<c01105ac>] (show_stack) from [<c07ce290>] (dump_stack+0x88/0x9c)
[  792.479515] [<c07ce290>] (dump_stack) from [<c012cc90>] (warn_slowpath_null+0xbc/0x114)
[  792.487594] [<c012cc90>] (warn_slowpath_null) from [<bf1e711c>] (ath10k_wmi_10_4_op_rx+0x4d4/0x6b0 [ath10k_core])
[  792.497961] [<bf1e711c>] (ath10k_wmi_10_4_op_rx [ath10k_core]) from [<bf1cfc18>] (ath10k_htc_rx_completion_handler+0x1a0/0x1b0 [ath10k_core])
[  792.510750] [<bf1cfc18>] (ath10k_htc_rx_completion_handler [ath10k_core]) from [<bf222b44>] (ath10k_pci_hif_fetch_cal_eeprom+0x744/0x758 [ath10k_pci])
[  792.524304] [<bf222b44>] (ath10k_pci_hif_fetch_cal_eeprom [ath10k_pci]) from [<bf227a24>] (ath10k_ce_per_engine_service_any+0x94/0x110 [ath10k_pci])
[  792.537682] [<bf227a24>] (ath10k_ce_per_engine_service_any [ath10k_pci]) from [<bf224a44>] (ath10k_pci_napi_poll+0x90/0x1ac [ath10k_pci])
[  792.550101] [<bf224a44>] (ath10k_pci_napi_poll [ath10k_pci]) from [<c06e7470>] (net_rx_action+0x134/0x374)
[  792.559804] [<c06e7470>] (net_rx_action) from [<c01018a0>] (__do_softirq+0xe0/0x2a4)
[  792.567587] [<c01018a0>] (__do_softirq) from [<c0133a18>] (irq_exit+0xb0/0xf8)
[  792.574848] [<c0133a18>] (irq_exit) from [<c01908d0>] (__handle_domain_irq+0xac/0xbc)
[  792.582716] [<c01908d0>] (__handle_domain_irq) from [<c010161c>] (gic_handle_irq+0x5c/0x90)
[  792.591106] [<c010161c>] (gic_handle_irq) from [<c011148c>] (__irq_svc+0x6c/0x90)
[  792.598622] Exception stack(0xc0b01f18 to 0xc0b01f60)
[  792.603697] 1f00:                                                       ce73c000 c0b00000
[  792.611915] 1f20: 00000000 0000f4b8 cfd98fc0 00000000 0f34c000 ce73c000 c0a4cfc0 00000002
[  792.620133] 1f40: c0b06c8c c0b01fac 00000000 c0b01f68 c015c2f8 c07ea5bc 60080013 ffffffff
[  792.628350] [<c011148c>] (__irq_svc) from [<c07ea5bc>] (__schedule+0x5c0/0x71c)
[  792.635696] [<c07ea5bc>] (__schedule) from [<c07eb138>] (schedule_idle+0x64/0x74)
[  792.643219] [<c07eb138>] (schedule_idle) from [<c0182e84>] (cpu_startup_entry+0x18/0x1c)
[  792.651355] [<c0182e84>] (cpu_startup_entry) from [<c0a00d74>] (start_kernel+0x494/0x4a0)
[  792.659588] ---[ end trace 5749a81eeedbf009 ]---
**[  792.664235] ath10k_pci 0000:02:00.0: Status code: 1**
[  797.418894] ath10k_pci 0000:02:00.0: failed to synchronize setup for vdev 0 restart 0: -110
[  797.427304] ath10k_pci 0000:02:00.0: failed to start vdev 0 addr 00:00:00:00:00:00 on freq 5200: -110

jhaakana, 2018-07-17:
Ben, I did some tests with older FW versions. FW-8 (10.4.3-ct-fW-008-ff28fae) does not have this issue yet, it seems to be first introduced in FW-9 (10.4-ct-9984-fW-009-dfa0083). I hope this helps.

greearb, 2018-07-17:
I will build a series of binaries for each git commit between those releases and maybe you can use those to bisect?  It will be a day or two before I have time to work on this...  If you can find the bad commit, it should make it easier for me to find the fix.

jhaakana, 2018-07-18:
Thanks, I can definitely do that. Just send me the binaries.

greearb, 2018-07-19:
Please try bisecting with these.  I might have rebased at least a bit since rls 8, so double check that the initial build works and the latest fails before wasting too much time testing the others.

https://www.candelatech.com/downloads/all_builds.9984-fW-july-2018.tar.gz

jhaakana, 2018-07-19:
The last version that does not have this problem is firmware-5-full-community-commit-699-d09449b.bin. Starting from firmware-5-full-community-commit-700-2262c75.bin the problem exists.

greearb, 2018-07-19:
Can you please retest this?  The commit 2262c75 just removes a single line that would not compile and is #ifdef'd out anyway.  The commit after it changes something in a file that is not used for 9984, and the one before just removes an assert that was not needed.  So, I cannot explain how that bisect could be correct.  If your retest shows the same thing, then I will manually build those images just in case my automatic series compile script somehow messed up the commit ids or something.

jhaakana, 2018-07-20:
Results below, showing commands I executed. It still shows the same. The first try is with 699 and the second with 700:

```
root@Eye-234:~# rmmod ath10k_pci
root@Eye-234:~# rmmod ath10k_core
root@Eye-234:~# rmmod ath
root@Eye-234:~# rmmod mac80211
root@Eye-234:~# rmmod cfg80211
root@Eye-234:~# cp all_builds/firmware-5-full-community-commit-699-d09449b.bin /lib/firmware/ath10k/QCA9984/hw1.0/firmware-5.bin
root@Eye-234:~# modprobe ath10k_pci
root@Eye-234:~# iw phy phy1 interface add ath1 type monitor
root@Eye-234:~# ifconfig ath1 up
root@Eye-234:~# iw dev ath1 set channel 40 HT40+
root@Eye-234:~# iw dev ath1 set channel 124 HT40-
root@Eye-234:~# iw dev ath1 set channel 157 HT40-
root@Eye-234:~# iw dev ath1 set channel 112 HT40+
root@Eye-234:~# dmesg 
[74567.590103] ath10k_pci 0000:02:00.0: removing stale peer 00:00:00:00:00:00 from vdev_id 0
[74567.598365] ath10k_pci 0000:02:00.0: removing peer mapping 00:00:00:00:00:00 peer-id: 0
[74567.606456] ath10k_pci 0000:02:00.0: peer-unmap-event: unknown peer id 0
[74567.930047] ath10k_pci 0000:01:00.0: removing stale peer 00:00:00:00:00:00 from vdev_id 0
[74567.938306] ath10k_pci 0000:01:00.0: removing peer mapping 00:00:00:00:00:00 peer-id: 0
[74567.946398] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 0
[74656.065495] ath10k 4.13 driver, optimized for CT firmware, probing pci device: 0x46.
[74656.074005] ath10k_pci 0000:01:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[74656.260237] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/fwcfg-pci-0000:01:00.0.txt failed with error -2
[74656.270837] ath10k_pci 0000:01:00.0: Falling back to user helper
[74656.288081] firmware ath10k!fwcfg-pci-0000:01:00.0.txt: firmware_loading_store: map pages failed
[74656.297150] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/pre-cal-pci-0000:01:00.0.bin failed with error -2
[74656.307937] ath10k_pci 0000:01:00.0: Falling back to user helper
[74656.325016] firmware ath10k!pre-cal-pci-0000:01:00.0.bin: firmware_loading_store: map pages failed
[74656.334229] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/cal-pci-0000:01:00.0.bin failed with error -2
[74656.344663] ath10k_pci 0000:01:00.0: Falling back to user helper
[74656.362457] firmware ath10k!cal-pci-0000:01:00.0.bin: firmware_loading_store: map pages failed
[74656.371336] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/QCA9984/hw1.0/ct-firmware-5.bin failed with error -2
[74656.382398] ath10k_pci 0000:01:00.0: Falling back to user helper
[74656.400115] firmware ath10k!QCA9984!hw1.0!ct-firmware-5.bin: firmware_loading_store: map pages failed
[74656.409578] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/QCA9984/hw1.0/ct-firmware-2.bin failed with error -2
[74656.420624] ath10k_pci 0000:01:00.0: Falling back to user helper
[74656.438049] firmware ath10k!QCA9984!hw1.0!ct-firmware-2.bin: firmware_loading_store: map pages failed
[74656.447517] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/QCA9984/hw1.0/firmware-6.bin failed with error -2
[74656.458329] ath10k_pci 0000:01:00.0: Falling back to user helper
[74656.475894] firmware ath10k!QCA9984!hw1.0!firmware-6.bin: firmware_loading_store: map pages failed
[74656.486482] ath10k_pci 0000:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[74656.496504] ath10k_pci 0000:01:00.0: kconfig debug 1 debugfs 1 tracing 0 dfs 1 testmode 0
[74656.507837] ath10k_pci 0000:01:00.0: firmware ver 10.4-ct-9984-fW-009-d09449b api 5 features peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT crc32 aec92d06
[74657.969665] ath10k_pci 0000:01:00.0: board_file api 2 bmi_id 0:30 crc32 bb2ad9d0
[74660.599023] ath10k_pci 0000:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[74660.606552] ath10k_pci 0000:01:00.0: msdu-desc: 2500  skid: 32
[74660.685250] ath10k_pci 0000:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0'
[74660.695360] ath10k_pci 0000:01:00.0: wmi print 'free: 95140 iram: 15892 sram: 26592'
[74660.857305] ath10k_pci 0000:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal otp max-sta 32 raw 0 hwcrypto 1
[74660.980295] ath: EEPROM regdomain: 0x0
[74660.980301] ath: EEPROM indicates default country code should be used
[74660.980304] ath: doing EEPROM country->regdmn map search
[74660.980310] ath: country maps to regdmn code: 0x3a
[74660.980314] ath: Country alpha2 being used: US
[74660.980317] ath: Regpair used: 0x3a
[74660.986048] ath10k 4.13 driver, optimized for CT firmware, probing pci device: 0x46.
[74660.995573] ath10k_pci 0000:02:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[74661.173965] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/fwcfg-pci-0000:02:00.0.txt failed with error -2
[74661.184620] ath10k_pci 0000:02:00.0: Falling back to user helper
[74661.372786] firmware ath10k!fwcfg-pci-0000:02:00.0.txt: firmware_loading_store: map pages failed
[74661.381912] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/pre-cal-pci-0000:02:00.0.bin failed with error -2
[74661.392712] ath10k_pci 0000:02:00.0: Falling back to user helper
[74661.410425] firmware ath10k!pre-cal-pci-0000:02:00.0.bin: firmware_loading_store: map pages failed
[74661.419614] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/cal-pci-0000:02:00.0.bin failed with error -2
[74661.430044] ath10k_pci 0000:02:00.0: Falling back to user helper
[... cut ...]
[74892.002698] [<c010d100>] (arch_cpu_idle) from [<c0182b24>] (do_idle+0xf0/0x1f4)
[74892.010044] [<c0182b24>] (do_idle) from [<c0182e84>] (cpu_startup_entry+0x18/0x1c)
[74892.017656] [<c0182e84>] (cpu_startup_entry) from [<c0a00d74>] (start_kernel+0x494/0x4a0)
[74892.025882] ---[ end trace 7db178330c89b3f1 ]---
[74892.030527] ath10k_pci 0000:02:00.0: Status code: 1
[74896.754916] ath10k_pci 0000:02:00.0: failed to synchronize setup for vdev 0 restart 0: -110
[74896.763317] ath10k_pci 0000:02:00.0: failed to start vdev 0 addr 00:00:00:00:00:00 on freq 5560: -110
```

greearb, 2018-07-20:
I see the issue with the bisect.  There were build bugs and my script that builds all commits is not smart enough to know when the build fails.  I am building new images...it might be easier to send them to you directly via email.  If you prefer, please email me direct at greearb@candelatech.com and I'll send you the binaries for bisecting.

greearb, 2018-07-20:
I uploaded a new set of images, please bisect these and let me know where the problem lies.  In additiona, please send me the dmesg logs showing the firmware version as it loads of both the good and bad images in case there were more build issues.  I think the bad one is probably within 50 commits of the starting one.

http://www.candelatech.com/downloads/all_builds.9984-fW-july-2018-2.tar.gz

jhaakana, 2018-07-23:
Thanks, I'll send bisect results over email.

greearb, 2018-07-24:
I resolved the problem by compiling out the 'WREG' logic in the firmware.  Will let host be in control of the regulatory domain channel limitations.  Available in current beta 9984 firmware, and will make it into other firmware next time I do builds.

closed 2018-07-24 by greearb