greearb/ath10k-ct #26: fw_stats: wrong phantom peer with some garbage stat information generated from interface mac address

klukonin, 2018-07-20:
fw_stats report wrong information about peers.
If we look intocat /sys/kernel/debug/ieee80211/phy0/ath10k/fw_stats
there will be a strange behavior.
Phantom peer is always present, even if there are no any stations connected.
It's always first and it's statistics depends on the first interface that was created.

Please look at the first peer and it's mac address. The first octet behave like a counter!
This behavior is reproducible on all 5 nodes in our lab at the same time.

         ath10k PEER stats (3)
         =================

          Peer MAC address 65:01:00:00:1e:d4
                 Peer RSSI 55633
              Peer TX rate 0
              Peer RX rate 6000
          Peer RX duration 0

          Peer MAC address 0a:24:68:48:51:d9
                 Peer RSSI 0
              Peer TX rate 6000
              Peer RX rate 0
          Peer RX duration 0

          Peer MAC address 3e:9d:75:48:54:6a
                 Peer RSSI 9
              Peer TX rate 58500
              Peer RX rate 175500
          Peer RX duration 0

         ath10k PEER stats (3)
         =================

          Peer MAC address 67:01:00:00:1e:d4
                 Peer RSSI 55633
              Peer TX rate 0
              Peer RX rate 6000
          Peer RX duration 0

          Peer MAC address 0a:24:68:48:51:d9
                 Peer RSSI 0
              Peer TX rate 6000
              Peer RX rate 0
          Peer RX duration 0

          Peer MAC address 3e:9d:75:48:54:6a
                 Peer RSSI 9
              Peer TX rate 58500
              Peer RX rate 175500
          Peer RX duration 0

         ath10k PEER stats (3)
         =================

          Peer MAC address 69:01:00:00:1e:d4
                 Peer RSSI 55633
              Peer TX rate 0
              Peer RX rate 6000
          Peer RX duration 0

          Peer MAC address 0a:24:68:48:51:d9
                 Peer RSSI 0
              Peer TX rate 6000
              Peer RX rate 0
          Peer RX duration 0

          Peer MAC address 3e:9d:75:48:54:6a
                 Peer RSSI 9
              Peer TX rate 58500
              Peer RX rate 175500
          Peer RX duration 0

         ath10k PEER stats (3)
         =================

          Peer MAC address 6b:01:00:00:1e:d4
                 Peer RSSI 55633
              Peer TX rate 0
              Peer RX rate 6000
          Peer RX duration 0

          Peer MAC address 0a:24:68:48:51:d9
                 Peer RSSI 0
              Peer TX rate 6000
              Peer RX rate 0
          Peer RX duration 0

          Peer MAC address 3e:9d:75:48:54:6a
                 Peer RSSI 9
              Peer TX rate 58500
              Peer RX rate 175500
          Peer RX duration 0

ip addr shows
ibss_5g_1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1564 qdisc mq master bat0 state UP group default qlen 1000
link/ether 3a:ed:ad:48:47:cf brd ff:ff:ff:ff:ff:ff

3a:ed is the first two octets of the interface mac address and the last two octets of phantom peer.
Also there is a strange static RSSI value.

printf %x\n 53063
cf47

It's a reverse order of last two octets from ibss interface mac.

AP interface have the same issue:

         ath10k PEER stats (1)
         =================

          Peer MAC address 2b:00:00:00:5a:be
                 Peer RSSI 45976
              Peer TX rate 0
              Peer RX rate 0
          Peer RX duration 0

ap_5g_1_1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
link/ether 5a:be:7b:80:98:b3 brd ff:ff:ff:ff:ff:ff

root@OpenWrt:~# printf %x/n 45976
b398

The same pieces of mac and last two octets with reverse order.

I can see a lot of (thousands) phy errors in fw_stats also.

Firmware ver 10.1-ct-8x-__fH-021-4fa9f30 api 2 features wmi-10.x,txstatus-noack,wmi-10.x-CT,ratemask-CT,get-temp-CT,tx-rc-CT,cust-stats-CT crc32 580c6146

Openwrt version is 18.06
Device: UBNT UniFi AP AC Lite (QCA988x)

klukonin, 2018-07-25:
I've just ckecked fwstats with the original firmware.
Looks like this bug is not present.
             ath10k PEER stats (4)
             =================

              Peer MAC address 22:a0:53:48:54:6a
                     Peer RSSI 0
                  Peer TX rate 0
                  Peer RX rate 0
              Peer RX duration 0

              Peer MAC address 7a:49:68:48:47:cf
                     Peer RSSI 27
                  Peer TX rate 325000
                  Peer RX rate 650000
              Peer RX duration 0

              Peer MAC address ee:1c:1b:48:44:e5
                     Peer RSSI 7
                  Peer TX rate 58500
                  Peer RX rate 117000
              Peer RX duration 0

              Peer MAC address 86:84:30:48:51:d9
                     Peer RSSI 11
                  Peer TX rate 175500
                  Peer RX rate 351000
              Peer RX duration 0

firmware ver 10.2.4.70.67 api 5 features no-p2p,raw-mode,mfp,allows-mesh-bcast crc32 c7476231

greearb, 2018-07-25:
Please try latest wave-1 ct beta firmware.  It should fix this problem.  It also stops showing self-peers (the ones with mac that had only 0x1 octet set.  There were several bugs, one of which had been around since 2016 when I backported some 10.2 code into my FW and made the pdev stats struct 4 bytes larger than the driver expected.

closed 2018-07-25 by greearb