greearb/ath10k-ct #28: kernel: ath10k_pci 0000:01:00.0: dropping dbg buffer due to crash since read

bill-qacafe, 2018-07-30:
Description of the problem (how to configure, how to reproduce, how often it happens).
Wifi virtual clients occasionally disassociate from the AP for some reason.  In /var/log/messages and dmesg, I see:

kernel: ath10k_pci 0000:01:00.0: dropping dbg buffer due to crash since read
and
No ack for nullfunc frame to AP f0:9f:c2:ff:2f:4a, disconnecting.

The client then reassociates and some time later this may happen again.  I see no pattern as to when this occurs, and it happens to all virtual clients eventually.  I've tried changing some of their MAC addresses to make them less similar, but no change.
I use the following commands to create the virtual clients:

iw dev wlan1 interface add cdrwifi1  type managed addr b0:75:0c:00:00:01; iw dev cdrwifi1  set power_save off; iw phy phy0 set coverage 0; sleep 1
iw dev wlan1 interface add cdrwifi2  type managed addr b0:75:0c:00:00:02; iw dev cdrwifi2  set power_save off; iw phy phy0 set coverage 0; sleep 1
iw dev wlan1 interface add cdrwifi3  type managed addr b0:75:0c:00:00:03; iw dev cdrwifi3  set power_save off; iw phy phy0 set coverage 0; sleep 1
(Repeat for additional clients, changing the name and MAC address.)

/usr/local/bin/wpa_supplicant -B -dd -Dnl80211 -c /home/qacafe/wifi/wpa.conf -i cdrwifi1 -N \
                                                         -Dnl80211 -c /home/qacafe/wifi/wpa.conf -i cdrwifi2 -N \
                                                         -Dnl80211 -c /home/qacafe/wifi/wpa.conf -i cdrwifi3
(One entry for each client.)
dhclient cdrwifi1 cdrwifi2 cdrwifi3
(Again, list each client.)

Software (OS, Firmware version, kernel, driver, etc)
# cat /etc/centos-release
CentOS Linux release 7.5.1804 (Core)
# ethtool -i wlan1
driver: ath10k_pci
version: 4.13.16-20180608.8eeb88f9.el7.q
firmware-version: 10.1-ct-8x-__xtW-020-0468ebf
expansion-rom-version: 
bus-info: 0000:01:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no

Hardware (NIC chipset, platform, etc)
System is an iBASE FWA8408, wifi NIC is WLE900VX 7AA

Logs (dmesg, maybe supplicant and/or hostap)
I'm using CandelaTech wpa_supplicant
# /usr/local/bin/wpa_supplicant -version:
wpa_supplicant v2.7-devel-hostap_2_3-5889-g91b1c54
Copyright (c) 2003-2017, Jouni Malinen <j@w1.fi> and contributors
I built it with the following .config:
# cat .config 
CONFIG_DRIVER_NL80211=y
CONFIG_IEEE8021X_EAPOL=y
CONFIG_EAP_MD5=y
CONFIG_EAP_MSCHAPV2=y
CONFIG_EAP_TLS=y
CONFIG_EAP_PEAP=y
CONFIG_EAP_TTLS=y
CONFIG_EAP_GTC=y
CONFIG_EAP_IKEV2=y
CONFIG_AP=y

Here's the wpa.conf file used:
# cat /home/qacafe/wifi/wpa.conf 
#ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=wheel
#p2p_disabled=1
network={
    ssid="Ubiquiti_NTAv6_testing"
    key_mgmt=WPA-PSK
    psk="qacafe123"
    ap_max_inactivity=36000
    #ap_max_inactivity=60
}

greearb, 2018-07-30:
Please attach full 'dmesg' output (or full 'journalctl -b0' or similar if that seems relevant), and if possible, the binary firmware crash dump file.  The dmesg output should have firmware crash logs that I can decode, and sometimes the binary dump file has additional info that can help me understand the crash.

bill-qacafe, 2018-07-30:
[dmesg.txt](https://github.com/greearb/ath10k-ct/files/2242611/dmesg.txt)
[fw_crash_dump.txt](https://github.com/greearb/ath10k-ct/files/2242612/fw_crash_dump.txt)
[journalctl.txt](https://github.com/greearb/ath10k-ct/files/2242613/journalctl.txt)

greearb, 2018-07-30:
This crash appears to be a WMI hang.  In general, this is a known bug, and I do not know any way to make more progress on fixing it.  I notice you are using WMI management firmware, and maybe that will exacerbate any WMI communication issues.  I suggest using the HTT mgt firmware version as communicated via email.

I am closing this bug, please open a new one if the htt-mgt firmware has issues.

closed 2018-07-30 by greearb