greearb/ath10k-ct #29: Problem with scanning hidden SSID

jhaakana, 2018-08-06:
WLAN module: WLE1216V2-20
Firmware: 10.4-ct-9984-fW-011-da1d791
board-2.bin: stock board-2.bin or board-2-ct.bin (from https://www.candelatech.com/downloads/ath10k-9984-10-4/board-2-ct.bin)
OS: LEDE latest, mvebu/cortexa9, arm_cortex-a9_vfpv3
Kernel: 4.14.51
Driver: ath10k-ct 4.13 (ath10k-ct-2018-05-30-127f9818)

**Problem description:**

ath10k-ct with CT firmware fails to scan a hidden SSID, when scanning on single channel. ath10k-ct driver with stock firmware does not have this problem.

**Setup:**

Interface ath0 in managed mode, interface ath1 in monitor mode. Cisco standalone AP, two SSIDs, "7signal" is beaconing, "7fast" is hidden. Channel 13 (2472MHz).

**Test case:**

```
# ifconfig ath0 down
# iw dev ath1 set channel 13
# ifconfig ath0 up
# iw dev ath0 scan freq 2472 ssid 7fast
# iw dev ath0 scan freq 2472 ssid 7fast
# iw dev ath0 scan freq 2472 ssid 7fast
# iw dev ath0 scan freq 2472 ssid 7fast
BSS 58:bc:27:5d:8b:b0(on ath0)
        TSF: 9491880875384 usec (109d, 20:38:00)
        freq: 2472
        beacon interval: 323 TUs
        capability: ESS (0x0431)
        signal: -56.00 dBm
        last seen: 0 ms ago
        Information elements from Probe Response frame:
        SSID: 7signal
        HT capabilities:
                Capabilities: 0x186e
                        HT20/HT40
                        SM Power Save disabled
                        RX HT20 SGI
                        RX HT40 SGI
                        No RX STBC
                        Max AMSDU length: 7935 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 8 usec (0x06)
                HT RX MCS rate indexes supported: 0-15
                HT TX MCS rate indexes are undefined
        RSN:     * Version: 1
                 * Group cipher: CCMP
                 * Pairwise ciphers: CCMP
                 * Authentication suites: PSK
                 * Capabilities: 4-PTKSA-RC 4-GTKSA-RC (0x0028)
        HT operation:
                 * primary channel: 13
                 * secondary channel offset: no secondary
                 * STA channel width: 20 MHz
```
tcpdump on ath1 shows only a beacon (not probe responses)

`10:52:21.277663 470035344us tsft 1.0 Mb/s 2472 MHz 11b -58dBm signal -69dBm signal antenna 0 -72dBm signal antenna 1 -59dBm signal antenna 2 -66dBm signal antenna 3 Beacon (7signal) [1.0* 2.0* 5.5* 6.0* 9.0* 11.0* 12.0* 18.0* Mbit] ESS CH: 13, PRIVACY`

With stock FW:

```
# ifconfig ath0 down
# iw dev ath1 set channel 13
# ifconfig ath0 up
# iw dev ath0 scan freq 2472 ssid 7fast
BSS 58:bc:27:5d:8b:b0(on ath0)
        TSF: 9492192296501 usec (109d, 20:43:12)
        freq: 2472
        beacon interval: 323 TUs
        capability: ESS (0x0431)
        signal: -56.00 dBm
        last seen: 50 ms ago
        SSID: 7fast
        HT capabilities:
                Capabilities: 0x186e
                        HT20/HT40
                        SM Power Save disabled
                        RX HT20 SGI
                        RX HT40 SGI
                        No RX STBC
                        Max AMSDU length: 7935 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 8 usec (0x06)
                HT RX MCS rate indexes supported: 0-15
                HT TX MCS rate indexes are undefined
        RSN:     * Version: 1
                 * Group cipher: CCMP
                 * Pairwise ciphers: CCMP
                 * Authentication suites: IEEE 802.1X
                 * Capabilities: 4-PTKSA-RC 4-GTKSA-RC (0x0028)
        HT operation:
                 * primary channel: 13
                 * secondary channel offset: no secondary
                 * STA channel width: 20 MHz
```

tcpdump on ath1 shows beacons and probe responses from the hidden SSID:

```
10:57:19.108469 136384411us tsft 1.0 Mb/s 2472 MHz 11b -56dBm signal -70dBm signal antenna 0 -72dBm signal antenna 1 -57dBm signal antenna 2 -64dBm signal antenna 3 Probe Response (7fast) [1.0* 2.0* 5.5* 6.0* 9.0* 11.0* 12.0* 18.0* Mbit] CH: 13, PRIVACY
10:57:22.699111 139975661us tsft bad-fcs 48.0 Mb/s 2472 MHz 11g -86dBm signal -93dBm signal antenna 0 -89dBm signal antenna 1 -93dBm signal antenna 2 -96dBm signal antenna 3 unknown 802.11 frame type (3)
10:57:29.943393 147216231us tsft 1.0 Mb/s 2472 MHz 11b -57dBm signal -67dBm signal antenna 0 -64dBm signal antenna 1 -60dBm signal antenna 2 -68dBm signal antenna 3 Probe Response (7fast) [1.0* 2.0* 5.5* 6.0* 9.0* 11.0* 12.0* 18.0* Mbit] CH: 13, PRIVACY
```

greearb, 2018-08-06:
Can you try the htt-mgt firmware and see if that acts differently?  That should help narrow down where the problem lies.

jhaakana, 2018-08-07:
Tested with 10.4b-ct-9984-fH-011-32616f6, no difference.

jhaakana, 2018-08-08:
By doing a packet capture in promiscuous mode on another device, I can see that probe requests are sent out, and the AP is responding with probe responses.

greearb, 2018-08-08:
Our 9984 NICs can only do 5Ghz, but hidden-ssid seems to work with it in station mode.  We used this supplicant config, you can comment out stuff your supplicant does not recognize.

Do you have a 5Ghz 9984 NIC you could test on to make sure it is not something related to
the frequency?

ctrl_interface=/var/run/wpa_supplicant
fast_reauth=1
driver_param="tx_legacy_rates=4080,adv_legacy_rates=4080"
concurrent_assoc_ok=1
accept_external_scan_results=1
scan_cur_freq=1
min_scan_gap=5
p2p_disabled=1
bss_max_count=2000
disable_ess_roaming=0

network={
    ssid="ubnt-pro-open"
    disable_ht=0
    disable_vht=0
    max_oper_chwidth=1
    ieee80211w=0
    disable_ht40=0
    disable_sgi=0
    disable_max_amsdu=-1
    ampdu_factor=-1
    ampdu_density=-1
    scan_ssid=1

    key_mgmt=NONE
    pairwise=NONE
    proactive_key_caching=0

}

klukonin, 2018-08-08:
Hello. I'm a little bit confused about **driver_param**.
Should it be like **driver_param="tx_legacy_rates=480,adv_legacy_rates=480**" ?

greearb, 2018-08-08:
Please comment that out, it is not needed for your test and won't work as requested unless you are using our modified kernel and supplicant.  The text I posted configures it for default values anyway.

jhaakana, 2018-08-09:
I've tested channels 11 and 13, but I also have a 5 GHz NIC and I'll test with it as well.

jhaakana, 2018-08-09:
No difference with 5 GHz NIC.

Notice that our use case is to scan single channel, not all channels of the band. 

If I scan all channels on 2.4 (leaving "freq" out from iw scan command), hidden SSID is returned in the scan results. This does not happen on 5GHz, hidden SSID is not returned even when scanning all channels.

greearb, 2018-08-09:
In your original report, you indicate you do not see a probe response.  But, you show no probe request either.  That seems more important.  If you sniff with proper filter, are you able to see probe requests for the hidden SSID?

And, since we only have a 5Ghz 9984 to test with, maybe you can do the test for probe-requests on 5Ghz instead?

jhaakana, 2018-08-10:
You are right - probe requests are not going out. I ran the scan command 8 times until it finally sent out a probe request:

```
...
# iw dev ath2 scan freq 5540 ssid 7fast | grep 7fast
# iw dev ath2 scan freq 5540 ssid 7fast | grep 7fast
# iw dev ath2 scan freq 5540 ssid 7fast | grep 7fast
# iw dev ath2 scan freq 5540 ssid 7fast | grep 7fast
        SSID: 7fast

```
On external device that executed tcpdump with correct filter:

```
# tcpdump -i ath1 wlan addr1 04:F0:21:39:12:AB or wlan addr2 04:F0:21:39:12:AB or wlan addr3 04:F0:21:39:12:AB
tcpdump: WARNING: ath1: no IPv4 address assigned
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on ath1, link-type IEEE802_11_RADIO (802.11 plus radiotap header), capture size 65535 bytes

13:53:09.372492 6.0 Mb/s 5540 MHz 11a -19dB signal antenna 0 Probe Request (7fast) [6.0 9.0 12.0 18.0 24.0 36.0 48.0 54.0 Mbit]
13:53:09.372887 6.0 Mb/s 5540 MHz 11a -33dB signal antenna 0 Probe Response (7fast) [6.0* 9.0 18.0 24.0* 36.0 48.0 54.0 Mbit], PRIVACY

```

So it looks like probe requests are not sent out reliably. This seems to with active scan, even without providing specific SSID. I cannot see probe requests with wildcard SSID on another device tcpdump. The FW version I used for this test was 10.4b-ct-9984-fH-011-32616f6.

greearb, 2018-08-10:
Please enable the firmware dbglog logging in the driver and re-run your test and then send me the 'dmesg' output (or more complete kernel logs if available).  I'm attaching a firmware with more scan related debugging enabled, please use that.  This firmware is for 9984 chipset.
[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2279129/firmware-5-full-htt-mgt-community.bin.gz)

Please double-check with a sniffer to see if probe-request hit the air or not.

jhaakana, 2018-08-13:
Did that. This time I did not see hidden SSID "7fast" in scan results at all, but I saw probe requests / responses occasionally on the sniffing device

**First attempt:**

```
# ifconfig ath2 down
# iw dev ath3 set channel 108
# ifconfig ath2 up
# iw dev ath2 scan freq 5540 ssid 7fast | grep fast
```

dmesg:

```
[  347.928578] ath10k_pci 0000:02:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[  347.928590] ath10k: [0000]: 0004CE67 14006402 71103331 00000000 00000100 00000001 00000000 0004CE67
[  347.928603] ath10k: [0008]: 14006402 71103332 00000000 00000100 0042E06C 00000000 0004CE67 14005854
[  347.928610] ath10k: [0016]: 51100002 000F4B38 00000057 00000000 004541A0 0004CE67 10005852 004305EC
[  347.928616] ath10k: [0024]: 004541A0 00000000 0042E06C 0004CE67 14005853 51100001 000F4B38 000003FC
[  347.928622] ath10k: [0032]: 00000001 004541A0 0004CE68 14005853 51100001 000F4AEC 00000408 00000002
[  347.928629] ath10k: [0040]: 004541A0 0004CE68 14005854 51100002 000F4AEC 0000008A 00000001 004541A0
[  347.928635] ath10k: [0048]: 0004CE68 10005851 004305EC 004541A0 00000000 00000000 0004CE68 14006403
[  347.928641] ath10k: [0056]: 00000000 00000000 00000000 00000001 004541A0 0004CE68 10005858 00003112
[  347.928647] ath10k: [0064]: 004305EC 00000003 FFFFFFFF 0004CE68 08003812 0045A8D4 004305EC 0004CE68
[  347.928653] ath10k: [0072]: 10003809 0000143C 00000003 00000000 00000000 0004CE68 0BFC5805 0000143C
[  347.928659] ath10k: [0080]: 00000000 0004CF89 17FC3C2A 61100001 01000001 01010000 143C0000 00000002
[  347.928665] ath10k_pci 0000:02:00.0: ATH10K_END
[  348.928590] ath10k_pci 0000:02:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[  348.928603] ath10k: [0000]: 0004CF8A 04003814 00000000 0004CF8A 13FC5808 00000A01 00000000 00000002
[  348.928610] ath10k: [0008]: 00000003 0004CF8A 14005808 00000003 00000000 00000000 00000000 00000001
[  348.928617] ath10k: [0016]: 0004CF8A 04005806 00000003 0004CF8A 10005858 00003112 004305EC 0000000A
[  348.928621] ath10k: [0024]: 00000000
[  348.928626] ath10k_pci 0000:02:00.0: ATH10K_END
[  525.412596] ath10k_pci 0000:02:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[  525.412609] ath10k: [0000]: 0007900D 08003813 0045A8D4 00000001 0007900D 08003812 0045A8D4 004305EC
[  525.412616] ath10k: [0008]: 0007900D 04005807 00000003 0007900D 13FC5808 00000A01 00000002 00000002
[  525.412623] ath10k: [0016]: FFFFFFFF 0007900D 0BFC3023 00000000 00000000 0007900D 08003812 0045A8D4
[  525.412630] ath10k: [0024]: 004305EC 0007900D 08003812 0045A8D4 004305EC 0007900D 10003809 000015A4
[  525.412635] ath10k: [0032]: 00000003 00000000 00000000 0007900D 0BFC5805 000015A4 00000000 0007911E
[  525.412642] ath10k: [0040]: 17FC3C2A 61100001 01000001 01010000 15A40000 00000002 0007911E 04003814
[  525.412647] ath10k: [0048]: 00000000 0007911E 13FC5808 00000A01 00000002 00000002 00000003 0007911E
[  525.412653] ath10k: [0056]: 14005808 00000003 00000000 00000000 00000000 00000001 0007911E 04005806
[  525.412659] ath10k: [0064]: 00000003 0007911E 10005858 00003112 004305EC 0000000A 00000000
[  525.412664] ath10k_pci 0000:02:00.0: ATH10K_END
[  527.415654] ath10k_pci 0000:02:00.0: removing stale peer 00:00:00:00:00:00 from vdev_id 0
[  527.423906] ath10k_pci 0000:02:00.0: removing peer mapping 00:00:00:00:00:00 peer-id: 0
[  527.432006] ath10k_pci 0000:02:00.0: peer-unmap-event: unknown peer id 0
[  528.413071] ath10k_pci 0000:02:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[  528.413084] ath10k: [0000]: 00079D78 08003813 0045A8D4 00000001 00079D78 08003812 0045A8D4 004305EC
[  528.413091] ath10k: [0008]: 00079D78 04005807 00000003 00079D78 13FC5808 00000A01 00000002 00000002
[  528.413098] ath10k: [0016]: FFFFFFFF 00079D78 0BFC3023 00000000 00000000 00079D78 08003812 0045A8D4
[  528.413104] ath10k: [0024]: 004305EC 00079D78 14006402 71103331 00000000 00000000 00000001 00000000
[  528.413110] ath10k: [0032]: 00079D78 14006402 71103332 00000000 00000000 0042E06C 00000000 00079D78
[  528.413116] ath10k: [0040]: 10005852 004305EC 004541A0 00000000 0042E06C 00079D78 14005854 51100002
[  528.413122] ath10k: [0048]: 000F4B38 00000001 00000000 004541A0 00079D90 14005853 51100001 000F4B38
[  528.413128] ath10k: [0056]: 000003FC 00000001 004541A0 00079D90 10005851 004305EC 004541A0 00000001
[  528.413134] ath10k: [0064]: 00000000 00079D90 14006403 00000000 01000000 00000000 00000000 004541A0
[  528.413140] ath10k: [0072]: 00079D90 10005858 00003112 004305EC 00000003 FFFFFFFF 00079D90 10005858
[  528.413145] ath10k: [0080]: 00003112 004305EC 0000000E 00000000
[  528.413151] ath10k_pci 0000:02:00.0: ATH10K_END
[  554.416966] ath10k_pci 0000:02:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[  554.416978] ath10k: [0000]: 000807A9 13FC4403 00000001 00000013 00000001 00000000 000807A9 0C00440F
[  554.416986] ath10k: [0008]: 00000001 0045A8D4 00000000 000807A9 1400380F 61100102 000015A4 000015A4
[  554.416992] ath10k: [0016]: 00000000 000F82C0 000807AF 1400440F 71109990 00000000 00000000 00000001
[  554.416998] ath10k: [0024]: 00000000 000807AF 1400440F 71109991 00000000 00000000 0042B318 00000000
[  554.417004] ath10k_pci 0000:02:00.0: ATH10K_END

```

Sniffer:
[... 51 lines cut ...]
10:35:59.644762 6.0 Mb/s 5540 MHz 11a -33dB signal antenna 0 Probe Response (7fast) [6.0* 9.0 18.0 24.0* 36.0 48.0 54.0 Mbit], PRIVACY

```

`# iw dev ath2 scan freq 5540 ssid 7fast | grep fast`

dmesg:

```
[  888.465753] ath10k_pci 0000:02:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[... cut ...]
[  888.465785] ath10k: [0024]: 00000000 000D3DC9 1400440F 71109991 00000000 00000000 0042B358 00000000

```

Sniffer output:

```
10:37:51.189788 6.0 Mb/s 5540 MHz 11a -19dB signal antenna 0 Probe Request (7fast) [6.0 9.0 12.0 18.0 24.0 36.0 48.0 54.0 Mbit]
10:37:51.190318 6.0 Mb/s 5540 MHz 11a -34dB signal antenna 0 Probe Response (7fast) [6.0* 9.0 18.0 24.0* 36.0 48.0 54.0 Mbit], PRIVACY
10:37:51.191160 6.0 Mb/s 5540 MHz 11a -32dB signal antenna 0 Probe Response (7fast) [6.0* 9.0 18.0 24.0* 36.0 48.0 54.0 Mbit], PRIVACY

```

greearb, 2018-08-13:
The logs look like the firmware is sending the probe requests as expected.  I guess there could be
issues farther down to the tx path.  But, maybe the faster way to debug this is by bisecting.  Can you try some of the older releases on our web site and see if any of the older ones work for this test case?

https://www.candelatech.com/ath10k-10.4.php

jhaakana, 2018-08-22:
Sorry for the delay, I had to work on some other stuff.

firmware-5-ct-full-htt-mgt-community-10.bin:
firmware-5-ct-full-htt-mgt-community-9.bin:
firmware-5-ct-full-htt-mgt-community-8.bin:

* Probe requests are sent out occasionally
* When probe requests are sent out, AP responds with probe response
* But hidden SSID is not visible in scan results

firmware-5-ct-full-htt-mgt-community-7.bin:
* Probe requests are sent out occasionally
* When probe requests are sent out, AP responds with probe response
* When the probe requests were sent out and responded, hidden SSID is always present in scan results

<= firmware-5-ct-full-htt-mgt-community-6.bin: Does not work:

```
[   17.223528] firmware ath10k!QCA9984!hw1.0!firmware-6.bin: firmware_loading_store: map pages failed
[   17.233765] ath10k_pci 0000:02:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[   17.243779] ath10k_pci 0000:02:00.0: kconfig debug 1 debugfs 1 tracing 0 dfs 1 testmode 0
[   17.254771] ath10k_pci 0000:02:00.0: firmware ver 10.4.3-ct-com-F-004-14b1705 api 5 features peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT crc32 c15d4922
[   20.320080] ath10k_pci 0000:02:00.0: unable to read from the device, execute bmi, ret: -110
[   20.328474] ath10k_pci 0000:02:00.0: could not execute otp for board id check: -110
[   20.336174] ath10k_pci 0000:02:00.0: failed to get board id from otp: -110
[   20.343171] ath10k_pci 0000:02:00.0: could not probe fw (-110)
```

So firmware-5-ct-full-htt-mgt-community-7.bin works better than more recent versions, but it still does not send out probe requests always. When it does, hidden SSID can be seen in scan results. This does not happen with more recent versions.

greearb, 2018-08-22:
I have been working on a rebase of my 10.4 tree onto a newer upstream QCA branch.

I have done builds for every commit.  The first one (0006) should be virtually identical to upstream QCA code.  Please see if the problem persists in these builds, and bisect it if so.

https://www.candelatech.com/downloads/ath10k-9984-10-4b/ath10k-fw-beta/all_builds-9984b-full-community-5-862.tar.gz

jhaakana, 2018-08-28:
Seems that commit firmware-5-full-community-commit-375-60e6ebf.bin broke it. Hidden SSID scan on single channel worked with all commit before that. Starting from 375-60e6ebf none of the versions have worked for me (so far my script has tested to commit 439-c9d9ef2, I'll let it run to the most recent commit).

jhaakana, 2018-08-29:
Tested all commits:
< firmware-5-full-community-commit-375-60e6ebf.bin: Hidden SSID scan on single channel works (probe requests are sent out)

firmware-5-full-community-commit-375-60e6ebf.bin - firmware-5-full-community-commit-862-f9581cd.bin: Hidden SSID scan on single channel does not work.

jhaakana, 2018-08-31:
I'm not changing MACs. For the test, I used 2.4GHz radio because results are more predictable (you are either getting hidden SSID in scan results during the couple of first scans, or not at all).

I've one VIF in managed mode (ath0) and one in monitor mode (ath1).

greearb, 2018-09-07:
Please try the attached firmware, and show me the dmesg output once you have run the test.  This binary is the problem commit with a bit of fixup and some additional logging.
[firmware-5-full-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2361781/firmware-5-full-community.bin.gz)

jhaakana, 2018-09-24:
Sorry for delay, I was on vacation. Re-ran the test with this FW version, dmesg output below:

```
# dmesg 
[   17.853763] ath: Country alpha2 being used: US
[   17.853766] ath: Regpair used: 0x3a
[   17.859569] ath10k 4.13 driver, optimized for CT firmware, probing pci device: 0x46.
[   17.872149] pci 0000:00:02.0: enabling device (0140 -> 0142)
[   17.878192] ath10k_pci 0000:02:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[   18.073359] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/fwcfg-pci-0000:02:00.0.txt failed with error -2
[   18.084011] ath10k_pci 0000:02:00.0: Falling back to user helper
[   18.414893] firmware ath10k!fwcfg-pci-0000:02:00.0.txt: firmware_loading_store: map pages failed
[   18.423989] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/pre-cal-pci-0000:02:00.0.bin failed with error -2
[   18.434766] ath10k_pci 0000:02:00.0: Falling back to user helper
[   18.451612] firmware ath10k!pre-cal-pci-0000:02:00.0.bin: firmware_loading_store: map pages failed
[   18.460862] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/cal-pci-0000:02:00.0.bin failed with error -2
[   18.471302] ath10k_pci 0000:02:00.0: Falling back to user helper
[   18.489089] firmware ath10k!cal-pci-0000:02:00.0.bin: firmware_loading_store: map pages failed
[   18.497953] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/QCA9984/hw1.0/ct-firmware-5.bin failed with error -2
[   18.508988] ath10k_pci 0000:02:00.0: Falling back to user helper
[   18.526869] firmware ath10k!QCA9984!hw1.0!ct-firmware-5.bin: firmware_loading_store: map pages failed
[   18.536345] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/QCA9984/hw1.0/ct-firmware-2.bin failed with error -2
[   18.547381] ath10k_pci 0000:02:00.0: Falling back to user helper
[   18.564260] firmware ath10k!QCA9984!hw1.0!ct-firmware-2.bin: firmware_loading_store: map pages failed
[   18.573742] ath10k_pci 0000:02:00.0: Direct firmware load for ath10k/QCA9984/hw1.0/firmware-6.bin failed with error -2
[   18.584519] ath10k_pci 0000:02:00.0: Falling back to user helper
[   18.601301] firmware ath10k!QCA9984!hw1.0!firmware-6.bin: firmware_loading_store: map pages failed
[   18.610515] ath10k_pci 0000:02:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[   18.620514] ath10k_pci 0000:02:00.0: kconfig debug 1 debugfs 1 tracing 0 dfs 1 testmode 0
[   18.631565] ath10k_pci 0000:02:00.0: firmware ver 10.4-ct-9984-fW-008-fdbdf33 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT crc32 0980c50f
[   19.927376] ath10k_pci 0000:02:00.0: board_file api 2 bmi_id 0:31 crc32 e807b522
[   22.438227] ath10k_pci 0000:02:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   22.445801] ath10k_pci 0000:02:00.0: msdu-desc: 2500  skid: 32
[   22.524941] ath10k_pci 0000:02:00.0: wmi print 'P 48/48 V 16 K 144 T 186  msdu-desc: 2500  sw-crypt: 0'
[   22.534434] ath10k_pci 0000:02:00.0: wmi print 'free: 73472 iram: 10260 sram: 18116'
[   22.733286] ath10k_pci 0000:02:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal otp max-sta 32 raw 0 hwcrypto 1
[   22.883748] ath: EEPROM regdomain: 0x0
[   22.883754] ath: EEPROM indicates default country code should be used
[   22.883757] ath: doing EEPROM country->regdmn map search
[   22.883762] ath: country maps to regdmn code: 0x3a
[   22.883766] ath: Country alpha2 being used: US
[   22.883769] ath: Regpair used: 0x3a
[   22.892628] ip_tables: (C) 2000-2006 Netfilter Core Team
[   23.413290] PPP generic driver version 2.4.2
[   23.418519] NET: Registered protocol family 24
[   23.424635] kmodloader: done loading kernel modules from /etc/modules.d/*
[   23.972381] UBIFS (ubi1:0): background thread "ubifs_bgt1_0" started, PID 2228
[   24.038725] UBIFS (ubi1:0): UBIFS: mounted UBI device 1, volume 0, name "log"
[   24.045943] UBIFS (ubi1:0): LEB size: 126976 bytes (124 KiB), min./max. I/O unit sizes: 2048 bytes/2048 bytes
[   24.055919] UBIFS (ubi1:0): FS size: 10158080 bytes (9 MiB, 80 LEBs), journal size 1015809 bytes (0 MiB, 6 LEBs)
[   24.066149] UBIFS (ubi1:0): reserved for root: 479791 bytes (468 KiB)
[   24.072630] UBIFS (ubi1:0): media format: w4/r0 (latest is w5/r0), UUID 4B0A4D71-7A90-4B6B-BACA-6C63726CEB4F, small LPT model
[   24.108666] Log partition mount is ok. No further checks needed.
[   28.792202] mvpp2 f10f0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[ 3610.325125] ath10k_pci 0000:01:00.0: Setting debug-mask to: 0x0  old: 0x10000000
[ 3647.415550] ath10k_pci 0000:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[ 3647.423125] ath10k_pci 0000:01:00.0: msdu-desc: 2500  skid: 32
[ 3647.502584] ath10k_pci 0000:01:00.0: wmi print 'P 48/48 V 16 K 144 T 186  msdu-desc: 2500  sw-crypt: 0'
[ 3647.512085] ath10k_pci 0000:01:00.0: wmi print 'free: 73472 iram: 10260 sram: 18116'
[ 3647.867808] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[ 3647.867821] ath10k: [0000]: 00000A46 13FC0007 60002070 00000010 00000030 000E0001 00000A46 13FC586C
[ 3647.867828] ath10k: [0008]: 00000001 00000001 0000096C 00000000 00000A8F 17FC582E 00000003 0000001E
[ 3647.867835] ath10k: [0016]: 0000094C 0000094C 00000000 00000A8F 17FC582E 00000012 0000001E 0000094C
[ 3647.867842] ath10k: [0024]: 0000094C 00000000 00000A8F 17FC582E 00000045 0000001E 0000094C 0000094C
[... cut ...]
[ 3658.421383] ath10k: [0168]: 00003471 1400587F 71109990 00432370 00000000 0045DF40 00098000 00003471
[ 3658.421389] ath10k: [0176]: 0C00587A 00432370 004585A8 00000001 00003471 0C006403 00000000 00000100
[ 3658.421396] ath10k: [0184]: 00000000 00003471 04003C25 00000000 00003471 1400587F 71109990 00432370
[ 3658.421402] ath10k: [0192]: 00000000 0045DF2C 01000000 00003472 10005881 31100002 00432370 00000003
[ 3658.421408] ath10k: [0200]: FFFFFFFF 00003472 10005881 31100002 00432370 0000000E 00000000
[ 3658.421415] ath10k_pci 0000:01:00.0: ATH10K_END

```

greearb, 2018-09-28:
I found a bug in how I was trying to apply the bssid mask.
Please test with this build and let me know the results (probe requests sent?  responses seen on air?  responses show up in scan results?)
[firmware-5-full-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2429904/firmware-5-full-community.bin.gz)

jhaakana, 2018-10-01:
That seems to fix the problem. I can now see the hidden SSID in scan results, and I can see probe requests / responses on the air.

Was this issue specific to wave 2 FW? If I recall right, we might have seen similar problem on wave 1 FW as well.

```
[  267.504098] ath10k_pci 0000:01:00.0: Setting debug-mask to: 0x0  old: 0x10000000
[  411.480456] ath10k_pci 0000:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[  411.488014] ath10k_pci 0000:01:00.0: msdu-desc: 2500  skid: 32
[  411.567509] ath10k_pci 0000:01:00.0: wmi print 'P 48/48 V 16 K 144 T 186  msdu-desc: 2500  sw-crypt: 0'
[  411.577033] ath10k_pci 0000:01:00.0: wmi print 'free: 73472 iram: 10372 sram: 18116'
[  412.208240] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[  412.208253] ath10k: [0000]: 00000A4F 13FC0007 60002070 00000010 00000030 000E0001 00000A4F 13FC586C
[  412.208260] ath10k: [0008]: 00000001 00000001 0000096C 00000000 00000A99 17FC582E 00000003 0000001E
[  412.208268] ath10k: [0016]: 0000094C 0000094C 00000000 00000A99 17FC582E 00000012 0000001E 0000094C
[  412.208274] ath10k: [0024]: 0000094C 00000000 00000A99 17FC582E 00000045 0000001E 0000094C 0000094C
[  412.208281] ath10k: [0032]: 00000000 00000A99 17FC582E 00000067 0000001E 0000094C 0000094C 00000000
[  412.208288] ath10k: [0040]: 00000AAC 1400587C 51100001 000F705C 000003FC 00000001 004585A8 00000AAC
[  412.208294] ath10k: [0048]: 1400587F 71109990 00432370 00000000 0045DF54 00001200 00000AAC 1400587F
[  412.208307] ath10k: [0056]: 71109990 00432370 00000000 0045DF40 00098000 00000AAC 0C00587A 00432370
[  412.208313] ath10k: [0064]: 004585A8 00000001 00000AAC 0C006403 00000000 00000100 00000000 00000AAC
[  412.208320] ath10k: [0072]: 04003C25 00000000 00000AAC 1400587F 71109990 00432370 00000000 0045DF2C
[  412.208326] ath10k: [0080]: 01000000 00000AAC 14006402 71103331 00000000 00000100 00000001 00000000
[  412.208332] ath10k: [0088]: 00000AAC 14006402 71103332 00000000 00000100 0042FBE8 00000000 00000AAC
[  412.208339] ath10k: [0096]: 1400587D 51100002 000F705C 00000002 00000000 004585A8 00000AAC 1000587B
[  412.208345] ath10k: [0104]: 00432370 004585A8 00000000 00000000 00000AAC 10005880 71109991 00432370
[  412.208352] ath10k: [0112]: 00000000 0045DF54 00000AAC 10005880 71109991 00432370 00000000 0045DF40
[  412.208358] ath10k: [0120]: 00000AAC 10005880 71109991 00432370 00000000 0045DF2C 00000AB3 13FC586C
[  412.208365] ath10k: [0128]: 00000002 00000002 0000096C 00000000 00000ACC 13FC5822 9110B000 00000001
[  412.208371] ath10k: [0136]: 00000001 00000001 00000AD2 13FC586C 00000002 00000002 0000096C 00000000
[  412.208378] ath10k: [0144]: 00000AEB 13FC5822 9110B000 00000001 00000001 00000001 00000AF2 13FC586C
[  412.208384] ath10k: [0152]: 00000002 00000002 0000096C 00000000 00000B0B 13FC5822 9110B000 00000001
[  412.208390] ath10k: [0160]: 00000001 00000001 00000B11 13FC586C 00000002 00000002 0000096C 00000000
[  412.208396] ath10k: [0168]: 00000B2A 13FC5822 9110B000 00000001 00000001 00000001 00000B31 13FC586C
[  412.208403] ath10k: [0176]: 00000002 00000002 0000096C 00000000 00000B4B 13FC5822 9110B000 00000001
[  412.208409] ath10k: [0184]: 00000001 00000001 00000B51 13FC586C 00000002 00000002 0000096C 00000000
[  412.208415] ath10k: [0192]: 00000B6A 13FC5822 9110B000 00000001 00000001 00000001 00000B70 13FC586C
[  412.208421] ath10k: [0200]: 00000002 00000002 0000096C 00000000 00000B8B 13FC5822 9110B000 00000001
[  412.208427] ath10k: [0208]: 00000001 00000001 00000B91 13FC586C 00000002 00000002 0000096C 00000000
[  412.208434] ath10k: [0216]: 00000BAB 13FC5822 9110B000 00000001 00000001 00000001 00000BB1 13FC586C
[  412.208440] ath10k: [0224]: 00000002 00000002 0000096C 00000000 00000BCA 13FC5822 9110B000 00000001
[  412.208446] ath10k: [0232]: 00000001 00000001 00000BD1 13FC586C 00000002 00000002 0000096C 00000000
[  412.208452] ath10k: [0240]: 00000BEB 13FC5822 9110B000 00000001 00000001 00000001 00000BF1 13FC586C
[  412.208458] ath10k: [0248]: 00000002 00000002 0000096C 00000000 00000C0A 13FC5822 9110B000 00000001
[  412.208464] ath10k: [0256]: 00000001 00000001 00000C0B 17FC5837 0000003A 0000003A 0000003A 00000010
[  412.208471] ath10k: [0264]: 00000010 00000C0B 17FC5837 0000003A 00000010 00000010 00000010 00000012
[  412.208477] ath10k: [0272]: 00000C0B 13FC6416 00000000 00000000 00000000 0000DE80 00000C0B 1400587C
[  412.208483] ath10k: [0280]: 51100001 000F705C 000003FC 00000001 004585A8 00000C0B 1400587C 51100001
[  412.208490] ath10k: [0288]: 000F7014 000003FE 00000002 004585A8 00000C0B 1400587F 71109990 00432370
[  412.208496] ath10k: [0296]: 00000000 0045DF2C 00001200 00000C0B 1400587F 71109990 00432370 00000000
[  412.208503] ath10k: [0304]: 0045DF40 00098000 00000C0B 0C00587A 00432370 004585A8 00000000 00000C0B
[  412.208509] ath10k: [0312]: 0C006403 00000000 00000000 00000000 00000C0B 04003C25 00000000 00000C0B
[  412.208515] ath10k: [0320]: 1400587F 71109990 00432370 00000000 0045DF54 01000000 00000C0C 10005881
[  412.208522] ath10k: [0328]: 31100002 00432370 00000003 FFFFFFFF 00000C0C 08003812 0045EBFC 00432370
[  412.208528] ath10k: [0336]: 00000C0C 10003809 0000096C 00000003 00000000 00000000 00000C12 0BFC5805
[  412.208535] ath10k: [0344]: 0001096C 00000000 00000C12 13FC586C 00000002 00000080 0000096C 00000001
[  412.208541] ath10k: [0352]: 00000D2A 13FC5808 711A0A01 00000000 00000002 00000003 00000D2A 14005808
[  412.208547] ath10k: [0360]: 00000003 00000000 00000000 00000000 00000001 00000D2A 04005806 00000003
[... cut ...]
[  564.739423] ath10k: [0080]: 00026E74 1400587F 71109990 00432370 00000000 0045DF40 00098000 00026E74
[  564.739429] ath10k: [0088]: 0C00587A 00432370 004585A8 00000001 00026E74 0C006403 00000000 00000100
[  564.739436] ath10k: [0096]: 00000000 00026E74 04003C25 00000000 00026E74 1400587F 71109990 00432370
[  564.739443] ath10k: [0104]: 00000000 0045DF2C 01000000 00026E74 10005881 31100002 00432370 00000003
[  564.739449] ath10k: [0112]: FFFFFFFF 00026E74 10005881 31100002 00432370 0000000E 00000000
[  564.739455] ath10k_pci 0000:01:00.0: ATH10K_END

```

jhaakana, 2018-10-02:
Ok thanks. When the fix will be available in official beta FW?

greearb, 2018-10-02:
Latest beta should have this now...please let me know if it does not work for you as expected.