greearb/ath10k-ct #31: bad performance compared to stock driver / firmware

stintel, 2018-08-13:
**Description of the problem (how to configure, how to reproduce, how often it happens).**
Being fed up with the instability of ath10k, I decided to try your driver / firmware. Unfortunately I'm experiencing quite terrible performance with it. I initially tried this a few months ago. My WiFi connection was so slow at that time, that I couldn't even stream a 720p video. As I wanted to continue watching, I reverted back to stock without further testing. After our conversation in #openwrt-devel a few weeks ago, I am now retesting everything.

I thought that with all combinations of ath10k stock with your firmware, ath10k-ct with -ct firmware and ath10k-ct with -ct-htt firmware were equally bad. Testing now shows that this is not true. It's with ath10k-ct and -ct-htt firmware that performance is so bad that it makes it unusable. See the iperf results below. The tests were done with both the AP and the client in the exact same location (in the bedroom with closed doors, doors have windows in them). The distance between the AP and client is roughly 9m.

/etc/config/wireless:
```
config wifi-device 'radio0'
        option type 'mac80211'
        option channel '36'
        option country 'BG'
        option htmode 'VHT80'
        option hwmode '11a'
        option legacy_rates '0'
        option path 'pci0000:00/0000:00:00.0'
        option disabled '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option ifname 'wlan0'
        option encryption 'psk2'
        option ieee80211w '1'
        option peerkey '1'
        option rsn_preauth '0'
        option network 'lan'
        option mode 'ap'
        option ssid 'XXXXXXXX'
        option key 'VerySecretPSK'
        option nasid 'wlan0.ap1'
        option ieee80211r '1'
        option ft_psk_generate_local '1'
        option wpa_disable_eapol_key_retries '1'

config wifi-iface 'guest_radio0'
        option device 'radio0'
        option ifname 'wlan2'
        option encryption 'psk2+ccmp'
        option ieee80211w '1'
        option network 'guest'
        option mode 'ap'
        option ssid 'XXXXXXXX-Guests'
        option key 'VerySecretGuestPSK'
        option wpa_disable_eapol_key_retries '1'
        option disabled '0'

config wifi-device 'radio1'
        option type 'mac80211'
        option channel '11'
        option country 'BG'
        option htmode 'HT20'
        option hwmode '11g'
        option legacy_rates '0'
        option path 'platform/ahb/ahb:apb/18100000.wmac'
        option txpower '20'
        option disabled '0'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option ifname 'wlan1'
        option encryption 'psk2'
        option ieee80211w '1'
        option peerkey '1'
        option rsn_preauth '0'
        option network 'lan'
        option mode 'ap'
        option ssid 'XXXXXXXX'
        option key 'VerySecretPSK'
        option nasid 'wlan1.ap1'
        option ieee80211r '1'
        option ft_psk_generate_local '1'
        option wpa_disable_eapol_key_retries '1'
        option macfilter 'deny'
        list maclist '2c:54:91:80:8c:5d'
        list maclist '58:82:a8:c0:cf:e4'

config wifi-iface 'guest_radio1'
        option device 'radio1'
        option ifname 'wlan3'
        option encryption 'psk2+ccmp'
        option ieee80211w '1'
        option network 'guest'
        option mode 'ap'
        option ssid 'XXXXXXXX-Guests'
        option key 'VerySecretGuestPSK'
        option wpa_disable_eapol_key_retries '1'
        option disabled '0'
```

**Software (OS, Firmware version, kernel, driver, etc)**
AP: Ubiquiti Unifi AP AC Pro running OpenWrt master (ath79) built from blogic's staging tree (mac80211 4.18-rc7)
Client: Dell XPS13 9343 w/ Intel 8265 running Gentoo Linux w/ 4.17.13 kernel

**Hardware (NIC chipset, platform, etc)**
Ubiquiti Unifi AP AC Pro
[   22.543045] ath10k_pci 0000:00:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000

**Logs (dmesg, maybe supplicant and/or hostap)**

**ath10k + stock firmware**
```
stijn@sylvester ~ $ sudo iw wlan0 link
Connected to f0:9f:c2:22:30:9b (on wlan0)
        SSID: XXXXXXXX
        freq: 5180
        RX: 28559 bytes (190 packets)
        TX: 14061 bytes (93 packets)
        signal: -67 dBm
        tx bitrate: 650.0 MBit/s VHT-MCS 7 80MHz short GI VHT-NSS 2

        bss flags:      short-slot-time
        dtim period:    2
        beacon int:     100
stijn@sylvester ~ $ iperf -c 192.168.54.10
------------------------------------------------------------
Client connecting to 192.168.54.10, TCP port 5001
TCP window size: 85.0 KByte (default)
------------------------------------------------------------
[  3] local 192.168.54.190 port 34866 connected with 192.168.54.10 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-10.0 sec   206 MBytes   172 Mbits/sec
```

**ath10k + ct firmware**
```
stijn@sylvester ~ $ sudo iw wlan0 link
Connected to f0:9f:c2:22:30:9b (on wlan0)
        SSID: XXXXXXXX
        freq: 5180
        RX: 12889 bytes (92 packets)
        TX: 4376 bytes (32 packets)
        signal: -71 dBm
        tx bitrate: 585.1 MBit/s VHT-MCS 6 80MHz short GI VHT-NSS 2

        bss flags:      short-slot-time
        dtim period:    2
        beacon int:     100
stijn@sylvester ~ $ iperf -c 192.168.54.10
------------------------------------------------------------
Client connecting to 192.168.54.10, TCP port 5001
TCP window size: 85.0 KByte (default)
------------------------------------------------------------
[  3] local 192.168.54.190 port 34868 connected with 192.168.54.10 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-10.0 sec  78.2 MBytes  65.4 Mbits/sec
```

**ath10k-ct + ct firmware**
```
stijn@sylvester ~ $ sudo iw wlan0 link
Connected to f0:9f:c2:22:30:9b (on wlan0)
        SSID: XXXXXXXX
        freq: 5180
        RX: 183728 bytes (1705 packets)
        TX: 5960591 bytes (3952 packets)
        signal: -72 dBm
        tx bitrate: 585.1 MBit/s VHT-MCS 6 80MHz short GI VHT-NSS 2

        bss flags:      short-slot-time
        dtim period:    2
        beacon int:     100
stijn@sylvester ~ $ iperf -c 192.168.54.10
------------------------------------------------------------
Client connecting to 192.168.54.10, TCP port 5001
TCP window size: 85.0 KByte (default)
------------------------------------------------------------
[  3] local 192.168.54.190 port 34880 connected with 192.168.54.10 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-10.1 sec  83.2 MBytes  69.4 Mbits/sec
```

**ath10k-ct + ct-htt firmware**
```
stijn@sylvester ~ $ sudo iw wlan0 link
Connected to f0:9f:c2:22:30:9b (on wlan0)
        SSID: XXXXXXXX
        freq: 5180
        RX: 11893 bytes (94 packets)
        TX: 6636 bytes (54 packets)
        signal: -72 dBm
        tx bitrate: 585.1 MBit/s VHT-MCS 6 80MHz short GI VHT-NSS 2

        bss flags:      short-slot-time
        dtim period:    2
        beacon int:     100
stijn@sylvester ~ $ iperf -c 192.168.54.10
------------------------------------------------------------
Client connecting to 192.168.54.10, TCP port 5001
TCP window size: 85.0 KByte (default)
------------------------------------------------------------
[  3] local 192.168.54.190 port 34884 connected with 192.168.54.10 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-10.0 sec  24.4 MBytes  20.4 Mbits/sec
```

stintel, 2018-08-13:
Apparently it seems to be related to 11W. After disabling 11W, the performance with -ct-htt seems to be very close to stock:

```
stijn@sylvester ~ $ sudo iw wlan0 link
Connected to f0:9f:c2:22:30:9b (on wlan0)
        SSID: XXXXXXXX
        freq: 5180
        RX: 10268 bytes (66 packets)
        TX: 2433 bytes (24 packets)
        signal: -71 dBm
        tx bitrate: 585.1 MBit/s VHT-MCS 6 80MHz short GI VHT-NSS 2

        bss flags:      short-slot-time
        dtim period:    2
        beacon int:     100
stijn@sylvester ~ $ iperf -c 192.168.54.10
------------------------------------------------------------
Client connecting to 192.168.54.10, TCP port 5001
TCP window size: 85.0 KByte (default)
------------------------------------------------------------
[  3] local 192.168.54.190 port 34888 connected with 192.168.54.10 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-10.0 sec   200 MBytes   168 Mbits/sec
```

stintel, 2018-08-13:
As requested, hostapd-phy0.conf:
```
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=BG
ieee80211d=1
ieee80211h=1
hw_mode=a
beacon_int=100
channel=36

ieee80211n=1
ht_coex=0
ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40]
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
ieee80211ac=1
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7]

interface=wlan0
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
nas_identifier=wlan0.ap1
wpa_passphrase=VerySecretPSK
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=XXXXXXXX
bridge=br-lan
mobility_domain=bb82
ft_psk_generate_local=1
ft_over_ds=1
reassociation_deadline=1000
wpa_disable_eapol_key_retries=1
wpa_key_mgmt=WPA-PSK FT-PSK WPA-PSK-SHA256
okc=0
disable_pmksa_caching=1
ieee80211w=1
group_mgmt_cipher=AES-128-CMAC
bssid=f0:9f:c2:22:30:9b

bss=wlan2
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
wpa_passphrase=VerySecretGuestPSK
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=XXXXXXXX-Guests
bridge=br-guest
wpa_disable_eapol_key_retries=1
wpa_key_mgmt=WPA-PSK WPA-PSK-SHA256
okc=0
disable_pmksa_caching=1
ieee80211w=1
group_mgmt_cipher=AES-128-CMAC
bssid=f2:9f:c2:22:30:9b
```

stintel, 2018-08-13:
**ath10k-ct with stock firmware**
```
stijn@sylvester ~ $ sudo iw wlan0 link
Connected to f0:9f:c2:22:30:9b (on wlan0)
        SSID: wlan.hq.adlevio.net
        freq: 5180
        RX: 846374 bytes (3661 packets)
        TX: 53299213 bytes (34782 packets)
        signal: -68 dBm
        tx bitrate: 650.0 MBit/s VHT-MCS 7 80MHz short GI VHT-NSS 2

        bss flags:      short-slot-time
        dtim period:    2
        beacon int:     100
stijn@sylvester ~ $ iperf -c 192.168.54.10
------------------------------------------------------------
Client connecting to 192.168.54.10, TCP port 5001
TCP window size: 85.0 KByte (default)
------------------------------------------------------------
[  3] local 192.168.54.190 port 34900 connected with 192.168.54.10 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-10.0 sec   213 MBytes   179 Mbits/sec
```

huaracheguarache, 2018-08-14:
I also experience this issue with ath10k-ct and the -htt firmware.

/ect/config/wireless:

```
config wifi-device 'radio0'
	option type 'mac80211'
	option channel '36'
	option hwmode '11a'
	option path 'soc/1b500000.pci/pci0000:00/0000:00:00.0/0000:01:00.0'
	option htmode 'VHT80'
	option txpower '23'
	option country 'NO'
	option legacy_rates '0'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'openwrt'
	option encryption 'psk2+ccmp'
	option key 'psk'
	option ieee80211w '1'

config wifi-device 'radio1'
	option type 'mac80211'
	option channel '11'
	option hwmode '11g'
	option path 'soc/1b700000.pci/pci0001:00/0001:00:00.0/0001:01:00.0'
	option htmode 'HT20'
	option txpower '19'
	option country 'NO'
	option legacy_rates '0'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'openwrt'
	option encryption 'psk2+ccmp'
	option key 'psk'
	option ieee80211w '1'
```

/var/run/hostapd-phy0.conf:

```
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=NO
ieee80211d=1
ieee80211h=1
hw_mode=a
beacon_int=100
channel=36

ieee80211n=1
ht_coex=0
ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40]
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
ieee80211ac=1
vht_capab=[RXLDPC][SHORT-GI-80][SHORT-GI-160][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][VHT160-80PLUS80][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7]

interface=wlan0
ctrl_interface=/var/run/hostapd
ap_isolate=1
[... 140 lines cut ...]
[  7]   6.00-7.00   sec  3.54 MBytes  29.7 Mbits/sec                  
[SUM]   6.00-7.00   sec  10.3 MBytes  86.8 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   7.00-8.00   sec  7.48 MBytes  62.7 Mbits/sec                  
[  7]   7.00-8.00   sec  4.05 MBytes  33.9 Mbits/sec                  
[SUM]   7.00-8.00   sec  11.5 MBytes  96.7 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   8.00-9.00   sec  7.58 MBytes  63.6 Mbits/sec                  
[  7]   8.00-9.00   sec  4.21 MBytes  35.3 Mbits/sec                  
[SUM]   8.00-9.00   sec  11.8 MBytes  98.9 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   9.00-10.00  sec  8.27 MBytes  69.4 Mbits/sec                  
[  7]   9.00-10.00  sec  4.78 MBytes  40.1 Mbits/sec                  
[SUM]   9.00-10.00  sec  13.0 MBytes   109 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  63.7 MBytes  53.4 Mbits/sec    1             sender
[  5]   0.00-10.00  sec  62.1 MBytes  52.1 Mbits/sec                  receiver
[  7]   0.00-10.00  sec  35.0 MBytes  29.4 Mbits/sec    2             sender
[  7]   0.00-10.00  sec  34.1 MBytes  28.6 Mbits/sec                  receiver
[SUM]   0.00-10.00  sec  98.7 MBytes  82.8 Mbits/sec    3             sender
[SUM]   0.00-10.00  sec  96.2 MBytes  80.7 Mbits/sec                  receiver

iperf Done.
```

greearb, 2018-08-24:
The root cause was block-ack failing in 11w mode.  I fixed this by backporting the appropriate code from 10.2 firmware and updating the driver to allow 11w mode in 10.1 firmware.  Appears to be fixed now.

closed 2018-08-24 by greearb