greearb/ath10k-ct #33: Firmware crashing on QCA9880

neheb, 2018-09-09:
Description of the problem (how to configure, how to reproduce, how often it happens).

Firmware keeps crashing on QCA9880. This has been an issue on normal firmwares and -ct firmwares. Although I haven't tested the very latest stock firmwares since I like having 11w support.

This is not related to 11w as it crashes with and without it enabled.

Initially I thought this was due to an iPhone. But as of the time of the log, only motorola devices were connected.

Software (OS, Firmware version, kernel, driver, etc)

Latest OpenWrt trunk.

Hardware (NIC chipset, platform, etc)

Turris Omnia - QCA9880 - Mikrotik 5HacT. I've had a very similar issue with a Compex card also based on QCA9880.

Logs (dmesg, maybe supplicant and/or hostap)

https://gist.github.com/neheb/4f83dc7267d929568fb01e8b6bc6e6d8

/etc/config/wireless is set to

config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11a'
        option path 'soc/soc:pcie/pci0000:00/0000:00:02.0/0000:02:00.0'
        option htmode 'VHT80'
        option channel '100'
        option country 'US'
        option legacy_rates '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'Neheb'
        option encryption 'psk2+ccmp'
        option key 'rgbfihfionikfgje'
        option wpa_disable_eapol_key_retries '1'
        option ieee80211w '1'

neheb, 2018-09-09:
https://drive.google.com/file/d/1cE1q0fvcAfDLW3cMikdVdNYis_ANl5NO/view?usp=sharing

neheb, 2018-09-10:
dump right after this one. 3 hours apart. although fewer clients are connected now...

https://drive.google.com/open?id=1hajULb1lw86ggdqYePTR0cS6fJpnP-xk

neheb, 2018-09-10:
And number 3

https://drive.google.com/open?id=1dzBZrj6mRcxwEuCs94d2I4iP2ZHHYLEO

klukonin, 2018-09-10:
@neheb 
@greearb 
Hello!
This bug is related to SPI flash operations (eg. with mtd write or read).

Please try to trigger this issue with such command
**dd if=/dev/mtdblock6 of=/dev/null**

klukonin, 2018-09-10:
We faced with this issue on UNBT AP-AC-LITE devices.

greearb, 2018-09-10:
All 3 dumps show setting a CLEAR key right before the crash, and I cannot see why the system should clear a key and then not (re)set a good key.  Possibly this crash is different from what klukonin reports, but if you can verify with the command he suggests, that would be helpful.

Please re-test this with MAC and WMI debugging enabled (so you should see key set logging and other such things in dmesg output), and send me the dmesg logs around the crash time as well as the binary dump.

neheb, 2018-09-10:
hmm? doesn't 0xc0000032 enable both?

greearb, 2018-09-10:
On 09/10/2018 09:53 AM, Rosen Penev wrote:
> hmm? doesn't 0xc0000032 enable both?

Probably...but it is only in the dmesg output, and you were attaching links only to
the binary crash dumps, which do not contain the dmesg output.

For that matter, if you could enable and include hostapd/supplicant logging as well
that might further help.

Thanks,
Ben

-- 
Ben Greear <greearb@candelatech.com>
Candela Technologies Inc  http://www.candelatech.com

klukonin, 2018-09-10:
> Possibly this crash is different from what klukonin reports, but if you can verify with the command he suggests, that would be helpful.

Hmmm. I bet, it's the same issue. Because dmesg output looks like ours. The same swba overruns with lost beacon, the same firmware crush. And we know how to avoid it again in the same tricky way, like beacon_int=10 with IBSS mode =).
We made a small patch that set tx-credits to 12 and limit vdevs parameter to 6 by default.
Swba overruns now don't hang the firmware. Than we made router to cache all file system in RAM right after the boot.

Looks like this issue is related to "tx-credits starvation" that was reported some month ago.
Please, try to fix it with my instructions or I can provide a patch.

Best regards, 
Kirill

neheb, 2018-09-10:
The issue is, that command looks like it's meant for a MIPS device. This is a Turria Omnia with an 8MB NOR flash chip and an 8GB!!! eMMC chip. The NOR chip is just used for U-Boot and a recovery partition:

mangix@TrapNet:~$ cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00100000 00010000 "U-Boot"
mtd1: 00700000 00010000 "Rescue system"

I'm guessing that mtdblock is where the wifi calibration data sits. On my miniPCIe card, it's on board or missing (no idea).

neheb, 2018-09-11:
@greearb I enabled log_level 1 in /etc/config/wireless and it seems logread has more output. Here's a paste (wlan0 is the relevant interface): 
https://gist.github.com/neheb/669fa39cf8ea47da0cdb0659877fed9f

and here's the dump (probably the same as the others): https://drive.google.com/open?id=1AqGT-AL7_t5oCaHzmaaWXCDYfNQdUaOa

klukonin, 2018-09-11:
[301-ideco-swba-overrun-fix-1.patch.txt](https://github.com/greearb/ath10k-ct/files/2370358/301-ideco-swba-overrun-fix-1.patch.txt)
Please try this patch.
Just put it to /package/kernel/ath10k-ct/patches/ and build openwrt
Than you will need to put fwcfg file here: (/lib/firmware/ath10k/) with vdevs = 6 parameter.

This will fix issue with swba overruns and firmware crash in most cases.

greearb, 2018-09-11:
The logread output is either skipping the ath10k 'mac', 'wmi' and other logging, or that logging is not enabled for whatever reason.  It does seem that deleting a station causes this...when I previously debugged the crash bin files, I saw the key delete but not the dis-assoc logic, even though hostap appears to show the disassoc before the key delete.  It would be very interesting to see the mac and wmi debugging to see what commands the driver is sending to the firmware and in what order.

neheb, 2018-09-13:
Another one: https://gist.github.com/neheb/305f71161d5ca6f9fd91f7499a27cbdb

This time with debug mask set to 0x12. I believe that sets only mac and wmi.

Gonna stop posting crashdumps since they all seem to be the same.

greearb, 2018-09-28:
One thing I noticed, you are using the 'WMI' ath10k-ct firmware instead of the HTT-MGT ath10k-ct firmware.  The htt-mgt firmware normally works better.  Also, I uploaded new betas today that do fewer software retries in the firmware, which might also help this problem.  And, were you able to test the patch nehab suggested to see if that helped?

Latest 10.1 beta firmware are here:
http://www.candelatech.com/downloads/ath10k-fw-beta/

neheb, 2018-09-28:
HTT-MGT has a different issue with my phone where rx on my phone or tx on the card stops randomly. Stuck queue maybe.

No patch testing for me for some time. Laptop died (bad BIOS flash, need an SPI programmer).

Will test new firmware later.

w01230, 2018-10-22:
replaced firmware in OPENWRT master with newest beta firmware 48 hours ago. And it seems fine until now.

neheb, 2018-10-31:
Small update, I put ct-htt firmware and I get no more crashing. No idea if I get that stuck queue issue. Will test later.

greearb, 2019-02-28:
One possible cause of this is DMA errors, and I found a fix for bad DMA access when deleting a beaconing interface.  Patches should hit OpenWRT soon.  I'm interested to know if these timeouts still happen with that firmware.

neheb, 2019-03-13:
Bad news: This is still an issue.

I was using an Archer C7v2 with ct-htt firmware for quite a while and noticed no problems. I recently migrated back to my Turris Omnia. Issue showed up again.

With ct-htt: Random traffic hangs.

With ct: Constant crashing. 3 Hardware restarts within 4 hours.

I would guess this is some kind of incompatibility with the specific pcie card that I use. I'm using a Mikrotik R11e-5HacT 802.11nac

greearb, 2019-03-13:
The crashes are all related to 'cannot communicate with firmware'?

For the random traffic hangs, can you enable DBGLOG debugging and send me dmesg?  Maybe there is something new there.

And, if you load stock driver/firmware, does it work better?

neheb, 2019-03-13:
I had a thought yesterday about the hangs.

When the firmware would crash (normal ct), I noticed traffic stopped on my phone. Basically mimicked behaviour with ct-htt. I wonder if ct-htt firmware is also supposed to crash but something is preventing it from doing so...

I'll try to get some debug output soon.

neheb, 2019-03-15:
Some output:

```
Thu Mar 14 19:17:12 2019 kern.debug kernel: [188085.374630] ath10k_pci 0000:02:00.0: ath10k_pci ATH10K_DBG_BUFFER:
Thu Mar 14 19:17:12 2019 kern.debug kernel: [188085.380924] ath10k: [0000]: 037B10DF 14004C4F 00000036 0041E680 00438008 00000000 00710000 037B10DF
Thu Mar 14 19:17:12 2019 kern.debug kernel: [188085.390094] ath10k: [0008]: 14004C56 00438008 009BDBEC 00110000 00003D20 00000001 037B10DF 14004C56
Thu Mar 14 19:17:12 2019 kern.debug kernel: [188085.399261] ath10k: [0016]: 00438008 0043C680 0000000C 00001D00 00000001 037B10DF 10004C4A F889DC54
Thu Mar 14 19:17:12 2019 kern.debug kernel: [188085.408420] ath10k: [0024]: D1640000 0000F889 DC54D164 037B10DF 14004C4A F889DC54 D1640000 0000F889
Thu Mar 14 19:17:12 2019 kern.debug kernel: [188085.417587] ath10k: [0032]: DC54D164 0000011D 037B10DF 14005C03 00000000 75A6B488 000040DA 009BBC68
Thu Mar 14 19:17:12 2019 kern.debug kernel: [188085.426744] ath10k: [0040]: 00438008 037B10E1 14004C52 00438008 00000080 00000001 00000002 00000208
Thu Mar 14 19:17:12 2019 kern.debug kernel: [188085.435907] ath10k: [0048]: 037B10E1 17FC2801 71100B07 010A0003 1601B003 0000FFFE 000000FF 037B10E1
Thu Mar 14 19:17:12 2019 kern.debug kernel: [188085.445064] ath10k: [0056]: 17FC2801 71108880 00000008 00448BC0 00960034 00000FF0 037B10E1 17FC2801
Thu Mar 14 19:17:13 2019 kern.debug kernel: [188085.454224] ath10k: [0064]: 71108880 00010008 00448BC0 00960034 000000FF 037B10E1 17FC2801 71108880
Thu Mar 14 19:17:13 2019 kern.debug kernel: [188085.463386] ath10k: [0072]: 00020008 00448BC0 00960034 000000FF 037B10E1 17FC2801 71108880 00030009
Thu Mar 14 19:17:13 2019 kern.debug kernel: [188085.472545] ath10k: [0080]: 00448BC0 00960034 000001FF 037B10E1 17FC2801 71108880 00040009 00448BC0
Thu Mar 14 19:17:13 2019 kern.debug kernel: [188085.481703] ath10k: [0088]: 00960034 000001FF 037B10E1 17FC2801 71108880 0005000A 00448BC0 00960034
Thu Mar 14 19:17:13 2019 kern.debug kernel: [188085.490856] ath10k: [0096]: 000003FF 037B10E2 10004C1D 00437EE8 00000001 00000000 00000140 037B10E2
Thu Mar 14 19:17:13 2019 kern.debug kernel: [188085.500013] ath10k: [0104]: 10004C1D 00437EE8 00000001 00000006 00000140 037B10E3 14004C56 00438008
Thu Mar 14 19:17:13 2019 kern.debug kernel: [188085.509172] ath10k: [0112]: 009BDC34 0006000C 00003D00 00000001 037B10F1 10004C1C 00438008 00000000
Thu Mar 14 19:17:13 2019 kern.debug kernel: [188085.518333] ath10k: [0120]: 00000006 000000C0 037B10F1 14004C52 00438008 000000E0 00000001 00000002
Thu Mar 14 19:17:13 2019 kern.debug kernel: [188085.527492] ath10k: [0128]: 00004208 037B10F3 17FC2801 71100B07 010A0003 1601B003 0000FFFE 000000FF
Thu Mar 14 19:17:13 2019 kern.debug kernel: [188085.536653] ath10k: [0136]: 037B10F3 17FC2801 71108880 00000008 00442998 00960034 00000FF0 037B10F3
Thu Mar 14 19:17:13 2019 kern.debug kernel: [188085.545812] ath10k: [0144]: 17FC2801 71108880 00010008 00442998 00960034 000000FF 037B10F3 17FC2801
Thu Mar 14 19:17:13 2019 kern.debug kernel: [188085.554969] ath10k: [0152]: 71108880 00020008 00442998 00960034 000000FF 037B10F3 17FC2801 71108880
Thu Mar 14 19:17:13 2019 kern.debug kernel: [188085.564127] ath10k: [0160]: 00030009 00442998 00960034 000001FF 037B10F3 17FC2801 71108880 00040009
Thu Mar 14 19:17:13 2019 kern.debug kernel: [188085.573284] ath10k: [0168]: 00442998 00960034 000001FF 037B10F3 17FC2801 71108880 0005000A 00442998
Thu Mar 14 19:17:13 2019 kern.debug kernel: [188085.582440] ath10k: [0176]: 00960034 000003FF
Thu Mar 14 19:17:13 2019 kern.debug kernel: [188085.586893] ath10k_pci 0000:02:00.0: ATH10K_END
Thu Mar 14 19:17:13 2019 kern.debug kernel: [188086.374609] ath10k_pci 0000:02:00.0: ath10k_pci ATH10K_DBG_BUFFER:
Thu Mar 14 19:17:13 2019 kern.debug kernel: [188086.380897] ath10k: [0000]: 037B11E3 10004C24 00438008 00010001 00000001 00000040
Thu Mar 14 19:17:13 2019 kern.debug kernel: [188086.388504] ath10k_pci 0000:02:00.0: ATH10K_END
Thu Mar 14 19:17:49 2019 daemon.info hostapd: wlan1: STA 88:b4:a6:75:da:40 IEEE 802.11: authenticated
Thu Mar 14 19:18:20 2019 daemon.info hostapd: wlan1: STA 88:b4:a6:75:da:40 IEEE 802.11: authenticated
Thu Mar 14 19:18:20 2019 daemon.info hostapd: wlan1: STA 88:b4:a6:75:da:40 IEEE 802.11: associated (aid 2)
Thu Mar 14 19:18:20 2019 daemon.debug hostapd: wlan0: STA 88:b4:a6:75:da:40 WPA: received EAPOL-Key 2/4 Pairwise with unexpected replay counter
Thu Mar 14 19:18:20 2019 daemon.debug hostapd: wlan0: STA 88:b4:a6:75:da:40 WPA: received EAPOL-Key 4/4 Pairwise with unexpected replay counter
Thu Mar 14 19:18:20 2019 daemon.info hostapd: wlan1: STA 88:b4:a6:75:da:40 RADIUS: starting accounting session 692ED341CB0749AB
Thu Mar 14 19:18:20 2019 daemon.info hostapd: wlan1: STA 88:b4:a6:75:da:40 WPA: pairwise key handshake completed (RSN)
Thu Mar 14 19:21:37 2019 daemon.err smbd[6413]: [2019/03/14 19:21:37.068577,  0] ../source3/printing/print_standard.c:71(std_pcap_cache_reload)
Thu Mar 14 19:21:37 2019 daemon.err smbd[6413]:   Unable to open printcap file /etc/printcap for read!
Thu Mar 14 19:23:27 2019 kern.debug kernel: [188460.375162] ath10k_pci 0000:02:00.0: ath10k_pci ATH10K_DBG_BUFFER:
Thu Mar 14 19:23:27 2019 kern.debug kernel: [188460.381465] ath10k: [0000]: 0380ECD9 14004C56 00438008 009BDC7C 0010001A 00003D02 00000001
Thu Mar 14 19:23:27 2019 kern.debug kernel: [188460.389845] ath10k_pci 0000:02:00.0: ATH10K_END
Thu Mar 14 19:23:30 2019 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 88:b4:a6:75:da:40
[... cut ...]
Thu Mar 14 19:23:41 2019 kern.debug kernel: [188473.699918] ath10k: [0024]: 0043F3D8 03811FA2 0C005C08 3330333B 00000000 00000064 03811FA2 08002007
Thu Mar 14 19:23:41 2019 kern.debug kernel: [188473.709076] ath10k: [0032]: 0043F364 009BD740 03811FA3 14004C4F 00000035 0041E648 00437EE8 00000001
Thu Mar 14 19:23:41 2019 kern.debug kernel: [188473.718233] ath10k: [0040]: 00730001 03811FA3 10004C1D 00437EE8 00000001 00000006 00000040
Thu Mar 14 19:23:41 2019 kern.debug kernel: [188473.726614] ath10k_pci 0000:02:00.0: ATH10K_END

```

greearb, 2019-03-15:
The firmware reports that it has not received messages from the host in a while and it attempts to go into CE polling mode, which sometimes lets things sort of keep working, but in this case, communication never resumes so driver restarts the firmware.  Previous debugging made me think that the NIC and/or host stop getting IRQs from each other and/or there is some other host to NIC transport issue.  I have no ideas on how to make progress with this unfortunately.

neheb, 2019-03-16:
It seems related to 802.11w. I turned it off and it seems no more firmware crashing.

greearb, 2019-03-18:
For the system that previously worked OK for you, was it using .11w?  Did you disable/change anything else except disabling .11w?  And, perhaps certain devices help trigger the problem, so if you can correlate the problem to any particular client devices, that might help.

neheb, 2019-03-18:
Last hardware restart was on March 16. So yeah, that fixed it.

On my Archer C7v2, I was using ct-htt, which does not crash. I also don't remember the stuck traffic issue (which I speculate to be some mini crash).

As far as devices go, I have a bunch of Motorola phones which do not support 11w (I don't even know why). The only devices I have that support 11w are an iPhone and maybe a Samsung phone.

One other thought I just had was that it could be PSK-SHA256 causing the issue. On OpenWrt, turning on 11w also turns on PSK-SHA256. I remember this option causing issues for Apple devices below iOS 6 where they wouldn't connect.

greearb, 2019-03-18:
I am primarily interested in ct-htt as wmi-mgt has some known issues that can degrade performance.

So, any reason you are not using ct-htt on your new device since it worked on your Archer?

Can you test w/out PSK-SHA256 to help determine if that is the culprit or not?

neheb, 2019-03-18:
ct-htt gives me stuck traffic, which is more inconvenient than having the firmware die (there's 2.4ghz to fall back on).

I'll try testing without PSK-SHA256.

greearb, 2019-03-18:
Can you describe the stuck traffic problem in more detail?  Is it only one station that is stuck, or do multiple get stuck at once?  Does re-associating the device cause it to be unstuck, or do you have to reboot the AP?  I can compile verbose tx-path debugging into the firmware, so as long as the system is otherwise responsive, if you can reproduce this with minimal other traffic going on, then maybe that would be a way to diagnose the problem.  Maybe open a new bug for this stuck traffic problem?

neheb, 2019-03-18:
Sure thing.

I'll be able to test tomorrow at the earliest.

Reassociating makes the stuck traffic go away for a little while. But it becomes stuck again. I usually get frustrated and switch to LTE. Maybe it has something to do with powersaving. Who knows.

I've tested this behavior with a moto x4, LG Nexus 5X, and iPhone 6S Plus.

I've also seen the same thing with mwlwifi, which is why I sold my WRT1900ACv2.

neheb, 2020-05-30:
Going to close this. I do not currently use this device. I can also avoid HTT firmwares and not hit this problem.

closed 2020-05-30 by neheb