greearb/ath10k-ct #34: offchannel timeout in QCA9888

HenryGiraldo, 2018-09-17:
driver: ath10k-4.13 CT 
firmware: CT full-community 10.4-ct-9888-fH-011-cf79c7f
Chipset: QCA9888.
OS: OpenWrt 18.06

Does this firmware as well as the 10.1 htt have offchannel support?
In the 10.1 htt I send packets in non-operating channels using the offchannel, in this firmware I cannot do it, I get timeouts:

> timed out waiting for offchannel skb %pK, len: %d\n

--

greearb, 2018-09-19:
I think I might have partially broken 10.1 offchannel support, and plan to fix it shortly.
But, I am not aware of any reason 10.4 would fail.  Does upstream 10.4 firmware work for you?

HenryGiraldo, 2018-09-20:
Capturing with Wireshark with 10.1 htt I can see the frames.

Firmware 10.4:
Your htt gives the warning and I can't see the frames.
With upstream firmware, frames don't appear in wireshark, but it doesn't get the warning either.
I use the command nl80211_send_frame_cmd to send the frames through hostapd.

It is very odd, because when scanning, every probe that is sent appears in Wireshark

greearb, 2018-09-28:
I uploaded new 10.4 beta images today.  Please test with it and see if it works better.  When checking a packet sniff, look for probe requests, responses, and ACKs for responses.  Let me know if any of that is missing if the test still fails.
http://www.candelatech.com/downloads/ath10k-9888-10-4/ath10k-fw-beta/

Stock firmware has issue sniffing probe requests I think, but otherwise it should properly do probe requests and responses.

HenryGiraldo, 2018-10-01:
I've just tried it and the off-channel frames still trigger timeout.

greearb, 2018-12-18:
Is offchannel broken for you in stock firmware/driver, or just in ath10k-ct driver/firmware?

HenryGiraldo, 2019-01-02:
I'm not sure, because I only tested it with your firmware. I habitually use package injection.

closed 2019-08-22 by HenryGiraldo