greearb/ath10k-ct #36: qca9984 wifi problem with deauth timer

Ansuel, 2018-09-29:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).
My phone, samsung galaxy s8 lost connection my times due to deauth timer.

Software (OS, Firmware version, kernel, driver, etc) Openwrt Trunk

Hardware (NIC chipset, platform, etc) qca9984

Logs (dmesg, maybe supplicant and/or hostap)
Sat Sep 29 21:16:01 2018 daemon.info hostapd: wlan1: STA 30:07:4d:1b:ee:6c IEEE 802.11: disconnected due to excessive missing ACKs
Sat Sep 29 21:16:01 2018 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 30:07:4d:1b:ee:6c
Sat Sep 29 21:16:31 2018 daemon.info hostapd: wlan1: STA 30:07:4d:1b:ee:6c IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sat Sep 29 21:19:45 2018 daemon.info hostapd: wlan1: STA 30:07:4d:1b:ee:6c IEEE 802.11: authenticated
Sat Sep 29 21:19:45 2018 daemon.info hostapd: wlan1: STA 30:07:4d:1b:ee:6c IEEE 802.11: associated (aid 1)

mocmocamoc, 2018-09-30:
I'm seeing the same issue with a Pixel 2 XL phone. My desktop with an Intel 9260 adapter has no problems and maintains a stable connection.

greearb, 2018-10-02:
Does this only happen with ath10k-ct firmware?

I'd disable the low-ack disconnect option, probably you can do this in LUCI or similar.  Does that make the problem go away?

ironpillow, 2019-04-09:
I am seeing the same problem with MacBook Pro 2019.

Hardware: QCA9948 and 
openwrt: snapshot compiled on 04_01_2019. 

driver: ath10k_pci
version: 4.14.109
firmware-version: 10.4b-ct-9984-fW-012-604ef8d6a
expansion-rom-version:
bus-info: 0000:00:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no

Thu Apr  4 23:43:20 2019 daemon.debug hostapd: wlan0-ap: STA 38:f9:d3:a1:2c:12 IEEE 802.11:authentication OK (open system)
Thu Apr  4 23:43:20 2019 daemon.debug hostapd: wlan0-ap: STA 38:f9:d3:a1:2c:12 MLME:
MLME-AUTHENTICATE.indication(38:f9:d3:a1:2c:12, OPEN_SYSTEM)
Thu Apr  4 23:43:20 2019 daemon.debug hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 MLME: MLME-DELETEKEYS.request(38:f9:d3:a1:2c:12)
Thu Apr  4 23:43:20 2019 daemon.debug hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 IEEE 802.11: Drop repeated authentication frame
seq_ctrl=0x42b0
Thu Apr  4 23:43:20 2019 daemon.debug hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 IEEE 802.11: Drop repeated authentication frame
seq_ctrl=0x42b0
Thu Apr  4 23:43:20 2019 daemon.notice hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 IEEE 802.11: did not acknowledge authentication
response
Thu Apr  4 23:43:20 2019 daemon.debug hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 IEEE 802.11: Drop repeated authentication frame
seq_ctrl=0x42b0
Thu Apr  4 23:43:20 2019 daemon.debug hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 IEEE 802.11: Drop repeated authentication frame
seq_ctrl=0x42b0
Thu Apr  4 23:43:20 2019 daemon.debug hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 IEEE 802.11: Drop repeated authentication frame
seq_ctrl=0x42b0
Thu Apr  4 23:43:20 2019 daemon.debug hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 IEEE 802.11: Drop repeated authentication frame
seq_ctrl=0x42b0
Thu Apr  4 23:43:20 2019 daemon.debug hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 IEEE 802.11: authentication OK (open system)
Thu Apr  4 23:43:20 2019 daemon.debug hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 MLME:
MLME-AUTHENTICATE.indication(38:f9:d3:a1:2c:12, OPEN_SYSTEM)
Thu Apr  4 23:43:20 2019 daemon.debug hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 MLME: MLME-DELETEKEYS.request(38:f9:d3:a1:2c:12)
Thu Apr  4 23:43:20 2019 daemon.notice hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 IEEE 802.11: did not acknowledge authentication
response
Thu Apr  4 23:43:21 2019 daemon.debug hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 IEEE 802.11: authentication OK (open system)
Thu Apr  4 23:43:21 2019 daemon.debug hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 MLME:
MLME-AUTHENTICATE.indication(38:f9:d3:a1:2c:12, OPEN_SYSTEM)
Thu Apr  4 23:43:21 2019 daemon.debug hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 MLME: MLME-DELETEKEYS.request(38:f9:d3:a1:2c:12)
Thu Apr  4 23:43:21 2019 daemon.notice hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 IEEE 802.11: did not acknowledge authentication
response
Thu Apr  4 23:43:21 2019 daemon.debug hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 IEEE 802.11: authentication OK (open system)
Thu Apr  4 23:43:21 2019 daemon.debug hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 MLME:
MLME-AUTHENTICATE.indication(38:f9:d3:a1:2c:12, OPEN_SYSTEM)
Thu Apr  4 23:43:21 2019 daemon.debug hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 MLME: MLME-DELETEKEYS.request(38:f9:d3:a1:2c:12)
Thu Apr  4 23:43:21 2019 daemon.notice hostapd: wlan0-ap: STA
[... 177 lines cut ...]
38:f9:d3:a1:2c:12 MLME: MLME-ASSOCIATE.indication(38:f9:d3:a1:2c:12)
Fri Apr  5 00:13:14 2019 daemon.debug hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 MLME: MLME-DELETEKEYS.request(38:f9:d3:a1:2c:12)
Fri Apr  5 00:13:14 2019 daemon.debug hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 IEEE 802.11: binding station to interface 'wlan0-ap'
Fri Apr  5 00:13:14 2019 daemon.debug hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 WPA: event 1 notification
Fri Apr  5 00:13:14 2019 daemon.debug hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 WPA: sending 1/4 msg of 4-Way Handshake
Fri Apr  5 00:13:14 2019 daemon.debug hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 WPA: received EAPOL-Key frame (2/4 Pairwise)
Fri Apr  5 00:13:14 2019 daemon.debug hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 WPA: sending 3/4 msg of 4-Way Handshake
Fri Apr  5 00:13:14 2019 daemon.debug hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 WPA: received EAPOL-Key frame (4/4 Pairwise)
Fri Apr  5 00:13:14 2019 daemon.notice hostapd: wlan0-ap:
AP-STA-CONNECTED 38:f9:d3:a1:2c:12
Fri Apr  5 00:13:14 2019 daemon.debug hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 IEEE 802.1X: authorizing port
Fri Apr  5 00:13:14 2019 daemon.info hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 RADIUS: starting accounting session F6D97094B7FC98D9
Fri Apr  5 00:13:14 2019 daemon.info hostapd: wlan0-ap: STA
38:f9:d3:a1:2c:12 WPA: pairwise key handshake completed (RSN)

I will check with disabling `disassoc_low_ack` and report back.