greearb/ath10k-ct #42: [IPQ4028] TCP sessions lock connected to some clients

barhom, 2018-11-14:
Device: [GL-B1300](https://www.gl-inet.com/products/gl-b1300/)
Their original firmware that is hosted [here](http://download.gl-inet.com.s3.amazonaws.com/firmware/b1300/v1/qsdk-b1300-2.272.bin) does not have the issue I am about to explain.

The issue is visible by compiling openwrt with the bare minimum packages (the defaults) in either 18.06 or trunk. Openwrt uses ath10k-ct afaik.

```
# ethtool -i wlan0
driver: ath10k_ahb
version: 4.14.78
firmware-version: 10.4-ct-4019-fW-011-cf79c7f
expansion-rom-version:
bus-info: a000000.wifi
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```

```
#ethtool -i wlan1
driver: ath10k_ahb
version: 4.14.78
firmware-version: 10.4-ct-4019-fW-011-cf79c7f
expansion-rom-version:
bus-info: a800000.wifi
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```

I am unsure which wifi-chipset it is using. What I know is the SoC that it uses (IPQ4028).
The easiest way to explain the problem is to watch the following [video](https://drive.google.com/file/d/1q57hzz6vrTfb9rEiosAqCVnCiiaM7emE/view?usp=sharing).

What you see in the video is a SSH session on my laptop to the openwrt router. The router's ip address is 10.255.0.1 which I am pining (i.e.) I am pinging the router from itself. This is however not required to reproduce the bug, I could be pinging 8.8.8.8 or 1.1.1.1 and the problem would show up anyway.
The issue lies somewhere between the communication of the router and the ssh client on my laptop (bad tcp session?).

Reproduced on:
Oneplus 6T (using JuiceSSH client)
Nexus 5X (Connect bot)
Macbook Pro 2011 (AirPort Extreme  (0x14E4, 0xD6) Broadcom BCM43xx 1.0 (5.106.98.102.30) (video recorded on this)

Unable to reproduce on (i.e. the issue did NOT present itself on these devices):

barhom, 2018-11-15:
after @greearb mentioned that I should try kmod-ath10k instead of kmod-ath10k-ct here are my results:

kmod-ath10k: issue does not present itself
kmod-ath10k-ct: issue easily reproduceable

greearb, 2018-12-18:
Please try this new firmware.  In the comment above, it shows that you were using upstream QCA firmware, not (just) upstream ath10k driver.  This firmware attached is a rebase on newer QCA firmware, and if problems persist, then I can generate a series of binaries that can be used to bisect.

[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2692049/firmware-5-full-htt-mgt-community.bin.gz)

barhom, 2018-12-22:
> Please try this new firmware. In the comment above, it shows that you were using upstream QCA firmware, not (just) upstream ath10k driver. This firmware attached is a rebase on newer QCA firmware, [...]

Do you mean that I first compile openwrt with ath10k-ct again and then after booting the router replace some firmware file with the one you linked to?

greearb, 2018-12-22:
You should not need to re-compile openwrt for this test.  Just copy the firmware into place in /lib/firmware/ath10k/.... and reboot.  You can check dmesg to make sure that the firmware version matches the hex bytes in the firmware image name (that is the git commit id).

greearb, 2019-01-24:
Several bugs that might have caused this have been fixed, and no further comments in this bug for a while, so closing it.  Please open new bugs if you find problems with latest firmware.

closed 2019-01-24 by greearb