greearb/ath10k-ct #43: [QCA9980] Drops packets

elfhack, 2018-12-07:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).
ath10k-ct firmware drops packets. I get tcp timeouts on some connections. The same exact config works fine on stock firmware and driver. ath10k-ct firmware and stock driver produces timeouts as well (both from the openwrt repo and the site)

Software (OS, Firmware version, kernel, driver, etc)
OpenWRT trunk r8614 on ipq8064 

Hardware (NIC chipset, platform, etc)
TP-Link C2600

greearb, 2018-12-07:
Please try this one, it should be virtually identical to stock QCA fw.  If it works, then we can bisect.

NOTE:  This 9980 'b' firmware.

[firmware-5-full-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2657558/firmware-5-full-community.bin.gz)

greearb, 2018-12-07:
From IRC comments, the initial one above appears to work fine.  Here is top of tree.  Please check, if it fails, then I will build images so you can bisect.
[firmware-5-full-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2657944/firmware-5-full-community.bin.gz)

elfhack, 2018-12-07:
And this one seems to work perfectly

closed 2018-12-20 by elfhack