greearb/ath10k-ct #44: [QCA4019] random wifi crashes

fishxz, 2018-12-07:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).
wifi crashes randomly. im running this router since 2 days and i already got 2 crashes.

Software (OS, Firmware version, kernel, driver, etc)
OpenWrt SNAPSHOT r8661-32bc733
firmware ver 10.4-ct-4019-fW-011-cf79c7
kernel 4.14.82

Hardware (NIC chipset, platform, etc)
Zyxel nbg6617 IPQ4018

Logs (dmesg, maybe supplicant and/or hostap)
`[32602.955700] ath10k_ahb a800000.wifi: firmware crashed! (guid cc7b680b-93b1-4b23-a532-522564d926d3)
[32602.955812] ath10k_ahb a800000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[32602.963590] ath10k_ahb a800000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[32602.976997] ath10k_ahb a800000.wifi: firmware ver 10.4-ct-4019-fW-011-cf79c7f api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,cust-stats-CT crc32 3aa71cb7
[32602.984271] ath10k_ahb a800000.wifi: board_file api 2 bmi_id 0:17 crc32 bcebe54c
[32603.003802] ath10k_ahb a800000.wifi: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[32603.012370] ath10k_ahb a800000.wifi: firmware register dump:
[32603.021137] ath10k_ahb a800000.wifi: [00]: 0x0000000B 0x000015B3 0x009A4669 0x00975B31
[32603.027042] ath10k_ahb a800000.wifi: [04]: 0x009A4669 0x00060130 0x00000005 0x00000007
[32603.034750] ath10k_ahb a800000.wifi: [08]: 0x0040FD60 0x004478A6 0x00000001 0x00000000
[32603.042664] ath10k_ahb a800000.wifi: [12]: 0x00000009 0xFFFFFFFF 0x00971518 0x00971523
[32603.050562] ath10k_ahb a800000.wifi: [16]: 0x00971238 0x009606B6 0x009606A4 0x00000000
[32603.058462] ath10k_ahb a800000.wifi: [20]: 0x409A4669 0x0040555C 0x000000FF 0xFFFFFFFF
[32603.066358] ath10k_ahb a800000.wifi: [24]: 0x809A5F83 0x004055BC 0x00000003 0xC09A4669
[32603.074244] ath10k_ahb a800000.wifi: [28]: 0x809A73CA 0x0040560C 0x004471BC 0x0040FD60
[32603.082161] ath10k_ahb a800000.wifi: [32]: 0x809A75D6 0x004056FC 0x00419CC8 0x004057BC
[32603.090058] ath10k_ahb a800000.wifi: [36]: 0x809859EB 0x0040576C 0x00000000 0x00443CA4
[32603.097958] ath10k_ahb a800000.wifi: [40]: 0x8099381F 0x0040578C 0x0041BDC8 0x00443CA4
[32603.105855] ath10k_ahb a800000.wifi: [44]: 0x8098EA08 0x004057AC 0x0041BDC8 0x00443CA4
[32603.113742] ath10k_ahb a800000.wifi: [48]: 0x809630B5 0x0040599C 0x0041BDC8 0x0098EA00
[32603.121657] ath10k_ahb a800000.wifi: [52]: 0x80960DF0 0x004059BC 0x0000001F 0x00400000
[32603.129554] ath10k_ahb a800000.wifi: [56]: 0x80960DC2 0x004059EC 0x00400000 0x00000000
[32603.137453] ath10k_ahb a800000.wifi: Copy Engine register dump:
[32603.145339] ath10k_ahb a800000.wifi: [00]: 0x0004a000  13  13   3   3
[32603.151169] ath10k_ahb a800000.wifi: [01]: 0x0004a400   4   4 362 363
[32603.157767] ath10k_ahb a800000.wifi: [02]: 0x0004a800  15  15  78  79
[32603.164174] ath10k_ahb a800000.wifi: [03]: 0x0004ac00  17  17  19  17
[32603.170614] ath10k_ahb a800000.wifi: [04]: 0x0004b000 6663 6663  83  45
[32603.177037] ath10k_ahb a800000.wifi: [05]: 0x0004b400  19  19 466 467
[32603.183444] ath10k_ahb a800000.wifi: [06]: 0x0004b800  11  11  11  11
[32603.190057] ath10k_ahb a800000.wifi: [07]: 0x0004bc00   1   1   1   1
[32603.196482] ath10k_ahb a800000.wifi: [08]: 0x0004c000   0   0 127   0
[32603.202889] ath10k_ahb a800000.wifi: [09]: 0x0004c400   1   1   1   1
[32603.209328] ath10k_ahb a800000.wifi: [10]: 0x0004c800   0   0   0   0
[32603.215748] ath10k_ahb a800000.wifi: [11]: 0x0004cc00   0   0   0   0
[32603.223185] ath10k_ahb a800000.wifi: debug log header, dbuf: 0x4187e8  dropped: 0
[32603.229619] ath10k_ahb a800000.wifi: [0] next: 0x4187d0 buf: 0x414c50 sz: 1500 len: 84 count: 3 free: 0
[32603.237081] ath10k_ahb a800000.wifi: ath10k_pci ATH10K_DBG_BUFFER:
[32603.245250] ath10k: [0000]: 01FAB841 17FC4C07 9110A004 FFFFFFFF 00000000 FFFFFFFF 00000001 01FAB841
[32603.251522] ath10k: [0008]: 17FC4C07 9110A005 00000043 000000FF 0008877B 00000055 01FAB841 17FC0001
[32603.260458] ath10k: [0016]: 009A4669 000015B3 000015B3 0040544C 91104569
[32603.269481] ath10k_ahb a800000.wifi: ATH10K_END
[32603.277437] ath10k_ahb a800000.wifi: [1] next: 0x4187e8 buf: 0x414660 sz: 1500 len: 0 count: 0 free: 0
[32603.346637] ath10k_ahb a800000.wifi: removing peer, cleanup-all, deleting: peer ce0c3800 vdev: 0 addr: 5c:49:7d:a3:27:a5
[32603.347099] ath10k_ahb a800000.wifi: removing peer, cleanup-all, deleting: peer ce0c2a00 vdev: 0 addr: 2c:59:8a:73:2b:6e
[32603.357367] ath10k_ahb a800000.wifi: removing peer, cleanup-all, deleting: peer ce58a600 vdev: 0 addr: f4:f5:d8:54:c6:aa
[32603.368159] ath10k_ahb a800000.wifi: removing peer, cleanup-all, deleting: peer cf8c3200 vdev: 0 addr: c4:6e:1f:14:18:0c
[32603.379064] ath10k_ahb a800000.wifi: removing peer, cleanup-all, deleting: peer cebfe400 vdev: 0 addr: b8:ec:a3:2e:dd:75
[32603.457712] ieee80211 phy1: Hardware restart was requested
[32603.527875] ath10k_ahb a800000.wifi: Invalid state: 3 in ath10k_htt_tx_32, warning will not be repeated.
[32603.527920] ------------[ cut here ]------------
[32603.537623] WARNING: CPU: 1 PID: 0 at /var/lib/buildbot/slaves/slashdirt-03/MAIN/build/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-2018-09-29-b9989fbd/ath10k-4.16/htt_tx.c:1250 ath10k_htt_tx_32+0xd0/0x8d0 [ath10k_core]
[32603.541218] Modules linked in: xt_tcpmss xt_statistic xt_recent xt_length xt_hl xt_helper xt_ecn xt_dscp xt_connmark xt_connlimit xt_connbytes xt_HL xt_DSCP xt_CLASSIFY iptable_raw ipt_ECN sch_cake act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ifb pppoe ppp_async ath10k_pci ath10k_core ath pppox ppp_generic nf_conntrack_ipv6 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle
[32603.612294]  iptable_filter ip_tables hwmon crc_ccitt compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_of_simple gpio_button_hotplug
[32603.634133] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 4.14.82 #0
[... 64 lines cut ...]
[32604.873421] ath10k_ahb a800000.wifi: failed to submit frame: -19
[32604.879527] ath10k_ahb a800000.wifi: failed to push frame: -19
[32604.897597] ath10k_ahb a800000.wifi: failed to transmit packet, dropping: -19
[32604.897644] ath10k_ahb a800000.wifi: failed to submit frame: -19
[32604.903719] ath10k_ahb a800000.wifi: failed to push frame: -19
[32604.911819] ath10k_ahb a800000.wifi: failed to submit frame: -19
[32604.915457] ath10k_ahb a800000.wifi: failed to push frame: -19
[32604.923499] ath10k_ahb a800000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[32604.927509] ath10k_ahb a800000.wifi: msdu-desc: 2500  skid: 32
[32604.977936] ath10k_ahb a800000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0'
[32604.978331] ath10k_ahb a800000.wifi: wmi print 'free: 65652 iram: 15164 sram: 43468'
[32604.990649] ath10k_ahb a800000.wifi: failed to submit frame: -19
[32604.994873] ath10k_ahb a800000.wifi: failed to push frame: -19
[32605.151539] ath10k_ahb a800000.wifi: failed to submit frame: -19
[32605.151587] ath10k_ahb a800000.wifi: failed to push frame: -19
[32605.204895] ath10k_ahb a800000.wifi: failed to submit frame: -19
[32605.205218] ath10k_ahb a800000.wifi: failed to push frame: -19
[32605.225411] ath10k_ahb a800000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[32605.353905] ath10k_ahb a800000.wifi: failed to submit frame: -19
[32605.353944] ath10k_ahb a800000.wifi: failed to push frame: -19
[32605.420865] ath10k_ahb a800000.wifi: rate-override:  Skipping un-supported device-id, hw-nss: 2 dev-id: 0x0
[32605.427943] ath10k_ahb a800000.wifi: rate-override:  Skipping un-supported device-id, hw-nss: 2 dev-id: 0x0
[32605.430455] ath10k_ahb a800000.wifi: rate-override:  Skipping un-supported device-id, hw-nss: 2 dev-id: 0x0
[32605.440850] ath10k_ahb a800000.wifi: rate-override:  Skipping un-supported device-id, hw-nss: 2 dev-id: 0x0
[32605.451206] ath10k_ahb a800000.wifi: device successfully recovered`

greearb, 2018-12-18:
Please see if this firmware improves the FW crashes...if not, please post new dmesg logs of the crash.

[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2691972/firmware-5-full-htt-mgt-community.bin.gz)

fishxz, 2018-12-19:
```
[ 2254.616533] ath10k_ahb a800000.wifi: firmware crashed! (guid 795816fe-9383-40f8-a1b2-6e4897e90ccc)
[ 2254.617111] ath10k_ahb a800000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[ 2254.625286] ath10k_ahb a800000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[ 2254.680155] ath10k_ahb a800000.wifi: firmware ver 10.4b-ct-4019-fH-012-648296d3 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT crc32 e9d5c818
[ 2254.720270] ath10k_ahb a800000.wifi: board_file api 2 bmi_id 0:17 crc32 bcebe54c
[ 2254.720643] ath10k_ahb a800000.wifi: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[ 2254.727595] ath10k_ahb a800000.wifi: firmware register dump:
[ 2254.737136] ath10k_ahb a800000.wifi: [00]: 0x0000000B 0x00000000 0x00996547 0x00000000
[ 2254.742751] ath10k_ahb a800000.wifi: [04]: 0x00000000 0x00060124 0x00000000 0x00000000
[ 2254.750726] ath10k_ahb a800000.wifi: [08]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 2254.758575] ath10k_ahb a800000.wifi: [12]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 2254.766524] ath10k_ahb a800000.wifi: [16]: 0x009C22D7 0x0A003EBF 0x009606B6 0x00996547
[ 2254.774409] ath10k_ahb a800000.wifi: [20]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 2254.782070] ath10k_ahb a800000.wifi: [24]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 2254.790213] ath10k_ahb a800000.wifi: [28]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 2254.798073] ath10k_ahb a800000.wifi: [32]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 2254.806011] ath10k_ahb a800000.wifi: [36]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 2254.813911] ath10k_ahb a800000.wifi: [40]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 2254.821567] ath10k_ahb a800000.wifi: [44]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 2254.829708] ath10k_ahb a800000.wifi: [48]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 2254.837558] ath10k_ahb a800000.wifi: [52]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 2254.845506] ath10k_ahb a800000.wifi: [56]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 2254.853375] ath10k_ahb a800000.wifi: Copy Engine register dump:
[ 2254.861073] ath10k_ahb a800000.wifi: [00]: 0x0004a000   3   3   3   3
[ 2254.867138] ath10k_ahb a800000.wifi: [01]: 0x0004a400   6   6  43  45
[ 2254.873736] ath10k_ahb a800000.wifi: [02]: 0x0004a800   7   7  70  71
[ 2254.879896] ath10k_ahb a800000.wifi: [03]: 0x0004ac00   5   5   7   5
[ 2254.886584] ath10k_ahb a800000.wifi: [04]: 0x0004b000 6418 6418 179 139
[ 2254.892752] ath10k_ahb a800000.wifi: [05]: 0x0004b400  14  14 397 398
[ 2254.899430] ath10k_ahb a800000.wifi: [06]: 0x0004b800  24  24  24  24
[ 2254.905975] ath10k_ahb a800000.wifi: [07]: 0x0004bc00   1   1   1   1
[ 2254.912191] ath10k_ahb a800000.wifi: [08]: 0x0004c000   0   0 127   0
[ 2254.918871] ath10k_ahb a800000.wifi: [09]: 0x0004c400   1   1   1   1
[ 2254.925294] ath10k_ahb a800000.wifi: [10]: 0x0004c800   0   0   0   0
[ 2254.931459] ath10k_ahb a800000.wifi: [11]: 0x0004cc00   0   0   0   0
[ 2254.938351] ath10k_ahb a800000.wifi: debug log header, dbuf: 0x418790  dropped: 0
[ 2254.944718] ath10k_ahb a800000.wifi: [0] next: 0x4187a8 buf: 0x414ba0 sz: 1500 len: 156 count: 6 free: 0
[ 2254.951861] ath10k_ahb a800000.wifi: ath10k_pci ATH10K_DBG_BUFFER:
[ 2254.961585] ath10k: [0000]: 0022DFC1 17FC5813 004419D0 0000180C 00000140 0000013C 00000140 0022DFCA
[ 2254.967515] ath10k: [0008]: 1400581D 00000000 004419D0 000F02B0 00000006 00000000 0022DFCA 1000581B
[ 2254.976496] ath10k: [0016]: 0000180C 00000000 00000000 000F02B0 0022DFCE 14006402 71103332 141F6EC4
[ 2254.985520] ath10k: [0024]: 00000C18 0041A2D0 00000001 0022DFCE 13FC4C07 211000A1 000001BF 00000009
[ 2254.994537] ath10k: [0032]: 0043417C 0022DFCE 1000587B 0041D65C 004419D0 00000001 0041A2D0
[ 2255.003529] ath10k_ahb a800000.wifi: ATH10K_END
[ 2255.011789] ath10k_ahb a800000.wifi: [1] next: 0x418790 buf: 0x415190 sz: 1500 len: 0 count: 0 free: 0
[ 2255.041256] ath10k_ahb a800000.wifi: removing peer, cleanup-all, deleting: peer cef1fa00 vdev: 0 addr: b8:ec:a3:2e:dd:75 
[ 2255.041528] ath10k_ahb a000000.wifi: SWBA overrun on vdev 0, skipped old beacon
[ 2255.053839] ieee80211 phy1: Hardware restart was requested
[ 2256.545251] ath10k_ahb a800000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[ 2256.545593] ath10k_ahb a800000.wifi: msdu-desc: 2500  skid: 32
[ 2256.592681] ath10k_ahb a800000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[ 2256.594611] ath10k_ahb a800000.wifi: wmi print 'free: 58076 iram: 22776 sram: 35968'
[ 2256.894817] ath10k_ahb a800000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[ 2257.091628] ath10k_ahb a800000.wifi: device successfully recovered
```

greearb, 2018-12-19:
The crash looks like null-ptr exception in stats gathering code after peer is deleted.  I added some debugging to hopefully narrow down exactly where this is happening..please try the attached and post more crash dumps when they happen.  Then hopefully I can fix it properly.

[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2695641/firmware-5-full-htt-mgt-community.bin.gz)

fishxz, 2018-12-19:
```
[13172.703914] ath10k_ahb a000000.wifi: firmware crashed! (guid eb9cabd1-1dc6-4dca-b201-d570fb3db47c)
[13172.705521] ath10k_ahb a000000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[13172.714503] ath10k_ahb a000000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[13172.728414] ath10k_ahb a000000.wifi: firmware ver 10.4b-ct-4019-fH-012-648296d3 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT crc32 42e22ec5
[13172.738998] ath10k_ahb a000000.wifi: board_file api 2 bmi_id 0:16 crc32 bcebe54c
[13172.757472] ath10k_ahb a000000.wifi: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[13172.766104] ath10k_ahb a000000.wifi: firmware register dump:
[13172.774869] ath10k_ahb a000000.wifi: [00]: 0x0000000B 0x00000000 0x0099214F 0x00000000
[13172.780751] ath10k_ahb a000000.wifi: [04]: 0x00000000 0x00060124 0x00000000 0x00000000
[13172.788497] ath10k_ahb a000000.wifi: [08]: 0x00000000 0x00000000 0x00000000 0x00000000
[13172.796388] ath10k_ahb a000000.wifi: [12]: 0x00000000 0x00000000 0x00000000 0x00000000
[13172.804293] ath10k_ahb a000000.wifi: [16]: 0x009C208B 0x0097F8C3 0x009606B6 0x0099214F
[13172.812173] ath10k_ahb a000000.wifi: [20]: 0x00000000 0x00000000 0x00000000 0x00000000
[13172.820091] ath10k_ahb a000000.wifi: [24]: 0x00000000 0x00000000 0x00000000 0x00000000
[13172.827989] ath10k_ahb a000000.wifi: [28]: 0x00000000 0x00000000 0x00000000 0x00000000
[13172.835885] ath10k_ahb a000000.wifi: [32]: 0x00000000 0x00000000 0x00000000 0x00000000
[13172.843788] ath10k_ahb a000000.wifi: [36]: 0x00000000 0x00000000 0x00000000 0x00000000
[13172.851672] ath10k_ahb a000000.wifi: [40]: 0x00000000 0x00000000 0x00000000 0x00000000
[13172.859588] ath10k_ahb a000000.wifi: [44]: 0x00000000 0x00000000 0x00000000 0x00000000
[13172.867482] ath10k_ahb a000000.wifi: [48]: 0x00000000 0x00000000 0x00000000 0x00000000
[13172.875392] ath10k_ahb a000000.wifi: [52]: 0x00000000 0x00000000 0x00000000 0x00000000
[13172.883286] ath10k_ahb a000000.wifi: [56]: 0x00000000 0x00000000 0x00000000 0x00000000
[13172.891166] ath10k_ahb a000000.wifi: Copy Engine register dump:
[13172.899084] ath10k_ahb a000000.wifi: [00]: 0x0004a000   2   2   3   3
[13172.904897] ath10k_ahb a000000.wifi: [01]: 0x0004a400  30  30  68  69
[13172.911478] ath10k_ahb a000000.wifi: [02]: 0x0004a800  52  52 115 116
[13172.917921] ath10k_ahb a000000.wifi: [03]: 0x0004ac00   3   3   5   3
[13172.924345] ath10k_ahb a000000.wifi: [04]: 0x0004b000 5520 5520 241 202
[13172.930750] ath10k_ahb a000000.wifi: [05]: 0x0004b400   1   1 256 257
[13172.937191] ath10k_ahb a000000.wifi: [06]: 0x0004b800  12  12  12  12
[13172.943789] ath10k_ahb a000000.wifi: [07]: 0x0004bc00   1   1   1   1
[13172.950192] ath10k_ahb a000000.wifi: [08]: 0x0004c000   0   0 127   0
[13172.956636] ath10k_ahb a000000.wifi: [09]: 0x0004c400   0   0   0   0
[13172.963061] ath10k_ahb a000000.wifi: [10]: 0x0004c800   0   0   0   0
[13172.969465] ath10k_ahb a000000.wifi: [11]: 0x0004cc00   0   0   0   0
[13172.976932] ath10k_ahb a000000.wifi: debug log header, dbuf: 0x417f48  dropped: 0
[13172.983351] ath10k_ahb a000000.wifi: [0] next: 0x417f30 buf: 0x415150 sz: 1500 len: 352 count: 13 free: 0
[13172.990799] ath10k_ahb a000000.wifi: ath10k_pci ATH10K_DBG_BUFFER:
[13172.999344] ath10k: [0000]: 00CD937A 1400587C 51100001 000F0180 000003FC 0000000E 00440BC8 00CD937A
[13173.005418] ath10k: [0008]: 1000587A 0041CDFC 00440BC8 01000000 00000005 00CD937A 14006403 A37D495C
[13173.014365] ath10k: [0016]: A5270000 00000005 00000000 00440BC8 00CD9380 10004C01 71100D08 00000000
[13173.023388] ath10k: [0024]: 00000005 00000077 00CD9380 14004C01 71104441 030527A5 00009003 00000000
[13173.032418] ath10k: [0032]: 0000FFFF 00CD9380 14004C01 71108880 00000000 00680012 00000000 0000000C
[13173.041428] ath10k: [0040]: 00CD9380 14004C01 71108880 00010000 00680012 00000000 0000FFFF 00CD9380
[13173.050474] ath10k: [0048]: 14004C01 71108880 00020000 00680012 00000000 00000000 00CD9380 14004C01
[13173.059501] ath10k: [0056]: 71108880 00030000 00680012 00000000 00000000 00CD9380 14004C01 71108880
[13173.068528] ath10k: [0064]: 00040000 00680012 00000000 00000000 00CD9380 14004C01 71108880 00050000
[13173.077552] ath10k: [0072]: 00680012 00000000 00000000 00CD9391 14006402 71103332 A37D495C 0000A527
[13173.086584] ath10k: [0080]: 00419BD0 00000005 00CD9391 1000587B 0041CDFC 00440BC8 00000001 00419BD0
[13173.095608] ath10k_ahb a000000.wifi: ATH10K_END
[13173.105654] ath10k_ahb a000000.wifi: [1] next: 0x417f48 buf: 0x414b60 sz: 1500 len: 0 count: 0 free: 0
[13173.132733] ath10k_ahb a000000.wifi: failed to find peer 5c:49:7d:a3:27:a5 on vdev 0 after creation
[13173.132810] ath10k_ahb a000000.wifi: failed to add peer 5c:49:7d:a3:27:a5 for vdev 0 when adding a new sta: -2
[13173.137047] ath10k_ahb a800000.wifi: SWBA overrun on vdev 0, skipped old beacon
[13173.150831] ath10k_ahb a000000.wifi: removing peer, cleanup-all, deleting: peer ce1ea200 vdev: 0 addr: 7c:bb:8a:8a:00:d2 
[13173.152464] ath10k_ahb a000000.wifi: Invalid state: 2 in ath10k_htt_tx_32, warning will not be repeated.
[13173.152470] ------------[ cut here ]------------
[13173.152574] WARNING: CPU: 2 PID: 70 at /builder/ipq40xx_generic/build/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-2018-12-11-812d90c0/ath10k-4.19/htt_tx.c:1250 ath10k_ht [... cut ...]
[14790.013441] ath10k_ahb a000000.wifi: rate-override:  Skipping un-supported device-id, hw-nss: 2 dev-id: 0x0
[14790.018397] ath10k_ahb a000000.wifi: rate-override:  Skipping un-supported device-id, hw-nss: 2 dev-id: 0x0
[14790.027998] ath10k_ahb a000000.wifi: rate-override:  Skipping un-supported device-id, hw-nss: 2 dev-id: 0x0
[14790.037963] ath10k_ahb a000000.wifi: rate-override:  Skipping un-supported device-id, hw-nss: 2 dev-id: 0x0
[14790.056434] ath10k_ahb a000000.wifi: device successfully recovered
```

greearb, 2018-12-19:
OK, I think I found it.  Please see if this works better.

[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2696504/firmware-5-full-htt-mgt-community.bin.gz)

fishxz, 2018-12-21:
hey, is this fix in the latest openwrt firmware? i tested your firmware for 2 days and updated today to the latest snapshot and havent noticed any crash so far.

robimarko, 2018-12-22:
Yes, it was merged 2 days ago in OpenWrt and fixes all of the crashes.
Also, there is a PR to update to current ath10k-ct to remove useless firmware warnings that fill dmesg

greearb, 2018-12-22:
Thanks for testing...I'm closing this bug, please open a new one if more crashes or instability is seen.

closed 2018-12-22 by greearb