greearb/ath10k-ct #45: [QCA9980] Firmware crash

elfhack, 2018-12-08:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).
Firmware crashed, due to my Air Conditioner. This is on a 2.4Ghz radio

Software (OS, Firmware version, kernel, driver, etc)
OpenWRT trunk, firmware 10.4b-ct-9980-fW-011-3ee5715a

Hardware (NIC chipset, platform, etc)
QCA9980 on TP-Link C2600

Logs (dmesg, maybe supplicant and/or hostap)
[ 4503.776651] ath10k_pci 0001:01:00.0: Spurious quick kickout for STA f8:04:2e:ef:b1:57
[ 7756.055467] ath10k_pci 0001:01:00.0: Unknown (10.4) eventid: 36925
[ 7756.058172] ath10k_pci 0001:01:00.0: Unknown (10.4) eventid: 36925
[ 7756.124286] ath10k_pci 0001:01:00.0: Unknown (10.4) eventid: 36925
[ 7756.125372] ath10k_pci 0001:01:00.0: Unknown (10.4) eventid: 36925
[ 7758.183354] ath10k_pci 0001:01:00.0: firmware crashed! (guid n/a)
[ 7758.183444] ath10k_pci 0001:01:00.0: qca99x0 hw2.0 target 0x01000000 chip_id 0x003b01ff sub 168c:0002
[ 7758.188479] ath10k_pci 0001:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[ 7758.202114] ath10k_pci 0001:01:00.0: firmware ver 10.4b-ct-9980-fW-011-3ee5715a api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT crc32 17b3a5f9
[ 7758.207757] ath10k_pci 0001:01:00.0: board_file api 2 bmi_id 1:2 crc32 08fa09f2
[ 7758.229621] ath10k_pci 0001:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[ 7758.238777] ath10k_pci 0001:01:00.0: firmware register dump:
[ 7758.246824] ath10k_pci 0001:01:00.0: [00]: 0x00000009 0x00000000 0x00995106 0x00000000
[ 7758.252788] ath10k_pci 0001:01:00.0: [04]: 0x00000000 0x00060124 0x00000000 0x00000000
[ 7758.260423] ath10k_pci 0001:01:00.0: [08]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 7758.268323] ath10k_pci 0001:01:00.0: [12]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 7758.276222] ath10k_pci 0001:01:00.0: [16]: 0x009BF067 0x0A0039FE 0x0A003A08 0x00995106
[ 7758.284121] ath10k_pci 0001:01:00.0: [20]: 0x00000000 0x00401C20 0x00000000 0x00000000
[ 7758.292021] ath10k_pci 0001:01:00.0: [24]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 7758.299918] ath10k_pci 0001:01:00.0: [28]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 7758.307819] ath10k_pci 0001:01:00.0: [32]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 7758.315737] ath10k_pci 0001:01:00.0: [36]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 7758.323616] ath10k_pci 0001:01:00.0: [40]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 7758.331516] ath10k_pci 0001:01:00.0: [44]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 7758.339415] ath10k_pci 0001:01:00.0: [48]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 7758.347315] ath10k_pci 0001:01:00.0: [52]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 7758.355214] ath10k_pci 0001:01:00.0: [56]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 7758.363112] ath10k_pci 0001:01:00.0: Copy Engine register dump:
[ 7758.371019] ath10k_pci 0001:01:00.0: [00]: 0x0004a000   3   3   3   3
[ 7758.376834] ath10k_pci 0001:01:00.0: [01]: 0x0004a400  16  16 501 503
[ 7758.383431] ath10k_pci 0001:01:00.0: [02]: 0x0004a800  14  14  13  14
[ 7758.389855] ath10k_pci 0001:01:00.0: [03]: 0x0004ac00  20  20  22  20
[ 7758.396278] ath10k_pci 0001:01:00.0: [04]: 0x0004b000 979 979  41   1
[ 7758.402703] ath10k_pci 0001:01:00.0: [05]: 0x0004b400  13  13  76  77
[ 7758.409124] ath10k_pci 0001:01:00.0: [06]: 0x0004b800   5   5   5   5
[ 7758.415548] ath10k_pci 0001:01:00.0: [07]: 0x0004bc00   1   1   1   1
[ 7758.421972] ath10k_pci 0001:01:00.0: [08]: 0x0004c000   0   0 127   0
[ 7758.428396] ath10k_pci 0001:01:00.0: [09]: 0x0004c400   0   0   0   0
[ 7758.434819] ath10k_pci 0001:01:00.0: [10]: 0x0004c800   0   0   0   0
[ 7758.441244] ath10k_pci 0001:01:00.0: [11]: 0x0004cc00   0   0   0   0
[ 7758.449696] ath10k_pci 0001:01:00.0: debug log header, dbuf: 0x418fa8  dropped: 0
[ 7758.455108] ath10k_pci 0001:01:00.0: [0] next: 0x418f90 buf: 0x415860 sz: 1500 len: 104 count: 4 free: 0
[ 7758.462576] ath10k_pci 0001:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[ 7758.471101] ath10k: [0000]: 00789A4D 1400581D 00000000 00447108 000F57C8 00000006 00000000 00789A4D
[ 7758.477091] ath10k: [0008]: 1000581B 00006464 00000000 00000000 000F57C8 00789A4D 14006402 71103332
[ 7758.486032] ath10k: [0016]: 0D67B104 00006464 00422A68 00000002 00789A4E 1000587B 00424CF8 00447108
[ 7758.495058] ath10k: [0024]: 00000001 00422A68
[ 7758.504076] ath10k_pci 0001:01:00.0: ATH10K_END
[ 7758.509617] ath10k_pci 0001:01:00.0: [1] next: 0x418fa8 buf: 0x415270 sz: 1500 len: 0 count: 0 free: 0
[ 7758.537870] ath10k_pci 0001:01:00.0: removing peer, cleanup-all, deleting: peer da578e00 vdev: 0 addr: f8:04:2e:ef:b1:57 
[ 7758.537933] ath10k_pci 0001:01:00.0: removing peer, cleanup-all, deleting: peer dbd4e200 vdev: 0 addr: ec:08:6b:84:19:ad 
[ 7758.631508] ieee80211 phy1: Hardware restart was requested
[ 7759.131227] ath10k_pci 0001:01:00.0: Invalid state: 3 in ath10k_htt_tx_32, warning will not be repeated.
[ 7759.131253] ------------[ cut here ]------------
[ 7759.139895] WARNING: CPU: 1 PID: 53 at /home/movi/LEDE/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/ath10k-ct-2018-09-29-b9989fbd/ath10k-4.16/htt_tx.c:1250 ath10k_htt_tx_32+0xe8/0x918 [ath10k_core]
[ 7759.144438] Modules linked in: pppoe ppp_async option ath10k_pci ath10k_core ath usb_wwan pppox ppp_generic nf_conntrack_netlink nf_conntrack_ipv6 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE dvb_usb_rtl28xxu cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY wireguard usbserial slhc rtl2832 rtl2830 nfnetlink nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables fc0013 dvb_usb_v2
[ 7759.213604]  dvb_core crc_ccitt compat sch_cake nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress i2c_mux ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ifb ip6_udp_tunnel udp_tunnel sit tunnel4 ip_tunnel tun nls_utf8 nls_iso8859_1 nls_cp437 md5 md4 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_of_simple ohci_platform ohci_hcd phy_qcom_dwc3 ehci_platform sd_mod scsi_mod ehci_hcd gpio_button_hotplug ext4 jbd2 mbcache crc32c_generic
[... 37 lines cut ...]
[ 7759.630366] [<bf51b0b0>] (drv_start [mac80211]) from [<bf54c108>] (ieee80211_reconfig+0xa30/0xac4 [mac80211])
[ 7759.639829] [<bf54c108>] (ieee80211_reconfig [mac80211]) from [<bf5187d8>] (ieee80211_restart_work+0xb8/0xd4 [mac80211])
[ 7759.649616] [<bf5187d8>] (ieee80211_restart_work [mac80211]) from [<c032fc0c>] (process_one_work+0x24c/0x404)
[ 7759.660520] [<c032fc0c>] (process_one_work) from [<c0330cb8>] (worker_thread+0x344/0x58c)
[ 7759.670328] [<c0330cb8>] (worker_thread) from [<c033587c>] (kthread+0x140/0x14c)
[ 7759.678487] [<c033587c>] (kthread) from [<c0307f48>] (ret_from_fork+0x14/0x2c)
[ 7759.686014] ---[ end trace 07ca7eeec8ac3a89 ]---
[ 7759.693049] ath10k_pci 0001:01:00.0: failed to transmit packet, dropping: -19
[ 7759.697828] ath10k_pci 0001:01:00.0: failed to submit frame: -19
[ 7759.704780] ath10k_pci 0001:01:00.0: failed to push frame: -19
[ 7760.170072] ath10k_pci 0001:01:00.0: failed to transmit packet, dropping: -19
[ 7760.170098] ath10k_pci 0001:01:00.0: failed to submit frame: -19
[ 7760.176177] ath10k_pci 0001:01:00.0: failed to push frame: -19
[ 7760.487400] ath10k_pci 0001:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[ 7760.487453] ath10k_pci 0001:01:00.0: msdu-desc: 2500  skid: 32
[ 7760.565129] ath10k_pci 0001:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[ 7760.565900] ath10k_pci 0001:01:00.0: wmi print 'free: 37620 iram: 36500 sram: 12808'
[ 7760.957593] ath10k_pci 0001:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[ 7761.195485] ath10k_pci 0001:01:00.0: failed to transmit packet, dropping: -19
[ 7761.195586] ath10k_pci 0001:01:00.0: failed to submit frame: -19
[ 7761.201633] ath10k_pci 0001:01:00.0: failed to push frame: -19
[ 7761.262041] ath10k_pci 0001:01:00.0: failed to transmit packet, dropping: -19
[ 7761.262136] ath10k_pci 0001:01:00.0: failed to submit frame: -19
[ 7761.268184] ath10k_pci 0001:01:00.0: failed to push frame: -19
[ 7761.354885] ath10k_pci 0001:01:00.0: device successfully recovered

greearb, 2018-12-20:
Please see if this firmware fixes the problem, and if not, submit new crash logs from this firmware.

[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2700070/firmware-5-full-htt-mgt-community.bin.gz)

elfhack, 2018-12-20:
I've been using 10.4b-ct-9980-fW-012-81e1edd (not-htt) for a couple of days now, and have yet to experience it crash again

greearb, 2018-12-20:
Ok, I think this issue is resolved, please open another one if you find more crashes/problems.

closed 2018-12-20 by greearb