greearb/ath10k-ct #46: (QCA9984) Firmware crash.

danny30au, 2018-12-17:
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.88 (daniel@daniel-VirtualBox) (gcc version 8.2.0 (OpenWrt GCC 8.2.0 r8773-3c04d10483)) #0 SMP Mon Dec 17 04:16:34 2018
[    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Netgear Nighthawk X4S R7800
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 122880
[    0.000000] free_area_init_node: node 0, pgdat c0b29c40, node_mem_map dda39000
[    0.000000]   Normal zone: 960 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 122880 pages, LIFO batch:31
[    0.000000] random: get_random_bytes called from start_kernel+0x88/0x404 with crng_init=0
[    0.000000] percpu: Embedded 15 pages/cpu @dda01000 s29324 r8192 d23924 u61440
[    0.000000] pcpu-alloc: s29324 r8192 d23924 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 121920
[    0.000000] Kernel command line: 
[    0.000000] Bootloader command line (ignored): console=ttyHSL1,115200n8
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 476352K/491520K available (4820K kernel code, 169K rwdata, 1276K rodata, 1024K init, 226K bss, 15168K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xde800000 - 0xff800000   ( 528 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xde000000   ( 480 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0208000 - 0xc07b53f0   (5813 kB)
[    0.000000]       .init : 0xc0a00000 - 0xc0b00000   (1024 kB)
[    0.000000]       .data : 0xc0b00000 - 0xc0b2a540   ( 170 kB)
[    0.000000]        .bss : 0xc0b2c000 - 0xc0b64b58   ( 227 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] clocksource: dg_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 305801671480 ns
[    0.000007] sched_clock: 32 bits at 6MHz, resolution 160ns, wraps every 343597383600ns
[    0.000023] Switching to timer-based delay loop, resolution 160ns
[    0.000220] Calibrating delay loop (skipped), value calculated using timer frequency.. 12.50 BogoMIPS (lpj=62500)
[    0.000246] pid_max: default: 32768 minimum: 301
[    0.000377] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000395] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000967] CPU: Testing write buffer coherency: ok
[    0.001739] Setting up static identity map for 0x42300000 - 0x42300060
[    0.001893] Hierarchical SRCU implementation.
[    0.002633] smp: Bringing up secondary CPUs ...
[    0.004429] smp: Brought up 1 node, 2 CPUs
[    0.004446] SMP: Total of 2 processors activated (25.00 BogoMIPS).
[    0.004457] CPU: All CPU(s) started in SVC mode.
[    0.014952] VFP support v0.3: implementor 51 architecture 64 part 4d variant 2 rev 0
[    0.015125] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.015152] futex hash table entries: 512 (order: 3, 32768 bytes)
[    0.015260] pinctrl core: initialized pinctrl subsystem
[    0.016260] NET: Registered protocol family 16
[    0.016505] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.017861] cpuidle: using governor ladder
[    0.017923] cpuidle: using governor menu
[    0.039354] msm_bus_fabric_init_driver
[    0.040813] usbcore: registered new interface driver usbfs
[    0.040900] usbcore: registered new interface driver hub
[    0.040983] usbcore: registered new device driver usb
[    0.041042] pps_core: LinuxPPS API ver. 1 registered
[    0.041053] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.041088] PTP clock support registered
[    0.042924] clocksource: Switched to clocksource dg_timer
[    0.046150] NET: Registered protocol family 2
[    0.046674] TCP established hash table entries: 4096 (order: 2, 16384 bytes)
[    0.046717] TCP bind hash table entries: 4096 (order: 3, 32768 bytes)
[... 440 lines cut ...]
[ 5732.526080] ath10k_pci 0000:01:00.0: [0] next: 0x423808 buf: 0x419010 sz: 1500 len: 192 count: 7 free: 0
[ 5732.533548] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[ 5732.542082] ath10k: [0000]: 0058DFC2 17FC5823 00000033 00455FF4 DEF60000 00000000 00000002 0058DFC2
[ 5732.548068] ath10k: [0008]: 17FC5823 00000033 00455FF4 DEF60001 00000000 00000002 0058DFC3 17FC5823
[ 5732.557012] ath10k: [0016]: 00000033 00455FF4 DEF60002 00000000 00000002 0058DFC4 1400581D 00000000
[ 5732.566036] ath10k: [0024]: 00455FF4 000F64E8 00000006 00000000 0058DFC4 1000581B 0000DEF6 00000000
[ 5732.575067] ath10k: [0032]: 00000000 000F64E8 0058DFC5 14006402 71103332 74E925C0 0000F6DE 0042D7A4
[ 5732.584097] ath10k: [0040]: 00000001 0058FFA8 17FC0001 0096353E 000015B3 000015B3 00401AE0 91104569
[ 5732.593124] ath10k_pci 0000:01:00.0: ATH10K_END
[ 5732.603156] ath10k_pci 0000:01:00.0: [1] next: 0x4237f0 buf: 0x419600 sz: 1500 len: 0 count: 0 free: 0
[ 5732.606955] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[ 5735.656096] ath10k_pci 0000:01:00.0: wmi command 36967 timeout, restarting hardware
[ 5738.708477] ath10k_pci 0000:01:00.0: wmi command 40859 timeout, restarting hardware
[ 5738.708503] ath10k_pci 0000:01:00.0: failed to send wmi nop: -11
[ 5738.720492] ath10k_pci 0000:01:00.0: failed to send pdev bss chan info request: -108
[ 5738.721414] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dc5d2200 vdev: 0 addr: 78:d2:94:44:00:a9 
[ 5738.823603] ieee80211 phy0: Hardware restart was requested
[ 5738.824212] ath10k_pci 0000:01:00.0: failed to set beacon mode for vdev 0: -108
[ 5738.828167] ath10k_pci 0000:01:00.0: failed to set dtim period for vdev 0: -108
[ 5744.757624] ath10k_pci 0000:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[ 5744.757672] ath10k_pci 0000:01:00.0: msdu-desc: 2500  skid: 32
[ 5744.839426] ath10k_pci 0000:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[ 5744.840270] ath10k_pci 0000:01:00.0: wmi print 'free: 87036 iram: 27252 sram: 18240'
[ 5745.193386] ath10k_pci 0000:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[ 5745.474889] ath10k_pci 0000:01:00.0: device successfully recovered

greearb, 2018-12-17:
This looks similar to some other crashes I have seen lately, I think I introduced an infinite loop in a previous attempt at fixing stale memory access.  Please see if the attached image works better for you.
[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2688252/firmware-5-full-htt-mgt-community.bin.gz)

danny30au, 2018-12-18:
@greearb No worries mate. I just loaded the firmware - you uploaded. I'll test it out and see if it crashes again. So far so good. :)

danny30au, 2018-12-20:
@greearb I confirm that the uploaded firmware you linked - it's not crashing at all in my environment and devices connected to it. You can close this ticket. :) Thanks again for your quick and super helpful reply. Merry Christmas. :)

closed 2018-12-20 by danny30au