greearb/ath10k-ct #48: alot of log spam on latest openwrt snapshot

fishxz, 2018-12-18:
hey,
since this commit : https://github.com/openwrt/openwrt/commit/4df3c71cd4c59d80374dceb5267ecee5b91931ad

i get ALOT more log spam, than i had before.
 i also had some crashes (i already opened another thread about this crashes) which i can't post, because my log gets spammed by other messages.

dmesg
```
[112218.795744] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112219.925946] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112221.030285] ath10k_ahb a000000.wifi: Invalid VHT mcs 15 peer stats
[112221.124398] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112221.907026] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112222.428945] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112222.528105] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112222.537025] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112223.046158] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112223.897118] ath10k_ahb a000000.wifi: Invalid VHT mcs 15 peer stats
[112224.245402] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112226.004547] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112226.601479] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112226.877431] ath10k_ahb a000000.wifi: Invalid VHT mcs 15 peer stats
[112227.046775] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112228.245692] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112229.285113] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112230.325900] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112231.366957] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112231.516079] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112231.926335] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112232.797105] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112235.711854] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112236.931263] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112238.436496] ath10k_ahb a000000.wifi: Invalid VHT mcs 15 peer stats
[112238.744447] ath10k_ahb a000000.wifi: Invalid VHT mcs 15 peer stats
[112239.829671] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112244.474681] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112245.398412] ath10k_ahb a000000.wifi: Invalid VHT mcs 15 peer stats
[112246.854437] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112247.091884] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112248.294723] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112248.431921] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112250.033342] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112250.065874] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112250.215704] ath10k_ahb a000000.wifi: Invalid VHT mcs 15 peer stats
[112250.257696] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112250.302049] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112250.342850] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112250.349770] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112250.354640] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112250.362327] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112250.372539] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112250.384327] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112250.397775] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112250.410674] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112250.428222] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112250.471170] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112250.516754] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112250.602795] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112250.734636] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112251.443813] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112251.836522] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112251.898473] ath10k_ahb a000000.wifi: Invalid VHT mcs 15 peer stats
[112251.929055] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112252.033945] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112252.038500] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112252.165119] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112252.238613] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112252.311609] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112252.419077] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112252.492748] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[... 751 lines cut ...]
[112571.778811] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112571.786836] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112571.872250] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112571.959135] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112572.046005] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112572.155884] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112572.180309] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112572.207665] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112572.223855] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112572.263542] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112572.437414] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112572.441832] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112572.566742] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112572.830421] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[... cut ...]
[112574.400275] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
[112574.879957] ath10k_ahb a800000.wifi: Invalid peer id 2 or peer stats buffer, peer:   (null)  sta:   (null)
```

another message i see very often is the following:
```
Wed Dec 5 20:49:06 2018 kern.warn kernel: [29226.781214] ath10k_ahb a000000.wifi: rate-override: Skipping un-supported device-id, hw-nss: 2 dev-id: 0x0```

OpenWrt: OpenWrt SNAPSHOT r8795-0d23fd2 / LuCI Master (git-18.350.50885-91a73be)
Router: Zyxel nbg6617

greearb, 2018-12-18:
Can you check if this method is being called on your system:

ath10k_fetch_10_2_tx_stats

or if instead it is this one:

ath10k_htt_fetch_peer_stats

(Both from ath10k/htt_rx.c)

fishxz, 2018-12-18:
excuse the question, but how can i check this?

greearb, 2018-12-18:
If you are not sure, then don't worry about it.  It would require editing code and recompiling.  I will add a patch to quieten these warnings...I did not see any obvious problems in the code, and probably it is OK if we drop a few corrupted tx-stats reports.