greearb/ath10k-ct #49: QCA9880 FW crash (phy0 only)

notmyrealhandle, 2018-12-20:
Using firmware you provided in issue #47 (same set up).  1 fw crash only on phy0.

Dmesg output [here](https://gist.github.com/notmyrealhandle/c4dd1c0a5baaed8578ba0771b12bbf64#file-logcat-20181220-scrub-after) (just below ps output) after the crash plus some other stuff.

Similar output just before the crash last night [here](https://gist.github.com/notmyrealhandle/c4dd1c0a5baaed8578ba0771b12bbf64#file-logcat-20181219-scrub-before).  

There are some firmware stats near the bottom of both outputs if that helps.

Since it looks like all you need is dmesg output, I'll only send that in the future unless you ask for more.

Thank you for being so responsive and I hope this helps.

greearb, 2018-12-20:
This should already be fixed in the firmware I uploaded yesterday.  It should be in OpenWRT any day now...maybe already there.  I'm attaching the binary in case that is more convenient.  I do just need dmesg output usually.

[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2700065/firmware-5-full-htt-mgt-community.bin.gz)

Please test this and close the bug if it does indeed fix the problem for you.

notmyrealhandle, 2018-12-21:
I'll try the attached firmware above now and reopen this if there are issues.

FWIW, I can't find this firmware [here](http://candelatech.com/downloads/) or from openwrt.

FYI the firmware attached above:
cfe244d82a5e473b45bc561cc211ce212c0bb1f6ee99737c75960b42f8ee36fa  firmware-5-full-htt-mgt-community.bin

Latest firmware from openwrt (as of last night) and also found [here](http://candelatech.com/downloads/ath10k-10-4b/):
435877ebba90224fe3b696c1c459da281230f527ea05c316b7eeceb1a13b5e19  firmware-5-ct-full-htt-mgt-community-12.bin-lede.002

notmyrealhandle, 2018-12-23:
I got about 2 days on 10.4b-ct-9980-fH-012-81e1edd before the router came to a halt.  

I wasn't around when it went down, the "natives" were restless and they reset it so I have not logs from the crash to share.  Whatever happened, it brought brought down both wired and wireless.  The router was power cycled as it apparently didn't reboot on its own.   its back up now and telling me all is well.

Up to this morning, fw_reset_stats reported no crashes and I saw nothing remarkable in the logs.  After about a day, one of my android devices (2.4 GHz only) disconnected several times with out reporting anything in the logs. For one event, the logs indicate the android's wifi was not likely asleep as it was talking to my www server (around 3 am in the morning of course).

If this happens again and I get a log, I'll post it here (if this issue is still open or in a new issue if you close this issue).

notmyrealhandle, 2018-12-26:
The router crash described in the previous post has not reoccurred after about 3 days uptime when using 10.4b-ct-9980-fH-012-81e1edd.  I made no changes other than starting remote logging.  I am running of master OpenWrt SNAPSHOT r8858-852fc0ba0a and I've seen hints elsewhere about issues with mac80211 so perhaps I ran into something else this time.

In addition, fw_reset_stats indicates no crashes or resets over the past 3 days so this ct-fw version looks pretty good so far.

The unexplained STA drops have happened but at much lower frequency with the last couple ct-fw updates.  I suspect there are multiple causes and the drops over the last 5 days or so seem like they correlate to an android app (now deleted).    

Regards,

nmrh

closed 2018-12-26 by notmyrealhandle