greearb/ath10k-ct #50: [QCA9980] PS4Pro can't communicate on 5Ghz

elfhack, 2018-12-20:
Description of the problem (how to configure, how to reproduce, how often it happens).
PS4Pro on fw 6.20 (latest) has problems communicating on 5Ghz, works perfectly on a 2.4Ghz networks. Both are bridged to LAN. When switching to stock firmware it works as well

Software (OS, Firmware version, kernel, driver, etc)
OpenWRT trunk, 10.4b-ct-9980-fW-012-81e1edd, ath10k 4.16 driver, optimized for CT firmware, probing pci device: 0x40.

Hardware (NIC chipset, platform, etc)
QCA9980, TP-Link C2600

Logs (dmesg, maybe supplicant and/or hostap)
Unfortunately it's hard to debug, since there's no shell on a PS4. On OWRT side i can see it connects to the NIC but then has problems. The console reports no connectivity, and I can't use anything internet related (like browser, seems it can't even get to the DNS server)

Thu Dec 20 20:15:51 2018 daemon.info hostapd: wlan0: STA e8:9e:b4:aa:7c:e7 IEEE 802.11: authenticated
Thu Dec 20 20:15:51 2018 daemon.info hostapd: wlan0: STA e8:9e:b4:aa:7c:e7 IEEE 802.11: associated (aid 1)
Thu Dec 20 20:15:51 2018 daemon.info hostapd: wlan0: STA e8:9e:b4:aa:7c:e7 RADIUS: starting accounting session EFAC944475672A41
Thu Dec 20 20:15:51 2018 daemon.info hostapd: wlan0: STA e8:9e:b4:aa:7c:e7 WPA: pairwise key handshake completed (RSN)
Thu Dec 20 20:15:51 2018 daemon.info dnsmasq-dhcp[3830]: DHCPDISCOVER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:15:51 2018 daemon.info dnsmasq-dhcp[3830]: DHCPOFFER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:15:54 2018 daemon.info dnsmasq-dhcp[3830]: DHCPDISCOVER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:15:54 2018 daemon.info dnsmasq-dhcp[3830]: DHCPOFFER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:15:55 2018 daemon.info dnsmasq-dhcp[3830]: DHCPDISCOVER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:15:55 2018 daemon.info dnsmasq-dhcp[3830]: DHCPOFFER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:15:57 2018 daemon.info dnsmasq-dhcp[3830]: DHCPDISCOVER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:15:57 2018 daemon.info dnsmasq-dhcp[3830]: DHCPOFFER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:15:59 2018 daemon.info dnsmasq-dhcp[3830]: DHCPDISCOVER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:15:59 2018 daemon.info dnsmasq-dhcp[3830]: DHCPOFFER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:16:01 2018 daemon.info dnsmasq-dhcp[3830]: DHCPDISCOVER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:16:01 2018 daemon.info dnsmasq-dhcp[3830]: DHCPOFFER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:16:03 2018 daemon.info dnsmasq-dhcp[3830]: DHCPDISCOVER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:16:03 2018 daemon.info dnsmasq-dhcp[3830]: DHCPOFFER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:16:04 2018 daemon.info dnsmasq-dhcp[3830]: DHCPDISCOVER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:16:04 2018 daemon.info dnsmasq-dhcp[3830]: DHCPOFFER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:16:07 2018 daemon.info dnsmasq-dhcp[3830]: DHCPDISCOVER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:16:07 2018 daemon.info dnsmasq-dhcp[3830]: DHCPOFFER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:16:08 2018 daemon.warn odhcpd[1149]: DHCPV6 SOLICIT IA_NA from 00030001e89eb4aa7ce7 on br-lan: ok 2001:470:71:461::864/128 fdcf:a3f3:45a2::864/128
Thu Dec 20 20:16:09 2018 daemon.info dnsmasq-dhcp[3830]: DHCPDISCOVER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:16:09 2018 daemon.info dnsmasq-dhcp[3830]: DHCPOFFER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:16:10 2018 daemon.warn odhcpd[1149]: DHCPV6 SOLICIT IA_NA from 00030001e89eb4aa7ce7 on br-lan: ok 2001:470:71:461::864/128 fdcf:a3f3:45a2::864/128
Thu Dec 20 20:16:11 2018 daemon.info dnsmasq-dhcp[3830]: DHCPDISCOVER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:16:11 2018 daemon.info dnsmasq-dhcp[3830]: DHCPOFFER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:16:12 2018 daemon.warn odhcpd[1149]: DHCPV6 SOLICIT IA_NA from 00030001e89eb4aa7ce7 on br-lan: ok 2001:470:71:461::864/128 fdcf:a3f3:45a2::864/128
Thu Dec 20 20:16:13 2018 daemon.info dnsmasq-dhcp[3830]: DHCPDISCOVER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:16:13 2018 daemon.info dnsmasq-dhcp[3830]: DHCPOFFER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:16:15 2018 daemon.info dnsmasq-dhcp[3830]: DHCPDISCOVER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:16:15 2018 daemon.info dnsmasq-dhcp[3830]: DHCPOFFER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:16:16 2018 daemon.warn odhcpd[1149]: DHCPV6 SOLICIT IA_NA from 00030001e89eb4aa7ce7 on br-lan: ok 2001:470:71:461::864/128 fdcf:a3f3:45a2::864/128
Thu Dec 20 20:16:17 2018 daemon.info dnsmasq-dhcp[3830]: DHCPDISCOVER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:16:17 2018 daemon.info dnsmasq-dhcp[3830]: DHCPOFFER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:16:19 2018 daemon.info dnsmasq-dhcp[3830]: DHCPDISCOVER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:16:19 2018 daemon.info dnsmasq-dhcp[3830]: DHCPOFFER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:16:22 2018 daemon.info dnsmasq-dhcp[3830]: DHCPDISCOVER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:16:22 2018 daemon.info dnsmasq-dhcp[3830]: DHCPOFFER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:16:24 2018 daemon.warn odhcpd[1149]: DHCPV6 SOLICIT IA_NA from 00030001e89eb4aa7ce7 on br-lan: ok 2001:470:71:461::864/128 fdcf:a3f3:45a2::864/128
Thu Dec 20 20:16:40 2018 daemon.warn odhcpd[1149]: DHCPV6 SOLICIT IA_NA from 00030001e89eb4aa7ce7 on br-lan: ok 2001:470:71:461::864/128 fdcf:a3f3:45a2::864/128
Thu Dec 20 20:16:42 2018 daemon.info dnsmasq-dhcp[3830]: DHCPDISCOVER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7
Thu Dec 20 20:16:42 2018 daemon.info dnsmasq-dhcp[3830]: DHCPOFFER(br-lan) 10.1.1.35 e8:9e:b4:aa:7c:e7

greearb, 2018-12-20:
Please see if this firmware works...it is very similar to stock QCA 10.4.3.something FW.

[firmware-5-full-community-commit-006-307bc99.bin.gz](https://github.com/greearb/ath10k-ct/files/2700325/firmware-5-full-community-commit-006-307bc99.bin.gz)

elfhack, 2018-12-20:
Doesn't seem like this one can even load

[   25.909581] ath10k_pci 0000:01:00.0: qca99x0 hw2.0 target 0x01000000 chip_id 0x003b01ff sub 168c:0002
[   25.909628] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   25.920616] ath10k_pci 0000:01:00.0: firmware ver 10.4b-ct-9980-fW-000-307bc99 api 5 features peer-flow-ctrl,wmi-10.x-CT crc32 74eef09e
[   25.984568] ath10k_pci 0000:01:00.0: board_file api 2 bmi_id 1:1 crc32 08fa09f2
[   28.086593] ath10k_pci 0000:01:00.0: 10.4 wmi init: vdevs: 16  peers: 528  tid: 102
[   28.086635] ath10k_pci 0000:01:00.0: msdu-desc: 2500  skid: 32
[   28.163916] ath10k_pci 0000:01:00.0: firmware crashed! (guid n/a)
[   28.163950] ath10k_pci 0000:01:00.0: qca99x0 hw2.0 target 0x01000000 chip_id 0x003b01ff sub 168c:0002
[   28.169016] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   28.191557] ath10k_pci 0000:01:00.0: firmware ver 10.4b-ct-9980-fW-000-307bc99 api 5 features peer-flow-ctrl,wmi-10.x-CT crc32 74eef09e
[   28.191832] ath10k_pci 0000:01:00.0: board_file api 2 bmi_id 1:1 crc32 08fa09f2
[   28.204771] ath10k_pci 0000:01:00.0: htt-ver 0.0 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 512 raw 0 hwcrypto 1
[   28.214376] ath10k_pci 0000:01:00.0: firmware register dump:
[   28.220260] ath10k_pci 0000:01:00.0: [00]: 0x00000009 0x000015B3 0x00941AC9 0x00955B31
[   28.226253] ath10k_pci 0000:01:00.0: [04]: 0x00941AC9 0x00060330 0x0000001A 0x004186F8
[   28.233630] ath10k_pci 0000:01:00.0: [08]: 0x00000003 0x00000898 0x000FFD14 0x004596B0
[   28.243738] ath10k_pci 0000:01:00.0: [12]: 0x00000009 0x00000000 0x00953718 0x00953723
[   28.249429] ath10k_pci 0000:01:00.0: [16]: 0x00953438 0x00953718 0x00000000 0x00000000
[   28.259603] ath10k_pci 0000:01:00.0: [20]: 0x40941AC9 0x0040648C 0x00000001 0x00000000
[   28.265228] ath10k_pci 0000:01:00.0: [24]: 0x809C30A5 0x004064EC 0x00406EA8 0xC0941AC9
[   28.275398] ath10k_pci 0000:01:00.0: [28]: 0x809C3233 0x0040651C 0x00000023 0x00000898
[   28.281027] ath10k_pci 0000:01:00.0: [32]: 0x809C9548 0x0040653C 0x0045A240 0x00000002
[   28.290985] ath10k_pci 0000:01:00.0: [36]: 0x809C01A2 0x0040655C 0x004065BC 0x00000000
[   28.296826] ath10k_pci 0000:01:00.0: [40]: 0x809C8A69 0x0040658C 0x004065BC 0x00406EA8
[   28.306929] ath10k_pci 0000:01:00.0: [44]: 0x80981330 0x004065BC 0x00406EA8 0x00406FF0
[   28.312623] ath10k_pci 0000:01:00.0: [48]: 0x809C8D9D 0x004065EC 0x00406EA8 0x004118A0
[   28.322907] ath10k_pci 0000:01:00.0: [52]: 0x8098011C 0x0040661C 0x00406EA8 0x00000008
[   28.328422] ath10k_pci 0000:01:00.0: [56]: 0x809C8F73 0x0040663C 0x00406EA8 0x00000008
[   28.338478] ath10k_pci 0000:01:00.0: Copy Engine register dump:
[   28.344226] ath10k_pci 0000:01:00.0: [00]: 0x0004a000  14  14   3   3
[   28.351802] ath10k_pci 0000:01:00.0: [01]: 0x0004a400   4   4  14  15
[   28.356636] ath10k_pci 0000:01:00.0: [02]: 0x0004a800   1   1   0   1
[   28.365087] ath10k_pci 0000:01:00.0: [03]: 0x0004ac00   2   2   3   2
[   28.369483] ath10k_pci 0000:01:00.0: [04]: 0x0004b000   0   0  40   0
[   28.378177] ath10k_pci 0000:01:00.0: [05]: 0x0004b400   1   1   0   1
[   28.382330] ath10k_pci 0000:01:00.0: [06]: 0x0004b800   0   0   0   0
[   28.390875] ath10k_pci 0000:01:00.0: [07]: 0x0004bc00   1   1   1   1
[   28.395177] ath10k_pci 0000:01:00.0: [08]: 0x0004c000   0   0 127   0
[   28.403715] ath10k_pci 0000:01:00.0: [09]: 0x0004c400   1   1   1   1
[   28.408024] ath10k_pci 0000:01:00.0: [10]: 0x0004c800   0   0   0   0
[   28.416337] ath10k_pci 0000:01:00.0: [11]: 0x0004cc00   0   0   0   0
[   28.425783] ath10k_pci 0000:01:00.0: debug log header, dbuf: 0x407ecc  dropped: 0
[   28.429753] ath10k_pci 0000:01:00.0: [0] next: 0x407ee4 buf: 0xeab2c sz: 1500 len: 896 count: 68 free: 0
[   28.437252] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[   28.444306] ath10k: [0000]: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   28.452471] ath10k: [0008]: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   28.459239] ath10k: [0016]: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   28.470413] ath10k: [0024]: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   28.477293] ath10k: [0032]: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   28.488230] ath10k: [0040]: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   28.495349] ath10k: [0048]: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   28.506387] ath10k: [0056]: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   28.513405] ath10k: [0064]: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   28.524213] ath10k: [0072]: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   28.531461] ath10k: [0080]: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   28.542690] ath10k: [0088]: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   28.549516] ath10k: [0096]: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   28.559802] ath10k: [0104]: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   28.567572] ath10k: [0112]: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   28.578332] ath10k: [0120]: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[... cut ...]
[   28.675904] ath10k: [0208]: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   28.687289] ath10k: [0216]: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   28.693960] ath10k_pci 0000:01:00.0: ATH10K_END
[   28.708330] ath10k_pci 0000:01:00.0: [1] next: 0x407ecc buf: 0xeb11c sz: 1500 len: 0 count: 0 free: 0
[   33.113810] ath10k_pci 0000:01:00.0: wmi unified ready event not received
[   33.203639] ath10k_pci 0000:01:00.0: could not init core (-110)
[   33.204082] ath10k_pci 0000:01:00.0: could not probe fw (-110)

greearb, 2018-12-21:
Firmware ran OOM.  Please try editing your modprobe.conf for ath10k and set vdevs to 4 (default is 16).

[root@lf0313-6477 lanforge]# cat /etc/modprobe.d/ath10k.conf
...
options ath10k_core num_vdevs_ct=4

And see if that will load and work...

elfhack, 2018-12-21:
Yep, this one works fine

greearb, 2018-12-21:
Ok, I have uploaded a series of binaries for each commit.  The commit-ZZZ in the name shows the commit number...you started with 006.  Please use bisect logic to find the bad commit.

https://www.candelatech.com/downloads/ath10k-10-4b/all_builds/all_builds-9980-htt-dec-21-2018.tar.gz

elfhack, 2018-12-22:
Right, so I went through these, and anything past 153 kills it - hostapd doesn't even acknowledge any connection attempt, and the PS4 complains on not getting a DHCP lease. Which is the first time htt is mentioned, which as i mentioned might not be the problem I have (but a problem nontheless). Because the one I have - using firmware-5-ct-full-community-12.bin-lede.002 (10.4b-ct-9980-fW-012-81e1edd), again, non-htt is that it is connecting, is getting a DHCP lease, but then quickly seemingly stopping any traffic soon after getting that lease. Might we try the non-htt versions then?
[image]
[image]

greearb, 2019-01-02:
I am working on building a new series...I'm having to backport/tweak some fixes since 9980 won't compile on a lot of the earlier commits in the tree....hopefully I'll have this done today.

greearb, 2019-01-03:
Please try bisecting with this set of binaries:
http://www.candelatech.com/downloads/ath10k-10-4b/all_builds/all_builds-9980-wmi-jan-3-2019.tar.gz

elfhack, 2019-01-10:
Sorry for the lag, but i finally got around to bisecting this. 
firmware-5-full-community-commit-910-3e379a7d.bin is ok
firmware-5-full-community-commit-911-4e96efac.bin onwards is broken

greearb, 2019-01-11:
I re-sorted the commits a bit and built new images.  Please try these.  Probably your problem is in the 900 to 915 range, but please double-check and re-bisect.

https://www.candelatech.com/downloads/ath10k-10-4b/all_builds/all_builds-9980-wmi-commit-900+jan-11-2019.tar.gz

elfhack, 2019-01-11:
firmware-5-full-community-commit-909-77c3d414.bin works
firmware-5-full-community-commit-910-2dfbe546.bin onwards is bad

greearb, 2019-01-12:
That is...interesting.  Commit 2dfbe546 is a rate-ctrl optimization that decreases the amount of probes when lots of peers are present..  I have a hard time seeing how it could really break things.  Any chance you can take an over-the-air packet capture with that commit enabled and your PS4Pro attempting to talk to it?  Preferably with encryption disabled so it is easier to understand the packet capture, but probably I can make use of and encrypted capture as well.

elfhack, 2019-01-12:
Sure, just tell me exactly how to do it

greearb, 2019-01-12:
You really need a third wifi device that can act as a monitor-mode sniffer.  Considering you have a 4x4 AP, then you would want a 4x4 sniffer for best results.  Ath10k-ct on 9980 or 9984 NIC would be a good option.  I think some broadcom chips in laptops can also sniff but I haven't used them myself.

This seems to be a fairly useful description of how to do it:
https://sandilands.info/sgordon/capturing-wifi-in-monitor-mode-with-iw

elfhack, 2019-01-13:
Hopefully this is what is needed. 2 captures, one with working 909 and bad 910. I turn on the console from the off state, then run a network test (it's supposed to reconnect during it), then try to pull up a webpage in the browser

[pcaps.zip](https://github.com/greearb/ath10k-ct/files/2753072/pcaps.zip)

greearb, 2019-01-16:
I believe the captures were made with the monitor mode sniffer in 20Mhz instead of 80Mhz, so a lot of the frames are missing.  And over all, I don't see any obvious differences in the sniff.  My co-worker has a PS4, he is giong to bring it in next week and we will see if we can reproduce the problem.

elfhack, 2019-01-16:
These should be better - i need to link them, since they're too big https://cloud.movishell.pl/s/iixBg58waknZYZP

greearb, 2019-01-22:
Please see if this image works for you.  It worked on our ps4-Pro system (when compiled for our 9984 chipset AP).  
[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2781227/firmware-5-full-htt-mgt-community.bin.gz)

elfhack, 2019-01-24:
Nope, no change in behaviour

greearb, 2019-01-24:
Can you show me dmesg after reboot with that firmware so I can make sure you are running the same image that I think you are?

elfhack, 2019-01-26:
Sure

```[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.82 (movi@Nucille) (gcc version 7.3.0 (OpenWrt GCC 7.3.0 r8614-78ca6a5578)) #0 SMP Fri Nov 30 08:04:29 2018
[    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: TP-Link Archer C2600
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 122880
[    0.000000] free_area_init_node: node 0, pgdat c0b258c0, node_mem_map ddc39000
[    0.000000]   Normal zone: 960 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 122880 pages, LIFO batch:31
[    0.000000] random: get_random_bytes called from start_kernel+0x88/0x408 with crng_init=0
[    0.000000] percpu: Embedded 15 pages/cpu @ddc01000 s29324 r8192 d23924 u61440
[    0.000000] pcpu-alloc: s29324 r8192 d23924 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 121920
[    0.000000] Kernel command line: 
[    0.000000] Bootloader command line (ignored): console=ttyHSL1,115200n8 root=mtd:rootfs rootfstype=squashfs
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 478412K/491520K available (4765K kernel code, 152K rwdata, 1264K rodata, 1024K init, 228K bss, 13108K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xde800000 - 0xff800000   ( 528 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xde000000   ( 480 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0208000 - 0xc07a76d8   (5758 kB)
[    0.000000]       .init : 0xc0a00000 - 0xc0b00000   (1024 kB)
[    0.000000]       .data : 0xc0b00000 - 0xc0b261c0   ( 153 kB)
[    0.000000]        .bss : 0xc0b28000 - 0xc0b61098   ( 229 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] clocksource: dg_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 305801671480 ns
[    0.000007] sched_clock: 32 bits at 6MHz, resolution 160ns, wraps every 343597383600ns
[    0.000020] Switching to timer-based delay loop, resolution 160ns
[    0.000231] Calibrating delay loop (skipped), value calculated using timer frequency.. 12.50 BogoMIPS (lpj=62500)
[    0.000255] pid_max: default: 32768 minimum: 301
[    0.000381] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000397] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000959] CPU: Testing write buffer coherency: ok
[    0.001725] Setting up static identity map for 0x42300000 - 0x42300060
[    0.001879] Hierarchical SRCU implementation.
[    0.002616] smp: Bringing up secondary CPUs ...
[    0.004392] smp: Brought up 1 node, 2 CPUs
[    0.004408] SMP: Total of 2 processors activated (25.00 BogoMIPS).
[    0.004417] CPU: All CPU(s) started in SVC mode.
[    0.015740] VFP support v0.3: implementor 51 architecture 64 part 4d variant 2 rev 0
[    0.015911] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.015934] futex hash table entries: 512 (order: 3, 32768 bytes)
[    0.016041] pinctrl core: initialized pinctrl subsystem
[    0.017045] NET: Registered protocol family 16
[    0.017292] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.018645] cpuidle: using governor ladder
[    0.018711] cpuidle: using governor menu
[    0.039673] msm_bus_fabric_init_driver
[    0.041051] usbcore: registered new interface driver usbfs
[    0.041133] usbcore: registered new interface driver hub
[    0.041231] usbcore: registered new device driver usb
[    0.041289] pps_core: LinuxPPS API ver. 1 registered
[    0.041304] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.041335] PTP clock support registered
[    0.043168] clocksource: Switched to clocksource dg_timer
[    0.046386] NET: Registered protocol family 2
[... 380 lines cut ...]
[   40.154660] ath10k_pci 0001:01:00.0: wmi print 'P 48/48 V 4 K 144 PH 152 T 162  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   40.154857] ath10k_pci 0001:01:00.0: wmi print 'free: 51680 iram: 37124 sram: 18804'
[   40.534678] ath10k_pci 0001:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[   40.534965] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   41.074595] ath10k_pci 0001:01:00.0: NOTE:  Firmware DBGLOG output disabled in debug_mask: 0x10000000
[   41.809005] ath10k_pci 0000:01:00.0: 10.4 wmi init: vdevs: 4  peers: 48  tid: 96
[   41.809040] ath10k_pci 0000:01:00.0: msdu-desc: 2500  skid: 32
[   41.884789] ath10k_pci 0000:01:00.0: wmi print 'P 48/48 V 4 K 144 PH 152 T 162  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[... cut ...]
[   42.256529] device wlan0 entered promiscuous mode
[   42.738964] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   42.739200] br-lan: port 2(wlan1) entered blocking state
[   42.744574] br-lan: port 2(wlan1) entered forwarding state
[   43.179687] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   43.179795] br-lan: port 3(wlan0) entered blocking state
[   43.185184] br-lan: port 3(wlan0) entered forwarding state
[   46.324216] usb 1-1: DVB: adapter 0 frontend 0 frequency 0 out of range (174000000..862000000)
```

greearb, 2019-01-28:
That commit built for 9984 works for us.  We will do some more testing with the PS4Pro and see if we notice more problems.  We don't have a 9980 NIC that we can easily use for this test.

sq-bb, 2019-02-15:
I am also running into PS4Pro troubles with 5Ghz wifi. In my case, I'm using a Netgear r7800 (QCA9984) running a recent build of OpenWRT master and ct htt firmware version 10.4b-ct-9984-fH-012-5815a26a3. I hesitated typing this because I'm afraid I can't add very much to what has been reported above, but I figured I could be an extra tester, if nothing else.

Just as reported by elfhack, PS4Pro connectivity works fine over 2.4Ghz on ct firmware and either frequency on stock firmware (version 10.4-3.9.0.2-00021), but seems to stall over 5Ghz on ct firmware. In that case, the PS4 can reliably connect to the AP and obtain a dhcp lease, but appears unable to communicate with the internet.

I bisected using the binaries from bug 66 (I had hoped to also test for bugs 65 and 66, but unfortunately could not reproduce them on even the latest commit - sorry!), and again like elfhack, found that:
firmware-5-full-htt-mgt-community-commit-909-77c3d414e.bin works
firmware-5-full-htt-mgt-community-commit-910-2dfbe5468.bin has the issue

It sounds like you were able to do some local testing with the 9984 and had a working build. I don't have the necessary hardware for packet captures, but if an extra test environment would be useful nonetheless, I'm certainly willing to help.

Thank you very much for all your work on this firmware! Other than this one issue, it is currently providing very stable connectivity for our home network. :)

mocmocamoc, 2020-01-17:
This issue is still present with a master branch build of OpenWrt and QCA9984, has there been any progress on reproducing/fixing the problem?

greearb, 2020-01-17:
What firmware version are you using?  And, assuming it is recent, would you be interested in bisecting firmware images to find where this regression was added?

mocmocamoc, 2020-01-17:
I'm currently using ath10k-9984-10-4b/firmware-5-ct-full-community-12.bin-lede.014 as specified by OpenWrt master. I've tried going back as far as firmware-5-ct-full-community-12.bin-lede.001 but that exhibits the same problem with a PS4 Pro.

mocmocamoc, 2020-01-17:
I also have a Surface Pro 6 which uses the same Marvell 88W8897 wifi chip as the PS4 Pro, and the Surface Pro 6 works fine with master OpenWrt using firmware-5-ct-full-community-12.bin-lede.014.

greearb, 2020-01-17:
Ok, that is recent enough that it should have the fix I put in to fix the problem we could reproduce locally.  If you want to bisect binaries, grab the tarball from here and let me know the last good commit, assuming there is one:  http://www.candelatech.com/downloads/ath10k-9984-10-4b/bisect

mocmocamoc, 2020-01-18:
I've just tried firmware-5-full-community-commit-152-f0b1d364c.bin (the highest-numbered non-htt version) from that archive and the PS4 Pro works correctly with that version.

greearb, 2020-01-18:
The HTT versions should work fine with the ath10k-ct driver in OpenWRT.

mocmocamoc, 2020-01-18:
Thanks, bisection shows that the problem started between these two versions:
Good: firmware-5-full-htt-mgt-community-commit-1084-0a05ef49a.bin
Bad: firmware-5-full-htt-mgt-community-commit-1085-7597bb62e.bin

greearb, 2020-01-18:
Can you do a capture with a 9984 NIC for the good and bad commit?  Open AUTH if you can so it is easier to understand what is going on.  The 'bad' commit changes rate-ctrl in a relatively minor way,
so I am not sure why that would totally break communication.

mocmocamoc, 2020-01-19:
I've sent you a mail at the address on your profile with the capture, please let me know if any further details are required.

greearb, 2020-01-24:
Please see if this image works better.  It is my latest code, but adjusted to make rate-ctrl less aggressive in ramping up rates.
[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/4110204/firmware-5-full-htt-mgt-community.bin.gz)

mocmocamoc, 2020-01-24:
The PS4 Pro still has the same connectivity problems with that firmware unfortunately.

greearb, 2020-01-26:
Here's a new series to bisect since I had rebased since the older series that you bisected against.

http://www.candelatech.com/downloads/ath10k-9984-10-4b/bisect/all_builds-9984-H-jan-27-2020.tar.gz

mocmocamoc, 2020-01-26:
With that set the problem now starts between 1085-cd52f229e (good) and 1086-34b49cc18 (bad).

greearb, 2020-01-26:
That's quite interesting, it bisects to the same patch (that I had moved later in my tree).  But, earlier I had tried to send you the equiv of 1085-cd52f229e and you indicated it had issues?  Can you please double-check that last bisect step, and be sure to use dmesg to confirm you are loading the firmware that you think you are loading?

mocmocamoc, 2020-01-26:
I've switched back and forth between 1085-cd52f229e and 1086-34b49cc18 a few times, confirming via dmesg each time, and it still appears that the former is good and the latter is bad. The two files you sent me by email on Friday contain the string "10.4b-ct-9984-fH-012-34b49cc18", which suggests they may be equivalent to 1086-34b49cc18?

greearb, 2020-01-28:
Thanks for all the help bisecting this.  Bug is now fixed.  It was bug introduced in rate-ctrl by me several years ago.  I'm attaching a top-of-tree working image (previously sent privately in email) in case others want to try it.  I'll send updates to openwrt soon, after hopefully fixing a few other bugs that are actively being worked on.
[firmware-5-full-htt-mgt-community.bin-master.gz](https://github.com/greearb/ath10k-ct/files/4120408/firmware-5-full-htt-mgt-community.bin-master.gz)

closed 2020-01-28 by greearb