greearb/ath10k-ct #51: question about ping on my qca4019 device

fishxz, 2018-12-22:
hey,
i know its probably not an actual issue, but i still want to be sure.

i was running a tp-link wdr3600 for many years with openwrt and my ping on 5ghz was always <1ms and in worst cases 1ms. after i switched to my current device (zyxel nbg6617) i noticed my ping is slightly higher there (jumping mostly in the 1-5ms range).

so i wonder if this is hardware related or could it be a driver related problem?

it may sound like nitpicking, but im courious why this happens.

mastum, 2018-12-27:
Hi, I have tried now with the AVM 4040 and OpenWrt SNAPSHOT r8883-5f978eec18 / LuCI Master (git-18.361.23068-be24925), ath10k-firmware-qca4019-ct 2018-10-10-d366b80d-1and kmod-ath10k-ct
4.14.90+2018-12-20-118e16da-1 compiled today.
My PC have Intel AC 3168 and ping is always <1ms /1ms with 5 GHz.

greearb, 2019-01-02:
This sounds like power-save behaviour to me, and probably is working as designed.  If you test with stock QCA firmware and see reliably worse ping time on CT firmware, then it might be a regression in CT firmware and be something we can bisect.  Otherwise, probably just close this issue.

greearb, 2019-02-05:
I think this is likely not a real problem so closing this bug.  Please re-open or open a new one if you find more problems.

closed 2019-02-05 by greearb