greearb/ath10k-ct #53: Options and current state of 5MHz and 10MHz channel widths

ae6xe, 2018-12-29:
An enhancement request and questions. 

1) Add 5MHz and 10MHz channel width options
2) Pointers to migrate to ath10k-ct in IBSS mode for http://arednmesh.org project

Background on #1)
My investigation so far has not found support for 5 and 10 MHz channels in ath10k or ath10k-ct and firmware.   This is a core need for the http://www.arednmesh.org project -- http://github.com/aredn/aredn_ar71xx .   Our community generally has link distances of many kilometers with 50 to 70km links common place now.    10MHz channel is the go-to width for a long distance link because this provides higher throughput than 20MHz at these distances with the commonly available hardware in use.

I've applied these changes with mac80211 backport version in openwrt 18.06.1 -- no success:

https://patchwork.kernel.org/patch/10066513/
Not finding any errors/warnings, just stays at 20MHz channel, with Openwrt UCI settings at chanbw = 10.  Where else could I look?  Or should these changes be sufficient and I need to debug further?  

Specific Issues on #2)
Our builds are currently on top of openwrt 18.06.1 with modified channels in the Amateur Radio allocations, e.g. up to channel 184 in 5GHz, using ath9k.  I'm working to do the same with ath10k.   We're using IBSS and OLSR right now, considering BATMAN in the future.    

I'm building with ath10k-4.13 same as default in 18.06.1 (which is a 4.9 kernel?).  Is there a better option to build with?   Not quite ready to go to openwrt master and ath79,  although would it be better if we did to get ath10k-ct working for our needs?

 Is there an option to see the rate selection table in defugfs?     .../phy0/netdev:wlan0/stations/<MAC>/rc-stats 

Any pointers what else might need to happen to define a new country code and extended channels, same as ath9k, for amateur radio US part 97 channel allocations?   If I apply similar ath9k patches, is there anything else that might need to happen?:
https://github.com/aredn/aredn_ar71xx/blob/develop/patches/701-extended-spectrum.patch
https://github.com/aredn/aredn_ar71xx/blob/develop/patches/702-enable-country-hx.patch

Would appreciate any advice to help short-cut the time we spend on figuring this out for ath10k, which will be a big learning curve for us.

Joe AE6XE (ae6xe@arrl.net)

greearb, 2019-01-02:
First, please open one issue for one question/bug in the future.

For 5/10 Mhz support, I have no current plans to implement it.  This would normally be work I would wait to do until some company wanted to fund it.  Contact me via email if you wish to pursue funding this.

For rate-selection, CT firmware reports the tx rates used, so you could gather stats either directly in the driver or higher in the mac80211 stack.  I have no current plans to implement this either, but some other upstream patches I have seen discussed for mac80211 might offer this.

Thanks,
Ben

CodeFetch, 2019-03-08:
@ae6xe It's so sad that radio amateurs often cook their own soup. Contribute to https://github.com/freifunk-gluon/gluon/ and make it more attractive for radio amateurs. Some of us are Batman developers thus it just works and we are working on proper Babel support, too. 
Vy 73 de DO6VW

brealin, 2020-09-04:
If it can support 5M/10M bandwidth support, it will get a wider range of application scenarios。