greearb/ath10k-ct #55: qca4019

fishxz, 2019-01-02:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).
random wifi crash. after this crash my wifi got very unstable even that the device successfully recovered

Software (OS, Firmware version, kernel, driver, etc)
OpenWrt SNAPSHOT r8917-ae622c93b3 / LuCI Master (git-18.361.48411-39aa247)
firmware ver 10.4b-ct-4019-fW-012-81e1edd5
kernel 4.14.90

Hardware (NIC chipset, platform, etc)
Zyxel nbg6617 IPQ4018

Logs (dmesg, maybe supplicant and/or hostap)
```
[232315.859317] ath10k_ahb a000000.wifi: firmware crashed! (guid 4346a77a-d8a3-4a1c-abc9-34aa8ba3bb22)
[232315.859401] ath10k_ahb a000000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[232315.867276] ath10k_ahb a000000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[232315.881301] ath10k_ahb a000000.wifi: firmware ver 10.4b-ct-4019-fW-012-81e1edd5 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT crc32 adeedaac
[232315.891346] ath10k_ahb a000000.wifi: board_file api 2 bmi_id 0:16 crc32 bcebe54c
[232315.909786] ath10k_ahb a000000.wifi: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[232315.918262] ath10k_ahb a000000.wifi: firmware register dump:
[232315.927462] ath10k_ahb a000000.wifi: [00]: 0x0000000B 0x000015B3 0x0099C12B 0x00975B31
[232315.933079] ath10k_ahb a000000.wifi: [04]: 0x0099C12B 0x00060530 0x00000005 0x00000007
[232315.940933] ath10k_ahb a000000.wifi: [08]: 0x0041DEF8 0x004336D4 0x00000000 0x0043418C
[232315.948900] ath10k_ahb a000000.wifi: [12]: 0x00000009 0x00000000 0x009C2B34 0x009C2B38
[232315.956890] ath10k_ahb a000000.wifi: [16]: 0x00971238 0x009606B6 0x009606B6 0x00000000
[232315.964875] ath10k_ahb a000000.wifi: [20]: 0x4099C12B 0x0040569C 0x00433694 0x00000000
[232315.972837] ath10k_ahb a000000.wifi: [24]: 0x8099EAD9 0x004056FC 0x00975A00 0xC099C12B
[232315.980848] ath10k_ahb a000000.wifi: [28]: 0x80997A2A 0x0040573C 0x0041DEF8 0x00432274
[232315.988829] ath10k_ahb a000000.wifi: [32]: 0x80994EFC 0x0040576C 0x00000001 0x0041DEF8
[232315.996821] ath10k_ahb a000000.wifi: [36]: 0x8098F950 0x004057AC 0x0041C50C 0x00446240
[232316.004807] ath10k_ahb a000000.wifi: [40]: 0x809630B5 0x0040599C 0x0041C50C 0x0098F948
[232316.012766] ath10k_ahb a000000.wifi: [44]: 0x80960DF0 0x004059BC 0x0000001F 0x00400000
[232316.020778] ath10k_ahb a000000.wifi: [48]: 0x80960DC2 0x004059EC 0x00400000 0x00000000
[232316.028760] ath10k_ahb a000000.wifi: [52]: 0x80960E0D 0x00405A0C 0x00000000 0x00400600
[232316.036751] ath10k_ahb a000000.wifi: [56]: 0x40960024 0x00405A2C 0x004033D8 0x004033D8
[232316.044734] ath10k_ahb a000000.wifi: Copy Engine register dump:
[232316.052698] ath10k_ahb a000000.wifi: [00]: 0x0004a000   4   4   3   3
[232316.058974] ath10k_ahb a000000.wifi: [01]: 0x0004a400   8   8  45  47
[232316.065308] ath10k_ahb a000000.wifi: [02]: 0x0004a800  25  25  24  25
[232316.071793] ath10k_ahb a000000.wifi: [03]: 0x0004ac00   1   1   2   1
[232316.078334] ath10k_ahb a000000.wifi: [04]: 0x0004b000 1605 1591  73  73
[232316.084841] ath10k_ahb a000000.wifi: [05]: 0x0004b400  15  15 428 431
[232316.091671] ath10k_ahb a000000.wifi: [06]: 0x0004b800  26  26  26  26
[232316.098033] ath10k_ahb a000000.wifi: [07]: 0x0004bc00   1   1   1   1
[232316.104546] ath10k_ahb a000000.wifi: [08]: 0x0004c000   0   0 127   0
[232316.111029] ath10k_ahb a000000.wifi: [09]: 0x0004c400   1   1   1   1
[232316.117566] ath10k_ahb a000000.wifi: [10]: 0x0004c800   0   0   0   0
[232316.124048] ath10k_ahb a000000.wifi: [11]: 0x0004cc00   0   0   0   0
[232316.131614] ath10k_ahb a000000.wifi: debug log header, dbuf: 0x4187a0  dropped: 0
[232316.137116] ath10k_ahb a000000.wifi: [0] next: 0x4187b8 buf: 0x414bb0 sz: 1500 len: 212 count: 8 free: 0
[232316.145670] ath10k_ahb a000000.wifi: ath10k_pci ATH10K_DBG_BUFFER:
[232316.154282] ath10k: [0000]: 062D0D2F 1400581D 00000000 004415A8 000F01CC 00000006 00000000 062D0D2F
[232316.160336] ath10k: [0008]: 1000581B 000027A5 00000000 00000000 000F01CC 062D0D2F 14006402 71103332
[232316.169739] ath10k: [0016]: A37D495C 0000A527 0041A4F0 00000007 062D0D30 13FC4C07 211000A1 00000939
[232316.178852] ath10k: [0024]: 00000009 004341FC 062D0D30 1000587B 0041D66C 004415A8 00000001 0041A4F0
[232316.187967] ath10k: [0032]: 062D0D30 17FC587D 51100002 000F01CC 00000001 0000000B 004415A8 062D0D30
[232316.197080] ath10k: [0040]: 17FC4C07 711057A2 00000939 00000000 0000093A 00000001 062D0D30 17FC0001
[232316.206193] ath10k: [0048]: 0099C12B 000015B3 000015B3 0040558C 91104569
[232316.215303] ath10k_ahb a000000.wifi: ATH10K_END
[232316.222982] ath10k_ahb a000000.wifi: [1] next: 0x4187a0 buf: 0x4151a0 sz: 1500 len: 0 count: 0 free: 0
[232316.251239] ath10k_ahb a000000.wifi: failed to find peer 5c:49:7d:a3:27:a5 on vdev 0 after creation
[232316.251295] ath10k_ahb a000000.wifi: failed to add peer 5c:49:7d:a3:27:a5 for vdev 0 when adding a new sta: -2
[232316.251546] ath10k_ahb a800000.wifi: SWBA overrun on vdev 0, skipped old beacon
[232316.259719] ath10k_ahb a000000.wifi: removing peer, cleanup-all, deleting: peer cf844800 vdev: 0 addr: 2c:59:8a:73:2b:6e 
[232316.270117] ath10k_ahb a000000.wifi: Invalid state: 2 in ath10k_htt_tx_32, warning will not be repeated.
[232316.276939] ath10k_ahb a000000.wifi: removing peer, cleanup-all, deleting: peer cf844600 vdev: 0 addr: b8:27:eb:e5:a6:1e 
[232316.287760] ------------[ cut here ]------------
[... 258 lines cut ...]
[236030.042394] ath10k_ahb a800000.wifi: SWBA overrun on vdev 0, skipped old beacon
[236030.085491] ath10k_ahb a000000.wifi: removing peer, cleanup-all, deleting: peer cef78800 vdev: 0 addr: 5c:49:7d:a3:27:a5 
[236030.085893] ath10k_ahb a000000.wifi: removing peer, cleanup-all, deleting: peer cef78400 vdev: 0 addr: f4:f5:d8:54:c6:aa 
[236030.096041] ath10k_ahb a000000.wifi: removing peer, cleanup-all, deleting: peer cc56da00 vdev: 0 addr: b8:27:eb:50:dc:da 
[236030.106622] ath10k_ahb a000000.wifi: removing peer, cleanup-all, deleting: peer cc56d600 vdev: 0 addr: b8:27:eb:e5:a6:1e 
[236030.117642] ath10k_ahb a000000.wifi: removing peer, cleanup-all, deleting: peer cf844c00 vdev: 0 addr: 2c:59:8a:73:2b:6e 
[236030.128689] ath10k_ahb a000000.wifi: removing peer, cleanup-all, deleting: peer cf844000 vdev: 0 addr: c0:ee:fb:df:60:bd 
[236030.139695] ath10k_ahb a000000.wifi: removing peer, cleanup-all, deleting: peer cf845200 vdev: 0 addr: 14:30:c6:e0:63:f6 
[236030.150716] ath10k_ahb a000000.wifi: removing peer, cleanup-all, deleting: peer cf845400 vdev: 0 addr: 6c:ad:f8:b8:64:e0 
[236030.161741] ath10k_ahb a000000.wifi: removing peer, cleanup-all, deleting: peer cf844600 vdev: 0 addr: b8:ec:a3:2e:dd:74 
[236030.185966] ieee80211 phy0: Hardware restart was requested
[236030.535592] net_ratelimit: 56 callbacks suppressed
[236030.535613] ath10k_ahb a000000.wifi: failed to transmit packet, dropping: -19
[236030.539318] ath10k_ahb a000000.wifi: failed to submit frame: -19
[236030.546651] ath10k_ahb a000000.wifi: failed to push frame: -19
[236031.591863] ath10k_ahb a000000.wifi: failed to transmit packet, dropping: -19
[236031.591916] ath10k_ahb a000000.wifi: failed to submit frame: -19
[236031.598029] ath10k_ahb a000000.wifi: failed to push frame: -19
[236031.675675] ath10k_ahb a000000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[236031.676017] ath10k_ahb a000000.wifi: msdu-desc: 2500  skid: 32
[236031.722954] ath10k_ahb a000000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[236031.728000] ath10k_ahb a000000.wifi: wmi print 'free: 58060 iram: 22488 sram: 35968'
[236032.105303] ath10k_ahb a000000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[236032.378024] ath10k_ahb a000000.wifi: device successfully recovered
```

greearb, 2019-01-02:
This is same root cause as bug 54.  Please try the attached binary and see if it works better for you.
[firmware-5-full-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2721661/firmware-5-full-community.bin.gz)

fishxz, 2019-01-02:
thanks. it will take while to find out, if the new firmware works better. its the first time i encountered this crash.

fishxz, 2019-01-03:
hey,
im not able to test this firmware... if i use your firmware, the wifi crashs in a loop and when its up for a short time im not able to access any website, not even luci... i was able to access ssh and move back to old one.

greearb, 2019-01-03:
Sorry about that, I had a logic flaw in the last patch. Please try this one instead. And, please run with debug-level of 0xc0000020 and send me 'dmesg' output after the system has been running for a bit even if it doesn't crash or have obvious issues.

[firmware-5-full-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2724918/firmware-5-full-community.bin.gz)

fishxz, 2019-01-03:
hey,
would be nice to know how to run it with this debug level.

greearb, 2019-01-03:
See first part of this page:  http://www.candelatech.com/ath10k-bugs.php

You would expect to see some DBGLOG entries spam your dmesg a bit...or maybe a lot with the 0xc0000020 setting.

fishxz, 2019-01-09:
sorry for the delayed answer. this firmware still dont let me access anything. i can connect to the router, but no connection, like accessing werbinterface will work.

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.90 (buildbot@buildslave) (gcc version 7.4.0 (OpenWrt GCC 7.4.0 r8917-ae622c93b3)) #0 SMP Fri Dec 28 22:09:38 2018
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: ZyXEL NBG6617
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 65024
[    0.000000] free_area_init_node: node 0, pgdat c0a23e40, node_mem_map cfdf9000
[    0.000000]   Normal zone: 512 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65024 pages, LIFO batch:15
[    0.000000] random: get_random_bytes called from start_kernel+0x88/0x3c4 with crng_init=0
[    0.000000] percpu: Embedded 15 pages/cpu @cfdaf000 s29388 r8192 d23860 u61440
[    0.000000] pcpu-alloc: s29388 r8192 d23860 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64512
[    0.000000] Kernel command line: board=NBG6617 root=mtd:rootfs zld_ver=2.02 mtdparts=spi0.0:0x60000@0x60000(0:QSEE)ro,0x80000@0xE0000(u-boot)ro,0x10000(env)ro,0x10000(0:ART)ro,0x400000(HLOS),0x10000(dualflag),0x10000(header),0x100000(romd),0x100000(rootfs_data),0x1860000(rootfs) mtdparts= root=31:13
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Memory: 249928K/260096K available (4369K kernel code, 145K rwdata, 1200K rodata, 1024K init, 235K bss, 10168K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xd0800000 - 0xff800000   ( 752 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xd0000000   ( 256 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0208000 - 0xc0744638   (5362 kB)
[    0.000000]       .init : 0xc0900000 - 0xc0a00000   (1024 kB)
[    0.000000]       .data : 0xc0a00000 - 0xc0a24740   ( 146 kB)
[    0.000000]        .bss : 0xc0a26000 - 0xc0a60c98   ( 236 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
[    0.000008] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
[    0.000022] Switching to timer-based delay loop, resolution 20ns
[    0.000263] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
[    0.000282] pid_max: default: 32768 minimum: 301
[    0.000418] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000436] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000996] CPU: Testing write buffer coherency: ok
[    0.001688] Setting up static identity map for 0x80300000 - 0x80300060
[    0.001830] Hierarchical SRCU implementation.
[    0.002475] smp: Bringing up secondary CPUs ...
[    0.005225] smp: Brought up 1 node, 4 CPUs
[    0.005246] SMP: Total of 4 processors activated (384.00 BogoMIPS).
[    0.005254] CPU: All CPU(s) started in SVC mode.
[    0.009672] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.009821] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.009844] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.010117] pinctrl core: initialized pinctrl subsystem
[    0.010992] NET: Registered protocol family 16
[    0.011317] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.012360] cpuidle: using governor ladder
[    0.012404] cpuidle: using governor menu
[    0.029351] usbcore: registered new interface driver usbfs
[    0.029436] usbcore: registered new interface driver hub
[    0.029518] usbcore: registered new device driver usb
[    0.029565] pps_core: LinuxPPS API ver. 1 registered
[    0.029575] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.029599] PTP clock support registered
[    0.030902] clocksource: Switched to clocksource arch_sys_counter
[    0.031731] NET: Registered protocol family 2
[... 203 lines cut ...]
[   24.942576] ath10k_ahb a000000.wifi: wmi print 'free: 60212 iram: 23160 sram: 35968'
[   25.329108] ath10k_ahb a000000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[   25.330867] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   25.889139] ath10k_ahb a000000.wifi: NOTE:  Firmware DBGLOG output disabled in debug_mask: 0x10000000
[... cut ...]
[   27.629217] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   27.629481] br-lan: port 2(wlan0) entered blocking state
[   27.634772] br-lan: port 2(wlan0) entered forwarding state
[   28.034907] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   28.035171] br-lan: port 3(wlan1) entered blocking state
[   28.040341] br-lan: port 3(wlan1) entered forwarding state
[   88.263967] pppoe-wan: renamed from ppp0
[  147.818717] ath10k_ahb a000000.wifi: Invalid VHT mcs 15 peer stats
```

greearb, 2019-01-11:
Closing this bug, I think I have made progress on bug 54 and that user appears to have an easier time testing code.

closed 2019-01-11 by greearb