greearb/ath10k-ct #56: 9980 crash in scheduler (same as 54 but for 9980)

greearb, 2019-01-03:
Reported in OpenWRT forum by escalade.

Same root cause as bug 54, but for chipset 9980.

greearb, 2019-01-03:
Please try this firmware. And, please run with debug-level of 0xc0000020 and send me 'dmesg' output after the system has been running for a bit even if it doesn't crash or have obvious issues.

[firmware-5-full-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2724983/firmware-5-full-community.bin.gz)

notmyrealhandle, 2019-01-09:
I see some updates on #54 for 9984 regarding this bug but not here.  If I'd like to test for 9980 (r7500v2) is the firmware above still relevant or is there a newer version? Do you still want dmesg output (log level 0xc0000020) after an hour or two up?

FWIW, similar to other reports I got 10+ days using 10.4b-ct-9980-fH-012-81e1edd.  The router just crashed and reset (sorry no dmesg/log output to the router reboot).

greearb, 2019-01-09:
Let me try to get this fixed in bug #54, and when it seems to work there, I'll update this bug with a new firmware to test with.

greearb, 2019-01-11:
Ok, I think I made some progress on bug 54, please try this image and send me logs if you see crashes or other problems.

[firmware-5-full-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2750504/firmware-5-full-community.bin.gz)

notmyrealhandle, 2019-01-11:
I only got about an hour and half with 10.4b-ct-9980-fW-012-4651c335 before the router crashed and rebooted.  I was remote logging on a wired connection (debug level 0xc0000020) and that file is [here](https://gist.github.com/notmyrealhandle/6bb0275c77d692eb9943a2aed39578db#file-r7500v2-log-scrub-out). 

There is no indication of the crash that I can see.

If it helps, there is a dmesg and logread output I took before the crash right after booting crash [here](https://gist.github.com/notmyrealhandle/6bb0275c77d692eb9943a2aed39578db#file-logcat-20190111-135359-scrub-out).

greearb, 2019-01-14:
I do not see any obvious errors in the logs.  The log messages are almost all about ERP being disabled and about short pre-amble being enable and disabled.  Looks like the driver just sending commands to do these things, and probably it is no big deal.  Perhaps the crash was not related to the firmware, or maybe the log captures just missed the important event.

notmyrealhandle, 2019-01-14:
thank you for responding.  I'll try again with it and report back; however, I don't have a serial connection so if I keep getting unexplained crashes I'll stop testing for now.

notmyrealhandle, 2019-01-17:
a little over two days up now and no issues this time.  I did run most of that time w/o at the default debug level and intermittent remote logging.  The last 3 hours or so, I used debug level 0xc0000020 with remote logging in the off chance that doing so precipitated the afore mentioned crash - no issues tho.

Unless there is a newer version you like to have tested, I'll let this go through the weekend when traffic is heavier and see how it does.

notmyrealhandle, 2019-01-21:
6+ days up now, no issues

greearb, 2019-01-22:
Ok, I'm going to consider this one fixed.  Please open a new bug if you see more problems.

closed 2019-01-22 by greearb