greearb/ath10k-ct #57: 988x:  Slightly slower download speed than with stock firmware.

greearb, 2019-01-03:
silentcreek reports on openwrt forum:

So, I took the time to test the ath10k-ct firmware and driver against their upstream counterparts. The platform I used is qca988x-based (TP-Link Archer C7 v2).

First things first, the gap that I saw in throughput measurements before (last time I did this was more than a year ago) is gone. The performance is very similar now. Both drivers/firmwares allow for more than 400 Mbit/s throughput between the access point and a 2x2 ac wireless client (a 2014 or 2015 MacBook Air). I tested throuput using iperf with repeated measurements and averaged their results. The server was connected by wire to the same switch as the Archer C7 v2 (which runs in access point mode).

I did notice a few things though:

With the upstream firmware and driver, there's not much difference between the client being the sender or receiver of data. The results are all between about 410-430Mbit/s (with UDP transfers more towards the upper boundary and TCP more towards the lower). This is not the case with the ct driver and firmware. Here performance is slightly lower when the client is receiving data (380-400Mbit/s). This is noting I'd be worried about at all, but the difference seems to be consistent in both TCP and UDP measurements.

I did test the combinations ath10k-ct driver with upstream firmware and ath10k-upstream driver with ct firmware as well, albeit not as long. While the combination ath10k-ct driver and upstream firmware worked fine, the upstream driver didn't seem to like the ct firmware. In fact, it was horrible. Packet loss was extremely high and the throughput only 8-10 Mbit/s!

A few more notes on the testing I did: Testing was done at night, so there wouldn't be as much wifi traffic in the neighborhood. The MacBook was places about 2-2.5m away from the access point (without obstructions) and was not moved during testing. The commands used for testing were:

iperf3 -c <server-ip>   # TCP client to server
iperf3 -R -c <server-ip>   # TCP server to client
iperf3 -u -b 867M -c <server-ip>   #UDP client to server
iperf3 -R -u -b 867M -c <server-ip>   #UDP server to client
I'm not an iperf expert, but I chose 867 Mbit/s here as it is the theoretical maximum bandwidth of the 2x2 client and then looked at the actual throughput taking the packet loss into account.

I used recent builds from the master branch (ar71xx) for testing.

Pentaphon, 2020-03-04:
> The platform I used is qca988x-based (TP-Link Archer C7 v2).

Same router and I also get the same results when switching to the stock firmware.

timkgh, 2021-02-15:
Old bug report but the issues is still there, 2 years later.