greearb/ath10k-ct #58: R7800 10.4b-ct-9984-fH-012-81e1edd54 crash

schwichtgit, 2019-01-07:
Description of the problem (how to configure, how to reproduce, how often it happens).
A couple of times a day.
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.415221] ath10k_pci 0000:01:00.0: firmware crashed! (guid 839f1fc6-d10c-42af-98f5-c084cab83afd)
Jan  7 15:37:46 LEDE-10-0-8-7 kernel: [ 4884.636383] ath10k_pci 0000:01:00.0: firmware crashed! (guid 6be439db-ceb0-4fab-a6ec-8e975d1355b5)
Jan  7 17:12:08 LEDE-10-0-8-7 kernel: [10547.834224] ath10k_pci 0000:01:00.0: firmware crashed! (guid a821b9ec-c03b-42b9-bae0-f5eb9a3ddc48)

Four APs (R7800) with 802.11r setup
I remote the syslog, so I could provide the complete context.

I tried both the htt and the regular ct firmware from openwrt-trunk.

Software (OS, Firmware version, kernel, driver, etc)
OpenWrt SNAPSHOT, r8982-a54129d8aa

root@LEDE-10-0-8-7:/lib/firmware/ath10k/QCA9984/hw1.0# ls -al

> drwxr-xr-x    1 root     root           824 Jan  7 18:50 .
> drwxr-xr-x    1 root     root           224 Jan  3 21:24 ..
> lrwxrwxrwx    1 root     root            42 Jan  7 18:50 board-2.bin -> board-2.bin-10.4b-ct-9984-fH-012-81e1edd54
> -rw-r--r--    1 root     root        171916 Jan  4 23:05 board-2.bin-10.4b-ct-9984-fH-012-81e1edd54
> -rw-r--r--    1 root     root        171916 Jan  4 23:00 board-2.bin-10.4b-ct-9984-fW-012-81e1edd54
> lrwxrwxrwx    1 root     root            47 Jan  4 23:05 ct-firmware-5.bin -> ct-firmware-5.bin-0.4b-ct-9984-fH-012-81e1edd54
> -rw-r--r--    1 root     root        618612 Jan  4 23:05 ct-firmware-5.bin-0.4b-ct-9984-fH-012-81e1edd54
> -rw-r--r--    1 root     root        618892 Jan  4 23:00 firmware-5.bin-10.4b-ct-9984-fW-012-81e1edd54
> 

Hardware (NIC chipset, platform, etc)
R7800
lspci |grep -i qual
0000:00:00.0 PCI bridge: Qualcomm Device 0101
0000:01:00.0 Network controller: Qualcomm Atheros QCA9984 802.11ac Wave 2 Wireless Network Adapter
0001:00:00.0 PCI bridge: Qualcomm Device 0101
0001:01:00.0 Network controller: Qualcomm Atheros QCA9984 802.11ac Wave 2 Wireless Network Adapter

ethtool -i wlan0
driver: ath10k_pci
version: 4.14.90
firmware-version: 10.4b-ct-9984-fH-012-81e1edd54
expansion-rom-version:
bus-info: 0000:01:00.0
Logs (dmesg, maybe supplicant and/or hostap)

> Jan  7 14:13:05 LEDE-10-0-8-7 hostapd: wlan0: STA e4:a4:71:9f:57:36 IEEE 802.11: associated (aid 7)
Jan  7 14:13:05 LEDE-10-0-8-7 hostapd: wlan0: STA-OPMODE-SMPS-MODE-CHANGED e4:a4:71:9f:57:36 off
Jan  7 14:13:05 LEDE-10-0-8-7 hostapd: wlan0: AP-STA-CONNECTED e4:a4:71:9f:57:36
Jan  7 14:13:05 LEDE-10-0-8-7 hostapd: wlan0: STA e4:a4:71:9f:57:36 RADIUS: starting accounting session 643B55C2FD6D77C3
Jan  7 14:13:05 LEDE-10-0-8-7 hostapd: wlan0: STA e4:a4:71:9f:57:36 WPA: pairwise key handshake completed (RSN)
Jan  7 14:13:22 LEDE-10-0-8-7 hostapd: wlan0: AP-STA-DISCONNECTED 54:33:cb:ef:0b:01
Jan  7 14:13:22 LEDE-10-0-8-7 hostapd: wlan0: STA 54:33:cb:ef:0b:01 IEEE 802.11: disassociated
Jan  7 14:13:23 LEDE-10-0-8-7 hostapd: wlan0: STA 54:33:cb:ef:0b:01 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Jan  7 14:13:28 LEDE-10-0-8-7 hostapd: wlan0: STA 48:a1:95:b5:ec:3e IEEE 802.11: authenticated
Jan  7 14:13:28 LEDE-10-0-8-7 hostapd: wlan0: STA 48:a1:95:b5:ec:3e IEEE 802.11: associated (aid 2)
Jan  7 14:13:28 LEDE-10-0-8-7 hostapd: wlan0: AP-STA-CONNECTED 48:a1:95:b5:ec:3e
Jan  7 14:13:28 LEDE-10-0-8-7 hostapd: wlan0: STA 48:a1:95:b5:ec:3e RADIUS: starting accounting session A1DEC828E551D692
Jan  7 14:13:28 LEDE-10-0-8-7 hostapd: wlan0: STA 48:a1:95:b5:ec:3e WPA: pairwise key handshake completed (RSN)
Jan  7 14:13:50 LEDE-10-0-8-7 hostapd: wlan0: STA dc:a9:04:75:db:0d IEEE 802.11: authenticated
Jan  7 14:13:51 LEDE-10-0-8-7 hostapd: wlan0: STA dc:a9:04:75:db:0d IEEE 802.11: associated (aid 12)
Jan  7 14:13:51 LEDE-10-0-8-7 hostapd: wlan0: AP-STA-CONNECTED dc:a9:04:75:db:0d
Jan  7 14:13:51 LEDE-10-0-8-7 hostapd: wlan0: STA dc:a9:04:75:db:0d RADIUS: starting accounting session 318A607A5662D38C
Jan  7 14:13:51 LEDE-10-0-8-7 hostapd: wlan0: STA dc:a9:04:75:db:0d WPA: pairwise key handshake completed (RSN)
Jan  7 14:14:03 LEDE-10-0-8-7 hostapd: wlan0: STA 54:33:cb:ef:0b:01 IEEE 802.11: authenticated
Jan  7 14:14:03 LEDE-10-0-8-7 hostapd: wlan0: STA 54:33:cb:ef:0b:01 IEEE 802.11: associated (aid 13)
Jan  7 14:14:04 LEDE-10-0-8-7 hostapd: wlan0: AP-STA-CONNECTED 54:33:cb:ef:0b:01
Jan  7 14:14:04 LEDE-10-0-8-7 hostapd: wlan0: STA 54:33:cb:ef:0b:01 RADIUS: starting accounting session BDBF78A31CFD8656
Jan  7 14:14:04 LEDE-10-0-8-7 hostapd: wlan0: STA 54:33:cb:ef:0b:01 WPA: pairwise key handshake completed (RSN)
Jan  7 14:15:17 LEDE-10-0-8-7 hostapd: wlan0: STA 80:b0:3d:23:02:f6 IEEE 802.11: authenticated
Jan  7 14:15:17 LEDE-10-0-8-7 hostapd: wlan0: STA 80:b0:3d:23:02:f6 IEEE 802.11: associated (aid 14)
Jan  7 14:15:17 LEDE-10-0-8-7 hostapd: wlan0: AP-STA-CONNECTED 80:b0:3d:23:02:f6
Jan  7 14:15:17 LEDE-10-0-8-7 hostapd: wlan0: STA 80:b0:3d:23:02:f6 RADIUS: starting accounting session 136271CE2161DAEA
Jan  7 14:15:17 LEDE-10-0-8-7 hostapd: wlan0: STA 80:b0:3d:23:02:f6 WPA: pairwise key handshake completed (RSN)
[... 29 lines cut ...]
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.646542] ath10k_pci 0000:01:00.0: [06]: 0x0004b800  30  30  30  30
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.653138] ath10k_pci 0000:01:00.0: [07]: 0x0004bc00   1   1   1   1
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.659563] ath10k_pci 0000:01:00.0: [08]: 0x0004c000   0   0 127   0
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.665986] ath10k_pci 0000:01:00.0: [09]: 0x0004c400   0   0   0   0
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.672411] ath10k_pci 0000:01:00.0: [10]: 0x0004c800   0   0   0   0
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.678836] ath10k_pci 0000:01:00.0: [11]: 0x0004cc00   0   0   0   0
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.687288] ath10k_pci 0000:01:00.0: debug log header, dbuf: 0x4237f0  dropped: 0
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.692701] ath10k_pci 0000:01:00.0: [0] next: 0x423808 buf: 0x419010 sz: 1500 len: 44 count: 2 free: 0
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.700167] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.708345] ath10k: [0000]: 02235695 08004C01 91104110 000000FF 02235695 17FC0001 009A5186 000015B3
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.714595] ath10k: [0008]: 000015B3 0040657C 91104569
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.723528] ath10k_pci 0000:01:00.0: ATH10K_END
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.729760] ath10k_pci 0000:01:00.0: [1] next: 0x4237f0 buf: 0x419600 sz: 1500 len: 0 count: 0 free: 0
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.806840] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dd783200 vdev: 0 addr: 80:b0:3d:23:02:f6
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.806891] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dd783400 vdev: 0 addr: 54:33:cb:ef:0b:01
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.816877] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dd782200 vdev: 0 addr: dc:a9:04:75:db:0d
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.827841] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dd782c00 vdev: 0 addr: 48:a1:95:b5:ec:3e
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.838776] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dd782800 vdev: 0 addr: e4:a4:71:9f:57:36
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.849706] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dd782400 vdev: 0 addr: 3c:2e:f9:d5:1e:94
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.860644] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dd783600 vdev: 0 addr: dc:a9:04:75:b1:e5
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.871574] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dac8bc00 vdev: 0 addr: 6c:96:cf:d8:a5:83
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.882508] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dac8a200 vdev: 0 addr: 78:4f:43:a7:87:c3
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.893441] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dc73f000 vdev: 0 addr: dc:ef:ca:64:46:39
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.904377] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dc374e00 vdev: 0 addr: 60:f8:1d:c1:0f:28
Jan  7 14:15:56 LEDE-10-0-8-7 kernel: [35068.915311] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dc374800 vdev: 0 addr: 7c:2e:bd:56:75:b0

greearb, 2019-01-09:
Thanks for the report.  It looks like rate-ctrl gets confused and picks zero possible rates and then asserts. I have a potential fix, and more logging in case it is not fixed in the build below.  This also contains the fixes I have been making in order to try to fix bug 54...that effort has been full of regressions but I once again have hope I have it fixed, so let me know how this one works.

[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2742820/firmware-5-full-htt-mgt-community.bin.gz)

schwichtgit, 2019-01-11:
not much luck.

while the firmware seems to load
```
ethtool -i wlan0
driver: ath10k_pci
version: 4.14.90
firmware-version: 10.4b-ct-9984-fH-012-4cce451d4
expansion-rom-version:
bus-info: 0000:01:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```

I never get ipv4 traffic to work.
Also 
```
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.175703] ath10k_pci 0000:01:00.0: firmware crashed! (guid 522964e1-5092-4639-94c7-dd4f3cbb6c85)
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.175741] ath10k_pci 0000:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.183633] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.195517] ath10k_pci 0000:01:00.0: firmware ver 10.4b-ct-9984-fH-012-4cce451d4 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT crc32 eaaf4b73
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.205888] ath10k_pci 0000:01:00.0: board_file api 2 bmi_id 0:1 crc32 cf58c3bc
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.227638] ath10k_pci 0000:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.236620] ath10k_pci 0000:01:00.0: firmware register dump:
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.244728] ath10k_pci 0000:01:00.0: [00]: 0x0000000A 0x000015B3 0x0098C337 0x00975B31
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.250624] ath10k_pci 0000:01:00.0: [04]: 0x0098C337 0x00060130 0x00000019 0x0046490C
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.258347] ath10k_pci 0000:01:00.0: [08]: 0x00000000 0x0045C63C 0x00020921 0x00000000
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.266192] ath10k_pci 0000:01:00.0: [12]: 0x00000009 0x00000003 0x0098C2D8 0x0098C357
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.274149] ath10k_pci 0000:01:00.0: [16]: 0x0098C337 0x0A003E1B 0x009606CA 0x00000000
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.282049] ath10k_pci 0000:01:00.0: [20]: 0x4098C337 0x0040652C 0x004625FC 0x004552C4
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.289953] ath10k_pci 0000:01:00.0: [24]: 0x80983B2F 0x0040658C 0x0000007B 0xC098C337
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.297857] ath10k_pci 0000:01:00.0: [28]: 0x809C8BA6 0x004065AC 0x0042E4B0 0x004552C4
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.305687] ath10k_pci 0000:01:00.0: [32]: 0x80983B58 0x004065CC 0x004552C4 0x0042F620
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.313633] ath10k_pci 0000:01:00.0: [36]: 0x809C8DC9 0x0040666C 0x004552C4 0x00415408
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.321538] ath10k_pci 0000:01:00.0: [40]: 0x80980AA6 0x0040669C 0x0045C82C 0x0042D0A4
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.329440] ath10k_pci 0000:01:00.0: [44]: 0x809C62EB 0x004066DC 0x0045C82C 0x0042D0A4
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.337345] ath10k_pci 0000:01:00.0: [48]: 0x80980B56 0x0040670C 0x0045C82C 0x00009023
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.345181] ath10k_pci 0000:01:00.0: [52]: 0x809C88A8 0x004068AC 0x00000000 0x00009023
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.353151] ath10k_pci 0000:01:00.0: [56]: 0x809B3E56 0x004068DC 0x000EC15C 0x00429F74
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.361057] ath10k_pci 0000:01:00.0: Copy Engine register dump:
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.368938] ath10k_pci 0000:01:00.0: [00]: 0x0004a000  10  10   3   3
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.374704] ath10k_pci 0000:01:00.0: [01]: 0x0004a400   0   0 134 135
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.381365] ath10k_pci 0000:01:00.0: [02]: 0x0004a800  41  41  40  41
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.387770] ath10k_pci 0000:01:00.0: [03]: 0x0004ac00   2   2   3   2
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.394143] ath10k_pci 0000:01:00.0: [04]: 0x0004b000 6308 6308 124  84
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.400644] ath10k_pci 0000:01:00.0: [05]: 0x0004b400   9   9  69  73
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.407047] ath10k_pci 0000:01:00.0: [06]: 0x0004b800  16  16  16  16
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.413586] ath10k_pci 0000:01:00.0: [07]: 0x0004bc00   1   1   1   1
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.420078] ath10k_pci 0000:01:00.0: [08]: 0x0004c000   0   0 127   0
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.426435] ath10k_pci 0000:01:00.0: [09]: 0x0004c400   1   1   1   1
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.432915] ath10k_pci 0000:01:00.0: [10]: 0x0004c800   0   0   0   0
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.439346] ath10k_pci 0000:01:00.0: [11]: 0x0004cc00   0   0   0   0
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.447722] ath10k_pci 0000:01:00.0: debug log header, dbuf: 0x422fa8  dropped: 0
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.453140] ath10k_pci 0000:01:00.0: [0] next: 0x422f90 buf: 0x4195c0 sz: 1500 len: 212 count: 8 free: 0
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.460662] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[... cut ...]
Jan 11 17:54:40 SchwichtenburgWrt kernel: [ 1285.537163] ath10k_pci 0000:01:00.0: [1] next: 0x422fa8 buf: 0x418fd0 sz: 1500 len: 0 count: 0 free: 0
Jan 11 17:54:41 SchwichtenburgWrt kernel: [ 1285.579053] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 1, skipped old beacon
Jan 11 17:54:41 SchwichtenburgWrt kernel: [ 1285.579136] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Jan 11 17:54:41 SchwichtenburgWrt kernel: [ 1285.585225] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 1, skipped old beacon

```

greearb, 2019-01-12:
I think you hit regressions in the work I did for bug 54.  I think that is at least mostly resolved now...please try this binary.

[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2752091/firmware-5-full-htt-mgt-community.bin.gz)

greearb, 2019-01-14:
Here is an updated image.  The user for bug 54 hit the same bug, and I added some additional logging to try to better understand the issue.
[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2756664/firmware-5-full-htt-mgt-community.bin.gz)

schwichtgit, 2019-01-14:
Thanks @greearb .

I build 
```
OpenWrt SNAPSHOT, r9059-6e104c63d6
```
and applied 
```
ethtool -i wlan0
driver: ath10k_pci
version: 4.14.93
firmware-version: 10.4b-ct-9984-fH-012-25e4434c1
expansion-rom-version:
bus-info: 0000:01:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```

```
[   23.502524] ath10k_pci 0001:01:00.0: firmware ver 10.4b-ct-9984-fH-012-25e4434c1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-al
l-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT crc32 a1004684
[   25.848483] ath10k_pci 0001:01:00.0: board_file api 2 bmi_id 0:2 crc32 cf58c3bc
[   31.718432] ath10k_pci 0001:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   31.718459] ath10k_pci 0001:01:00.0: msdu-desc: 2500  skid: 32
[   31.802462] ath10k_pci 0001:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   31.803315] ath10k_pci 0001:01:00.0: wmi print 'free: 87700 iram: 27412 sram: 18240'
[   32.132762] ath10k_pci 0001:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[   32.235899] ath: EEPROM regdomain: 0x0
[   32.235913] ath: EEPROM indicates default country code should be used
[   32.235924] ath: doing EEPROM country->regdmn map search
[   32.235943] ath: country maps to regdmn code: 0x3a
[   32.235957] ath: Country alpha2 being used: US
[   32.235967] ath: Regpair used: 0x3a
[   32.245555] kmodloader: done loading kernel modules from /etc/modules.d/*
[   36.293767] Generic PHY fixed-0:00: attached PHY driver [Generic PHY] (mii_bus:phy_addr=fixed-0:00, irq=POLL)
```

```
[  169.763270] ath10k_pci 0001:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[  169.763298] ath10k_pci 0001:01:00.0: msdu-desc: 2500  skid: 32
[  169.847265] ath10k_pci 0001:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[  169.848121] ath10k_pci 0001:01:00.0: wmi print 'free: 87700 iram: 27412 sram: 18240'
[  170.277861] ath10k_pci 0001:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[  170.278137] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[  170.761976] ath10k_pci 0001:01:00.0: NOTE:  Firmware DBGLOG output disabled in debug_mask: 0x10000000
[  170.782560] ------------[ cut here ]------------
[  170.782661] WARNING: CPU: 0 PID: 1529 at /home/schwicht/projects/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/ath10k-ct-2018-12-20-118e16da/ath10k-4.19/h
tt_rx.c:903 ath10k_htt_t2h_msg_handler+0xf2c/0x265c [ath10k_core]
[  170.786364] Modules linked in: pppoe ppp_async ath10k_pci ath10k_core ath pppox ppp_generic nf_conntrack_ipv6 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp
 xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4
 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables c
rc_ccitt compat ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip6_vti ip_vti sit ipcomp6 xfrm6_tunnel xfrm6_mod
e_tunnel xfrm6_mode_transport xfrm6_mode_beet esp6 ah6 ipcomp xfrm4_tunnel
[  170.857006]  xfrm4_mode_tunnel xfrm4_mode_transport xfrm4_mode_beet esp4 ah4 ip6_tunnel tunnel6 tunnel4 ip_tunnel tun af_key xfrm_user xfrm_ipcomp xfrm_algo sha1_generic md5 echainiv
authenc leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_of_simple ohci_platform ohci_hcd phy_qcom_dwc3 ahci ehci_platform sd_mod ahci_platform libahci_platform libahci libata scsi_mo
d ehci_hcd gpio_button_hotplug
[  170.892053] CPU: 0 PID: 1529 Comm: hostapd Not tainted 4.14.93 #0
[  170.914195] Hardware name: Generic DT based system
[  170.920468] [<c030f58c>] (unwind_backtrace) from [<c030b7a4>] (show_stack+0x14/0x20)
[  170.925142] [<c030b7a4>] (show_stack) from [<c07b2cb8>] (dump_stack+0x88/0x9c)
[  170.933045] [<c07b2cb8>] (dump_stack) from [<c0318ecc>] (__warn+0xf0/0x11c)
[  170.940068] [<c0318ecc>] (__warn) from [<c0318fb8>] (warn_slowpath_null+0x20/0x28)
[  170.946967] [<c0318fb8>] (warn_slowpath_null) from [<bf4b303c>] (ath10k_htt_t2h_msg_handler+0xf2c/0x265c [ath10k_core])
[  170.954686] [<bf4b303c>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf4b322c>] (ath10k_htt_t2h_msg_handler+0x111c/0x265c [ath10k_core])
[  170.965314] [<bf4b322c>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf4b45d0>] (ath10k_htt_t2h_msg_handler+0x24c0/0x265c [ath10k_core])
[  170.978160] [<bf4b45d0>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf4b4e70>] (ath10k_htt_txrx_compl_task+0x6f8/0xbb8 [ath10k_core])
[  170.990899] [<bf4b4e70>] (ath10k_htt_txrx_compl_task [ath10k_core]) from [<bf506758>] (ath10k_pci_napi_poll+0x7c/0x11c [ath10k_pci])
[... 37 lines cut ...]
[  183.363464] ath10k_pci 0000:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[  183.364292] ath10k_pci 0000:01:00.0: wmi print 'free: 87700 iram: 27412 sram: 18240'
[  183.732638] ath10k_pci 0000:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[... cut ...]
[  190.271708] br-lan: port 3(wlan1) entered disabled state
[  190.276726] device wlan1 entered promiscuous mode
[  199.313320] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[  199.313667] br-lan: port 3(wlan1) entered blocking state
[  199.318771] br-lan: port 3(wlan1) entered forwarding state
[  274.469179] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  274.469322] br-lan: port 2(wlan0) entered blocking state
[  274.474687] br-lan: port 2(wlan0) entered forwarding state
```

no crash visible yet

schwichtgit, 2019-01-15:
[dmesg.log](https://github.com/greearb/ath10k-ct/files/2757670/dmesg.log)
sadly in my home config with 802.11r I get crashes again.
```
ethtool -i wlan0
driver: ath10k_pci
version: 4.14.93
firmware-version: 10.4b-ct-9984-fH-012-25e4434c1
expansion-rom-version:
bus-info: 0000:01:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```

schwichtgit, 2019-01-15:
also
```
[  718.717573] ath10k_pci 0000:01:00.0: firmware crashed! (guid cc007e3c-6d88-4f7e-a945-8f6d03cf8bc8)
[  718.717679] ath10k_pci 0000:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[  718.725480] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[  718.740379] ath10k_pci 0000:01:00.0: firmware ver 10.4b-ct-9984-fH-012-25e4434c1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT crc32 a1004684
[  718.748546] ath10k_pci 0000:01:00.0: board_file api 2 bmi_id 0:1 crc32 cf58c3bc
[  718.769481] ath10k_pci 0000:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[  718.778574] ath10k_pci 0000:01:00.0: firmware register dump:
[  718.786570] ath10k_pci 0000:01:00.0: [00]: 0x0000000A 0x000015B3 0x009A5240 0x00975B31
[  718.792473] ath10k_pci 0000:01:00.0: [04]: 0x009A5240 0x00060930 0x00000005 0x00000001
[  718.800198] ath10k_pci 0000:01:00.0: [08]: 0x0042D04C 0x00000000 0x000000FF 0x00406740
[  718.808098] ath10k_pci 0000:01:00.0: [12]: 0x00000009 0x00000000 0x009A7C2C 0x009A7C34
[  718.815996] ath10k_pci 0000:01:00.0: [16]: 0x00973AB0 0x009606CA 0x00960E65 0x00000000
[  718.823896] ath10k_pci 0000:01:00.0: [20]: 0x409A5240 0x0040668C 0xFFFFFFFF 0x00412CE0
[  718.831796] ath10k_pci 0000:01:00.0: [24]: 0x809A7BB0 0x004066EC 0x00000007 0xC09A5240
[  718.839696] ath10k_pci 0000:01:00.0: [28]: 0x809A8D8A 0x0040672C 0x0042D04C 0x00406850
[  718.847594] ath10k_pci 0000:01:00.0: [32]: 0x809A8DBA 0x0040684C 0x0042D04C 0x0042E700
[  718.855494] ath10k_pci 0000:01:00.0: [36]: 0x80985E47 0x0040689C 0x00000000 0x0045A73C
[  718.863392] ath10k_pci 0000:01:00.0: [40]: 0x80994B1B 0x004068BC 0x0042E4B0 0x0045A73C
[  718.871292] ath10k_pci 0000:01:00.0: [44]: 0x8098FC9C 0x004068DC 0x0042E4B0 0x00000000
[  718.879191] ath10k_pci 0000:01:00.0: [48]: 0x80963AD3 0x00406A7C 0x0042E4B0 0x0098FC94
[  718.887090] ath10k_pci 0000:01:00.0: [52]: 0x80960E80 0x00406A9C 0x0000001F 0x00400000
[  718.894991] ath10k_pci 0000:01:00.0: [56]: 0x80960E51 0x00406ACC 0x00400000 0x00000000
[  718.902889] ath10k_pci 0000:01:00.0: Copy Engine register dump:
[  718.910792] ath10k_pci 0000:01:00.0: [00]: 0x0004a000  10  10   3   3
[  718.916607] ath10k_pci 0000:01:00.0: [01]: 0x0004a400  13  13 467 468
[  718.923205] ath10k_pci 0000:01:00.0: [02]: 0x0004a800   5   5  68  69
[  718.929628] ath10k_pci 0000:01:00.0: [03]: 0x0004ac00  14  14  16  14
[  718.936050] ath10k_pci 0000:01:00.0: [04]: 0x0004b000 5716 5716  83  44
[  718.942474] ath10k_pci 0000:01:00.0: [05]: 0x0004b400  25  25 280 281
[  718.948899] ath10k_pci 0000:01:00.0: [06]: 0x0004b800  31  31  31  31
[  718.955496] ath10k_pci 0000:01:00.0: [07]: 0x0004bc00   1   1   1   1
[  718.961987] ath10k_pci 0000:01:00.0: [08]: 0x0004c000   0   0 127   0
[  718.968342] ath10k_pci 0000:01:00.0: [09]: 0x0004c400   0   0   0   0
[  718.974766] ath10k_pci 0000:01:00.0: [10]: 0x0004c800   0   0   0   0
[  718.981273] ath10k_pci 0000:01:00.0: [11]: 0x0004cc00   0   0   0   0
[  718.989641] ath10k_pci 0000:01:00.0: debug log header, dbuf: 0x422f90  dropped: 0
[  718.995056] ath10k_pci 0000:01:00.0: [0] next: 0x422fa8 buf: 0x418fd0 sz: 1500 len: 132 count: 5 free: 0
[  719.002527] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[  719.011047] ath10k: [0000]: 000A8AA9 14004C07 91108003 00000000 00000FF0 00000000 000000FF 000A8AA9
[  719.017036] ath10k: [0008]: 14004C07 91108004 00000000 00000000 00000000 00000000 000A8AA9 10004C07
[  719.025978] ath10k: [0016]: 91108005 00000007 00000000 00000001 000A8AA9 10004C01 91104110 000000FF
[  719.035005] ath10k: [0024]: 00000000 00000004 000A8AA9 17FC0001 009A5240 000015B3 000015B3 0040657C
[  719.044032] ath10k: [0032]: 91104569
[  719.053049] ath10k_pci 0000:01:00.0: ATH10K_END
[  719.057894] ath10k_pci 0000:01:00.0: [1] next: 0x422f90 buf: 0x4195c0 sz: 1500 len: 0 count: 0 free: 0
[  719.100407] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 1, skipped old beacon
[  719.100555] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  719.106723] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 1, skipped old beacon
[  719.113930] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  719.194353] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer daa3b400 vdev: 0 addr: f0:18:98:2a:00:ee
[  719.194384] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dbc1d600 vdev: 0 addr: 54:33:cb:bc:d1:ea
[  719.204376] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dbc90600 vdev: 0 addr: 54:60:09:d5:72:f2
[  719.215280] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dbc91800 vdev: 0 addr: 54:60:09:d5:78:f0
[  719.226212] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dcaa4200 vdev: 0 addr: ac:bc:32:80:37:cd
[  719.237148] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dc401000 vdev: 0 addr: 18:b4:30:c5:a3:40
[  719.248080] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dbf1dc00 vdev: 0 addr: 64:16:66:05:b0:58
[  719.259045] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dc546000 vdev: 0 addr: 64:16:66:04:76:88
[  719.269981] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dbf82800 vdev: 0 addr: 78:d2:94:a3:96:9e
[... cut ...]
[  726.472712] ath10k_pci 0000:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[  726.492645] ath10k_pci 0000:01:00.0: failed to submit frame: -19
[  726.492691] ath10k_pci 0000:01:00.0: failed to push frame: -19
[  726.717843] ath10k_pci 0000:01:00.0: failed to submit frame: -19
[  726.717931] ath10k_pci 0000:01:00.0: failed to push frame: -19
[  726.742358] ath10k_pci 0000:01:00.0: failed to submit frame: -19
[  726.742400] ath10k_pci 0000:01:00.0: failed to push frame: -19
[  726.800964] ath10k_pci 0000:01:00.0: device successfully recovered
```

schwichtgit, 2019-01-15:
and
```
6134.390396] ath10k_pci 0000:01:00.0: firmware crashed! (guid e47c4e8e-f7b1-4ca6-afda-45b37a429c41)
[ 6134.390445] ath10k_pci 0000:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[ 6134.398262] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[ 6134.410145] ath10k_pci 0000:01:00.0: firmware ver 10.4b-ct-9984-fH-012-25e4434c1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT crc32 a1004684
[ 6134.420533] ath10k_pci 0000:01:00.0: board_file api 2 bmi_id 0:1 crc32 cf58c3bc
[ 6134.442261] ath10k_pci 0000:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[ 6134.451320] ath10k_pci 0000:01:00.0: firmware register dump:
[ 6134.459357] ath10k_pci 0000:01:00.0: [00]: 0x0000000A 0x000015B3 0x009A5240 0x00975B31
[ 6134.465271] ath10k_pci 0000:01:00.0: [04]: 0x009A5240 0x00060930 0x00000005 0x00000001
[ 6134.472988] ath10k_pci 0000:01:00.0: [08]: 0x0042D0A4 0x00000000 0x000000FF 0x00406740
[ 6134.480888] ath10k_pci 0000:01:00.0: [12]: 0x00000009 0x00000000 0x009A7C2C 0x009A7C34
[ 6134.488786] ath10k_pci 0000:01:00.0: [16]: 0x00973AB0 0x00960E51 0x009606CA 0x00000000
[ 6134.496686] ath10k_pci 0000:01:00.0: [20]: 0x409A5240 0x0040668C 0xFFFFFFFF 0x00412CE0
[ 6134.504585] ath10k_pci 0000:01:00.0: [24]: 0x809A7BB0 0x004066EC 0x00000007 0xC09A5240
[ 6134.512483] ath10k_pci 0000:01:00.0: [28]: 0x809A8D8A 0x0040672C 0x0042D0A4 0x00406850
[ 6134.520383] ath10k_pci 0000:01:00.0: [32]: 0x809A8DBA 0x0040684C 0x0042D0A4 0x0042E700
[ 6134.528284] ath10k_pci 0000:01:00.0: [36]: 0x80985E47 0x0040689C 0x00000000 0x0045A5EC
[ 6134.536184] ath10k_pci 0000:01:00.0: [40]: 0x80994B1B 0x004068BC 0x0042E4B0 0x0045A5EC
[ 6134.544081] ath10k_pci 0000:01:00.0: [44]: 0x8098FC9C 0x004068DC 0x0042E4B0 0x00000000
[ 6134.551980] ath10k_pci 0000:01:00.0: [48]: 0x80963AD3 0x00406A7C 0x0042E4B0 0x0098FC94
[ 6134.559879] ath10k_pci 0000:01:00.0: [52]: 0x80960E80 0x00406A9C 0x0000001F 0x00400000
[ 6134.567779] ath10k_pci 0000:01:00.0: [56]: 0x80960E51 0x00406ACC 0x00400000 0x00000000
[ 6134.575680] ath10k_pci 0000:01:00.0: Copy Engine register dump:
[ 6134.583581] ath10k_pci 0000:01:00.0: [00]: 0x0004a000  10  10   3   3
[ 6134.589397] ath10k_pci 0000:01:00.0: [01]: 0x0004a400  12  12 274 275
[ 6134.595992] ath10k_pci 0000:01:00.0: [02]: 0x0004a800  48  48 111 112
[ 6134.602416] ath10k_pci 0000:01:00.0: [03]: 0x0004ac00  16  16  18  16
[ 6134.608848] ath10k_pci 0000:01:00.0: [04]: 0x0004b000 7438 7438 177 137
[ 6134.615264] ath10k_pci 0000:01:00.0: [05]: 0x0004b400   5   5  68  69
[ 6134.621687] ath10k_pci 0000:01:00.0: [06]: 0x0004b800  17  17  17  17
[ 6134.628284] ath10k_pci 0000:01:00.0: [07]: 0x0004bc00   1   1   1   1
[ 6134.634707] ath10k_pci 0000:01:00.0: [08]: 0x0004c000   0   0 127   0
[ 6134.641132] ath10k_pci 0000:01:00.0: [09]: 0x0004c400   0   0   0   0
[ 6134.647555] ath10k_pci 0000:01:00.0: [10]: 0x0004c800   0   0   0   0
[ 6134.653978] ath10k_pci 0000:01:00.0: [11]: 0x0004cc00   0   0   0   0
[ 6134.662417] ath10k_pci 0000:01:00.0: debug log header, dbuf: 0x422fa8  dropped: 0
[ 6134.667836] ath10k_pci 0000:01:00.0: [0] next: 0x422f90 buf: 0x4195c0 sz: 1500 len: 132 count: 5 free: 0
[ 6134.675304] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[ 6134.683837] ath10k: [0000]: 00069B3A 14004C07 91108003 00000000 00000FF0 00000000 000000FF 00069B3A
[ 6134.689830] ath10k: [0008]: 14004C07 91108004 00000000 00000000 00000000 00000000 00069B3A 10004C07
[ 6134.698770] ath10k: [0016]: 91108005 00000007 00000000 00000001 00069B3A 10004C01 91104110 000000FF
[ 6134.707798] ath10k: [0024]: 00000000 00000004 00069B3A 17FC0001 009A5240 000015B3 000015B3 0040657C
[ 6134.716825] ath10k: [0032]: 91104569
[ 6134.725845] ath10k_pci 0000:01:00.0: ATH10K_END
[ 6134.730677] ath10k_pci 0000:01:00.0: [1] next: 0x422fa8 buf: 0x418fd0 sz: 1500 len: 0 count: 0 free: 0
[ 6134.772624] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 1, skipped old beacon
[ 6134.772656] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[ 6134.778781] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 1, skipped old beacon
[ 6134.786121] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[ 6134.832596] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer daaa5200 vdev: 0 addr: f0:18:98:2a:00:ee
[ 6134.832628] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer daa4e000 vdev: 0 addr: 54:60:09:d5:78:f0
[ 6134.842604] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer db45be00 vdev: 0 addr: 98:b8:e3:9c:75:96
[ 6134.853535] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer daaee200 vdev: 0 addr: 54:60:09:d5:72:f2
[ 6134.864487] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dc990000 vdev: 0 addr: 64:16:66:05:b0:58
[ 6134.875411] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dc991e00 vdev: 0 addr: 18:b4:30:c5:a3:40
[ 6134.886340] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer daa3a200 vdev: 0 addr: 54:33:cb:bc:d1:ea
[ 6134.897269] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer daa3b000 vdev: 0 addr: 64:16:66:04:76:88
[ 6134.902656] net_ratelimit: 32 callbacks suppressed
[ 6134.902675] ath10k_pci 0000:01:00.0: failed to transmit packet, dropping: -19
[... cut ...]
[ 6141.386548] ath10k_pci 0000:01:00.0: failed to push frame: -19
[ 6141.492966] ath10k_pci 0000:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[ 6141.515263] ath10k_pci 0000:01:00.0: failed to submit frame: -19
[ 6141.515315] ath10k_pci 0000:01:00.0: failed to push frame: -19
[ 6141.723020] ath10k_pci 0000:01:00.0: failed to submit frame: -19
[ 6141.723104] ath10k_pci 0000:01:00.0: failed to push frame: -19
[ 6141.819340] ath10k_pci 0000:01:00.0: device successfully recovered
```

greearb, 2019-01-16:
Ok, you can reproduce the problem at least!  The problem station is a 1x1 20Mhz HT device from what I can tell.  That should work, of course.  Here is an image with more debugging to try to track down the crash.
[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2765523/firmware-5-full-htt-mgt-community.bin.gz)

schwichtgit, 2019-01-17:
{code}
driver: ath10k_pci
version: 4.14.93
firmware-version: 10.4b-ct-9984-fH-012-587105aba
expansion-rom-version:
bus-info: 0000:01:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
{code}

schwichtgit, 2019-01-17:
@greearb do you care for a specific ath10k debug log setting ...
I have two AP using the latest 10.4b-ct-9984-fH-012-587105aba

greearb, 2019-01-17:
For this case, I think default debug log setting is fine.

shelterx, 2019-01-17:
I run this aswell now, no crash so far. :(

schwichtgit, 2019-01-18:
```
driver: ath10k_pci
version: 4.14.93
firmware-version: 10.4b-ct-9984-fH-012-587105aba
```
not using 802.11r seems to be stable up2now.
7 STA on wlan0 (802.11nac)
7 STA on wlan1 (802.11bgn)
uptime 18:42, no crashes yet.

may re-enable 802.11r soon

schwichtgit, 2019-01-18:
I decided to apply the firmware on one of the 4 AP configured for 802.11r.
so far so good it seems.
6 STA on wlan0 (802.11nac)
0 STA on wlan1 (802.11bgn)

shelterx, 2019-01-18:
So it doesn't crash for you either now? None of the debug builds crash for me, I haven't changed anything else than the images.

shelterx, 2019-01-18:
I spoke too soon, here's a crash with the image above.
[crash-2019-01-18.txt](https://github.com/greearb/ath10k-ct/files/2774164/crash-2019-01-18.txt)

schwichtgit, 2019-01-18:
I will update this once I see issues. up2now both scenarios are working (moderate usage).
If I put more active STA on the 802.11r setup (by disabling wifi on some other APs) I may get errors
no crash in log up2now about 8 STA/radio each

greearb, 2019-01-18:
That debugging is the driver twiddling vdev params for one reason or another:

Fri Jan 18 20:28:32 2019 kern.debug kernel: [57425.457556] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
Decoding dbglog buffer, len: 48
  ath10k-fw: ts: 484067 args: 4       WAL(22) vid:   0                MISC(129)  (0x3112) wal-vdev-set-param:  vdev: 0x42f620  type: 10(ERP_PROTECTION)  val: 0x0
  ath10k-fw: ts: 484067 args: 4       WAL(22) vid:   0                MISC(129)  (0x3112) wal-vdev-set-param:  vdev: 0x42f620  type: 14(SHORT_PREAMBLE)  val: 0x0

greearb, 2019-01-18:
shelterx, you hit one of the original problems in bug 54.  Here is an image with a very specific attempt at work-around for this.  Hopefully no more egregious regressions this time!

[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2774713/firmware-5-full-htt-mgt-community.bin.gz)

shelterx, 2019-01-19:
> shelterx, you hit one of the original problems in bug 54. 

Ok, I went here because I thought the issue i created was fixed and the bug I last  had was related to this issue. Anyway, let's see how things go with this image.

schwichtgit, 2019-01-19:
after a period of stability ... I added 
```
 option ieee80211v '1'
 option bss_transition '1'
```
on 
```
driver: ath10k_pci
version: 4.14.93
firmware-version: 10.4b-ct-9984-fH-012-587105aba
```

and got
```
166390.877212] ath10k_pci 0000:01:00.0: firmware crashed! (guid be9fd3d9-40c9-436e-a2e3-6903235c784f)
[166390.877289] ath10k_pci 0000:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[166390.885220] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[166390.900149] ath10k_pci 0000:01:00.0: firmware ver 10.4b-ct-9984-fH-012-587105aba api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT crc32 738f84c7
[166390.908501] ath10k_pci 0000:01:00.0: board_file api 2 bmi_id 0:1 crc32 cf58c3bc
[166390.929500] ath10k_pci 0000:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[166390.938943] ath10k_pci 0000:01:00.0: firmware register dump:
[166390.947065] ath10k_pci 0000:01:00.0: [00]: 0x0000000A 0x000015B3 0x009A5240 0x00975B31
[166390.952714] ath10k_pci 0000:01:00.0: [04]: 0x009A5240 0x00060930 0x00000005 0x00000001
[166390.960429] ath10k_pci 0000:01:00.0: [08]: 0x0042D0FC 0x00000000 0x000000FF 0x00406740
[166390.968504] ath10k_pci 0000:01:00.0: [12]: 0x00000009 0x00000000 0x009A7C34 0x009A7C3D
[166390.976486] ath10k_pci 0000:01:00.0: [16]: 0x00973AB0 0x00963BBB 0x009606CA 0x00000000
[166390.984482] ath10k_pci 0000:01:00.0: [20]: 0x409A5240 0x0040667C 0xFFFFFFFF 0x00412CE0
[166390.992448] ath10k_pci 0000:01:00.0: [24]: 0x809A7BB5 0x004066DC 0x00000007 0xC09A5240
[166391.000358] ath10k_pci 0000:01:00.0: [28]: 0x809A8DCE 0x0040671C 0x0042D0FC 0x00406850
[166391.008443] ath10k_pci 0000:01:00.0: [32]: 0x809A8DFE 0x0040684C 0x0042D0FC 0x0042E700
[166391.016409] ath10k_pci 0000:01:00.0: [36]: 0x80985E47 0x0040689C 0x00000000 0x0045A42C
[166391.024407] ath10k_pci 0000:01:00.0: [40]: 0x80994B1B 0x004068BC 0x0042E4B0 0x0045A42C
[166391.032373] ath10k_pci 0000:01:00.0: [44]: 0x8098FC9C 0x004068DC 0x0042E4B0 0x00000000
[166391.040289] ath10k_pci 0000:01:00.0: [48]: 0x80963AD3 0x00406A7C 0x0042E4B0 0x0098FC94
[166391.048368] ath10k_pci 0000:01:00.0: [52]: 0x80960E80 0x00406A9C 0x0000001F 0x00400000
[166391.056334] ath10k_pci 0000:01:00.0: [56]: 0x80960E51 0x00406ACC 0x00400000 0x00000000
[166391.064331] ath10k_pci 0000:01:00.0: Copy Engine register dump:
[166391.072307] ath10k_pci 0000:01:00.0: [00]: 0x0004a000  10  10   3   3
[166391.078488] ath10k_pci 0000:01:00.0: [01]: 0x0004a400  23  23 285 286
[166391.084892] ath10k_pci 0000:01:00.0: [02]: 0x0004a800  38  38 101 102
[166391.091422] ath10k_pci 0000:01:00.0: [03]: 0x0004ac00  25  25  27  25
[166391.097845] ath10k_pci 0000:01:00.0: [04]: 0x0004b000 5222 5222  93  53
[166391.104449] ath10k_pci 0000:01:00.0: [05]: 0x0004b400  23  23 470 471
[166391.111289] ath10k_pci 0000:01:00.0: [06]: 0x0004b800  18  18  18  18
[166391.117548] ath10k_pci 0000:01:00.0: [07]: 0x0004bc00   1   1   1   1
[166391.124130] ath10k_pci 0000:01:00.0: [08]: 0x0004c000   0   0 127   0
[166391.130570] ath10k_pci 0000:01:00.0: [09]: 0x0004c400   1   1   1   1
[166391.137157] ath10k_pci 0000:01:00.0: [10]: 0x0004c800   0   0   0   0
[166391.143655] ath10k_pci 0000:01:00.0: [11]: 0x0004cc00   0   0   0   0
[166391.152140] ath10k_pci 0000:01:00.0: debug log header, dbuf: 0x422f90  dropped: 0
[166391.157634] ath10k_pci 0000:01:00.0: [0] next: 0x422fa8 buf: 0x418fd0 sz: 1500 len: 220 count: 8 free: 0
[166391.165261] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[166391.173892] ath10k: [0000]: 0007A167 17FC4C01 00008051 000000FF 00000003 000000FF 004551D8 0007A167
[166391.179875] ath10k: [0008]: 17FC4C01 00008051 000000FF 00000004 000000FF 004551D8 0007A167 17FC4C01
[166391.189321] ath10k: [0016]: 00008051 000000FF 00000005 000000FF 004551D8 0007A167 14044C07 91108003
[166391.198443] ath10k: [0024]: 00000000 00000FF0 00000000 000000FF 0007A167 14044C07 91108004 00000000
[166391.207565] ath10k: [0032]: 00000000 00000000 00000000 0007A167 14044C07 91108005 00000007 00000000
[166391.216688] ath10k: [0040]: 00000001 00000002 0007A167 10044C01 91104110 000000FF 00000000 00000004
[166391.225780] ath10k: [0048]: 0007A167 17FC0001 009A5240 000015B3 000015B3 0040656C 91104569
[166391.234899] ath10k_pci 0000:01:00.0: ATH10K_END
[166391.243984] ath10k_pci 0000:01:00.0: [1] next: 0x422f90 buf: 0x4195c0 sz: 1500 len: 0 count: 0 free: 0
[166391.287006] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 1, skipped old beacon
[166391.287154] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[166391.293684] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 1, skipped old beacon
[166391.301001] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[166391.331344] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer daa08600 vdev: 0 addr: a4:4e:31:0c:b9:80
[166391.331377] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dbd5dc00 vdev: 0 addr: f0:18:98:2a:00:ee
[166391.341354] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dbd5d000 vdev: 0 addr: 98:5a:eb:8e:b5:a0
[166391.352383] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dbd5c000 vdev: 0 addr: 54:33:cb:ef:0b:01
[... cut ...]
[166398.850414] ath10k_pci 0000:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[166398.912548] ath10k_pci 0000:01:00.0: failed to submit frame: -19
[166398.912650] ath10k_pci 0000:01:00.0: failed to push frame: -19
[166398.963528] ath10k_pci 0000:01:00.0: failed to submit frame: -19
[166398.963572] ath10k_pci 0000:01:00.0: failed to push frame: -19
[166399.121276] ath10k_pci 0000:01:00.0: failed to submit frame: -19
[166399.121308] ath10k_pci 0000:01:00.0: failed to push frame: -19
[166399.196245] ath10k_pci 0000:01:00.0: device successfully recovered
```

schwichtgit, 2019-01-19:
```
[167012.442280] ath10k_pci 0000:01:00.0: firmware crashed! (guid 0f601e9b-d1db-4256-bbc3-225c5ecf8713)
[167012.442360] ath10k_pci 0000:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[167012.450258] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[167012.465327] ath10k_pci 0000:01:00.0: firmware ver 10.4b-ct-9984-fH-012-587105aba api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT crc32 738f84c7
[167012.473480] ath10k_pci 0000:01:00.0: board_file api 2 bmi_id 0:1 crc32 cf58c3bc
[167012.494554] ath10k_pci 0000:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[167012.504014] ath10k_pci 0000:01:00.0: firmware register dump:
[167012.512125] ath10k_pci 0000:01:00.0: [00]: 0x0000000A 0x000015B3 0x009A5F34 0x00975B31
[167012.517778] ath10k_pci 0000:01:00.0: [04]: 0x009A5F34 0x00060B30 0x00000005 0x00000001
[167012.525588] ath10k_pci 0000:01:00.0: [08]: 0x0045F8B4 0x0000003A 0x0045F8EC 0x00000001
[167012.533484] ath10k_pci 0000:01:00.0: [12]: 0x00000009 0x00000000 0x009A58DE 0x009A58EE
[167012.541564] ath10k_pci 0000:01:00.0: [16]: 0x00973AB0 0x009606CA 0x009606CA 0x00000000
[167012.549549] ath10k_pci 0000:01:00.0: [20]: 0x409A5F34 0x0040661C 0x0045ECFC 0x00000001
[167012.557515] ath10k_pci 0000:01:00.0: [24]: 0x809A6E0E 0x0040667C 0x00007FFF 0xC09A5F34
[167012.565513] ath10k_pci 0000:01:00.0: [28]: 0x809A75C0 0x0040670C 0x00412CE0 0x00406850
[167012.573415] ath10k_pci 0000:01:00.0: [32]: 0x809A8D02 0x0040676C 0x00406850 0x0045ECFC
[167012.581476] ath10k_pci 0000:01:00.0: [36]: 0x809A8DFE 0x0040684C 0x0042D204 0x004068DC
[167012.589474] ath10k_pci 0000:01:00.0: [40]: 0x80985E47 0x0040689C 0x00000000 0x00459FCC
[167012.597471] ath10k_pci 0000:01:00.0: [44]: 0x80994B1B 0x004068BC 0x0042E4B0 0x00459FCC
[167012.605438] ath10k_pci 0000:01:00.0: [48]: 0x8098FC9C 0x004068DC 0x0042E4B0 0x00000000
[167012.613345] ath10k_pci 0000:01:00.0: [52]: 0x80963AD3 0x00406A7C 0x0042E4B0 0x0098FC94
[167012.621401] ath10k_pci 0000:01:00.0: [56]: 0x80960E80 0x00406A9C 0x0000001F 0x00400000
[167012.629398] ath10k_pci 0000:01:00.0: Copy Engine register dump:
[167012.637404] ath10k_pci 0000:01:00.0: [00]: 0x0004a000  10  10   3   3
[167012.643556] ath10k_pci 0000:01:00.0: [01]: 0x0004a400   7   7 205 206
[167012.649990] ath10k_pci 0000:01:00.0: [02]: 0x0004a800  18  18  81  82
[167012.656488] ath10k_pci 0000:01:00.0: [03]: 0x0004ac00  13  13  15  13
[167012.662913] ath10k_pci 0000:01:00.0: [04]: 0x0004b000 3830 3830 165 125
[167012.669516] ath10k_pci 0000:01:00.0: [05]: 0x0004b400  22  22 373 374
[167012.676358] ath10k_pci 0000:01:00.0: [06]: 0x0004b800  26  26  26  26
[167012.682617] ath10k_pci 0000:01:00.0: [07]: 0x0004bc00   1   1   1   1
[167012.689228] ath10k_pci 0000:01:00.0: [08]: 0x0004c000   0   0 127   0
[167012.695725] ath10k_pci 0000:01:00.0: [09]: 0x0004c400   0   0   0   0
[167012.702150] ath10k_pci 0000:01:00.0: [10]: 0x0004c800   0   0   0   0
[167012.708753] ath10k_pci 0000:01:00.0: [11]: 0x0004cc00   0   0   0   0
[167012.717279] ath10k_pci 0000:01:00.0: debug log header, dbuf: 0x422f90  dropped: 0
[167012.722699] ath10k_pci 0000:01:00.0: [0] next: 0x422fa8 buf: 0x418fd0 sz: 1500 len: 328 count: 12 free: 0
[167012.730329] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[167012.738958] ath10k: [0000]: 0006A25A 17FC4C07 9110A004 00000001 00000001 00000001 00000001 0006A25A
[167012.745111] ath10k: [0008]: 17FC4C07 9110A005 00000083 0000003A 00088774 0000003B 0006A25A 17FC4C07
[167012.754406] ath10k: [0016]: 0000A0B5 0045F2B8 0045ECFC 0042D204 00000004 0006A25A 0FFC4C07 0000A0B6
[167012.763603] ath10k: [0024]: 0042D204 00000006 0006A25A 14044C01 71108880 00000000 00000000 00000000
[167012.772726] ath10k: [0032]: 00000FF0 0006A25A 14044C01 71108880 00010000 00000000 00000000 0000FFFF
[167012.781816] ath10k: [0040]: 0006A25A 14044C01 71108880 00020000 00000000 00000000 0000FFFF 0006A25A
[167012.790939] ath10k: [0048]: 14044C01 71108880 00030000 00000000 00000000 0007FDFF 0006A25A 14044C01
[167012.800060] ath10k: [0056]: 71108880 00040000 00000000 00000000 0007FDFF 0006A25A 14044C01 71108880
[167012.809183] ath10k: [0064]: 00050000 00000000 00000000 000FFFFF 0006A25A 14044C01 71108880 00060000
[167012.818304] ath10k: [0072]: 00000000 00000000 00000000 0006A25A 17FC0001 009A5F34 000015B3 000015B3
[167012.827394] ath10k: [0080]: 0040650C 91104569
[167012.836515] ath10k_pci 0000:01:00.0: ATH10K_END
[167012.841709] ath10k_pci 0000:01:00.0: [1] next: 0x422f90 buf: 0x4195c0 sz: 1500 len: 0 count: 0 free: 0
[167012.885001] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 1, skipped old beacon
[167012.885281] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[167012.891685] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 1, skipped old beacon
[167012.899148] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[167012.906443] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 1, skipped old beacon
[167012.945305] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer db027c00 vdev: 0 addr: f0:18:98:2a:00:ee
[167012.945337] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dc50d000 vdev: 1 addr: 54:60:09:d5:72:f2
[... cut ...]
[167014.335168] ath10k_pci 0000:01:00.0: failed to submit frame: -19
[167014.335190] ath10k_pci 0000:01:00.0: failed to push frame: -19
[167014.340246] ath10k_pci 0000:01:00.0: failed to submit frame: -19
[167014.346022] ath10k_pci 0000:01:00.0: failed to push frame: -19
[167014.403951] ath10k_pci 0000:01:00.0: failed to submit frame: -19
[167014.403972] ath10k_pci 0000:01:00.0: failed to push frame: -19
[167014.463799] ath10k_pci 0000:01:00.0: failed to submit frame: -19
[167014.463822] ath10k_pci 0000:01:00.0: failed to push frame: -19
```

schwichtgit, 2019-01-19:
kicked out
```
 option ieee80211v '1'
 option bss_transition '1'
```

just using
```
 option ieee80211r '1'
```
still getting these ...

```
[168353.441423] ath10k_pci 0000:01:00.0: firmware crashed! (guid bc4654ec-b73c-4cc3-8f0b-a2da13a081c0)
[168353.441500] ath10k_pci 0000:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[168353.449456] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[168353.464444] ath10k_pci 0000:01:00.0: firmware ver 10.4b-ct-9984-fH-012-587105aba api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT crc32 738f84c7
[168353.472646] ath10k_pci 0000:01:00.0: board_file api 2 bmi_id 0:1 crc32 cf58c3bc
[168353.493662] ath10k_pci 0000:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[168353.503075] ath10k_pci 0000:01:00.0: firmware register dump:
[168353.511276] ath10k_pci 0000:01:00.0: [00]: 0x0000000A 0x000015B3 0x009A5240 0x00975B31
[168353.516897] ath10k_pci 0000:01:00.0: [04]: 0x009A5240 0x00060930 0x00000005 0x00000001
[168353.524708] ath10k_pci 0000:01:00.0: [08]: 0x0042D25C 0x00000000 0x000000FF 0x00406740
[168353.532624] ath10k_pci 0000:01:00.0: [12]: 0x00000009 0x00000000 0x009A7C34 0x009A7C3D
[168353.540702] ath10k_pci 0000:01:00.0: [16]: 0x00973AB0 0x009606CA 0x009606CA 0x00000000
[168353.548668] ath10k_pci 0000:01:00.0: [20]: 0x409A5240 0x0040667C 0xFFFFFFFF 0x00412CE0
[168353.556678] ath10k_pci 0000:01:00.0: [24]: 0x809A7BB5 0x004066DC 0x00000007 0xC09A5240
[168353.564663] ath10k_pci 0000:01:00.0: [28]: 0x809A8DCE 0x0040671C 0x0042D25C 0x00406850
[168353.572555] ath10k_pci 0000:01:00.0: [32]: 0x809A8DFE 0x0040684C 0x0042D25C 0x0042E700
[168353.580625] ath10k_pci 0000:01:00.0: [36]: 0x80985E47 0x0040689C 0x00000000 0x00459E7C
[168353.588624] ath10k_pci 0000:01:00.0: [40]: 0x80994B1B 0x004068BC 0x0042E4B0 0x00459E7C
[168353.596590] ath10k_pci 0000:01:00.0: [44]: 0x8098FC9C 0x004068DC 0x0042E4B0 0x00000000
[168353.604588] ath10k_pci 0000:01:00.0: [48]: 0x80963AD3 0x00406A7C 0x0042E4B0 0x0098FC94
[168353.612484] ath10k_pci 0000:01:00.0: [52]: 0x80960E80 0x00406A9C 0x0000001F 0x00400000
[168353.620551] ath10k_pci 0000:01:00.0: [56]: 0x80960E51 0x00406ACC 0x00400000 0x00000000
[168353.628548] ath10k_pci 0000:01:00.0: Copy Engine register dump:
[168353.636523] ath10k_pci 0000:01:00.0: [00]: 0x0004a000  10  10   3   3
[168353.642698] ath10k_pci 0000:01:00.0: [01]: 0x0004a400  18  18 504 505
[168353.649110] ath10k_pci 0000:01:00.0: [02]: 0x0004a800  20  20  19  20
[168353.655639] ath10k_pci 0000:01:00.0: [03]: 0x0004ac00  22  22  24  22
[168353.662054] ath10k_pci 0000:01:00.0: [04]: 0x0004b000 2838 2838 181 141
[168353.668635] ath10k_pci 0000:01:00.0: [05]: 0x0004b400  30  30 445 446
[168353.675506] ath10k_pci 0000:01:00.0: [06]: 0x0004b800  16  16  16  16
[168353.681759] ath10k_pci 0000:01:00.0: [07]: 0x0004bc00   1   1   1   1
[168353.688347] ath10k_pci 0000:01:00.0: [08]: 0x0004c000   0   0 127   0
[168353.694876] ath10k_pci 0000:01:00.0: [09]: 0x0004c400   0   0   0   0
[168353.701289] ath10k_pci 0000:01:00.0: [10]: 0x0004c800   0   0   0   0
[168353.707872] ath10k_pci 0000:01:00.0: [11]: 0x0004cc00   0   0   0   0
[168353.716430] ath10k_pci 0000:01:00.0: debug log header, dbuf: 0x422fa8  dropped: 0
[168353.721841] ath10k_pci 0000:01:00.0: [0] next: 0x422f90 buf: 0x4195c0 sz: 1500 len: 220 count: 8 free: 0
[168353.729479] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[168353.738077] ath10k: [0000]: 0012C238 17FC4C01 00008051 000000FF 00000003 000000FF 00454B78 0012C238
[168353.744166] ath10k: [0008]: 17FC4C01 00008051 000000FF 00000004 000000FF 00454B78 0012C238 17FC4C01
[168353.753539] ath10k: [0016]: 00008051 000000FF 00000005 000000FF 00454B78 0012C238 14044C07 91108003
[168353.762574] ath10k: [0024]: 00000000 00000FF0 00000000 000000FF 0012C238 14044C07 91108004 00000000
[168353.771782] ath10k: [0032]: 000000FF 00000000 00000000 0012C238 14044C07 91108005 00000007 00000000
[168353.780873] ath10k: [0040]: 00000001 00000002 0012C238 10044C01 91104110 000000FF 00000000 00000004
[168353.789997] ath10k: [0048]: 0012C238 17FC0001 009A5240 000015B3 000015B3 0040656C 91104569
[168353.799115] ath10k_pci 0000:01:00.0: ATH10K_END
[168353.808201] ath10k_pci 0000:01:00.0: [1] next: 0x422fa8 buf: 0x418fd0 sz: 1500 len: 0 count: 0 free: 0
[168353.855059] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 1, skipped old beacon
[168353.855196] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[168353.861625] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 1, skipped old beacon
[168353.869106] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[168353.876476] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 1, skipped old beacon
[168353.903417] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dc361c00 vdev: 0 addr: f0:18:98:2a:00:ee
[168353.903448] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dc565200 vdev: 1 addr: 54:60:09:d5:72:f2
[168353.913420] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer daa9b600 vdev: 1 addr: 98:b8:e3:9c:75:96
[168353.924442] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer da93b200 vdev: 1 addr: 54:60:09:d5:78:f0
[... cut ...]
[168354.001608] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer db027200 vdev: 0 addr: 78:d2:94:a3:96:9e
[168354.082594] net_ratelimit: 30 callbacks suppressed
[168354.082809] ath10k_pci 0000:01:00.0: failed to transmit packet, dropping: -19
[168354.086384] ath10k_pci 0000:01:00.0: failed to submit frame: -19
[168354.093615] ath10k_pci 0000:01:00.0: failed to push frame: -19
[168354.106874] ieee80211 phy0: Hardware restart was requested
[168354.308965] ath10k_pci 0000:01:00.0: failed to transmit packet, dropping: -19
```

greearb, 2019-01-21:
These are rate-ctrl bugs where phymode does not match expected ratemasks.  Several other bugs have this same root cause.  I have added debugging for one of the cases, the other one has previously proven hard to debug, so hopefully you will hit the first case and I can make progress on that front.  Please try this firmware and report crashes.  If you can get longer logs (logread, etc) and enable ath10k debugging (i0xc0000020) it might help, but crash alone may be enough if you don't know how to do that.

[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2779895/firmware-5-full-htt-mgt-community.bin.gz)

schwichtgit, 2019-01-21:
```
driver: ath10k_pci
version: 4.14.93
firmware-version: 10.4b-ct-9984-fH-012-02c4d66f9
```
applied ... lets see

schwichtgit, 2019-01-21:
@greearb as it seems that we seem to be more vulnerable in the 802.11r config case ...
I wonder if
```
These are rate-ctrl bugs where phymode does not match expected ratemasks. Several other bugs have this same root cause
```
are somehow related to (fast) transitions?

schwichtgit, 2019-01-22:
[LEDE-10-0-8-8-dmesg.log](https://github.com/greearb/ath10k-ct/files/2785008/LEDE-10-0-8-8-dmesg.log)

schwichtgit, 2019-01-23:
the crash happend, before I had been able to apply the debug_level settings, so we have default dmesg logs ...
that does not change the scenario, but the usable infos for @greearb 
I have been busy this morning ... but I have heard just a single WIFI complain ...
will look if something reportable happend in a bit

schwichtgit, 2019-01-23:
@greearb 
this is the complete set of syslogs of the 4 APs involved.
you see many crashes happening
```
system-LEDE-10-0-8-7.log:Jan 22 13:52:18 LEDE-10-0-8-7 kernel: [33651.463211] ath10k_pci 0000:01:00.0: firmware crashed! (guid 41f67d73-6f28-437e-baf2-ffcffbfe304d)
system-LEDE-10-0-8-7.log:Jan 22 14:30:39 LEDE-10-0-8-7 kernel: [35952.595988] ath10k_pci 0000:01:00.0: firmware crashed! (guid 43516203-320b-4311-b650-c0c298318a39)
system-LEDE-10-0-8-7.log:Jan 22 14:40:29 LEDE-10-0-8-7 kernel: [36542.934122] ath10k_pci 0000:01:00.0: firmware crashed! (guid da659adb-2134-4257-9432-6567c5c7f0bc)
system-LEDE-10-0-8-7.log:Jan 22 16:21:33 LEDE-10-0-8-7 kernel: [42606.853434] ath10k_pci 0000:01:00.0: firmware crashed! (guid d8516d9d-c91e-4286-918d-09fdffe1461f)
system-LEDE-10-0-8-7.log:Jan 22 16:43:30 LEDE-10-0-8-7 kernel: [ 1285.636575] ath10k_pci 0000:01:00.0: firmware crashed! (guid 31bfdfdc-0116-42e8-b692-65dc751dbfe6)
system-LEDE-10-0-8-7.log:Jan 22 17:02:27 LEDE-10-0-8-7 kernel: [ 2421.593628] ath10k_pci 0000:01:00.0: firmware crashed! (guid bddc0da9-a0c9-4402-bf4a-b1da3faa870e)
system-LEDE-10-0-8-7.log:Jan 22 17:13:41 LEDE-10-0-8-7 kernel: [ 3095.857802] ath10k_pci 0000:01:00.0: firmware crashed! (guid cf68d1cc-36af-47ff-84ff-7082d4e3e056)
system-LEDE-10-0-8-7.log:Jan 22 17:50:38 LEDE-10-0-8-7 kernel: [ 5313.532417] ath10k_pci 0001:01:00.0: firmware crashed! (guid 9be1a78d-485e-4da3-a936-64ec2d4aab5f)
system-LEDE-10-0-8-7.log:Jan 22 18:05:59 LEDE-10-0-8-7 kernel: [ 6234.338934] ath10k_pci 0000:01:00.0: firmware crashed! (guid 86774df3-a8ec-4fea-aa63-a4f918b318d3)
system-LEDE-10-0-8-7.log:Jan 22 18:12:41 LEDE-10-0-8-7 kernel: [ 6635.896974] ath10k_pci 0000:01:00.0: firmware crashed! (guid 65454729-1cab-4d7d-bec2-f71bb55a3dec)
system-LEDE-10-0-8-7.log:Jan 22 18:14:42 LEDE-10-0-8-7 kernel: [ 6756.430882] ath10k_pci 0000:01:00.0: firmware crashed! (guid 1c937064-24db-41f5-9a08-5c826a49c5e9)
system-LEDE-10-0-8-7.log:Jan 22 18:17:42 LEDE-10-0-8-7 kernel: [ 6937.421382] ath10k_pci 0000:01:00.0: firmware crashed! (guid 1e6aba3c-bef8-48db-9e6a-23eaa2e297ca)
system-LEDE-10-0-8-7.log:Jan 22 18:18:10 LEDE-10-0-8-7 kernel: [ 6964.859267] ath10k_pci 0000:01:00.0: firmware crashed! (guid 8733f5a5-a847-4e7e-9b8b-cd317ece637b)
system-LEDE-10-0-8-7.log:Jan 22 18:20:36 LEDE-10-0-8-7 kernel: [ 7110.751683] ath10k_pci 0000:01:00.0: firmware crashed! (guid 8ce76e0c-7088-46cb-9cfd-2f07789da8b7)
system-LEDE-10-0-8-7.log:Jan 22 18:21:36 LEDE-10-0-8-7 kernel: [ 7170.471056] ath10k_pci 0001:01:00.0: firmware crashed! (guid 6df96710-3e99-456f-9fb9-ac2b8512e12b)
system-LEDE-10-0-8-7.log:Jan 22 18:40:00 LEDE-10-0-8-7 kernel: [ 8275.118519] ath10k_pci 0000:01:00.0: firmware crashed! (guid 30c22698-2136-4460-8ee0-2ed2ce784e3c)
system-LEDE-10-0-8-7.log:Jan 22 19:10:39 LEDE-10-0-8-7 kernel: [10113.627529] ath10k_pci 0000:01:00.0: firmware crashed! (guid 3222ac9d-b1ee-4e5b-92af-ffdf565f756f)
system-LEDE-10-0-8-7.log:Jan 22 19:53:29 LEDE-10-0-8-7 kernel: [12683.584261] ath10k_pci 0000:01:00.0: firmware crashed! (guid 6a44b7b7-fa17-4ad3-9b1e-d8166e55e2d4)
system-LEDE-10-0-8-7.log:Jan 22 20:37:34 LEDE-10-0-8-7 kernel: [15328.529121] ath10k_pci 0000:01:00.0: firmware crashed! (guid 21badcbb-1cf5-4eaa-9618-153e25cb203f)
system-LEDE-10-0-8-7.log:Jan 22 20:59:20 LEDE-10-0-8-7 kernel: [16635.207542] ath10k_pci 0000:01:00.0: firmware crashed! (guid feca61fa-5cc1-4eef-9e8f-ce41af7c31ce)
system-LEDE-10-0-8-7.log:Jan 22 21:19:30 LEDE-10-0-8-7 kernel: [17844.965758] ath10k_pci 0000:01:00.0: firmware crashed! (guid 1b34eb1a-d879-4c4a-9e79-23b6169d082a)
system-LEDE-10-0-8-7.log:Jan 22 22:00:58 LEDE-10-0-8-7 kernel: [20332.917325] ath10k_pci 0000:01:00.0: firmware crashed! (guid b4530d42-7fb3-4699-a464-710ff0bb9ad9)
system-LEDE-10-0-8-7.log:Jan 22 22:43:42 LEDE-10-0-8-7 kernel: [22896.692352] ath10k_pci 0000:01:00.0: firmware crashed! (guid 7900119d-512c-4a48-922a-16fa622e2b21)
system-LEDE-10-0-8-7.log:Jan 22 23:09:23 LEDE-10-0-8-7 kernel: [24437.971673] ath10k_pci 0000:01:00.0: firmware crashed! (guid 68befbf7-1dc5-424e-977c-b1f344803f67)
system-LEDE-10-0-8-7.log:Jan 22 23:14:17 LEDE-10-0-8-7 kernel: [24731.667910] ath10k_pci 0000:01:00.0: firmware crashed! (guid db15e482-55ac-4693-8ef0-c4602d545015)
system-LEDE-10-0-8-7.log:Jan 22 23:18:14 LEDE-10-0-8-7 kernel: [24969.049471] ath10k_pci 0000:01:00.0: firmware crashed! (guid 36f00fff-b842-452f-b2fa-d428d79a2cf4)
system-LEDE-10-0-8-7.log:Jan 23 00:32:29 LEDE-10-0-8-7 kernel: [29424.424188] ath10k_pci 0000:01:00.0: firmware crashed! (guid 10832adb-6a2c-422d-bd60-e3b1638a64c5)
system-LEDE-10-0-8-7.log:Jan 23 14:22:08 LEDE-10-0-8-7 kernel: [35440.855122] ath10k_pci 0000:01:00.0: firmware crashed! (guid 99d08142-5f16-4b0c-87b5-9cd388759a51)
system-LEDE-10-0-8-7.log:Jan 23 14:24:50 LEDE-10-0-8-7 kernel: [35602.926441] ath10k_pci 0000:01:00.0: firmware crashed! (guid ac309335-fb5b-48bc-ab90-4cf3573c4a82)
system-LEDE-10-0-8-7.log:Jan 23 15:59:23 LEDE-10-0-8-7 kernel: [41276.118870] ath10k_pci 0000:01:00.0: firmware crashed! (guid 5e083aed-eb63-4ebe-a8e0-5eea43b06ac4)
system-LEDE-10-0-8-7.log:Jan 23 17:25:06 LEDE-10-0-8-7 kernel: [46419.336356] ath10k_pci 0000:01:00.0: firmware crashed! (guid 3477bd93-f508-422f-9baa-34c19e015fca)
system-LEDE-10-0-8-7.log:Jan 23 19:06:58 LEDE-10-0-8-7 kernel: [52530.860005] ath10k_pci 0000:01:00.0: firmware crashed! (guid 04e3512b-4511-463d-93f1-9415552d91bd)
[... cut ...]
system-LEDE-10-0-8-8.log:Jan 23 16:35:05 LEDE-10-0-8-8 kernel: [43418.965098] ath10k_pci 0000:01:00.0: firmware crashed! (guid 75140bb2-5ade-4f99-9ed1-810036dd6fa8)
system-LEDE-10-0-8-8.log:Jan 23 18:00:38 LEDE-10-0-8-8 kernel: [48551.587577] ath10k_pci 0000:01:00.0: firmware crashed! (guid 9e33d09f-8b99-437e-afc1-35bf4911a0e9)
system-LEDE-10-0-8-9.log:Jan 23 14:24:37 LEDE-10-0-8-9 kernel: [54294.410211] ath10k_pci 0000:01:00.0: firmware crashed! (guid 3717b8c4-440f-491b-be3d-fbdb1e1a59dc)
```
LEDE-10-0-8-7 runs the latest openwrt release with default firmware.

schwichtgit, 2019-01-23:
[system-LEDE-10-0-8-x.zip](https://github.com/greearb/ath10k-ct/files/2788887/system-LEDE-10-0-8-x.zip)

greearb, 2019-01-24:
Thank you for the detailed logs.  The system-LEDE-10-0-8-7.log shows that device with last two MAC of: db:0d associates as HT, but then a while later, the code starts thinking it is VHT (probably mem corruption I guess?) and then it asserts because there are zero rates enabled for VHT.  This is the same bug I have been chasing for a while, but the better logs hopefully give me a clue that I can use to track it down.

Here is an un-tested image that may catch this mem corruption very near where it happens.  If you have time, please test it, or wait a bit until I get a chance test it later today.  This image will probably run at least a bit slower, but hopefully it will at least be functional.

[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2792028/firmware-5-full-htt-mgt-community.bin.gz)

schwichtgit, 2019-01-24:
Working from home atm and applied the firmware here.
```
driver: ath10k_pci
version: 4.14.93
firmware-version: 10.4b-ct-9984-fHB-012-5f9cc1a57
```
[dmesg.log](https://github.com/greearb/ath10k-ct/files/2792530/dmesg.log)

greearb, 2019-01-24:
My debug logic raced with an IRQ...I think I have that resolved now.  Please see if this works better.
[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2793191/firmware-5-full-htt-mgt-community.bin.gz)

schwichtgit, 2019-01-24:
```
driver: ath10k_pci
version: 4.14.93
firmware-version: 10.4b-ct-9984-fHB-012-c6285e0a2
```
applied

loaded ok. will keep you posted, @greearb

schwichtgit, 2019-01-24:
@greearb  ... posting comes quicker than expected ;-)
```
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.722088] ath10k_pci 0000:01:00.0: firmware crashed! (guid f3989fb2-6884-447c-a955-cdd0436031cf)
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.722208] ath10k_pci 0000:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.730011] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.744883] ath10k_pci 0000:01:00.0: firmware ver 10.4b-ct-9984-fHB-012-c6285e0a2 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT crc32 2249f586
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.753047] ath10k_pci 0000:01:00.0: board_file api 2 bmi_id 0:1 crc32 cf58c3bc
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.773992] ath10k_pci 0000:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.783144] ath10k_pci 0000:01:00.0: firmware register dump:
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.791168] ath10k_pci 0000:01:00.0: [00]: 0x0000000A 0x000015B3 0x00980614 0x00975B31
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.797074] ath10k_pci 0000:01:00.0: [04]: 0x00980614 0x00060033 0x00000005 0x00000007
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.804814] ath10k_pci 0000:01:00.0: [08]: 0x00400000 0x004179F8 0x00000268 0x00000007
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.812696] ath10k_pci 0000:01:00.0: [12]: 0x00000009 0x00000000 0x009C4221 0x009C422A
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.820598] ath10k_pci 0000:01:00.0: [16]: 0x00980614 0x00972009 0x009606CA 0x00000000
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.828498] ath10k_pci 0000:01:00.0: [20]: 0x40980614 0x00401120 0x00001080 0x0045ED78
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.836395] ath10k_pci 0000:01:00.0: [24]: 0x809BB2D1 0x00401180 0x00000000 0xC0980614
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.844294] ath10k_pci 0000:01:00.0: [28]: 0x809C0EBF 0x004011C0 0x77790002 0x009635B0
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.852192] ath10k_pci 0000:01:00.0: [32]: 0x809608C1 0x004011F0 0x00000008 0x0000001F
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.860094] ath10k_pci 0000:01:00.0: [36]: 0x40960345 0x00401210 0x00000008 0x00000000
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.867993] ath10k_pci 0000:01:00.0: [40]: 0x00000000 0x00401230 0x00050022 0x004445D0
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.875893] ath10k_pci 0000:01:00.0: [44]: 0x00000000 0x00000000 0x00000000 0x00000000
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.883792] ath10k_pci 0000:01:00.0: [48]: 0x00000000 0x00000000 0x00000000 0x00000000
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.891691] ath10k_pci 0000:01:00.0: [52]: 0x00000000 0x00000000 0x00000000 0x00000000
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.899588] ath10k_pci 0000:01:00.0: [56]: 0x00000000 0x00000000 0x00000000 0x00000000
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.907488] ath10k_pci 0000:01:00.0: Copy Engine register dump:
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.915393] ath10k_pci 0000:01:00.0: [00]: 0x0004a000  14  14   3   3
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.921207] ath10k_pci 0000:01:00.0: [01]: 0x0004a400   1   1   7   8
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.927804] ath10k_pci 0000:01:00.0: [02]: 0x0004a800   8   8  71  72
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.934226] ath10k_pci 0000:01:00.0: [03]: 0x0004ac00   9   9  11   9
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.940651] ath10k_pci 0000:01:00.0: [04]: 0x0004b000 6926 6926 177 137
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.947074] ath10k_pci 0000:01:00.0: [05]: 0x0004b400  29  29 412 413
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.953497] ath10k_pci 0000:01:00.0: [06]: 0x0004b800  17  17  17  17
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.960094] ath10k_pci 0000:01:00.0: [07]: 0x0004bc00   1   1   1   1
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.966520] ath10k_pci 0000:01:00.0: [08]: 0x0004c000   0   0 127   0
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.972943] ath10k_pci 0000:01:00.0: [09]: 0x0004c400   1   1   1   1
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.979366] ath10k_pci 0000:01:00.0: [10]: 0x0004c800   0   0   0   0
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.985790] ath10k_pci 0000:01:00.0: [11]: 0x0004cc00   0   0   0   0
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.994240] ath10k_pci 0000:01:00.0: debug log header, dbuf: 0x422fa8  dropped: 0
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  376.999658] ath10k_pci 0000:01:00.0: [0] next: 0x422f90 buf: 0x4195c0 sz: 1500 len: 84 count: 3 free: 0
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  377.007122] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  377.015299] ath10k: [0000]: 00052743 14000007 9110DE77 77790002 009635B0 00000007 00000000 00052743
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  377.021550] ath10k: [0008]: 14000007 80003333 00000000 00000000 00000000 00000000 00052743 17FC0001
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  377.030492] ath10k: [0016]: 00980614 000015B3 000015B3 00401010 91104569
[... cut ...]
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  377.132211] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dbfef800 vdev: 0 addr: ac:bc:32:80:37:cd
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  377.132244] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dbfff600 vdev: 0 addr: 18:b4:30:c5:a3:40
Jan 24 21:07:05 SchwichtenburgWrt kernel: [  377.142220] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dad00800 vdev: 0 addr: 54:33:cb:ef:0b:01
```
more context in syslog below:

[system-SchwichtenburgWrt.log.zip](https://github.com/greearb/ath10k-ct/files/2793621/system-SchwichtenburgWrt.log.zip)

greearb, 2019-01-25:
So, some timer called from IRQ causes this corruption.  It is going to take me a bit to figure out how to patch into the ROM and install my own timer irq handler to add debugging....

greearb, 2019-01-25:
Please try this one, hopefully it will tell me what timer method is causing the issue...
[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2794224/firmware-5-full-htt-mgt-community.bin.gz)

schwichtgit, 2019-01-25:
```
driver: ath10k_pci
version: 4.14.93
firmware-version: 10.4b-ct-9984-fHB-012-ee8a097dd
```
applied

loaded ok. will keep you posted, @greearb

schwichtgit, 2019-01-25:
```
[  202.009954] ath10k_pci 0000:01:00.0: ATH10K_END
[  202.709719] ath10k_pci 0000:01:00.0: firmware crashed! (guid 7e0e3afd-242d-4f13-b600-66ca09401b08)
[  202.709764] ath10k_pci 0000:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[  202.717658] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[  202.729467] ath10k_pci 0000:01:00.0: firmware ver 10.4b-ct-9984-fHB-012-ee8a097dd api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT crc32 532f1cb3
[  202.739849] ath10k_pci 0000:01:00.0: board_file api 2 bmi_id 0:1 crc32 cf58c3bc
[  202.761636] ath10k_pci 0000:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[  202.770728] ath10k_pci 0000:01:00.0: firmware register dump:
[  202.778767] ath10k_pci 0000:01:00.0: [00]: 0x0000000A 0x000015B3 0x00980614 0x00975B31
[  202.784672] ath10k_pci 0000:01:00.0: [04]: 0x00980614 0x00060530 0x00000005 0x00000007
[  202.792397] ath10k_pci 0000:01:00.0: [08]: 0x00400000 0x004179F8 0x00000160 0x00000004
[  202.800297] ath10k_pci 0000:01:00.0: [12]: 0x00000009 0x00000000 0x00963B71 0x00963B83
[  202.808198] ath10k_pci 0000:01:00.0: [16]: 0x00973AB0 0x009606CA 0x00960370 0x00000000
[  202.816097] ath10k_pci 0000:01:00.0: [20]: 0x40980614 0x0040693C 0x00001080 0x0045EE7C
[  202.823992] ath10k_pci 0000:01:00.0: [24]: 0x809BB2D1 0x0040699C 0x00000000 0xC0980614
[  202.831891] ath10k_pci 0000:01:00.0: [28]: 0x809B330A 0x004069DC 0x777A0001 0x0045F03C
[  202.839790] ath10k_pci 0000:01:00.0: [32]: 0x8096EE81 0x00406A0C 0x0045F228 0x0045F03C
[  202.847689] ath10k_pci 0000:01:00.0: [36]: 0x8096F857 0x00406A2C 0x00000000 0x00000000
[  202.855589] ath10k_pci 0000:01:00.0: [40]: 0x80963A26 0x00406A4C 0x00406C00 0x00405528
[  202.863489] ath10k_pci 0000:01:00.0: [44]: 0x80960E80 0x00406A9C 0x00000010 0x00400000
[  202.871389] ath10k_pci 0000:01:00.0: [48]: 0x80960E51 0x00406ACC 0x00400000 0x00000000
[  202.879286] ath10k_pci 0000:01:00.0: [52]: 0x80960E9D 0x00406AEC 0x00000000 0x00400600
[  202.887187] ath10k_pci 0000:01:00.0: [56]: 0x40960024 0x00406B0C 0x00403D08 0x00403D08
[  202.895088] ath10k_pci 0000:01:00.0: Copy Engine register dump:
[  202.902988] ath10k_pci 0000:01:00.0: [00]: 0x0004a000   1   1   3   3
[  202.908803] ath10k_pci 0000:01:00.0: [01]: 0x0004a400  10  10 144 145
[  202.915399] ath10k_pci 0000:01:00.0: [02]: 0x0004a800  44  44 107 108
[  202.921824] ath10k_pci 0000:01:00.0: [03]: 0x0004ac00  20  20  22  20
[  202.928248] ath10k_pci 0000:01:00.0: [04]: 0x0004b000 6752 6752  90  50
[  202.934672] ath10k_pci 0000:01:00.0: [05]: 0x0004b400  31  31 478 479
[  202.941095] ath10k_pci 0000:01:00.0: [06]: 0x0004b800  29  29  29  29
[  202.947691] ath10k_pci 0000:01:00.0: [07]: 0x0004bc00   1   1   1   1
[  202.954115] ath10k_pci 0000:01:00.0: [08]: 0x0004c000   0   0 127   0
[  202.960549] ath10k_pci 0000:01:00.0: [09]: 0x0004c400   0   0   0   0
[  202.966963] ath10k_pci 0000:01:00.0: [10]: 0x0004c800   0   0   0   0
[  202.973386] ath10k_pci 0000:01:00.0: [11]: 0x0004cc00   0   0   0   0
[  202.981823] ath10k_pci 0000:01:00.0: debug log header, dbuf: 0x422ff8  dropped: 0
[  202.987243] ath10k_pci 0000:01:00.0: [0] next: 0x422fe0 buf: 0x4195c0 sz: 1500 len: 84 count: 3 free: 0
[  202.994711] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[  203.002898] ath10k: [0000]: 00027A91 14000007 9110DE77 777A0001 0045F03C 00000004 00000000 00027A91
[  203.009148] ath10k: [0008]: 14000007 80003333 00000000 00000000 00000000 00000000 00027A91 17FC0001
[  203.018090] ath10k: [0016]: 00980614 000015B3 000015B3 0040682C 91104569
[  203.027116] ath10k_pci 0000:01:00.0: ATH10K_END
[  203.035068] ath10k_pci 0000:01:00.0: [1] next: 0x422ff8 buf: 0x418fd0 sz: 1500 len: 0 count: 0 free: 0
[  203.077040] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 1, skipped old beacon
[  203.077099] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  203.083259] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 1, skipped old beacon
[  203.112698] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dad44800 vdev: 0 addr: 54:33:cb:ef:0b:01
[  203.112746] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dad45e00 vdev: 0 addr: a4:4e:31:0c:b9:80
[  203.122736] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dad44c00 vdev: 0 addr: 70:81:eb:52:c5:4b
[  203.133657] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dbf78e00 vdev: 0 addr: 64:16:66:05:b0:58
[  203.144608] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dbf79800 vdev: 0 addr: 54:60:09:d5:72:f2
[  203.155549] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer da418a00 vdev: 0 addr: 18:b4:30:c5:a3:40
[  203.166476] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer da418200 vdev: 0 addr: 64:16:66:04:76:88
[  203.177411] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer da419600 vdev: 0 addr: 54:60:09:d5:78:f0
[  203.188344] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer dc2d0600 vdev: 0 addr: 78:d2:94:a3:96:9e
[  203.231637] ath10k_pci 0000:01:00.0: Invalid state: 2 in ath10k_htt_tx_32, warning will not be repeated.
[  203.231664] ------------[ cut here ]------------
[... cut ...]
[  207.959075] ath10k_pci 0000:01:00.0: failed to submit frame: -19
[  207.959160] ath10k_pci 0000:01:00.0: failed to push frame: -19
[  207.965285] ath10k_pci 0000:01:00.0: failed to submit frame: -19
[  207.969818] ath10k_pci 0000:01:00.0: failed to push frame: -19
[  208.783840] net_ratelimit: 11 callbacks suppressed
```
and syslog 
[system-SchwichtenburgWrt.log.zip](https://github.com/greearb/ath10k-ct/files/2794370/system-SchwichtenburgWrt.log.zip)

good thing that it happens often, @greearb
(hopefully the same issue ... )

shelterx, 2019-01-25:
@schwichtgit Do you run irqbalance by any chance?

Anyway @greearb , my original #54 issue seems to be gone, uptime of slightly over 6 days without crashes.

And I don't seem to run into this (#58) issue anymore, @schwichtgit seems to be able to trigger it, I don't use multiple AP's or fast transition if that's what's trigger this crash. I had an extender for a while tho' but I removed it because it sucked.

greearb, 2019-01-25:
Well, the problem appears to have moved, and may be related to DMA transfer callback from the host.  Or maybe my debugging logic is funky.  Please report several of the crashes when you see more, maybe I can find a pattern.  Here is a new image that might provide more insight.  Perhaps interestingly, these crashes hit right after some warnings in the pfsched logic that I debugged in bug 54.  Maybe some systems/situations have timing differences that exacerbate this issue since you can reproduce it so easily and others never see it.

[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2796325/firmware-5-full-htt-mgt-community.bin.gz)

greearb, 2019-01-25:
Regarding the 'previous' crash, that one also shows corruption after DMA transfer from the host, but before the DMA transfer handler code actually runs in the firmware.  That would seem to indicate that the DMA transfer itself may have corrupted the memory that my bug-check logic is trying to verify, and that may be a real pain to debug/fix!  There was no indication of the pfsched warning in the second crash, so that may just be a symptom and not a cause.  Let me see what you get with the latest binary and maybe I can think of more ways to debug this issue...

schwichtgit, 2019-01-25:
a new set of logs.

please take note of
```
Jan 25 21:36:02 LEDE-10-0-8-7 kernel: [ 4860.418595] ath10k_pci 0000:01:00.0: firmware crashed! (guid c1dd3bed-f3c8-408f-8e47-361cc21e3d6c)
```
and
```
Jan 25 15:35:46 LEDE-10-0-8-9 kernel: [71722.524783] ath10k_pci 0000:01:00.0: firmware crashed! (guid bb29d1c9-0684-40c5-85eb-6e9647213e39)
Jan 25 17:02:44 LEDE-10-0-8-9 kernel: [76940.233136] ath10k_pci 0000:01:00.0: firmware crashed! (guid b2d28b98-a77d-45fc-a623-34a2caaa3c28)
Jan 25 17:02:44 LEDE-10-0-8-9 kernel: [76940.773311] ath10k_pci 0000:01:00.0: firmware crashed! (guid 4855f965-860e-4d19-8537-92b771f3127d)
Jan 25 17:02:44 LEDE-10-0-8-9 kernel: [76940.856435] ath10k_pci 0000:01:00.0: in crash-regs-harder
Jan 25 17:02:45 LEDE-10-0-8-9 kernel: [76941.610856] ath10k_pci 0000:01:00.0: in crash-regs-harder, firmware did not provide indicator: 0xdeadbeef
```

as it had been used too much I did not reboot system-LEDE-10-0-8-8 yet so the firmware is older
and as before system-LEDE-10-0-8-6 runs latest stock openwrt firmware.

[system-LEDE-10-0-8-x.zip](https://github.com/greearb/ath10k-ct/files/2798003/system-LEDE-10-0-8-x.zip)

schwichtgit, 2019-01-27:
more crash log for you, @greearb 
```
driver: ath10k_pci
version: 4.14.93
firmware-version: 10.4b-ct-9984-fHB-012-41cd84839
```
```
Jan 26 13:38:33 SchwichtenburgWrt kernel: [83546.841507] ath10k_pci 0000:01:00.0: firmware crashed! (guid bcbbde60-beb2-41d1-bd38-1680d57899cf)
Jan 26 13:39:40 SchwichtenburgWrt kernel: [83614.382962] ath10k_pci 0000:01:00.0: firmware crashed! (guid 9829dfb1-08b5-4dd6-88cd-fababaf96551)
Jan 26 13:42:05 SchwichtenburgWrt kernel: [83758.868723] ath10k_pci 0000:01:00.0: firmware crashed! (guid 0adff0af-218c-491e-8060-605d1272e435)
Jan 26 13:42:16 SchwichtenburgWrt kernel: [83770.579590] ath10k_pci 0000:01:00.0: firmware crashed! (guid b3e3ae8c-d89f-4c7b-8588-fdd2a4b38c97)
Jan 26 13:55:11 SchwichtenburgWrt kernel: [84545.128410] ath10k_pci 0000:01:00.0: firmware crashed! (guid 131d92f1-3ef3-49af-a5bd-907e64a50120)
Jan 26 13:56:34 SchwichtenburgWrt kernel: [84628.316273] ath10k_pci 0000:01:00.0: firmware crashed! (guid f73b53ed-0c77-406e-8e8e-3cfcc870a243)
Jan 26 14:20:08 SchwichtenburgWrt kernel: [86041.859320] ath10k_pci 0000:01:00.0: firmware crashed! (guid d09a20d6-bfc1-427f-bed1-3d80bd45be26)
Jan 26 22:21:57 SchwichtenburgWrt kernel: [114951.024384] ath10k_pci 0001:01:00.0: firmware crashed! (guid 04b594b3-b560-4524-9676-0f11e3bde768)
Jan 27 00:34:59 SchwichtenburgWrt kernel: [122933.455802] ath10k_pci 0000:01:00.0: firmware crashed! (guid 3456db07-8204-4795-9e03-ee001eaa5d92)
Jan 27 00:51:46 SchwichtenburgWrt kernel: [123940.690142] ath10k_pci 0000:01:00.0: firmware crashed! (guid 517dd868-902c-4935-88f5-7ce09c9bf286)
Jan 27 00:52:21 SchwichtenburgWrt kernel: [123975.709947] ath10k_pci 0000:01:00.0: firmware crashed! (guid 6b29216f-1dd9-43f7-b1ff-2c7a3b2392a1)
Jan 27 01:02:30 SchwichtenburgWrt kernel: [124584.620215] ath10k_pci 0000:01:00.0: firmware crashed! (guid db5c9792-26b7-472f-aa77-3e760f99f2fa)
Jan 27 01:02:47 SchwichtenburgWrt kernel: [124601.287818] ath10k_pci 0000:01:00.0: firmware crashed! (guid 1c4c807b-3749-4ed5-9cc9-eb7d05939dd2)
Jan 27 01:10:45 SchwichtenburgWrt kernel: [125079.516631] ath10k_pci 0000:01:00.0: firmware crashed! (guid 0777a1fd-c4d0-48b3-b67b-1460c0ef332e)
```
[system-SchwichtenburgWrt.log.zip](https://github.com/greearb/ath10k-ct/files/2800327/system-SchwichtenburgWrt.log.zip)

greearb, 2019-01-28:
Thanks for the logs.  I added more debugging, please try the attached and send more logs.
[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2803228/firmware-5-full-htt-mgt-community.bin.gz)

schwichtgit, 2019-01-28:
```
10.4b-ct-9984-fHB-012-68feaa5bd
```
```
Jan 28 16:04:17 SchwichtenburgWrt kernel: [   56.083764] ath10k_pci 0000:01:00.0: firmware crashed! (guid 878b330f-9e25-4652-81ff-3967dc75e2c3)
Jan 28 16:05:31 SchwichtenburgWrt kernel: [  100.177092] ath10k_pci 0001:01:00.0: firmware crashed! (guid c88f5d02-5fe5-4842-a953-0927a33c19da)
Jan 28 16:05:45 SchwichtenburgWrt kernel: [  114.002169] ath10k_pci 0001:01:00.0: firmware crashed! (guid 83c9aacf-4bca-4c0f-a77c-2f01f6083b72)
Jan 28 16:06:01 SchwichtenburgWrt kernel: [  130.200587] ath10k_pci 0000:01:00.0: firmware crashed! (guid 6abd183d-6718-4800-976c-82fc49ce3735)
Jan 28 16:06:10 SchwichtenburgWrt kernel: [  139.100715] ath10k_pci 0000:01:00.0: firmware crashed! (guid 3023d2de-8e60-44cc-85b5-f6ed27489c1b)
Jan 28 16:06:20 SchwichtenburgWrt kernel: [  149.090982] ath10k_pci 0000:01:00.0: firmware crashed! (guid 3f0bcd6f-2621-4ed4-96a2-258db9962ac2)
Jan 28 16:06:29 SchwichtenburgWrt kernel: [  157.610954] ath10k_pci 0000:01:00.0: firmware crashed! (guid 432f8e72-19b6-4830-9c72-32cd40f1feed)
Jan 28 16:06:30 SchwichtenburgWrt kernel: [  158.510906] ath10k_pci 0000:01:00.0: firmware crashed! (guid 40489d59-2fa4-4254-bf6d-531ce53a3bff)
Jan 28 16:06:30 SchwichtenburgWrt kernel: [  158.594176] ath10k_pci 0000:01:00.0: in crash-regs-harder
Jan 28 16:06:31 SchwichtenburgWrt kernel: [  159.343094] ath10k_pci 0000:01:00.0: in crash-regs-harder, firmware did not provide indicator: 0xdeadbeef
Jan 28 16:07:05 SchwichtenburgWrt kernel: [  193.537615] ath10k_pci 0001:01:00.0: firmware crashed! (guid a57441b6-1a04-4d92-95e6-d01c811519b7)
```
[system-SchwichtenburgWrt.log.zip](https://github.com/greearb/ath10k-ct/files/2803827/system-SchwichtenburgWrt.log.zip)

greearb, 2019-01-28:
Please try this one, it has more debugging, and a potential fix.

[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2804397/firmware-5-full-htt-mgt-community.bin.gz)

greearb, 2019-01-28:
And another, this may narrow down crash from latest, reported in bug 64.
[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2804972/firmware-5-full-htt-mgt-community.bin.gz)

schwichtgit, 2019-01-28:
```
driver: ath10k_pci
version: 4.14.93
firmware-version: 10.4b-ct-9984-fHB-012-663356dd5
```
applied

shelterx, 2019-01-29:
@greearb Any idea what triggers this bug? I don't seem to encounter it,

schwichtgit, 2019-01-29:
@greearb 

same config as 15 hours ago ... but more crashes in the meantime...
(family suffering bad WIFI)
```
Jan 28 22:19:36 SchwichtenburgWrt kernel: [  134.647001] ath10k_pci 0000:01:00.0: firmware crashed! (guid b5d09e6f-fca7-4364-b032-45766514467b)
Jan 28 22:21:24 SchwichtenburgWrt kernel: [  243.299499] ath10k_pci 0000:01:00.0: firmware crashed! (guid 3ccd00de-8aa0-45ff-9277-32e127703d00)
Jan 29 10:05:37 SchwichtenburgWrt kernel: [42496.305166] ath10k_pci 0000:01:00.0: firmware crashed! (guid 862f2d50-8972-4fd9-8b2e-70f0cb4b862e)
Jan 29 10:48:03 SchwichtenburgWrt kernel: [45042.145325] ath10k_pci 0000:01:00.0: firmware crashed! (guid 1bf50011-93c5-4a86-975b-96cf3294dd3e)
Jan 29 10:50:02 SchwichtenburgWrt kernel: [45160.926433] ath10k_pci 0000:01:00.0: firmware crashed! (guid 6578c3ae-9001-40b8-901f-6688268d259b)
Jan 29 11:02:01 SchwichtenburgWrt kernel: [45880.038305] ath10k_pci 0000:01:00.0: firmware crashed! (guid f5723041-dad7-4b65-9d6d-a1d13bc17784)
Jan 29 11:04:18 SchwichtenburgWrt kernel: [46017.413187] ath10k_pci 0000:01:00.0: firmware crashed! (guid b37f045f-1c51-43e1-a319-4f511cb6ffbd)
Jan 29 11:08:57 SchwichtenburgWrt kernel: [46296.523462] ath10k_pci 0000:01:00.0: firmware crashed! (guid ce1cd5af-2af1-462e-98ef-76e9dd94c73c)
Jan 29 11:17:20 SchwichtenburgWrt kernel: [46798.714282] ath10k_pci 0000:01:00.0: firmware crashed! (guid 1e38cf26-9a34-4f89-9309-b318722cb532)
Jan 29 11:26:09 SchwichtenburgWrt kernel: [47327.797854] ath10k_pci 0000:01:00.0: firmware crashed! (guid 6f393c73-2ef1-49a6-a363-6351c5e51717)
Jan 29 11:38:54 SchwichtenburgWrt kernel: [48092.872762] ath10k_pci 0000:01:00.0: firmware crashed! (guid eed927fc-93d9-478a-b256-d6cad96239f5)
Jan 29 11:39:13 SchwichtenburgWrt kernel: [48112.591654] ath10k_pci 0000:01:00.0: firmware crashed! (guid c0575022-5a0f-4b52-ad52-5952d8bf48a9)
Jan 29 11:54:53 SchwichtenburgWrt kernel: [49052.004720] ath10k_pci 0000:01:00.0: firmware crashed! (guid bdb7db96-b2c0-4f06-bfdb-83540be06b82)
Jan 29 11:57:56 SchwichtenburgWrt kernel: [49235.678175] ath10k_pci 0000:01:00.0: firmware crashed! (guid c3554afd-5f29-424d-90a6-0743e57155e8)
Jan 29 12:01:31 SchwichtenburgWrt kernel: [49450.624954] ath10k_pci 0000:01:00.0: firmware crashed! (guid 091c764f-4f27-420e-a294-959fdaaa1b0d)
Jan 29 12:31:54 SchwichtenburgWrt kernel: [51273.645093] ath10k_pci 0000:01:00.0: firmware crashed! (guid f83c25ed-8c6c-4e3b-ae9f-eb00e62074e6)
Jan 29 12:34:48 SchwichtenburgWrt kernel: [51447.152817] ath10k_pci 0000:01:00.0: firmware crashed! (guid 8a458adc-c76c-49e1-92a9-2b4c4a17c8a7)
Jan 29 12:46:13 SchwichtenburgWrt kernel: [52132.954806] ath10k_pci 0000:01:00.0: firmware crashed! (guid dfc6765a-b112-4367-a43f-32b9fe13e7d1)
Jan 29 12:46:25 SchwichtenburgWrt kernel: [52144.924627] ath10k_pci 0000:01:00.0: firmware crashed! (guid e25daf1a-8086-40da-9d8b-87dff84d8ba2)
Jan 29 12:47:00 SchwichtenburgWrt kernel: [52179.255824] ath10k_pci 0000:01:00.0: firmware crashed! (guid 51084c78-f145-4ca5-96bb-ed600e707d1c)
```
[system-SchwichtenburgWrt.log.zip](https://github.com/greearb/ath10k-ct/files/2807726/system-SchwichtenburgWrt.log.zip)

greearb, 2019-01-29:
sheterx, seems several different but related rate-ctrl bugs.  I think I fixed one that managed to access unitialized rate-ctrl related memory in certain racy situations.  Looks like there are others, hopefully the logs from schwichtgit will let me make more progress.

shelterx, 2019-01-29:
Well, seems like my days of no crashes are gone, here ya go. :) Forgot which firmware this is but it's from this bug report anyway.
[boom.txt](https://github.com/greearb/ath10k-ct/files/2809355/boom.txt)

schwichtgit, 2019-01-30:
as my WIFI AP reputation is suffering on my test audience .. I have been using
```
driver: ath10k_pci
version: 4.14.93
firmware-version: 10.4-3.9.0.2-00021
```
for the last 8 hours in identical (home) scenario w/o any crash
(https://github.com/kvalo/ath10k-firmware/tree/master/QCA9984/hw1.0/3.9.0.2)
it seems the 802.11 firmware/ device driver scenarios are very complex. is there a way we can optimize the triaging on this?

schwichtgit, 2019-01-30:
nevertheless I put it on one out of 4 APs at work
```
driver: ath10k_pci
version: 4.14.93
firmware-version: 10.4b-ct-9984-fHB-012-ae62cf394
```

```
ath10k_pci 0000:01:00.0: firmware crashed! (guid e81a8f64-672f-47b7-9ae0-eb088d3ef72e)
```
[LEDE-10-0-8-7.dmesg.log.zip](https://github.com/greearb/ath10k-ct/files/2810393/LEDE-10-0-8-7.dmesg.log.zip)

schwichtgit, 2019-01-30:
[system-LEDE-10-0-8-x.zip](https://github.com/greearb/ath10k-ct/files/2810443/system-LEDE-10-0-8-x.zip)

greearb, 2019-01-30:
I think only logs from the latest binaries will help me at this point.  And, based on clues from previous reported crashes, I have some ideas how I might could reproduce this.  But, if you all can do the testing and bug reports, it frees up time for me to work on the code.  The last crash from ae62cf394 looks new, but probably related to some changes I made recently.  Probably a null-pointer dereference.  I added an assert to catch it, which should give me a better stack trace if that is the issue.

[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2810887/firmware-5-full-htt-mgt-community.bin.gz)

shelterx, 2019-01-30:
That image crashed right on boot for me.
[boom-again.txt](https://github.com/greearb/ath10k-ct/files/2812023/boom-again.txt)

greearb, 2019-01-30:
Thanks, I believe I found the problem in the previous commit.  See if this one works better:

[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2812611/firmware-5-full-htt-mgt-community.bin.gz)

[21 more replies]

closed 2019-02-01 by schwichtgit