greearb/ath10k-ct #60: Zyxel Z2 ath10k-ct crashes

msoltyspl, 2019-01-10:
Adding issue as requested on openwrt forums:

This is freshly bought ZyXEL NBG6817 running on yesterday's openwrt snapshot:

```
OpenWrt SNAPSHOT, r9010-adc8b374e3
Linux hs14 4.14.91 #0 SMP Tue Jan 8 08:17:11 2019 armv7l GNU/Linux

driver: ath10k_pci
version: 4.14.91
firmware-version: 10.4b-ct-9984-fW-012-81e1edd54
expansion-rom-version: 
bus-info: 0000:01:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```

Basically the driver is reporting crashes every few hours or so. With default debug level, it looks this way:
```
Jan 10 11:17:35 hs14 kernel: [60769.690442] ath10k_pci 0001:01:00.0: firmware crashed! (guid 3504e360-299d-49a8-ba82-da5d44232c49)
Jan 10 11:17:35 hs14 kernel: [60769.690504] ath10k_pci 0001:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
Jan 10 11:17:35 hs14 kernel: [60769.698307] ath10k_pci 0001:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
Jan 10 11:17:35 hs14 kernel: [60769.710559] ath10k_pci 0001:01:00.0: firmware ver 10.4b-ct-9984-fW-012-81e1edd54 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT crc32 0391c067
Jan 10 11:17:35 hs14 kernel: [60769.719645] ath10k_pci 0001:01:00.0: board_file api 2 bmi_id 0:2 crc32 cf58c3bc
Jan 10 11:17:35 hs14 kernel: [60769.741353] ath10k_pci 0001:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
Jan 10 11:17:35 hs14 kernel: [60769.750423] ath10k_pci 0001:01:00.0: firmware register dump:
Jan 10 11:17:35 hs14 kernel: [60769.758456] ath10k_pci 0001:01:00.0: [00]: 0x0000000A 0x000015B3 0x009A582B 0x00975B31
Jan 10 11:17:35 hs14 kernel: [60769.764361] ath10k_pci 0001:01:00.0: [04]: 0x009A582B 0x00060B30 0x00000000 0x00406570
Jan 10 11:17:35 hs14 kernel: [60769.772087] ath10k_pci 0001:01:00.0: [08]: 0x00000000 0x0046187C 0x00406764 0x00000000
Jan 10 11:17:35 hs14 kernel: [60769.779987] ath10k_pci 0001:01:00.0: [12]: 0x00000009 0x00000000 0x00973ABC 0x00973AD2
Jan 10 11:17:35 hs14 kernel: [60769.787886] ath10k_pci 0001:01:00.0: [16]: 0x00973AB0 0x009BACF5 0x009606CA 0x00000000
Jan 10 11:17:35 hs14 kernel: [60769.795786] ath10k_pci 0001:01:00.0: [20]: 0x409A582B 0x0040660C 0x00000004 0x000000FF
Jan 10 11:17:35 hs14 kernel: [60769.803685] ath10k_pci 0001:01:00.0: [24]: 0x809A698A 0x0040666C 0x00000004 0xC09A582B
Jan 10 11:17:35 hs14 kernel: [60769.811584] ath10k_pci 0001:01:00.0: [28]: 0x809A7EF9 0x0040669C 0x0045EF2C 0x00461280
Jan 10 11:17:35 hs14 kernel: [60769.819485] ath10k_pci 0001:01:00.0: [32]: 0x809A8D32 0x0040672C 0x004612BC 0x00406850
Jan 10 11:17:35 hs14 kernel: [60769.827383] ath10k_pci 0001:01:00.0: [36]: 0x809A8D62 0x0040684C 0x0042D96C 0x0042EF70
Jan 10 11:17:35 hs14 kernel: [60769.835283] ath10k_pci 0001:01:00.0: [40]: 0x80985E4B 0x0040689C 0x00000000 0x0045A57C
Jan 10 11:17:35 hs14 kernel: [60769.843183] ath10k_pci 0001:01:00.0: [44]: 0x80994A73 0x004068BC 0x0042ED20 0x0045A57C
Jan 10 11:17:35 hs14 kernel: [60769.851080] ath10k_pci 0001:01:00.0: [48]: 0x8098FC30 0x004068DC 0x0042ED20 0x00000000
Jan 10 11:17:35 hs14 kernel: [60769.858980] ath10k_pci 0001:01:00.0: [52]: 0x80963AD3 0x00406A7C 0x0042ED20 0x0098FC28
Jan 10 11:17:35 hs14 kernel: [60769.866879] ath10k_pci 0001:01:00.0: [56]: 0x80960E80 0x00406A9C 0x0000001F 0x00400000
Jan 10 11:17:35 hs14 kernel: [60769.874777] ath10k_pci 0001:01:00.0: Copy Engine register dump:
Jan 10 11:17:35 hs14 kernel: [60769.882683] ath10k_pci 0001:01:00.0: [00]: 0x0004a000  13  13   3   3
Jan 10 11:17:35 hs14 kernel: [60769.888498] ath10k_pci 0001:01:00.0: [01]: 0x0004a400  19  19 345 346
Jan 10 11:17:35 hs14 kernel: [60769.895096] ath10k_pci 0001:01:00.0: [02]: 0x0004a800  35  35  98  99
Jan 10 11:17:35 hs14 kernel: [60769.901518] ath10k_pci 0001:01:00.0: [03]: 0x0004ac00  22  22  24  22
Jan 10 11:17:35 hs14 kernel: [60769.907943] ath10k_pci 0001:01:00.0: [04]: 0x0004b000 5965 5965  67  28
Jan 10 11:17:35 hs14 kernel: [60769.914365] ath10k_pci 0001:01:00.0: [05]: 0x0004b400  31  31 350 351
Jan 10 11:17:35 hs14 kernel: [60769.920790] ath10k_pci 0001:01:00.0: [06]: 0x0004b800   4   4   4   4
Jan 10 11:17:35 hs14 kernel: [60769.927387] ath10k_pci 0001:01:00.0: [07]: 0x0004bc00   1   1   1   1
Jan 10 11:17:35 hs14 kernel: [60769.933821] ath10k_pci 0001:01:00.0: [08]: 0x0004c000   0   0 127   0
Jan 10 11:17:35 hs14 kernel: [60769.940234] ath10k_pci 0001:01:00.0: [09]: 0x0004c400   0   0   0   0
Jan 10 11:17:35 hs14 kernel: [60769.946658] ath10k_pci 0001:01:00.0: [10]: 0x0004c800   0   0   0   0
Jan 10 11:17:35 hs14 kernel: [60769.953131] ath10k_pci 0001:01:00.0: [11]: 0x0004cc00   0   0   0   0
Jan 10 11:17:35 hs14 kernel: [60769.961523] ath10k_pci 0001:01:00.0: debug log header, dbuf: 0x423818  dropped: 0
Jan 10 11:17:35 hs14 kernel: [60769.966939] ath10k_pci 0001:01:00.0: [0] next: 0x423800 buf: 0x419610 sz: 1500 len: 28 count: 1 free: 0
Jan 10 11:17:35 hs14 kernel: [60769.974406] ath10k_pci 0001:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
Jan 10 11:17:35 hs14 kernel: [60769.982589] ath10k: [0000]: 001DA3ED 17FC0001 009A582B 000015B3 000015B3 004064FC 91104569
Jan 10 11:17:35 hs14 kernel: [60769.988839] ath10k_pci 0001:01:00.0: ATH10K_END
Jan 10 11:17:35 hs14 kernel: [60769.998096] ath10k_pci 0001:01:00.0: [1] next: 0x423818 buf: 0x419020 sz: 1500 len: 0 count: 0 free: 0
Jan 10 11:17:35 hs14 kernel: [60770.041112] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 1, skipped old beacon
Jan 10 11:17:35 hs14 kernel: [60770.041167] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Jan 10 11:17:35 hs14 kernel: [60770.054006] ath10k_pci 0001:01:00.0: removing peer, cleanup-all, deleting: peer dc133000 vdev: 0 addr: f8:59:71:30:67:bd
Jan 10 11:17:35 hs14 kernel: [60770.054528] ath10k_pci 0001:01:00.0: removing peer, cleanup-all, deleting: peer db1dc400 vdev: 0 addr: 88:d7:f6:9f:51:10
Jan 10 11:17:35 hs14 kernel: [60770.065710] ath10k_pci 0001:01:00.0: removing peer, cleanup-all, deleting: peer db1dd600 vdev: 0 addr: 84:c7:ea:8f:41:c7
Jan 10 11:17:35 hs14 kernel: [60770.076655] ath10k_pci 0001:01:00.0: removing peer, cleanup-all, deleting: peer dbd42e00 vdev: 0 addr: 64:a2:f9:31:0f:98
Jan 10 11:17:35 hs14 kernel: [60770.087578] ath10k_pci 0001:01:00.0: removing peer, cleanup-all, deleting: peer dbd42a00 vdev: 0 addr: 48:45:20:c2:27:84
[... 94 lines cut ...]
Jan 10 11:17:41 hs14 kernel: [60776.321315] ath10k_pci 0001:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
Jan 10 11:17:41 hs14 kernel: [60776.322142] ath10k_pci 0001:01:00.0: wmi print 'free: 87020 iram: 26788 sram: 18240'
Jan 10 11:17:41 hs14 kernel: [60776.397285] ath10k_pci 0001:01:00.0: failed to transmit packet, dropping: -19
Jan 10 11:17:41 hs14 kernel: [60776.397333] ath10k_pci 0001:01:00.0: failed to submit frame: -19
Jan 10 11:17:41 hs14 kernel: [60776.403419] ath10k_pci 0001:01:00.0: failed to push frame: -19
Jan 10 11:17:41 hs14 kernel: [60776.413594] ath10k_pci 0001:01:00.0: failed to transmit packet, dropping: -19
Jan 10 11:17:41 hs14 kernel: [60776.415131] ath10k_pci 0001:01:00.0: failed to submit frame: -19
Jan 10 11:17:41 hs14 kernel: [60776.422406] ath10k_pci 0001:01:00.0: failed to push frame: -19
Jan 10 11:17:41 hs14 kernel: [60776.429235] ath10k_pci 0001:01:00.0: failed to transmit packet, dropping: -19
Jan 10 11:17:41 hs14 kernel: [60776.434058] ath10k_pci 0001:01:00.0: failed to submit frame: -19
Jan 10 11:17:41 hs14 kernel: [60776.441331] ath10k_pci 0001:01:00.0: failed to push frame: -19
Jan 10 11:17:41 hs14 kernel: [60776.447565] ath10k_pci 0001:01:00.0: failed to transmit packet, dropping: -19
Jan 10 11:17:41 hs14 kernel: [60776.453053] ath10k_pci 0001:01:00.0: failed to submit frame: -19
Jan 10 11:17:41 hs14 kernel: [60776.460255] ath10k_pci 0001:01:00.0: failed to push frame: -19
Jan 10 11:17:41 hs14 kernel: [60776.466363] ath10k_pci 0001:01:00.0: failed to submit frame: -19
Jan 10 11:17:41 hs14 kernel: [60776.471981] ath10k_pci 0001:01:00.0: failed to push frame: -19
Jan 10 11:17:41 hs14 kernel: [60776.543289] ath10k_pci 0001:01:00.0: failed to submit frame: -19
Jan 10 11:17:41 hs14 kernel: [60776.543333] ath10k_pci 0001:01:00.0: failed to push frame: -19
Jan 10 11:17:41 hs14 kernel: [60776.548657] ath10k_pci 0001:01:00.0: failed to submit frame: -19
Jan 10 11:17:41 hs14 kernel: [60776.554015] ath10k_pci 0001:01:00.0: failed to push frame: -19
Jan 10 11:17:41 hs14 kernel: [60776.778568] ath10k_pci 0001:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
Jan 10 11:17:42 hs14 kernel: [60777.133697] ath10k_pci 0001:01:00.0: device successfully recovered
```

This is the default debug_level, for now I've set 0xffffffff and I'm waiting for another one.

greearb, 2019-01-10:
This is an assert in the rate-ctrl logic.  Please try the attached binary, it adds some debugging in the code that was asserting.  This is unlikely to fix the bug, but hopefully the new crash info will provide more clues to the root cause of this issue.  For debug_level, I suggest this:  0xc0000020

Please note, this binary is on top of work to try to fix bug 54, which has been full of regressions.  Let me know how it goes...

[firmware-5-full-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2746409/firmware-5-full-community.bin.gz)

msoltyspl, 2019-01-14:
Short update: so far this firmware has refused to crash. On the recommended debug_level, the only suspicious thing I get are those kinds of messages:

```
Jan 14 16:50:57 hs14 kernel: [ 1122.531421] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Jan 14 16:50:57 hs14 kernel: [ 1122.557540] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Jan 14 16:50:57 hs14 kernel: [ 1122.557589] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
```

greearb, 2019-01-14:
Those SWBA overrun warnings are likely not a problem.  Lets leave the bug open a while longer to make sure the crash is really gone...

greearb, 2019-01-24:
Closing this bug, there are at least some remaining rate-ctrl bugs, but will track those in other bugs.

closed 2019-01-24 by greearb

msoltyspl, 2019-01-24:
All right (also - no crashes so far with the firmware you linked earlier).

msoltyspl, 2019-01-28:
Managed to get the crash:

[crash.txt](https://github.com/greearb/ath10k-ct/files/2803085/crash.txt)