greearb/ath10k-ct #61: after few days uptime, wifi becomes very laggy

fishxz, 2019-01-15:
Please provide this info. See this link for more info on how to gather debug
info: http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).
hey, sorry that i have to open another bug... but i noticed, that my wifi gets super laggy after few days of uptime. maybe this is related to some other bug?

Software (OS, Firmware version, kernel, driver, etc)
OpenWrt SNAPSHOT r8917-ae622c93b3 / LuCI Master (git-18.361.48411-39aa247)
firmware ver 10.4b-ct-4019-fW-012-81e1edd5
kernel 4.14.90

Hardware (NIC chipset, platform, etc)
Zyxel nbg6617 IPQ4018

Logs (dmesg, maybe supplicant and/or hostap)
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.90 (buildbot@buildslave) (gcc version 7.4.0 (OpenWrt GCC 7.4.0 r8917-ae622c93b3)) #0 SMP Fri Dec 28 22:09:38 2018
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: ZyXEL NBG6617
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 65024
[    0.000000] free_area_init_node: node 0, pgdat c0a23e40, node_mem_map cfdf9000
[    0.000000]   Normal zone: 512 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65024 pages, LIFO batch:15
[    0.000000] random: get_random_bytes called from start_kernel+0x88/0x3c4 with crng_init=0
[    0.000000] percpu: Embedded 15 pages/cpu @cfdaf000 s29388 r8192 d23860 u61440
[    0.000000] pcpu-alloc: s29388 r8192 d23860 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64512
[    0.000000] Kernel command line: board=NBG6617 root=mtd:rootfs zld_ver=2.02 mtdparts=spi0.0:0x60000@0x60000(0:QSEE)ro,0x80000@0xE0000(u-boot)ro,0x10000(env)ro,0x10000(0:ART)ro,0x400000(HLOS),0x10000(dualflag),0x10000(header),0x100000(romd),0x100000(rootfs_data),0x1860000(rootfs) mtdparts= root=31:13
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Memory: 249928K/260096K available (4369K kernel code, 145K rwdata, 1200K rodata, 1024K init, 235K bss, 10168K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xd0800000 - 0xff800000   ( 752 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xd0000000   ( 256 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0208000 - 0xc0744638   (5362 kB)
[    0.000000]       .init : 0xc0900000 - 0xc0a00000   (1024 kB)
[    0.000000]       .data : 0xc0a00000 - 0xc0a24740   ( 146 kB)
[    0.000000]        .bss : 0xc0a26000 - 0xc0a60c98   ( 236 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
[    0.000008] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
[    0.000021] Switching to timer-based delay loop, resolution 20ns
[    0.000263] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
[    0.000282] pid_max: default: 32768 minimum: 301
[    0.000422] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000440] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.001002] CPU: Testing write buffer coherency: ok
[    0.001690] Setting up static identity map for 0x80300000 - 0x80300060
[    0.001836] Hierarchical SRCU implementation.
[    0.002490] smp: Bringing up secondary CPUs ...
[    0.005215] smp: Brought up 1 node, 4 CPUs
[    0.005236] SMP: Total of 4 processors activated (384.00 BogoMIPS).
[    0.005244] CPU: All CPU(s) started in SVC mode.
[    0.009650] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.009800] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.009825] futex hash table entries: 1024 (order: 4, 65536 bytes)
[... 237 lines cut ...]
[   26.209255] br-lan: port 2(wlan0) entered blocking state
[   26.209794] br-lan: port 2(wlan0) entered forwarding state
[   26.657270] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   26.657643] br-lan: port 3(wlan1) entered blocking state
[   26.662807] br-lan: port 3(wlan1) entered forwarding state
[   49.684245] ath10k_ahb a000000.wifi: Invalid VHT mcs 15 peer stats
[   55.581135] ath10k_ahb a000000.wifi: Invalid peer id 4 or peer stats buffer, peer:   (null)  sta:   (null)
[508049.385628] device wlan1 left promiscuous mode
[508049.387260] br-lan: port 3(wlan1) entered disabled state
[508049.486367] ath10k_ahb a800000.wifi: peer-unmap-event: unknown peer id 0
[508049.486715] ath10k_ahb a800000.wifi: peer-unmap-event: unknown peer id 0
[508049.492432] ath10k_ahb a800000.wifi: peer-unmap-event: unknown peer id 0
[508051.546699] ath10k_ahb a800000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[508051.546753] ath10k_ahb a800000.wifi: msdu-desc: 2500  skid: 32
[508051.594091] ath10k_ahb a800000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[508051.597766] ath10k_ahb a800000.wifi: wmi print 'free: 58060 iram: 22488 sram: 35968'
[508051.897259] ath10k_ahb a800000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[508051.897530] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[508051.912048] br-lan: port 3(wlan1) entered blocking state
[508051.913459] br-lan: port 3(wlan1) entered disabled state
[508051.919209] device wlan1 entered promiscuous mode
[508052.615001] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[508052.615439] br-lan: port 3(wlan1) entered blocking state
[508052.620570] br-lan: port 3(wlan1) entered forwarding state
```

greearb, 2019-01-16:
If you use stock QCA firmware, do you see the same problem?

If you disconnect and reconnect the station device(s), does the problem remain?

There are no obvious errors in the logs.

fishxz, 2019-01-16:
what u mean by stock qca firmware? the firmware from https://github.com/kvalo/ath10k-firmware/tree/master/QCA4019/hw1.0/3.6?

if i restart the interface interface, the problem is instant solved.

```
Ping wird ausgeführt für openwrt.lan [192.168.1.1] mit 32 Bytes Daten:
Antwort von 192.168.1.1: Bytes=32 Zeit=1ms TTL=64
Zeitüberschreitung der Anforderung.
Antwort von 192.168.1.1: Bytes=32 Zeit=2ms TTL=64
Antwort von 192.168.1.1: Bytes=32 Zeit=2ms TTL=64
Antwort von 192.168.1.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.1.1: Bytes=32 Zeit=5ms TTL=64
Antwort von 192.168.1.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.1.1: Bytes=32 Zeit=3ms TTL=64
Antwort von 192.168.1.1: Bytes=32 Zeit=1ms TTL=64
Zeitüberschreitung der Anforderung.
Antwort von 192.168.1.1: Bytes=32 Zeit=1ms TTL=64
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Antwort von 192.168.1.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.1.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.1.1: Bytes=32 Zeit=1ms TTL=64`
```
happened again like 1 hour ago.

greearb, 2019-01-16:
Yes, like that firmware from kvalo (or the QCA one packaged in openwrt).

Can you show me ping time for when it is messing up?  And, if you do a larger download, will it work OK (ie, is it just ping that is weird?)

fishxz, 2019-01-16:
this is the ping when it is messing up. i will check next time a download, but i can say discord lags like hell then, same for counter-strike. it is like im just dropping alot of pakages.

fishxz, 2019-01-27:
i tested for a while the non -ct pakage and i couldnt reproduce it, so i switched back to -ct, to see if it will happen again...

i also noticed a another problem: its not possible to run 2 1080p streams over wlan in my network. as soon the second stream starts, both will start to buffer frequently. (ct and non -ct)

fishxz, 2019-02-03:
just happened again... after wifi restart all is fine again

vamanea, 2019-02-03:
Just wanted to add my +1 for this problem.

I have exactly the same router and faced pretty much the same problems with the -ct firmwares. I downloaded the ones from candelatech.com as well but the problems described above persisted. Once I downloaded and started using the non -ct firmware everything started working properly again, no timeouts, no need to restart wifi, and my ESPxxx devices did not randomly disconnect.

For reference this is the firmware I'm currently using:
```
[   12.566301] ath10k_ahb a000000.wifi: firmware ver 10.4-3.6-00140 api 5 features no-p2p,mfp,peer-flow-ctrl,btcoex-param,allows-mesh-bcast,no-ps crc32 ba79b746
[   12.768511] ath10k_ahb a000000.wifi: board_file api 2 bmi_id 0:16 crc32 bcebe54c
[   14.139131] ath10k_ahb a000000.wifi: 10.4 wmi init: vdevs: 16  peers: 528  tid: 102
[   14.139185] ath10k_ahb a000000.wifi: msdu-desc: 2500  skid: 32
[   14.189408] ath10k_ahb a000000.wifi: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 512 raw 0 hwcrypto 1
```

greearb, 2019-02-05:
Please describe how your systems are configured.  Can you reproduce it if you disable encryption, for instance?  Are you using PMF/MFP?  Is it only certain devices that have this connection problem, or do all of them exhibit the same behaviour?

Are you able to take an RF capture in the case where the problem is happening?

vamanea, 2019-02-11:
@greearb I can't speak for OP but for my cases:

* tried with both PMF enabled and disabled, did not fix or change the problem
* some devices like ESP32 based IoT cheap stuff can't stay connected more than 30 minutes, they just disconnect
* some USB Wlan dongles 8192cu didn't want to connect at all!
* but both the above seemed to get fixed when I switched to the firmware from candelatech.com(the for personal use only firmware)
* some devices like laptop with bcm wlan running Linux didn't seem to be affected
* most mobile devices(eg android) with Qualcomm or Broadcom WLAN were affected. In the morning after long time sleeping(the devices) were quite laggy. Running a ping seemed to loose the first couple of packets in a consistent matter.

I am able to take a RF capture, is it something specific I'm looking for?

greearb, 2019-03-25:
I just uploaded a new beta firmware that fixes a problem that could cause tx-scheduling problems in the firmware.  Maybe retest with that most recent firmware to see if it works any better in your case?

vamanea, 2019-03-26:
@greearb I suppose I need to update the driver too, the current driver I have is spamming my console with this:
`ath10k_ahb a800000.wifi: Unknown (10.4) eventid: 40955`

vamanea, 2019-04-22:
Updated to beta ct firmware: 10.4b-ct-4019-fH-012-c886a6af and also updated to last openwrt and it seems much better.

Only problem seem to be some kernel warnings:
`
[175100.198548] WARNING: CPU: 2 PID: 1805 at backports-4.19.32-1/net/wireless/util.c:1146 cfg80211_calculate_bitrate+0x1d8/0x24c [cfg80211]
[175100.202120] invalid rate bw=0, mcs=15, nss=4
[175100.214473] Modules linked in: pppoe ppp_async ath10k_pci ath10k_core ath pppox ppp_generic nf_conntrack_ipv6 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT wireguard slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables hwmon crc_ccitt compat ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ip6_udp_tunnel udp_tunnel leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_of_simple gpio_button_hotplug
[175100.268563] CPU: 2 PID: 1805 Comm: hostapd Not tainted 4.14.111 #0
[175100.290040] Hardware name: Generic DT based system
[175100.296055] [<c030e45c>] (unwind_backtrace) from [<c030a968>] (show_stack+0x10/0x14)
[175100.300909] [<c030a968>] (show_stack) from [<c07393c8>] (dump_stack+0x88/0x9c)
[175100.308892] [<c07393c8>] (dump_stack) from [<c0320e20>] (__warn+0xe4/0x100)
[175100.316250] [<c0320e20>] (__warn) from [<c0320e74>] (warn_slowpath_fmt+0x38/0x48)
[175100.325032] [<c0320e74>] (warn_slowpath_fmt) from [<bf20072c>] (cfg80211_calculate_bitrate+0x1d8/0x24c [cfg80211])
[175100.331855] [<bf20072c>] (cfg80211_calculate_bitrate [cfg80211]) from [<bf2287e4>] (nl80211_put_sta_rate+0x48/0x2e4 [cfg80211])
[175100.341359] [<bf2287e4>] (nl80211_put_sta_rate [cfg80211]) from [<bf212328>] (nl80211_update_connect_params+0x794/0x1034 [cfg80211])
[175100.353144] [<bf212328>] (nl80211_update_connect_params [cfg80211]) from [<bf212d68>] (cfg80211_del_sta_sinfo+0x98/0x188 [cfg80211])
[175100.365197] [<bf212d68>] (cfg80211_del_sta_sinfo [cfg80211]) from [<bf25df20>] (sta_set_sinfo+0xc54/0xcb0 [mac80211])
[175100.378521] [<bf25df20>] (sta_set_sinfo [mac80211]) from [<bf25df9c>] (__sta_info_destroy+0x20/0x28 [mac80211])
[175100.388940] [<bf25df9c>] (__sta_info_destroy [mac80211]) from [<bf25e00c>] (sta_info_destroy_addr_bss+0x28/0x40 [mac80211])
[175100.399090] [<bf25e00c>] (sta_info_destroy_addr_bss [mac80211]) from [<bf20a4f8>] (nl80211_del_station+0xe8/0xf8 [cfg80211])
[175100.409534] [<bf20a4f8>] (nl80211_del_station [cfg80211]) from [<c066b620>] (genl_rcv_msg+0x2e8/0x388)
[175100.420352] [<c066b620>] (genl_rcv_msg) from [<c066a800>] (netlink_rcv_skb+0x8c/0xfc)
[175100.429537] [<c066a800>] (netlink_rcv_skb) from [<c066b328>] (genl_rcv+0x24/0x34)
[175100.437514] [<c066b328>] (genl_rcv) from [<c0669fd4>] (netlink_unicast+0x158/0x208)
[175100.445056] [<c0669fd4>] (netlink_unicast) from [<c066a478>] (netlink_sendmsg+0x334/0x37c)
[175100.452984] [<c066a478>] (netlink_sendmsg) from [<c0619b64>] (sock_sendmsg+0x14/0x24)
[175100.461049] [<c0619b64>] (sock_sendmsg) from [<c061a098>] (___sys_sendmsg+0x204/0x22c)
[175100.469041] [<c061a098>] (___sys_sendmsg) from [<c061aea0>] (__sys_sendmsg+0x40/0x64)
[175100.476960] [<c061aea0>] (__sys_sendmsg) from [<c0307680>] (ret_fast_syscall+0x0/0x54)
[175100.486713] ---[ end trace 60ecb48db228e521 ]---
`