greearb/ath10k-ct #62: driver crash on startup

jekkos, 2019-01-18:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).
use latest openwrt ipq40xx build for bl1300 gl.inet

Software (OS, Firmware version, kernel, driver, etc)
OpenWrt snapshot build, 4.14.90, kmod-ath10k-4.14.90+2018-12-20-118e16da-1

Hardware (NIC chipset, platform, etc)
Gl.Inet B1300 device

Logs (dmesg, maybe supplicant and/or hostap)
`''
[   23.901758] WARNING: CPU: 0 PID: 1577 at /builder/ipq40xx_generic/build/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-2018-12-20-118e16da/ath10k-4.19/htt_rx.c:903 ath10k_htt_t2h_ms
g_handler+0xf00/0x2524 [ath10k_core]                                                                                                                                                                                 
[   23.905298] Modules linked in: rtl8192cu rtl8192c_common rtl_usb pppoe ppp_async ath10k_pci ath10k_core ath rtlwifi pppox ppp_generic nf_conntrack_ipv6 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt
_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrac
k_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables hwmon crc_ccitt compat ip6t_REJECT nf_re
ject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_of_simple gpio_button_hotplug                                            
[   23.975200] CPU: 0 PID: 1577 Comm: ip Not tainted 4.14.90 #0                                                                                                                                                      
[   23.997297] Hardware name: Generic DT based system                                                                                                                                                                
[   24.003074] [<c030e774>] (unwind_backtrace) from [<c030ac9c>] (show_stack+0x10/0x14)                                                                                                                              
[   24.007653] [<c030ac9c>] (show_stack) from [<c072bb68>] (dump_stack+0x88/0x9c)                                                                                                                                    
[   24.015575] [<c072bb68>] (dump_stack) from [<c0317448>] (__warn+0xe4/0x100)                                                                                                                                       
[   24.022579] [<c0317448>] (__warn) from [<c0317514>] (warn_slowpath_null+0x20/0x28)                                                                                                                                
[   24.029491] [<c0317514>] (warn_slowpath_null) from [<bf2f41e8>] (ath10k_htt_t2h_msg_handler+0xf00/0x2524 [ath10k_core])                                                                                           
[   24.037159] [<bf2f41e8>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf2f43b8>] (ath10k_htt_t2h_msg_handler+0x10d0/0x2524 [ath10k_core])                                                                    
[   24.047800] [<bf2f43b8>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf2f56bc>] (ath10k_htt_t2h_msg_handler+0x23d4/0x2524 [ath10k_core])                                                                    
[   24.060646] [<bf2f56bc>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf2f5ee4>] (ath10k_htt_txrx_compl_task+0x6cc/0xb78 [ath10k_core])                                                                      
[   24.073387] [<bf2f5ee4>] (ath10k_htt_txrx_compl_task [ath10k_core]) from [<bf3452d4>] (ath10k_pci_napi_poll+0x70/0x100 [ath10k_pci]) 
```

jekkos, 2019-01-18:
The device does survive this first crash, but eventually also locks up. This does not happen with the openwrt stable release that uses regular ath10k driver.

jekkos, 2019-01-18:
Also seeing firmware crash 

```
[ 2254.565470] ath10k_ahb a800000.wifi: firmware crashed! (guid 26340abb-03b6-40ab-8483-20bf6bb5cca7)
[ 2254.568010] ath10k_ahb a800000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003900ff sub 0000:0000
[ 2254.576930] ath10k_ahb a800000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[ 2254.589541] ath10k_ahb a800000.wifi: firmware ver 10.4b-ct-4019-fW-012-81e1edd5 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT crc32 adeedaac
[ 2254.599740] ath10k_ahb a800000.wifi: board_file api 2 bmi_id 0:17 crc32 bcebe54c
[ 2254.619223] ath10k_ahb a800000.wifi: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[ 2254.627587] ath10k_ahb a800000.wifi: firmware register dump:
[ 2254.636399] ath10k_ahb a800000.wifi: [00]: 0x0000000B 0x000015B3 0x00962B2C 0x00975B31
[ 2254.642194] ath10k_ahb a800000.wifi: [04]: 0x00962B2C 0x00060134 0x0000001B 0x00000000
[ 2254.649983] ath10k_ahb a800000.wifi: [08]: 0x00000001 0x00962AF4 0x00000000 0x00401CD0
[ 2254.657818] ath10k_ahb a800000.wifi: [12]: 0x00000009 0x00000000 0x00971518 0x00971523
[ 2254.665759] ath10k_ahb a800000.wifi: [16]: 0x00971238 0x0096FA45 0x0096317D 0x0096D254
[ 2254.673610] ath10k_ahb a800000.wifi: [20]: 0x40962B2C 0x00401BF0 0x00975A00 0x0000001F
[ 2254.681488] ath10k_ahb a800000.wifi: [24]: 0x80963530 0x00401C50 0x00000000 0xC0962B2C
[ 2254.689440] ath10k_ahb a800000.wifi: [28]: 0x809C2214 0x00401C70 0x0000001E 0x004022B0
[ 2254.697392] ath10k_ahb a800000.wifi: [32]: 0x809608AD 0x00401C90 0x0000001E 0x00400000
[ 2254.705207] ath10k_ahb a800000.wifi: [36]: 0x409605E1 0x00401CB0 0x0000001E 0x0000001E
[ 2254.713155] ath10k_ahb a800000.wifi: [40]: 0x00000000 0x00401CD0 0x00050024 0x0000001F
[ 2254.721026] ath10k_ahb a800000.wifi: [44]: 0x00418600 0x0000001F 0x0000001E 0x000FFFFF
[ 2254.728927] ath10k_ahb a800000.wifi: [48]: 0x00000000 0x00401CD0 0x00050024 0x0000001F
[ 2254.736807] ath10k_ahb a800000.wifi: [52]: 0x0000001F 0x00405694 0x00405698 0x0040569C
[ 2254.744730] ath10k_ahb a800000.wifi: [56]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 2254.752615] ath10k_ahb a800000.wifi: Copy Engine register dump:
[ 2254.760541] ath10k_ahb a800000.wifi: [00]: 0x0004a000   4   4   3   3
[ 2254.766357] ath10k_ahb a800000.wifi: [01]: 0x0004a400   7   7 491 494
[ 2254.772892] ath10k_ahb a800000.wifi: [02]: 0x0004a800  35  35  64  99
[ 2254.779338] ath10k_ahb a800000.wifi: [03]: 0x0004ac00   6   6   8   6
[ 2254.785792] ath10k_ahb a800000.wifi: [04]: 0x0004b000 7770 7770 129  89
[ 2254.792196] ath10k_ahb a800000.wifi: [05]: 0x0004b400  30  30  62  94
[ 2254.798652] ath10k_ahb a800000.wifi: [06]: 0x0004b800  20  20  20  20
[ 2254.805280] ath10k_ahb a800000.wifi: [07]: 0x0004bc00   0   0   0   0
[ 2254.811606] ath10k_ahb a800000.wifi: [08]: 0x0004c000   0   0 127   0
[ 2254.818053] ath10k_ahb a800000.wifi: [09]: 0x0004c400   0   0   0   0
[ 2254.824509] ath10k_ahb a800000.wifi: [10]: 0x0004c800   0   0   0   0
[ 2254.830877] ath10k_ahb a800000.wifi: [11]: 0x0004cc00   0   0   0   0
[ 2254.838388] ath10k_ahb a800000.wifi: debug log header, dbuf: 0x4187a0  dropped: 0
[ 2254.844803] ath10k_ahb a800000.wifi: [0] next: 0x4187b8 buf: 0x414bb0 sz: 1500 len: 52 count: 4 free: 0
[ 2254.852206] ath10k_ahb a800000.wifi: ath10k_pci ATH10K_DBG_BUFFER:
```

greearb, 2019-01-18:
This looks like it might be a bug I have already fixed (first part of bug 54).  When reporting crash dumps, please include the text can comes after the ATH10K_DBG_BUFFER: line, it is often very important for understanding the crash.
Please try the binary below and let me know if you see more problems.
[firmware-5-full-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2772892/firmware-5-full-community.bin.gz)

jekkos, 2019-01-19:
also still seeing this trace on startup

```
[   27.865601] ------------[ cut here ]------------
[   27.867022] WARNING: CPU: 0 PID: 27 at /var/lib/buildbot/slaves/slashdirt-03/MAIN/build/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-2018-12-20-118e16da/ath10k-4.19/htt_rx.c:903 ath10k_htt_t2h_msg_handler+0xf00/0x2524 [ath10k_core]
[   27.871265] Modules linked in: rtl8192cu rtl8192c_common rtl_usb pppoe ppp_async ath10k_pci ath10k_core ath rtlwifi pppox ppp_generic nf_conntrack_ipv6 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables hwmon crc_ccitt compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_of_simple gpio_button_hotplug
[   27.942034] CPU: 0 PID: 27 Comm: kworker/0:1 Not tainted 4.14.93 #0
[   27.963302] Hardware name: Generic DT based system
[   27.969426] Workqueue: events dbs_work_handler
[   27.974430] [<c030e774>] (unwind_backtrace) from [<c030ac9c>] (show_stack+0x10/0x14)
[   27.978817] [<c030ac9c>] (show_stack) from [<c072c148>] (dump_stack+0x88/0x9c)
[   27.986700] [<c072c148>] (dump_stack) from [<c03173e4>] (__warn+0xe4/0x100)
[   27.993734] [<c03173e4>] (__warn) from [<c03174b0>] (warn_slowpath_null+0x20/0x28)
[   28.001166] [<c03174b0>] (warn_slowpath_null) from [<bf2f41e8>] (ath10k_htt_t2h_msg_handler+0xf00/0x2524 [ath10k_core])
[   28.009367] [<bf2f41e8>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf2f43b8>] (ath10k_htt_t2h_msg_handler+0x10d0/0x2524 [ath10k_core])
[   28.019858] [<bf2f43b8>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf2f56bc>] (ath10k_htt_t2h_msg_handler+0x23d4/0x2524 [ath10k_core])
[   28.032692] [<bf2f56bc>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf2f5ee4>] (ath10k_htt_txrx_compl_task+0x6cc/0xb78 [ath10k_core])
[   28.045100] [<bf2f5ee4>] (ath10k_htt_txrx_compl_task [ath10k_core]) from [<bf3452d4>] (ath10k_pci_napi_poll+0x70/0x100 [ath10k_pci])
[   28.057484] [<bf3452d4>] (ath10k_pci_napi_poll [ath10k_pci]) from [<c062ae4c>] (net_rx_action+0x138/0x2fc)
[   28.069171] [<c062ae4c>] (net_rx_action) from [<c0301520>] (__do_softirq+0xe0/0x240)
[   28.078646] [<c0301520>] (__do_softirq) from [<c031b35c>] (irq_exit+0xd4/0x138)
[   28.086534] [<c031b35c>] (irq_exit) from [<c0352b24>] (__handle_domain_irq+0x9c/0xac)
[   28.093540] [<c0352b24>] (__handle_domain_irq) from [<c030140c>] (gic_handle_irq+0x5c/0x90)
[   28.101515] [<c030140c>] (gic_handle_irq) from [<c030b80c>] (__irq_svc+0x6c/0x90)
[   28.109592] Exception stack(0xcf8eddb8 to 0xcf8ede00)
[   28.117238] dda0:                                                       80000000 000e2b32
[   28.122364] ddc0: 00070000 00000000 000aece0 c0939128 c0a03d48 00000003 000aece0 cf8eded0
[   28.130523] dde0: 04eca000 00000050 00004000 cf8ede08 c030d4a0 c0729c74 00000013 ffffffff
[   28.138713] [<c030b80c>] (__irq_svc) from [<c0729c74>] (__do_div64+0x70/0xe8)
[   28.146858] [<c0729c74>] (__do_div64) from [<c030d4a0>] (cpufreq_callback+0xf4/0x164)
[   28.153975] [<c030d4a0>] (cpufreq_callback) from [<c0332d24>] (notifier_call_chain+0x2c/0x6c)
[   28.161798] [<c0332d24>] (notifier_call_chain) from [<c0332fa8>] (__srcu_notifier_call_chain+0x44/0x60)
[   28.170308] [<c0332fa8>] (__srcu_notifier_call_chain) from [<c0332fdc>] (srcu_notifier_call_chain+0x18/0x20)
[   28.179527] [<c0332fdc>] (srcu_notifier_call_chain) from [<c05ecb64>] (cpufreq_notify_transition+0xb8/0x120)
[   28.189598] [<c05ecb64>] (cpufreq_notify_transition) from [<c05ecce0>] (cpufreq_freq_transition_begin+0x114/0x11c)
[   28.199406] [<c05ecce0>] (cpufreq_freq_transition_begin) from [<c05ed3d8>] (__cpufreq_driver_target+0x3e8/0x490)
[   28.209576] [<c05ed3d8>] (__cpufreq_driver_target) from [<c05f0efc>] (od_dbs_update+0xdc/0x160)
[   28.219868] [<c05f0efc>] (od_dbs_update) from [<c05f131c>] (dbs_work_handler+0x2c/0x54)
[   28.228269] [<c05f131c>] (dbs_work_handler) from [<c032c49c>] (process_one_work+0x240/0x3d0)
[   28.236272] [<c032c49c>] (process_one_work) from [<c032d438>] (worker_thread+0x330/0x560)
[   28.244967] [<c032d438>] (worker_thread) from [<c0331980>] (kthread+0x12c/0x138)
[   28.253039] [<c0331980>] (kthread) from [<c0307828>] (ret_from_fork+0x14/0x2c)

```

jekkos, 2019-02-05:
The driver seemed more stable with the firmware you posted. But throughput was really bad. I use two b1300 in a mesh setup so went back to official ath10k which gives me 144Mbit in both directions.

greearb, 2019-02-05:
What is the exact version of the firmware that is stable but slow?  Is your mesh using .11s?  Can you do 'iw dev wlanX station dump' to see the tx/rx rates in good and bad case and send me that info?  Can you simplify your test case so that you are not using mesh and see if the slowness still happens?

greearb, 2019-09-16:
I fixed a bunch of bugs since this one was submitted, so closing this one.  Please open new bugs if any problems are found with the latest upstream OpenWRT and ath10k-ct drivers/firmware.

closed 2019-09-16 by greearb