greearb/ath10k-ct #63: qca4019, nbg6617 firmware 10.4b-ct-4019-fW-012-81e1edd5 crashed

ynezz, 2019-01-21:
* running `OpenWrt SNAPSHOT r8997-488af51` from `Sun Jan 6 20:48:11 2019`
* crash log:

```
[11908.836949] ath10k_ahb a000000.wifi: firmware crashed! (guid 47d077ca-bdb4-48c2-abfe-d7795f390cd0)
[11908.837532] ath10k_ahb a000000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[11908.845506] ath10k_ahb a000000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[11908.900627] ath10k_ahb a000000.wifi: firmware ver 10.4b-ct-4019-fW-012-81e1edd5 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT crc32 adeedaac
[11908.940202] ath10k_ahb a000000.wifi: board_file api 2 bmi_id 0:16 crc32 bcebe54c
[11908.940580] ath10k_ahb a000000.wifi: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[11908.947522] ath10k_ahb a000000.wifi: firmware register dump:
[11908.957120] ath10k_ahb a000000.wifi: [00]: 0x0000000B 0x000015B3 0x0099C12B 0x00975B31
[11908.962690] ath10k_ahb a000000.wifi: [04]: 0x0099C12B 0x00060530 0x00000005 0x00000007
[11908.970680] ath10k_ahb a000000.wifi: [08]: 0x0041DEF8 0x004336D4 0x00000000 0x004341C4
[11908.978564] ath10k_ahb a000000.wifi: [12]: 0x00000009 0x00000000 0x009C2B34 0x009C2B38
[11908.986462] ath10k_ahb a000000.wifi: [16]: 0x00971238 0x00962F6B 0x009C50DF 0x00000000
[11908.994115] ath10k_ahb a000000.wifi: [20]: 0x4099C12B 0x0040569C 0x00433694 0x00000000
[11909.002270] ath10k_ahb a000000.wifi: [24]: 0x8099EAD9 0x004056FC 0x00975A00 0xC099C12B
[11909.010164] ath10k_ahb a000000.wifi: [28]: 0x80997A2A 0x0040573C 0x0041DEF8 0x004322AC
[11909.018070] ath10k_ahb a000000.wifi: [32]: 0x80994EFC 0x0040576C 0x00000001 0x0041DEF8
[11909.025927] ath10k_ahb a000000.wifi: [36]: 0x8098F950 0x004057AC 0x0041C50C 0x00446170
[11909.033610] ath10k_ahb a000000.wifi: [40]: 0x809630B5 0x0040599C 0x0041C50C 0x0098F948
[11909.041759] ath10k_ahb a000000.wifi: [44]: 0x80960DF0 0x004059BC 0x0000001F 0x00400000
[11909.049661] ath10k_ahb a000000.wifi: [48]: 0x80960DC2 0x004059EC 0x00400000 0x00000000
[11909.057560] ath10k_ahb a000000.wifi: [52]: 0x80960E0D 0x00405A0C 0x00000000 0x00400600
[11909.065452] ath10k_ahb a000000.wifi: [56]: 0x40960024 0x00405A2C 0x004033D8 0x004033D8
[11909.073069] ath10k_ahb a000000.wifi: Copy Engine register dump:
[11909.081259] ath10k_ahb a000000.wifi: [00]: 0x0004a000   4   4   3   3
[11909.087077] ath10k_ahb a000000.wifi: [01]: 0x0004a400  24  24  94  95
[11909.093411] ath10k_ahb a000000.wifi: [02]: 0x0004a800  27  27  90  91
[11909.100093] ath10k_ahb a000000.wifi: [03]: 0x0004ac00  22  22  23  22
[11909.106518] ath10k_ahb a000000.wifi: [04]: 0x0004b000 901 901 189 150
[11909.112687] ath10k_ahb a000000.wifi: [05]: 0x0004b400  31  31 318 319
[11909.119364] ath10k_ahb a000000.wifi: [06]: 0x0004b800  31  31  31  31
[11909.125780] ath10k_ahb a000000.wifi: [07]: 0x0004bc00   1   1   1   1
[11909.131953] ath10k_ahb a000000.wifi: [08]: 0x0004c000   0   0 127   0
[11909.138634] ath10k_ahb a000000.wifi: [09]: 0x0004c400   1   1   1   1
[11909.144802] ath10k_ahb a000000.wifi: [10]: 0x0004c800   0   0   0   0
[11909.151240] ath10k_ahb a000000.wifi: [11]: 0x0004cc00   0   0   0   0
[11909.158062] ath10k_ahb a000000.wifi: debug log header, dbuf: 0x4187b8  dropped: 0
[11909.164185] ath10k_ahb a000000.wifi: [0] next: 0x4187a0 buf: 0x4151a0 sz: 1500 len: 320 count: 12 free: 0
[11909.171938] ath10k_ahb a000000.wifi: ath10k_pci ATH10K_DBG_BUFFER:
[11909.181346] ath10k: [0000]: 00B9D5A7 1400581D 00000000 004415A8 000F01CC 00000006 00000000 00B9D5A7
[11909.187411] ath10k: [0008]: 1000581B 000051AB 00000000 00000000 000F01CC 00B9D5A8 14006402 71103332
[11909.196349] ath10k: [0016]: 5AAA4B48 0000AB51 0041A3E8 00000004 00B9D5A8 13FC4C07 211000A1 00000879
[11909.205372] ath10k: [0024]: 00000009 004341C4 00B9D5A8 1000587B 0041D66C 004415A8 00000001 0041A3E8
[11909.214152] ath10k: [0032]: 00B9D5A9 1400587C 51100001 000F0218 000003FC 00000008 00441710 00B9D5A9
[11909.223438] ath10k: [0040]: 1000587A 0041D66C 00441710 01000000 00000004 00B9D5A9 14006403 5AAA4B48
[11909.232471] ath10k: [0048]: AB510000 00000004 00000000 00441710 00B9D5AA 17FC587D 51100002 000F01CC
[11909.241489] ath10k: [0056]: 00000001 00000007 004415A8 00B9D5AA 17FC4C07 711050A2 00000879 00000000
[11909.250510] ath10k: [0064]: 0000087A 0000F744 00B9D5AA 17FC4C07 711057A2 00000879 00000000 0000087A
[11909.259536] ath10k: [0072]: 00000001 00B9D5AA 17FC0001 0099C12B 000015B3 000015B3 0040558C 91104569
[11909.268523] ath10k_ahb a000000.wifi: ATH10K_END
[11909.277800] ath10k_ahb a000000.wifi: [1] next: 0x4187b8 buf: 0x414bb0 sz: 1500 len: 0 count: 0 free: 0
[11909.304582] ath10k_ahb a800000.wifi: SWBA overrun on vdev 0, skipped old beacon
[11912.325372] ath10k_ahb a000000.wifi: bss channel survey timed out
[11912.326160] ath10k_ahb a000000.wifi: failed to set beacon mode for vdev 0: -108
[11912.330533] ath10k_ahb a000000.wifi: failed to set dtim period for vdev 0: -108
[11912.337841] ath10k_ahb a000000.wifi: failed to recalculate rts/cts prot for vdev 0: -108
[11912.345011] ath10k_ahb a000000.wifi: failed to set cts protection for vdev 0: -108
[11912.353273] ath10k_ahb a000000.wifi: failed to set erp slot for vdev 0: -108
[11912.360693] ath10k_ahb a000000.wifi: failed to set preamble for vdev 0: -108
[11912.384823] ath10k_ahb a000000.wifi: removing peer, cleanup-all, deleting: peer ce169c00 vdev: 0 addr: 48:4b:aa:5a:51:ab 
[11912.384871] ath10k_ahb a000000.wifi: removing peer, cleanup-all, deleting: peer ce169000 vdev: 0 addr: 08:d4:2b:17:bd:8a 
[11912.394798] ath10k_ahb a000000.wifi: removing peer, cleanup-all, deleting: peer cebbbc00 vdev: 0 addr: 98:fe:94:61:59:47 
[11912.405830] ath10k_ahb a000000.wifi: removing peer, cleanup-all, deleting: peer cebba800 vdev: 0 addr: c4:12:f5:22:16:17 
[11912.416677] ath10k_ahb a000000.wifi: removing peer, cleanup-all, deleting: peer ceedb400 vdev: 0 addr: b8:ec:a3:d8:2c:a0 
[11912.440479] ieee80211 phy0: Hardware restart was requested
[11913.932065] ath10k_ahb a000000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[11913.932111] ath10k_ahb a000000.wifi: msdu-desc: 2500  skid: 32
[... 30 lines cut ...]
[11914.601300] [<c066881c>] (ip_forward) from [<c06673cc>] (ip_rcv+0x210/0x278)
[11914.608334] [<c06673cc>] (ip_rcv) from [<c06258ec>] (__netif_receive_skb_core+0xa1c/0xba4)
[11914.615627] [<c06258ec>] (__netif_receive_skb_core) from [<c06275b0>] (process_backlog+0xa0/0x140)
[11914.623698] [<c06275b0>] (process_backlog) from [<c062ad3c>] (net_rx_action+0x138/0x2fc)
[11914.632640] [<c062ad3c>] (net_rx_action) from [<c0301520>] (__do_softirq+0xe0/0x240)
[11914.640887] [<c0301520>] (__do_softirq) from [<c031b3c4>] (irq_exit+0xd4/0x138)
[11914.648612] [<c031b3c4>] (irq_exit) from [<c030da40>] (handle_IPI+0xa4/0x18c)
[11914.655640] [<c030da40>] (handle_IPI) from [<c030143c>] (gic_handle_irq+0x8c/0x90)
[11914.662930] [<c030143c>] (gic_handle_irq) from [<c030b80c>] (__irq_svc+0x6c/0x90)
[11914.670391] Exception stack(0xcf859f80 to 0xcf859fc8)
[11914.677945] 9f80: 00000001 00000000 00000000 c0313c60 ffffe000 c0a03c74 c0a03c28 00000000
[11914.682984] 9fa0: 00000000 410fc075 00000000 00000000 cf859fc8 cf859fd0 c0308108 c030810c
[11914.691137] 9fc0: 60000013 ffffffff
[11914.699299] [<c030b80c>] (__irq_svc) from [<c030810c>] (arch_cpu_idle+0x34/0x38)
[11914.702605] [<c030810c>] (arch_cpu_idle) from [<c03498d4>] (do_idle+0xdc/0x1a0)
[11914.710241] [<c03498d4>] (do_idle) from [<c0349bf4>] (cpu_startup_entry+0x18/0x1c)
[11914.717270] [<c0349bf4>] (cpu_startup_entry) from [<8030170c>] (0x8030170c)
[11914.725011] ---[ end trace e0e53ee0395ce0b9 ]---
[11914.731768] ath10k_ahb a000000.wifi: failed to transmit packet, dropping: -19
[11914.736658] ath10k_ahb a000000.wifi: failed to submit frame: -19
[11914.743654] ath10k_ahb a000000.wifi: failed to push frame: -19
[11914.750064] ath10k_ahb a800000.wifi: SWBA overrun on vdev 0, skipped old beacon
[11914.756706] ath10k_ahb a000000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[11915.010905] ath10k_ahb a000000.wifi: device successfully recovered
```

greearb, 2019-01-22:
This looks like the prefetch scheduler bug I fixed recently.  Please see if the attached FW works better.
[firmware-5-full-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2781253/firmware-5-full-community.bin.gz)

ynezz, 2019-01-24:
@greearb Ben, not sure if it's related to that new firmware `10.4b-ct-4019-fW-012-c2d65911`, but I've just discovered following in my logs:

```
16:57:13 hostapd: wlan1: interface state DISABLED->COUNTRY_UPDATE
16:57:13 hostapd: wlan1: interface state COUNTRY_UPDATE->HT_SCAN
16:57:14 hostapd: wlan1: interface state HT_SCAN->DFS
16:57:14 hostapd: wlan1: DFS-CAC-START freq=5500 chan=100 sec_chan=1, width=1, seg0=106, seg1=0, cac_time=60s
16:57:18 hostapd: wlan1: DFS-CAC-COMPLETED success=0 freq=5500 ht_enabled=0 chan_offset=0 chan_width=3 cf1=5530 cf2=0
16:57:18 hostapd: wlan1: DFS-RADAR-DETECTED freq=5500 ht_enabled=0 chan_offset=0 chan_width=3 cf1=5530 cf2=0
16:57:18 hostapd: wlan1: DFS-NEW-CHANNEL freq=5180 chan=36 sec_chan=1
16:57:18 hostapd: Using interface wlan1 with hwaddr b8:ec:a3:xx:xx:xx and ssid "fresh-rum-daily"
16:57:19 kernel: [212134.380357] ------------[ cut here ]------------
16:57:19 kernel: [212134.380532] WARNING: CPU: 0 PID: 0 at /var/lib/buildbot/slaves/slashdirt-03/MAIN/build/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-2018-12-20-118e16da/ath10k-4
16:57:19 kernel: [212134.384909] Modules linked in: pppoe ppp_async ath10k_pci ath10k_core ath pppox ppp_generic nf_conntrack_ipv6 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_stat
16:57:19 kernel: [212134.453336] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.14.94 #0
16:57:19 kernel: [212134.475462] Hardware name: Generic DT based system
16:57:19 kernel: [212134.481482] [<c030e774>] (unwind_backtrace) from [<c030ac9c>] (show_stack+0x10/0x14)
16:57:19 kernel: [212134.486159] [<c030ac9c>] (show_stack) from [<c072c248>] (dump_stack+0x88/0x9c)
16:57:19 kernel: [212134.494142] [<c072c248>] (dump_stack) from [<c03173e4>] (__warn+0xe4/0x100)
16:57:19 kernel: [212134.501260] [<c03173e4>] (__warn) from [<c03174b0>] (warn_slowpath_null+0x20/0x28)
16:57:19 kernel: [212134.508623] [<c03174b0>] (warn_slowpath_null) from [<bf2e71e8>] (ath10k_htt_t2h_msg_handler+0xf00/0x2530 [ath10k_core])
16:57:19 kernel: [212134.516064] [<bf2e71e8>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf2e73b8>] (ath10k_htt_t2h_msg_handler+0x10d0/0x2530 [ath10k_core])
16:57:19 kernel: [212134.527137] [<bf2e73b8>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf2e86b4>] (ath10k_htt_t2h_msg_handler+0x23cc/0x2530 [ath10k_core])
16:57:19 kernel: [212134.539723] [<bf2e86b4>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf2e8ef0>] (ath10k_htt_txrx_compl_task+0x6cc/0xb78 [ath10k_core])
16:57:19 kernel: [212134.552541] [<bf2e8ef0>] (ath10k_htt_txrx_compl_task [ath10k_core]) from [<bf3382d4>] (ath10k_pci_napi_poll+0x70/0x100 [ath10k_pci])
16:57:19 kernel: [212134.565348] [<bf3382d4>] (ath10k_pci_napi_poll [ath10k_pci]) from [<c062af3c>] (net_rx_action+0x138/0x2fc)
16:57:19 kernel: [212134.577304] [<c062af3c>] (net_rx_action) from [<c0301520>] (__do_softirq+0xe0/0x240)
16:57:19 kernel: [212134.586853] [<c0301520>] (__do_softirq) from [<c031b35c>] (irq_exit+0xd4/0x138)
16:57:19 kernel: [212134.594838] [<c031b35c>] (irq_exit) from [<c0352b24>] (__handle_domain_irq+0x9c/0xac)
16:57:19 kernel: [212134.602300] [<c0352b24>] (__handle_domain_irq) from [<c030140c>] (gic_handle_irq+0x5c/0x90)
16:57:19 kernel: [212134.610025] [<c030140c>] (gic_handle_irq) from [<c030b80c>] (__irq_svc+0x6c/0x90)
16:57:19 kernel: [212134.618610] Exception stack(0xc0a01f48 to 0xc0a01f90)
16:57:19 kernel: [212134.625992] 1f40:                   00000001 00000000 00000000 c0313c60 ffffe000 c0a03c74
16:57:19 kernel: [212134.631118] 1f60: c0a03c28 00000000 00000000 c092da28 cfffcb00 00000000 c0a01f90 c0a01f98
16:57:19 kernel: [212134.639361] 1f80: c0308108 c030810c 60000013 ffffffff
16:57:19 kernel: [212134.647610] [<c030b80c>] (__irq_svc) from [<c030810c>] (arch_cpu_idle+0x34/0x38)
16:57:19 kernel: [212134.652737] [<c030810c>] (arch_cpu_idle) from [<c034986c>] (do_idle+0xdc/0x1a0)
16:57:19 kernel: [212134.660287] [<c034986c>] (do_idle) from [<c0349b8c>] (cpu_startup_entry+0x18/0x1c)
16:57:19 kernel: [212134.667752] [<c0349b8c>] (cpu_startup_entry) from [<c0900c94>] (start_kernel+0x3b8/0x3c4)
16:57:19 kernel: [212134.675259] ---[ end trace 6004b92667e9bf9f ]---
16:57:19 kernel: [212134.872082] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
16:57:19 kernel: [212134.874435] br-lan: port 3(wlan1) entered blocking state
16:57:19 kernel: [212134.878243] br-lan: port 3(wlan1) entered forwarding state
16:57:19 netifd: Network device 'wlan1' link is up
16:57:19 hostapd: wlan1: interface state DFS->ENABLED
16:57:19 hostapd: wlan1: AP-ENABLED
```

greearb, 2019-01-24:
The line in the warning that ends with this:
 "alinux-ipq40xx/ath10k-ct-2018-12-20-118e16da/ath10k-4" should show a line number if it were not truncated.  Can you check to see if you can get that full line out of dmesg or logread or something?

ynezz, 2019-01-24:
Here it is

```
[212134.380357] ------------[ cut here ]------------
[212134.380532] WARNING: CPU: 0 PID: 0 at /var/lib/buildbot/slaves/slashdirt-03/MAIN/build/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-2018-12-20-118e16da/ath10k-4.19/htt_rx.c:903 ath10k_htt_t2h_msg_handler+0xf00/0x2530 [ath10k_core]
[212134.384909] Modules linked in: pppoe ppp_async ath10k_pci ath10k_core ath pppox ppp_generic nf_conntrack_ipv6 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables hwmon crc_ccitt compat ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_of_simple gpio_button_hotplug
[212134.453336] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.14.94 #0
[212134.475462] Hardware name: Generic DT based system
[212134.481482] [<c030e774>] (unwind_backtrace) from [<c030ac9c>] (show_stack+0x10/0x14)
[212134.486159] [<c030ac9c>] (show_stack) from [<c072c248>] (dump_stack+0x88/0x9c)
[212134.494142] [<c072c248>] (dump_stack) from [<c03173e4>] (__warn+0xe4/0x100)
[212134.501260] [<c03173e4>] (__warn) from [<c03174b0>] (warn_slowpath_null+0x20/0x28)
[212134.508623] [<c03174b0>] (warn_slowpath_null) from [<bf2e71e8>] (ath10k_htt_t2h_msg_handler+0xf00/0x2530 [ath10k_core])
[212134.516064] [<bf2e71e8>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf2e73b8>] (ath10k_htt_t2h_msg_handler+0x10d0/0x2530 [ath10k_core])
[212134.527137] [<bf2e73b8>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf2e86b4>] (ath10k_htt_t2h_msg_handler+0x23cc/0x2530 [ath10k_core])
[212134.539723] [<bf2e86b4>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf2e8ef0>] (ath10k_htt_txrx_compl_task+0x6cc/0xb78 [ath10k_core])
[212134.552541] [<bf2e8ef0>] (ath10k_htt_txrx_compl_task [ath10k_core]) from [<bf3382d4>] (ath10k_pci_napi_poll+0x70/0x100 [ath10k_pci])
[212134.565348] [<bf3382d4>] (ath10k_pci_napi_poll [ath10k_pci]) from [<c062af3c>] (net_rx_action+0x138/0x2fc)
[212134.577304] [<c062af3c>] (net_rx_action) from [<c0301520>] (__do_softirq+0xe0/0x240)
[212134.586853] [<c0301520>] (__do_softirq) from [<c031b35c>] (irq_exit+0xd4/0x138)
[212134.594838] [<c031b35c>] (irq_exit) from [<c0352b24>] (__handle_domain_irq+0x9c/0xac)
[212134.602300] [<c0352b24>] (__handle_domain_irq) from [<c030140c>] (gic_handle_irq+0x5c/0x90)
[212134.610025] [<c030140c>] (gic_handle_irq) from [<c030b80c>] (__irq_svc+0x6c/0x90)
[212134.618610] Exception stack(0xc0a01f48 to 0xc0a01f90)
[212134.625992] 1f40:                   00000001 00000000 00000000 c0313c60 ffffe000 c0a03c74
[212134.631118] 1f60: c0a03c28 00000000 00000000 c092da28 cfffcb00 00000000 c0a01f90 c0a01f98
[212134.639361] 1f80: c0308108 c030810c 60000013 ffffffff
[212134.647610] [<c030b80c>] (__irq_svc) from [<c030810c>] (arch_cpu_idle+0x34/0x38)
[212134.652737] [<c030810c>] (arch_cpu_idle) from [<c034986c>] (do_idle+0xdc/0x1a0)
[212134.660287] [<c034986c>] (do_idle) from [<c0349b8c>] (cpu_startup_entry+0x18/0x1c)
[212134.667752] [<c0349b8c>] (cpu_startup_entry) from [<c0900c94>] (start_kernel+0x3b8/0x3c4)
[212134.675259] ---[ end trace 6004b92667e9bf9f ]---
```

greearb, 2019-01-24:
That code is the last warning in this method:

static struct ieee80211_channel *
ath10k_htt_rx_h_peer_channel(struct ath10k *ar, struct htt_rx_desc *rxd)
{
	struct ath10k_peer *peer;
	struct ath10k_vif *arvif;
	struct cfg80211_chan_def def;
	u16 peer_id;

	lockdep_assert_held(&ar->data_lock);

	if (!rxd)
		return NULL;

	if (rxd->attention.flags &
	    __cpu_to_le32(RX_ATTENTION_FLAGS_PEER_IDX_INVALID))
		return NULL;

	if (!(rxd->msdu_end.common.info0 &
	      __cpu_to_le32(RX_MSDU_END_INFO0_FIRST_MSDU)))
		return NULL;

	peer_id = MS(__le32_to_cpu(rxd->mpdu_start.info0),
		     RX_MPDU_START_INFO0_PEER_IDX);

	peer = ath10k_peer_find_by_id(ar, peer_id);
	if (!peer)
		return NULL;

	arvif = ath10k_get_arvif(ar, peer->vdev_id);
	if (WARN_ON_ONCE(!arvif))
		return NULL;

	if (WARN_ON_ONCE(ath10k_mac_vif_chan(arvif->vif, &def)))
		return NULL;

	return def.chan;
}

I am guessing this is not a real problem, and it looks like the ath10k_htt_rx_h_channel method
expects it can fail and deals with it.

ynezz, 2019-02-08:
Just an update, OpenWrt snapshot `r9086-8c9b182` from `Mon Jan 21 15:35:40 2019` and `10.4b-ct-4019-fW-012-c2d65911` has been running stable for 14 days and 16 hours so far :+1:

Thanks!

closed 2019-02-08 by ynezz