greearb/ath10k-ct #64: QCA9984 reboots when 2.4GHz and 5GHz are both in use

ah-, 2019-01-27:
Hi,

I'm trying to debug an issue I've had for a while. If I enable both 2.4GHz and 5GHz wifi and have clients using both the router hard reboots after a few seconds. If I enable only one I have no issues at all, and same while both are enabled but only one is in use.

It's perfectly repeatable and happens within seconds of starting to use the 5GHz wifi.

Router is a Netgear R7800 running OpenWrt SNAPSHOT r9071-27be78ef46, I tested various ath10-ct versions.

Running ath10k-ct, dmesg during startup: 
```
[   17.884028] ath10k_pci 0000:01:00.0: assign IRQ: got 67
[   17.884049] ath10k 4.19 driver, optimized for CT firmware, probing pci device: 0x46.
[   17.884421] ath10k_pci 0000:01:00.0: enabling device (0140 -> 0142)
[   17.890913] ath10k_pci 0000:01:00.0: enabling bus mastering
[   17.891367] ath10k_pci 0000:01:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[   18.067639] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/fwcfg-pci-0000:01:00.0.txt failed with error -2
[   18.067670] ath10k_pci 0000:01:00.0: Falling back to user helper
[   18.407396] firmware ath10k!fwcfg-pci-0000:01:00.0.txt: firmware_loading_store: map pages failed
[   18.414850] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/QCA9984/hw1.0/ct-firmware-5.bin failed with error -2
[   18.415259] ath10k_pci 0000:01:00.0: Falling back to user helper
[   18.461281] firmware ath10k!QCA9984!hw1.0!ct-firmware-5.bin: firmware_loading_store: map pages failed
[   18.461669] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/QCA9984/hw1.0/ct-firmware-2.bin failed with error -2
[   18.469580] ath10k_pci 0000:01:00.0: Falling back to user helper
[   18.510036] firmware ath10k!QCA9984!hw1.0!ct-firmware-2.bin: firmware_loading_store: map pages failed
[   18.510398] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/QCA9984/hw1.0/firmware-6.bin failed with error -2
[   18.518364] ath10k_pci 0000:01:00.0: Falling back to user helper
[   18.559712] firmware ath10k!QCA9984!hw1.0!firmware-6.bin: firmware_loading_store: map pages failed
[   18.735485] ath10k_pci 0000:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[   18.735524] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   18.746831] ath10k_pci 0000:01:00.0: firmware ver 10.4-3.9.0.2-00018 api 5 features no-p2p,mfp,peer-flow-ctrl,btcoex-param,allows-mesh-bcast,no-ps crc32 63cd3e75
[   21.023636] ath10k_pci 0000:01:00.0: board_file api 2 bmi_id 0:1 crc32 cf58c3bc
[   26.861628] ath10k_pci 0000:01:00.0: 10.4 wmi init: vdevs: 16  peers: 528  tid: 102
[   26.861660] ath10k_pci 0000:01:00.0: msdu-desc: 2500  skid: 32
[   26.945584] ath10k_pci 0000:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 512 raw 0 hwcrypto 1
[   27.046380] ath: EEPROM regdomain: 0x0
[   27.046395] ath: EEPROM indicates default country code should be used
[   27.046405] ath: doing EEPROM country->regdmn map search
[   27.046424] ath: country maps to regdmn code: 0x3a
[   27.046437] ath: Country alpha2 being used: US
[   27.046448] ath: Regpair used: 0x3a
[   27.054873] ath10k_pci 0001:01:00.0: assign IRQ: got 100
[   27.054897] ath10k 4.19 driver, optimized for CT firmware, probing pci device: 0x46.
[   27.055405] ath10k_pci 0001:01:00.0: enabling device (0140 -> 0142)
[   27.061766] ath10k_pci 0001:01:00.0: enabling bus mastering
[   27.062279] ath10k_pci 0001:01:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[   27.246768] ath10k_pci 0001:01:00.0: Direct firmware load for ath10k/fwcfg-pci-0001:01:00.0.txt failed with error -2
[   27.246800] ath10k_pci 0001:01:00.0: Falling back to user helper
[   27.511068] firmware ath10k!fwcfg-pci-0001:01:00.0.txt: firmware_loading_store: map pages failed
[   27.513610] ath10k_pci 0001:01:00.0: Direct firmware load for ath10k/QCA9984/hw1.0/ct-firmware-5.bin failed with error -2
[   27.518925] ath10k_pci 0001:01:00.0: Falling back to user helper
[   27.580816] firmware ath10k!QCA9984!hw1.0!ct-firmware-5.bin: firmware_loading_store: map pages failed
[   27.581084] ath10k_pci 0001:01:00.0: Direct firmware load for ath10k/QCA9984/hw1.0/ct-firmware-2.bin failed with error -2
[   27.589138] ath10k_pci 0001:01:00.0: Falling back to user helper
[   27.632036] firmware ath10k!QCA9984!hw1.0!ct-firmware-2.bin: firmware_loading_store: map pages failed
[   27.632312] ath10k_pci 0001:01:00.0: Direct firmware load for ath10k/QCA9984/hw1.0/firmware-6.bin failed with error -2
[   27.640360] ath10k_pci 0001:01:00.0: Falling back to user helper
[   27.683820] firmware ath10k!QCA9984!hw1.0!firmware-6.bin: firmware_loading_store: map pages failed
[   27.684037] ath10k_pci 0001:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[   27.691688] ath10k_pci 0001:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   27.704299] ath10k_pci 0001:01:00.0: firmware ver 10.4-3.9.0.2-00018 api 5 features no-p2p,mfp,peer-flow-ctrl,btcoex-param,allows-mesh-bcast,no-ps crc32 63cd3e75
[   29.987289] ath10k_pci 0001:01:00.0: board_file api 2 bmi_id 0:2 crc32 cf58c3bc
[   35.831437] ath10k_pci 0001:01:00.0: 10.4 wmi init: vdevs: 16  peers: 528  tid: 102
[   35.831469] ath10k_pci 0001:01:00.0: msdu-desc: 2500  skid: 32
[   35.918055] ath10k_pci 0001:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 512 raw 0 hwcrypto 1
[   36.016303] ath: EEPROM regdomain: 0x0
[   36.016317] ath: EEPROM indicates default country code should be used
[   36.016327] ath: doing EEPROM country->regdmn map search
[   36.016346] ath: country maps to regdmn code: 0x3a
[   36.016359] ath: Country alpha2 being used: US
[   36.016370] ath: Regpair used: 0x3a
```

I'm struggling to get some indication what might be wrong since it resets pretty quickly. I tried following https://www.candelatech.com/ath10k-bugs.php and set the debug_level to 0xc0000032 and kept watching dmesg in a loop over a wired connection, resulting in this debug output:
```
[  319.979465] ath10k_pci 0000:01:00.0: Setting debug-mask to: 0xc0000032  old: 0x10000000
[  349.005042] ath10k_pci 0001:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[  349.005068] ath10k: [0000]: 0004AAD1 07FC581A 00457FAC 0004AAD1 17FC4C01 07005949 00000005 00019003
[  349.010200] ath10k: [0008]: 0FFF0000 00FFFFFF 0004AADD 03FC581D 0004AADD 0BFC581B 00005949 00000000
[  349.019125] ath10k: [0016]: 0004AADD 07FC581A 00457FAC
[  349.028123] ath10k_pci 0001:01:00.0: ATH10K_END
[  354.005748] ath10k_pci 0001:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[  354.005776] ath10k: [0000]: 0004BC6D 17FC5823 00457FAC 59490006 00000002 00000040 00000001 0004BC6E
[  354.010911] ath10k: [0008]: 17FC4C01 07005949 00000005 00019003 0FFF0000 00FFFFFF 0004BD95 17FC5823
[  354.019856] ath10k: [0016]: 00457FAC 59490000 00000000 00000040 00000001
[  354.028833] ath10k_pci 0001:01:00.0: ATH10K_END
[  376.005734] ath10k_pci 0001:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[  376.005769] ath10k: [0000]: 0005170C 0BFC581B 00005949 00000000 0005170C 07FC581A 00457FAC
[  376.010928] ath10k_pci 0001:01:00.0: ATH10K_END
[  376.853923] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[  376.853957] ath10k: [0000]: 00053067 07FC581A 004581E8 00053067 17FC4C01 07005949 0000000A 0601B003
[  376.859109] ath10k: [0008]: 0FF0FFEA 00FFFFFF 0005306A 07FC581C 00000001 0005306B 03FC581D 0005306B
[  376.868043] ath10k: [0016]: 07FC581C 00000001 00053070 03FC581D 00053070 0BFC581B 00005949 00000000
[  376.877068] ath10k: [0024]: 00053070 07FC581A 004581E8 00053078 17FC5823 004581E8 59490006 00000002
[  376.886095] ath10k: [0032]: 00000040 00000001 0005307A 17FC4C01 07005949 0000000F 0601B003 0FF0FFEA
[  376.895043] ath10k: [0040]: 00FFFFFF 0005307B 17FC5823 004581E8 59490000 00000000 00000040 00000001
[  376.904144] ath10k_pci 0000:01:00.0: ATH10K_END
```

The last part at 376.853923 is just before it resets.

I have had this issue for a long time and seen it with various firmware versions. Is there any way of finding out what the DBG_BUFFER means? Or are there other ways of getting more debug output or stopping the reboot in order to inspect what state it is in?

I also tried various different channels etc., with no change.

greearb, 2019-01-28:
This crash is from stock firmware:   firmware ver 10.4-3.9.0.2-00018
You are running the ath10k-ct driver, and possibly it has issues with stock FW or other bugs,
or maybe you are hitting a bug in stock firmware.  You might have to attach a serial interface to your
router and log that to get a full picture of why it reboots.

I cannot properly decode the DBGLOG messages from stock FW.  If you have instability with CT firmware, please post logs and I should be able to decode them.

ah-, 2019-01-28:
Apologies, I was slightly confused by the various firmware versions. I'm now running the latest -ct from here: https://www.candelatech.com/ath10k-10.4.php, I believe it's https://www.candelatech.com/downloads/ath10k-9984-10-4b/ath10k-fw-beta/firmware-5-ct-full-community.bin. 

Now I am getting a more verbose crash:
```
[  544.362484] ath10k_pci 0001:01:00.0: firmware crashed! (guid n/a)
[  544.362554] ath10k_pci 0001:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[  544.362563] ath10k_pci 0001:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[  544.364874] ath10k_pci 0001:01:00.0: firmware ver 10.4-ct-9984-fW-011-33ec52c8 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT crc32 2f2caa6e
[  544.365503] ath10k_pci 0001:01:00.0: board_file api 2 bmi_id 0:2 crc32 cf58c3bc
[  544.365515] ath10k_pci 0001:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[  544.367561] ath10k_pci 0001:01:00.0: firmware register dump:
[  544.368796] ath10k_pci 0001:01:00.0: [00]: 0x0000000A 0x000015B3 0x009A4377 0x00975B31
[  544.374644] ath10k_pci 0001:01:00.0: [04]: 0x009A4377 0x00060B30 0x00000005 0x00000007
[  544.382251] ath10k_pci 0001:01:00.0: [08]: 0x00000000 0x0045FA88 0x0040679C 0x00000000
[  544.390265] ath10k_pci 0001:01:00.0: [12]: 0x00000009 0x00000000 0x009C3BDC 0x009C3BEA
[  544.398132] ath10k_pci 0001:01:00.0: [16]: 0x00973AB0 0x009606CA 0x009606B8 0x00000000
[  544.406026] ath10k_pci 0001:01:00.0: [20]: 0x409A4377 0x0040663C 0x00000005 0x00000000
[  544.413848] ath10k_pci 0001:01:00.0: [24]: 0x809A5778 0x0040669C 0x00000000 0xC09A4377
[  544.421833] ath10k_pci 0001:01:00.0: [28]: 0x809A6AD6 0x004066DC 0x0045D708 0x0045E1A8
[  544.429737] ath10k_pci 0001:01:00.0: [32]: 0x809A7875 0x0040676C 0x00406860 0x0007D15D
[  544.437608] ath10k_pci 0001:01:00.0: [36]: 0x809A78A6 0x0040685C 0x0042EA0C 0x004068EC
[  544.445513] ath10k_pci 0001:01:00.0: [40]: 0x80985B37 0x004068AC 0x00000000 0x00459C6C
[  544.453346] ath10k_pci 0001:01:00.0: [44]: 0x809932F3 0x004068CC 0x0042FEC8 0x00459C6C
[  544.461321] ath10k_pci 0001:01:00.0: [48]: 0x8098EEA8 0x004068EC 0x0042FEC8 0x00000000
[  544.469225] ath10k_pci 0001:01:00.0: [52]: 0x80963AD3 0x00406A7C 0x0042FEC8 0x0098EEA0
[  544.477127] ath10k_pci 0001:01:00.0: [56]: 0x80960E80 0x00406A9C 0x0000001F 0x00400000
[  544.485032] ath10k_pci 0001:01:00.0: Copy Engine register dump:
[  544.492846] ath10k_pci 0001:01:00.0: [00]: 0x0004a000  10  10   3   3
[  544.498753] ath10k_pci 0001:01:00.0: [01]: 0x0004a400  24  24 126 127
[  544.505344] ath10k_pci 0001:01:00.0: [02]: 0x0004a800  43  43 106 107
[  544.511684] ath10k_pci 0001:01:00.0: [03]: 0x0004ac00  23  23  25  23
[  544.518184] ath10k_pci 0001:01:00.0: [04]: 0x0004b000 681 681 167 127
[  544.524618] ath10k_pci 0001:01:00.0: [05]: 0x0004b400   6   6 325 326
[  544.530953] ath10k_pci 0001:01:00.0: [06]: 0x0004b800   2   2   2   2
[  544.537458] ath10k_pci 0001:01:00.0: [07]: 0x0004bc00   1   1   1   1
[  544.543800] ath10k_pci 0001:01:00.0: [08]: 0x0004c000   0   0 127   0
[  544.550298] ath10k_pci 0001:01:00.0: [09]: 0x0004c400   1   1   1   1
[  544.556734] ath10k_pci 0001:01:00.0: [10]: 0x0004c800   0   0   0   0
[  544.563072] ath10k_pci 0001:01:00.0: [11]: 0x0004cc00   0   0   0   0
[  544.571594] ath10k_pci 0001:01:00.0: debug log header, dbuf: 0x4237b8  dropped: 0
[  544.577020] ath10k_pci 0001:01:00.0: [0] next: 0x4237a0 buf: 0x419010 sz: 1500 len: 52 count: 2 free: 0
[  544.584403] ath10k_pci 0001:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[  544.592581] ath10k: [0000]: 0007D15D 13FC4C07 00000911 00000000 00000005 00000000 0007D15D 17FC0001
[  544.598902] ath10k: [0008]: 009A4377 000015B3 000015B3 0040652C 91104569
[  544.607834] ath10k_pci 0001:01:00.0: ATH10K_END
[  544.615819] ath10k_pci 0001:01:00.0: [1] next: 0x4237b8 buf: 0x418a20 sz: 1500 len: 0 count: 0 free: 0
[  544.674423] ath10k_pci 0001:01:00.0: removing peer, cleanup-all, deleting: peer da356e00 vdev: 0 addr: 64:a2:f9:4b:c9:f3 
[  544.674467] ath10k_pci 0001:01:00.0: removing peer, cleanup-all, deleting: peer db8dd400 vdev: 0 addr: 7c:1c:4e:18:9e:98 
[  544.684503] ath10k_pci 0001:01:00.0: removing peer, cleanup-all, deleting: peer dbbfda00 vdev: 0 addr: 68:c6:3a:a2:fe:37 
[  544.695377] ath10k_pci 0001:01:00.0: removing peer, cleanup-all, deleting: peer dbc0ee00 vdev: 0 addr: b0:b9:8a:60:04:bc 
[  544.799669] ieee80211 phy1: Hardware restart was requested
```

More complete dmesg here: https://gist.github.com/ah-/fbab3d5714907d4bca420fd3bfdfcac3

It seems to last longer now and doesn't reboot immediately but works for a minute or two before rebooting.

I hope this is a useful bug report, otherwise I'm happy to try again. I've also tried your debug firmware from https://github.com/greearb/ath10k-ct/issues/58, however that one crashes really early for me, I suspect my kernel doesn't support htt-mgt? In case it's useful, log from that test is here: https://gist.github.com/ah-/d6c1f41cab7959db2f25978209cb674f

greearb, 2019-01-28:
The FW from your dmesg is quite old.  The last FW in bug 58 was buggy, I'm about to upload a new one there...please test with it when it is uploaded in a few minutes.

ah-, 2019-01-28:
Thanks, that one does work for a while but then also crashes. Log here: https://gist.github.com/ah-/e26080c2cf7cba844d283668ed4d0366

I also noticed that throughput is a bit lower than usual.

greearb, 2019-01-28:
There is noisy logging, but no crashes in that image.  I just posted a less spammy one to bug 58, that will likely improve performance, but there is still a lot of debugging compiled into the bug-58 binary to try to track down its problem...

ah-, 2019-01-28:
Looks pretty similar: https://gist.github.com/ah-/67bd5ebae823a76bec090ce18be1aa5c

I'll try that one a few more times later to check if I see anything else. I can try and get a serial console working, but probably that'll be after next weekend.

greearb, 2019-01-28:
If SQM is implicated, maybe a new bug for that, but that might not be ath10k related at all.  The crash above is same thing I'm tracking for bug 58, so I'm going to close this bug, please report further crashes in bug 58 (and try the firmware I just posted there)...I think it may narrow down the site of the bug a bit more.

closed 2019-01-28 by greearb

ah-, 2019-01-28:
Yep that sounds great, I agree if it turns out to be SQM it might have nothing to do with ath10k. Or there were two bugs and just switching to recent ath10k-ct fixed one of them already.

Thanks for your help!