greearb/ath10k-ct #65: R7800  10.4b-ct-9984-fH-012-d6de75e0d SWBA overrun on vdev 0, skipped old beacon

schwichtgit, 2019-02-01:
Description of the problem (how to configure, how to reproduce, how often it happens).
R7800 multi AP setup using 802.11r.

Software (OS, Firmware version, kernel, driver, etc)
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r9059-6e104c63d6'
DISTRIB_TARGET='ipq806x/generic'
DISTRIB_ARCH='arm_cortex-a15_neon-vfpv4'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r9059-6e104c63d6'
DISTRIB_TAINTS=''
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="http://openwrt.org/"
BUG_URL="http://bugs.openwrt.org/"
SUPPORT_URL="http://forum.lede-project.org/"
BUILD_ID="r9059-6e104c63d6"
LEDE_BOARD="ipq806x/generic"
LEDE_ARCH="arm_cortex-a15_neon-vfpv4"
LEDE_TAINTS=""
LEDE_DEVICE_MANUFACTURER="OpenWrt"
LEDE_DEVICE_MANUFACTURER_URL="http://openwrt.org/"
LEDE_DEVICE_PRODUCT="Generic"
LEDE_DEVICE_REVISION="v0"
LEDE_RELEASE="OpenWrt SNAPSHOT r9059-6e104c63d6"
```
driver: ath10k_pci
version: 4.14.93

Firmware: 10.4b-ct-9984-fH-012-d6de75e0d 

Hardware (NIC chipset, platform, etc)
R7800

Logs (dmesg, maybe supplicant and/or hostap)
[system-SchwichtenburgWrt.log.zip](https://github.com/greearb/ath10k-ct/files/2822378/system-SchwichtenburgWrt.log.zip)

schwichtgit, 2019-02-01:
@greearb will you pick this one up, too ?

greearb, 2019-02-05:
All these bugs are belong to me.  Do you see similar warnings with stock firmware?  And if not, would you like to bisect the issue if I build you an image for each of my commits?

schwichtgit, 2019-02-05:
on

```
driver: ath10k_pci
version: 4.14.93
firmware-version: 10.4-3.9.0.2-00021
```
```
c38f8d67845d83986d347e6f7436f65b2e0323dad5911dbc96895ce575b451f2  firmware-5.bin_10.4-3.9.0.2-00021
```
I do not see a single occurrence.

however on
```
driver: ath10k_pci
version: 4.14.93
firmware-version: 10.4b-ct-9984-fH-012-d6de75e0d
```
I see
```
grep -i swba  system-SchwichtenburgWrt.log|wc -l
**915**
```

I do not mind being helpful to address this issue, as both the CT firmware and this AP R7800 are popular, so I guess many would benefit.

but the backside is, that these victim systems are in real use, so it would be good to have a focused triaging strategy to keep the pain level low on the STA users.

greearb, 2019-02-05:
It is liable to be disruptive to users to bisect this problem.  Several other users have reported the same basic problem, so maybe one of them will step up with an offer to bisect.

schwichtgit, 2019-02-05:
please provide the images ... I see what I can do

greearb, 2019-02-06:
Here are images to bisect with:

https://www.candelatech.com/downloads/ath10k-9984-10-4b/ath10k-fw-beta/all_builds-9984b-H-feb-6-2019.tar.gz

schwichtgit, 2019-02-06:
@greearb, I see a tar ball with 923 firmwares.
what is the process you suggest I should follow?

pgera, 2019-02-07:
Binary search would be the best approach. Should take fewer than 10 tests.

greearb, 2019-02-08:
Please see section on how to bisect toward the bottom of this page:
https://www.candelatech.com/ath10k-bugs.php

sq-bb, 2019-02-16:
@schwichtgit Is there a specific debug level at which you see these messages, or have you been able to link them to a specific aspect of your config/network? I would be able to do the bisect, but I can't seem to reproduce the issue on my r7800 running openwrt master and firmware 10.4b-ct-9984-fH-012-5815a26a3.

shelterx, 2019-02-25:
I saw this error too now however it doesn't seem very freqeuent for me so catching this error is next to impossible.

shelterx, 2019-12-20:
I can only speak for myself but I haven't seen any SWBA overruns in a long time now.

greearb, 2020-01-23:
Closing this one, hopefully it is fixed.

closed 2020-01-23 by greearb