greearb/ath10k-ct #66: FW crash - QCA9984 - OpenWRT commit Mon, 4 Feb 2019 12:09:35

shelterx, 2019-02-05:
Firmware crash on startup.
This [commit](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=5a8d03ceebaf50e82ac21dc5f2111387cc361ba0) causes it

OpenWrt SNAPSHOT, r9258-420376ab50
firmware ver 10.4b-ct-9984-fW-012-ddc348c0
Linux nighthawk 4.14.97 #0 SMP Tue Feb 5 08:07:54 2019 armv7l GNU/Linux

This is dmesg output, doesn't say anything more than this, if you want more the debug mask must be set earlier but I'm not sure where.
[   49.767176] ath10k_pci 0001:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   49.767230] ath10k_pci 0001:01:00.0: msdu-desc: 2500  skid: 32
[   49.850971] ath10k_pci 0001:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   49.851839] ath10k_pci 0001:01:00.0: wmi print 'free: 85528 iram: 26340 sram: 18240'
[   50.227594] ath10k_pci 0001:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[   50.227849] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   50.764167] ath10k_pci 0001:01:00.0: NOTE:  Firmware DBGLOG output disabled in debug_mask: 0x10000000
[   51.819425] ------------[ cut here ]------------
[   51.819558] WARNING: CPU: 0 PID: 0 at /home/ftp/Archive_2/.build/openwrt-basedir/master/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/ath10k-ct-2018-12-20-118e16da/ath10k-4.19/htt_rx.c:903 ath10k_htt_t2h_msg_handler+0xf2c/0x2650 [ath10k_core]
[   51.823892] Modules linked in: ath10k_pci ath10k_core ath mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_bpf xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY ts_kmp ts_fsm ts_bm nf_reject_ipv4 nf_nat_rtsp nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv4 nf_conntrack_rtsp nf_conntrack_rtcache nf_conntrack_netlink iptable_raw iptable_mangle iptable_filter ipt_ah ipt_ECN ip_tables crc_ccitt compat chaoskey fuse sch_cake act_skbedit act_mirred em_u32 cls_u32 cls_tcindex
[   51.894787]  cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_NPT ip6t_MASQUERADE nf_nat_masquerade_ipv6 ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 nf_nat nf_conntrack ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables msdos ip_gre gre ifb sit tunnel4 ip_tunnel tun vfat fat cifs nls_utf8 nls_iso8859_15 nls_iso8859_1 nls_cp850 nls_cp437 nls_cp1250 sha1_generic md5 md4 usb_storage leds_gpio xhci_plat_hcd
[   51.966276]  xhci_pci xhci_hcd dwc3 dwc3_of_simple ohci_platform ohci_hcd phy_qcom_dwc3 ahci ehci_platform sd_mod ahci_platform libahci_platform libahci libata scsi_mod ehci_hcd gpio_button_hotplug ext4 jbd2 mbcache exfat crc32c_generic
[   51.988514] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.14.97 #0
[   52.009394] Hardware name: Generic DT based system
[   52.015414] [<c030f58c>] (unwind_backtrace) from [<c030b7a4>] (show_stack+0x14/0x20)
[   52.020008] [<c030b7a4>] (show_stack) from [<c0791958>] (dump_stack+0x88/0x9c)
[   52.027904] [<c0791958>] (dump_stack) from [<c0318ecc>] (__warn+0xf0/0x11c)
[   52.034929] [<c0318ecc>] (__warn) from [<c0318fb8>] (warn_slowpath_null+0x20/0x28)
[   52.041829] [<c0318fb8>] (warn_slowpath_null) from [<bf7d5fbc>] (ath10k_htt_t2h_msg_handler+0xf2c/0x2650 [ath10k_core])
[   52.049586] [<bf7d5fbc>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf7d61ac>] (ath10k_htt_t2h_msg_handler+0x111c/0x2650 [ath10k_core])
[   52.060192] [<bf7d61ac>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf7d7558>] (ath10k_htt_t2h_msg_handler+0x24c8/0x2650 [ath10k_core])
[   52.073039] [<bf7d7558>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf7d7de4>] (ath10k_htt_txrx_compl_task+0x6f8/0xbb8 [ath10k_core])
[   52.085770] [<bf7d7de4>] (ath10k_htt_txrx_compl_task [ath10k_core]) from [<bf826040>] (ath10k_pci_napi_poll+0x7c/0x11c [ath10k_pci])
[   52.098532] [<bf826040>] (ath10k_pci_napi_poll [ath10k_pci]) from [<c0690240>] (net_rx_action+0x144/0x31c)
[   52.110371] [<c0690240>] (net_rx_action) from [<c03015c8>] (__do_softirq+0xf0/0x264)
[   52.119832] [<c03015c8>] (__do_softirq) from [<c031d288>] (irq_exit+0xdc/0x148)
[   52.127728] [<c031d288>] (irq_exit) from [<c0359f28>] (__handle_domain_irq+0xa8/0xc8)
[   52.134754] [<c0359f28>] (__handle_domain_irq) from [<c0301488>] (gic_handle_irq+0x6c/0xb8)
[   52.142738] [<c0301488>] (gic_handle_irq) from [<c030c38c>] (__irq_svc+0x6c/0x90)
[   52.150890] Exception stack(0xc0b01f48 to 0xc0b01f90)
[   52.158542] 1f40:                   00000001 00000000 00000000 c0315300 ffffe000 c0b03c74
[   52.163586] 1f60: c0b03c28 00000000 00000000 c0a2ca28 00000000 00000000 c0b01f90 c0b01f98
[   52.171734] 1f80: c030884c c0308850 60000013 ffffffff
[   52.179892] [<c030c38c>] (__irq_svc) from [<c0308850>] (arch_cpu_idle+0x38/0x44)
[   52.184932] [<c0308850>] (arch_cpu_idle) from [<c034fd58>] (do_idle+0xe8/0x1bc)
[   52.192394] [<c034fd58>] (do_idle) from [<c03500a0>] (cpu_startup_entry+0x1c/0x20)
[   52.199435] [<c03500a0>] (cpu_startup_entry) from [<c0a00ce4>] (start_kernel+0x3fc/0x408)
[   52.207069] ---[ end trace f64100c7a0d6a534 ]---
[   56.161853] ath10k_pci 0000:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   56.161881] ath10k_pci 0000:01:00.0: msdu-desc: 2500  skid: 32
[   56.243703] ath10k_pci 0000:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   56.244546] ath10k_pci 0000:01:00.0: wmi print 'free: 85528 iram: 26340 sram: 18240'
[   56.598949] ath10k_pci 0000:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4

shelterx, 2019-02-05:
No, not with 10.4-3.9.0.2-00021

greearb, 2019-02-06:
Please see if you can bisect to find which commit caused this regression.

https://www.candelatech.com/downloads/ath10k-9984-10-4b/ath10k-fw-beta/all_builds-9984b-H-feb-6-2019.tar.gz

shelterx, 2019-02-08:
Any idea how far back i need to look? The firmware last posted in #58 do not output this warning, if that helps.

greearb, 2019-02-08:
I added a section on how to bisect toward the bottom of this page.  If very latest firmware works, then I guess you can just close the bug.  Or you could bisect to see what commit fixed it and maybe we will understand the issue better in that case.

https://www.candelatech.com/ath10k-bugs.php

shelterx, 2019-02-08:
If I can I will have a look at it, I have more people than myself using the router.
I'm not sure what you mean with latest firmware, is the #58 firmware newer than the one in the OpenWRT commit?

sq-bb, 2019-02-16:
@shelterx Do you see the warning on every boot? I'm running a r7800 and firmware 10.4b-ct-9984-fH-012-5815a26a3, and this warning has shown up once for me, but on subsequent restarts everything looks fine. If you happen to know of a way to reliably reproduce the warning, I'd be happy to go through the bisect. :)

shelterx, 2019-02-20:
Updated the openwrt master build and  firmware ver 10.4b-ct-9984-fW-012-5815a26a3
Now seems to boot without warnings, unless the warning is due to a race condition, then it might be a hard catch.

greearb, 2019-02-20:
Lets call this fixed then, please open a new bug if you see it again.

closed 2019-02-20 by greearb