greearb/ath10k-ct #67: WDS (4addr) doesn't work (wlanX.staY interface doesn't get created)

enryIT, 2019-02-05:
_Possibly related to issue #13_ 

**Description of the problem (how to configure, how to reproduce, how often it happens).**
Set up WDS (4addr) AP. On station connect nor wlanX.staY gets created nor added to the specified bridge in hostapd configuration.

**Software (OS, Firmware version, kernel, driver, etc)**
- OWRT 18.0.6 with stock ath10k driver and `firmware-version: 10.4-3.5.3-00057` works
- OWRT 18.0.6 with stock ath10k driver and `firmware-version: 10.4-3.9.0.2-00021` works
- OWRT 18.0.6 and trunk, with ath10k-ct driver, latest stable and beta firmware (community, non-commercial, diet and no-diet) doesn't work

**Hardware (NIC chipset, platform, etc)**
Netgear R7800 - IPQ8065 SoC with QCA9984

**Logs (dmesg, maybe supplicant and/or hostap)**
No useful log in syslog nor dmesg.

Tomorrow I'll try with atheros debug on and with OWRT 18.0.6 stable with ct driver and firmware.

greearb, 2019-02-05:
Can you see if ct driver and stock firmware works?  I'm curious to know if bug is in the firmware or driver.

enryIT, 2019-02-06:
OWRT 18.0.6, with ath10k-ct driver and firmware `10.4b-ct-9984-fW-012-ddc348c02` the wlanX.staY interface gets created but then the firmware crashes.

This, I think, is the main error:
> Wed Feb  6 21:34:20 2019 kern.info kernel: [  137.871183] wlan0.sta1: HW problem - can not stop rx aggregation for 30:b5:c2:08:b9:e0 tid 0

Some other errors:
> daemon.err  hostapd: Failed to set beacon parameters
ath10k_pci 0000:01:00.0: failed to send pdev bss chan info request

Attached full log + crash dump.
[log_and_dump.zip](https://github.com/greearb/ath10k-ct/files/2838152/log_and_dump.zip)

greearb, 2019-02-08:
Please see if this is a regression in my firmware and/or driver by checking if QCA firmware has the same issue.

enryIT, 2019-02-08:
OWRT 18.0.6, with ath10k-ct driver and stock firmware 10.4-3.9.0.2-00021 works.

greearb, 2019-02-08:
OK, then please try to bisect to find out where I introduced the bug:

A tarball of images to bisect is here:
http://www.candelatech.com/downloads/ath10k-9984-10-4b/ath10k-fw-beta/all_builds-9984b-H-feb-6-2019.tar.gz

How to bisect is described at bottom of this page:
http://www.candelatech.com/ath10k-bugs.php

enryIT, 2019-02-08:
Found it, the last working commit is number 61.
60 works, 61 works, 62 crash.

greearb, 2019-02-10:
Thanks for the bisect.  The change in 62 reorders some struct objects for better packing to save memory.  And, it appears your crash is due to an assumption that the mac-addr is 32-bit aligned.  Please try this image and let me know how it goes.  Post the crash log if it still crashes.

[firmware-5-full-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2848433/firmware-5-full-community.bin.gz)

enryIT, 2019-02-10:
Thank you, using your attached firmware `10.4b-ct-9984-fW-012-1a8f0f760` it works flawlessly.
Let me test this version for a couple days in order to check for instabilities but I think you've already resolved the issue.

djangoa, 2019-02-10:
I'm experiencing a similar issue with a D-Link DAP-2695/Devolo 1750e - QCA9880 and firmware `10.1-ct-8x-__fW-022-1bbfa151` on OWRT master:

> Sun Feb 10 13:15:33 2019 daemon.info hostapd: wlan0: STA 00:11:22:33:44:55 IEEE 802.11: authenticated
> Sun Feb 10 13:15:33 2019 daemon.info hostapd: wlan0: STA 00:11:22:33:44:55 IEEE 802.11: associated (aid 1)
> Sun Feb 10 13:15:33 2019 daemon.err hostapd: Failed to create interface wlan0.sta1: -122 (Not supported)
> Sun Feb 10 13:15:42 2019 daemon.info hostapd: wlan0: STA 00:11:22:33:44:55 IEEE 802.11: deauthenticated due to local deauth request
> Sun Feb 10 13:15:42 2019 daemon.err hostapd: nl80211: NL80211_ATTR_STA_VLAN (addr=00:11:22:33:44:55 ifname=wlan0 vlan_id=0) failed: -2 (No such file or directory)
> Sun Feb 10 13:15:42 2019 daemon.err hostapd: Failed to remove interface (ifidx=0)
> Sun Feb 10 13:15:42 2019 daemon.notice hostapd: wlan0: WDS-STA-INTERFACE-REMOVED ifname=wlan0.sta1 sta_addr=00:11:22:33:44:55

I've also tried compiling with stock ath10k driver and firmware and get the same errors in the log. Is the CT 10.1 firmware effected by the same issue?

greearb, 2019-02-10:
The wave-1 issue looks different, probably missing support instead of buggy support.  Please open a new bug, but since stock FW doesn't support it either, it is at least not a regression.  I'm not adding a lot of new features to wave-1 these days unless someone wants to fund it, so not sure if or when I'll get that feature supported.

enryIT, 2019-02-10:
Ok, now we have a stable experience on stable OpenWRT 18.0.6.

The same firmware `10.4b-ct-9984-fW-012-1a8f0f760` still crashes on trunk version
> [  120.827254] ------------[ cut here ]------------
[  120.827328] WARNING: CPU: 0 PID: 3357 at /home/enrico/owrt_r7800/trunk/master/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/ath10k-ct-2018-12-20-118e16da/ath10k-4.19/htt_rx.c:903 ath10k_htt_t2h_msg_handler+0x11c8/0x2c68 [ath10k_core]
_[...omitted linked modules...]_
[  121.153565] CPU: 0 PID: 3357 Comm: hostapd Not tainted 4.14.98 #0
[  121.175722] Hardware name: Generic DT based system
[  121.181902] [<c030f58c>] (unwind_backtrace) from [<c030b7a4>] (show_stack+0x14/0x20)
[  121.186590] [<c030b7a4>] (show_stack) from [<c0794338>] (dump_stack+0x88/0x9c)
[  121.194485] [<c0794338>] (dump_stack) from [<c0318ecc>] (__warn+0xf0/0x11c)
[  121.201510] [<c0318ecc>] (__warn) from [<c0318fb8>] (warn_slowpath_null+0x20/0x28)
[  121.208392] [<c0318fb8>] (warn_slowpath_null) from [<bfd4a074>] (ath10k_htt_t2h_msg_handler+0x11c8/0x2c68 [ath10k_core])
[  121.216145] [<bfd4a074>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bfd4a264>] (ath10k_htt_t2h_msg_handler+0x13b8/0x2c68 [ath10k_core])
[  121.227075] [<bfd4a264>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bfd4b744>] (ath10k_htt_t2h_msg_handler+0x2898/0x2c68 [ath10k_core])
[  121.239660] [<bfd4b744>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bfd4c230>] (ath10k_htt_txrx_compl_task+0x710/0xc5c [ath10k_core])
[  121.252409] [<bfd4c230>] (ath10k_htt_txrx_compl_task [ath10k_core]) from [<bfda639c>] (ath10k_pci_napi_poll+0x7c/0x11c [ath10k_pci])
[  121.265179] [<bfda639c>] (ath10k_pci_napi_poll [ath10k_pci]) from [<c0691ef8>] (net_rx_action+0x144/0x31c)
[  121.277036] [<c0691ef8>] (net_rx_action) from [<c03015c8>] (__do_softirq+0xf0/0x264)
[  121.286499] [<c03015c8>] (__do_softirq) from [<c031d2d4>] (irq_exit+0xdc/0x148)
[  121.294395] [<c031d2d4>] (irq_exit) from [<c0359f78>] (__handle_domain_irq+0xa8/0xc8)
[  121.301423] [<c0359f78>] (__handle_domain_irq) from [<c0301488>] (gic_handle_irq+0x6c/0xb8)
[  121.309408] [<c0301488>] (gic_handle_irq) from [<c030c730>] (__irq_usr+0x50/0x80)
[  121.317566] Exception stack(0xdb113fb0 to 0xdb113ff8)
[  121.325207] 3fa0:                                     b6f4f2c0 00000000 b6f50448 00000000
[  121.330248] 3fc0: 00000064 b6eb2a4a 00000145 00000000 00000014 00000001 00000014 00000b1f
[  121.338406] 3fe0: 00800000 becb8f04 00000000 b6f01734 60000010 ffffffff
[  121.346633] ---[ end trace 081d8f295394393e ]---
[  121.353198] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  121.357866] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  121.477496] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  121.477880] br-lan: port 3(wlan0) entered blocking state
[  121.483076] br-lan: port 3(wlan0) entered listening state
[  123.511484] br-lan: port 3(wlan0) entered learning state
[  125.591228] br-lan: port 3(wlan0) entered forwarding state
[  125.591308] br-lan: topology change detected, propagating

and after the crash hostapd reports this error:
> Sun Feb 10 19:05:40 2019 kern.info kernel: [  123.511484] br-lan: port 3(wlan0) entered learning state
> Sun Feb 10 19:05:41 2019 daemon.info hostapd: wlan0: STA 30:b5:c2:08:b9:e0 IEEE 802.11: authenticated
> Sun Feb 10 19:05:41 2019 daemon.info hostapd: wlan0: STA 30:b5:c2:08:b9:e0 IEEE 802.11: associated (aid 1)
> **Sun Feb 10 19:05:41 2019 daemon.err hostapd: Failed to create interface wlan0.sta1: -95 (Not supported)**
> Sun Feb 10 19:05:41 2019 daemon.info hostapd: wlan0: STA 30:b5:c2:08:b9:e0 RADIUS: starting accounting session 7BCE2EC8A5348F0D
> Sun Feb 10 19:05:41 2019 daemon.info hostapd: wlan0: STA 30:b5:c2:08:b9:e0 WPA: pairwise key handshake completed (RSN)
> Sun Feb 10 19:05:43 2019 kern.info kernel: [  125.591228] br-lan: port 3(wlan0) entered forwarding state
> Sun Feb 10 19:05:43 2019 kern.info kernel: [  125.591308] br-lan: topology change detected, propagating
> Sun Feb 10 19:07:56 2019 daemon.notice hostapd: wlan0: STA 94:65:2d:b2:6c:9d IEEE 802.11: did not acknowledge authentication response

In this configuration the radio link on the station is up and running, the only thing missing is the wlan0.sta1 interface on AP side.

greearb, 2019-02-10:
That is a driver warning, not a firmware crash.  And, hostapd error is coming from userspace.  So, likely not a firmware issue since it works on the different OpenWRT.  You will need to bisect openwrt and/or deal with bugs in the stack to fix the problem above.

enryIT, 2019-02-10:
That `ath10k_htt_t2h_msg_handler` trace is not a crash?

Anyway, do you have any suggestion on how to proceed? I checked openwrt issue tracker but it's really not used, no real activity there, Do you know of another platform?

greearb, 2019-02-15:
comments in OpenWRT IRC indicate that this is a kernel bug and someone has a fix they are working to get upstream.  Not sure if or when it will be in the OpenWRT repo.  Not a FW bug now though, so closing this bug.  Please ask on #openwrt-devel (freenode IRC) and maybe you can get a patch to try.

closed 2019-02-15 by greearb