greearb/ath10k-ct #69: 9980 10.4b-ct-9980-fH-012-ddc348c0 (2019.02.02) crash

notmyrealhandle, 2019-02-08:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).

Software (OS, Firmware version, kernel, driver, etc)

Hardware (NIC chipset, platform, etc)

Logs (dmesg, maybe supplicant and/or hostap)

greearb, 2019-02-08:
This is a crash in rate-ctrl.  Not the corruption thankfully, and I think I see the fix.  Please try the attached firmware.

[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2846427/firmware-5-full-htt-mgt-community.bin.gz)

notmyrealhandle, 2019-02-08:
applied: 10.4b-ct-9980-fH-012-1a8f0f76

I'll report back in ~7 days or so unless issues come up or you'd like feedback sooner

notmyrealhandle, 2019-02-15:
With respect to this bug, it looks like it went 6+ days.  Thank you.

The router reset last night - no log of what happened so nothing I can report about it.

closed 2019-02-15 by notmyrealhandle