greearb/ath10k-ct #71: QCA9862 missing BTCOEX

psyborg55, 2019-03-04:
Using 2.4GHz wlan interface while bluetooth connection is active (audio streaming) interferes heavily with audio playback and web-browsing is slow as well. Is there a possibility to enable WMI_SERVICE_COEX_GPIO in firmware?

linux kernel 4.9.135; ath10k-ct for 4.9; firmware 10.1-ct-8x-__xfH-021-bc02e17b
QCA9862 mPCIe card - identifies as qca988x hw2.0

greearb, 2019-03-06:
Please see if this attached firmware works.
[firmware-2-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2938815/firmware-2-full-htt-mgt-community.bin.gz)

psyborg55, 2019-03-07:
99% error free BT audio playback but page loading slow, iperf from card to router doesn't start, from router to card about 4Mbps if using single stream, multiple streams distort audio heavily again and don't pass 16Mbps. with stopped playback iperf reaches about 70Mbps

greearb, 2019-03-07:
Can you try upstream firmware and see if it works any better?  I think I backported all of the pertinent COEX changes from 10.2 firmware source into that firmware I posted above, but maybe my 10.2 is still missing something?

psyborg55, 2019-03-08:
firmware-5.bin_10.2.4-1.0-00045 and firmware-5.bin_10.2.4.70.70 both have problems, playback is fine when doing speedtest download test, as soon as upload test start audio almost completely freezes.

looks like the driver needs some support (like it's done in ath9k with btcoex paramater)

when running ath9k card with btcoex_enable=0 the router connected to will report identical RSSI value as it reports for ath10k card. in such case bluetooth misbehaves on ath9k too.
when passed 1 to btcoex_enable bluetooth works fine as well as wifi but RSSI reported in router is about 6dB lower, and this weaker txpower is noticeable only on router that is very close to the laptop, same change makes no difference on my other router on upper floor through walls.

greearb, 2019-03-08:
Did the ath10k-ct firmware I posted work similarly to the upstream firmware?

Did the ath10k-ct firmware I posted work better than the ath10k-ct firmware that was previously included in openwrt?

psyborg55, 2019-03-08:
bluetooth worked better with above posted firmware than with upstream fw, but wifi was slower.
i did not compare it with openwrt versions as these are updated quite often, and i'm not sure which version should be used for this card, firmware-2* or firmware-5*. if you want me to test it just write the commit hash of the firmware or it's suffix

greearb, 2019-03-08:
Please compare against this firmware, which does not have the BTCOEX feature enabled:

http://www.candelatech.com/downloads/ath10k-fw-beta/firmware-2-ct-full-htt-mgt-community.bin

The firmware I posted earlier in this bug does have the BTCOEX feature enabled.  I want to make sure that enabling the BTCOEX feature actually changed things for the better before committing the change, and I have no good way to test BTCOEX locally currently.

psyborg55, 2019-03-17:
it did change things for the better regarding bluetooth, but wifi speed is heavily crippled, in speedtest dl reach max 5Mbps ul starts, reaches 0.20Mbps and Upload Test Error pops out

greearb, 2019-03-18:
So, does my firmware with the BTCOEX feature act similar to upstream firmware, or is upstream firmware noticeably better?  If they are the same, then probably  there is not much I can do about fixing this issue since I don't really have time to dig into it in detail.

Is the BTCOEX ct firmware better over-all than the older non-btcoex firmware?  If not, I can revert the
change.

psyborg55, 2019-03-18:
keep the change if it's already committed but look ahead for wifi improvement. at least what i noticed in ath9k the output power is dropped 5db for 20Mhz and 7db for 40Mhz mode on one chain if bluetooth is active. not sure it would be worth trying to do the same in ath10k without firmware changes so i didn't try.

greearb, 2019-03-25:
I uploaded my firmware with this fix applied.  I'll update this bug title to indicate that more work may be needed in the future.