greearb/ath10k-ct #72: Dynamic VLANs Broken - ath10k

promatow, 2019-03-05:
This is in parts re-opened #13.
In general from early 2018, ath10k drivers won't allow for dynamic vlan set-up.
Looking into discussions and finally, patch there: [kernel.org thread 10692385](https://patchwork.kernel.org/patch/10692385/) as well as to other ath10k related it seems that it is an issue with encryption keys that has to be used in sw mode for non-unicast and in hw mode for unicast/direct comm. Anyway there is patch in place.

It looks like resolved now with patch finally being [merged - example file](https://git.kernel.org/pub/scm/linux/kernel/git/kvalo/ath.git/commit/drivers/net/wireless/ath/ath10k/wmi.h?id=4920ce3bf7e0d533441b3c8e50b40a6f1897492b ) into Linux ath10k driver. 

I've been using dynamic vlan feature for a long time (with OpenWrt) - but now it looks like without this patch I need to ditch whole Qualcomm chipset (ath10k) devices family switching to other manufactures.

Kindly ask to merge patch into CT firmware(s) - especially to QCA9984 and QCA988X  :D  

As for debugging info, it is nothing spectacular but well known (to this thread) error:
- running on QCA9984 (same goes for QCA9880)
- daemon.err hostapd: Failed to create interface wlanX.XXX: -95 (Not supported)
- My drivers are latest snapshot (05.03.2019, CT non-commercial): ath10k_pci 0000:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0'

greearb, 2019-03-06:
Looks like that kernel patch uses some firmware feature that is not in my current firmware code.  At some point I'll probably pull in some more recent firmware changes, at least for wave-2 stuff, but not sure when.  Maybe stock-firmware will need to be used for this scenario until then.

makkrnic, 2019-05-27:
Hello, is there any progress on this issue on driver/firmware. I've been trying to get it to work on 8devices jalapeno (IPQ4018), but without much luck I keep getting error -22 indicating that the firmware lacks features.
I've tried both ct and broadcom firmware but with the same result.

hmronline, 2019-07-10:
I have a custom built openwrt 18.x running on Archer C7 v2 and I'm also affected by this issue.
I will be glad to help at least testing your patches.

Thank you @greearb!

ASoTNetworks, 2019-07-11:
I am also affected by this issue and I am too running a custom openwrt 18.x build with those fixes above.

I would be happy to test the patches on Archer C7 V2 and a IPQ4028 device.

zekica, 2019-07-11:
I'm also affected by this - @greearb said that "Maybe stock-firmware will need to be used for this scenario until then." which I tried (on a custom openwrt build) and it works but for some reason, on Archer C6 v2 EU the AP stops sending all unicast packets when under load (doesn't matter if dynamic vlans are used or not) so I'm waiting for this feature in -ct firmware.

I would also be happy to test on Archer C6 v2 EU (using the QCA9888/hw2.0 firmware).

Thank you @greearb for all the work you've done so far.

m68008, 2019-07-12:
I am affected by this issue. Currently using an Archer C7 v2 with 18.01. Build the Firnware after the description from https://bugs.openwrt.org/index.php?do=details&task_id=488 or rather https://github.com/chrislimpach/openwrt/commit/0ba96c35132e60c95dd42d25035154856910f907 and
https://github.com/chrislimpach/openwrt/commit/31284c3aa55a7f71eb5e127b9e891a1745ec8e09
from C. Limpach. With this patched firmware my C7 runs both radios with dynamic Vlans.

dfranusic, 2019-08-29:
I have compiled those commits into a single patch, but I still cannot get **AP / VLAN** support. Does anyone have a valid **firmware** that works with [IPQ4019](https://www.qualcomm.com/products/ipq4019)?
```sh
$ iw list
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * monitor
                 * mesh point
                 * P2P-client
                 * P2P-GO
                 * P2P-device
```
I have also attached the patch mentioned earlier.

Thanks
[999-ath10k_dynamic_vlan.patch.gz](https://github.com/greearb/ath10k-ct/files/3555880/999-ath10k_dynamic_vlan.patch.gz)

dfranusic, 2019-09-11:
This has been fixed by using the latest **firmware-5-ct-full-community-12.bin-lede.011** firmware and manually patching both **firmware/ath10k-firmware** and **kernel/ath10k-ct** packages in **OpenWrt**.

hmronline, 2019-09-11:
thanks @dfranusic, do you know how to get OpenWrt developers aware of this ?

dfranusic, 2019-09-11:
I'll make a pull request for OpemWrt tomorrow, as I still have some patches to add for 4.20 kernel.

dfranusic, 2019-09-13:
@hmronline I'm sorry, but the feature is firmware dependent and since @greearb has no plans to implement it for 9880, there's nothing I can do.

dfranusic, 2019-09-16:
For users of IPQ4019, AP/VLAN support has been merged to upstream:
openwrt/openwrt@3fe30b28ae7ff6a7b57cdeb7b846a0e247234c9a
openwrt/openwrt@7c930990af911f6634b422d7253f09df2bb164bf

greearb, 2019-09-16:
Closing bug, thanks for the testing and upstream patching.

closed 2019-09-16 by greearb

ianbmacdonald, 2020-01-28:
This issue is impacting OpenWRT 19.07 upgrades using Ath10k.   Any chance this wil be addressed for popular Ath10k builds like Archer C7? 

This includes a bunch of devices now part of the ath79 target to modernize these wifi implementations, but leaving out this interface mode used in almost all captive portals. 

[   16.007715] ath10k_pci 0000:00:00.0: firmware ver 10.1-ct-8x-__fW-022-b0e1b7cd api 2 features wmi-10.x,has-wmi-mgmt-tx,mfp,txstatus-noack,wmi-10.x-CT,ratemask-CT,txrate-CT,get-temp-CT,tx-rc-CT,cust-stats-CT,retry-gt2-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT crc32 1e527180

	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * monitor
		 * mesh point
		 * P2P-client
		 * P2P-GO
		 * P2P-device

greearb, 2020-01-29:
The ath10k-ct wave-1 firmware actually can do at least some sorts of raw transmit, so if someone wants to fix up the driver to send raw frames to the firmware (or whatever vlan needs), then I'll try to fix any firmware bugs that show up.

NerdyProjects, 2020-02-10:
Can you elaborate on your last comment? I was happy to having seen the dynamic vlan fix in openwrt 19.07 but it seems as if it does not work because the firmware for wave-1 devices does not announce raw packet support? So what fixup are you asking for? I run a setup using 6 archer c5 v1 access points in a small community project and am up for helping getting dynamic vlan support into openwrt mainline :-)

I am referring to this commit in openwrt:

```
commit 92953ae99f240a0b45ab66defde5f7b8eeb94e35
Author: Koen Vandeputte <koen.vandeputte@ncentric.com>
Date:   Tue Sep 24 11:58:54 2019 +0200

    ath10k-ct: update to version 2019-09-09
    
    5e8cd86f90da ath10k-ct: Backport ap-vlan code from 5.2 to 4.20 and 4.19 drivers.
    0c518586bd7f ath10k-ct: Fix a few warning splats.
    
    Adds AP VLAN.
    Refreshed all patches.
```

ianbmacdonald, 2020-02-13:
I also maintain a bunch of C7s that do captive portal nicely via packetfence for public wifi access and would be willing to commit whatever resources I can to assist.

greearb, 2020-02-14:
This needs someone(s) that can do ath10k driver work.  I think the steps would be:
1) Make driver advertise support vlans if ath10k-ct firmware/driver is loaded (assume it can do raw tx as needed, but no extra module parameters).  ath10k-ct driver already has some raw-tx support, hopefully that is enough for vlans.
2) See what fails, if it is driver related, maybe fix it yourself, or maybe report to me.
3) If firmware fails, write up good bug report for me and *maybe* I have time to fix it.

NerdyProjects, 2020-02-14:
I did a test:

1. patched ath-10kct driver to always advertise AP/Vlan support:
```
-- a/ath10k-4.19/mac.c
+++ b/ath10k-4.19/mac.c
9998,10001c9998,9999
<       if (test_bit(WMI_SERVICE_PER_PACKET_SW_ENCRYPT, ar->wmi.svc_map)) {
<               ar->hw->wiphy->interface_modes |= BIT(NL80211_IFTYPE_AP_VLAN);
<               ar->hw->wiphy->software_iftypes |= BIT(NL80211_IFTYPE_AP_VLAN);
<       }
---
>       ar->hw->wiphy->interface_modes |= BIT(NL80211_IFTYPE_AP_VLAN);
>       ar->hw->wiphy->software_iftypes |= BIT(NL80211_IFTYPE_AP_VLAN);
```

2. used it on an access point that has a WPA2 Enterprise network (and an open network) that has dynamic vlan assignment via radius as well as most traffic not assigning a vlan.

I did not observe any warnings/errors in kernel log, also everything worked fine.

I used OpenWRT 19.07 branch, commit `b6c01fec921bb094416e70abad4c706b6cb030fd` as a base.

So what kind of testing needs to be done to enable the feature flag inside the firmware to not need this hack? :-)

NerdyProjects, 2020-02-15:
Just to clarify, I redid the test now, using the proper 5ghz wifi. No indications of any errors in the logfile. Speed and latency seemed good (although I did not use any more detailled measures than doing an internet speedtest with a 50 mbps connection).

greearb, 2020-02-15:
Ok, that sounds promising.  For a real patch, check if this feature flag is supported (which will indicate recent ath10k-ct firmware) || existing check for the WMI flag.
Search for this in the code: ATH10K_FW_FEATURE_CONSUME_BLOCK_ACK_CT

NerdyProjects, 2020-02-15:
Thanks for that information, I can try that.

Still, I wonder if the right fix would not be to set WMI_SERVICE_PER_PACKET_SW_ENCRYPT in the wave 1 firmware as well?

greearb, 2020-02-16:
I'm not sure that will be simple to do.  The patch will never make it upstream anyway, so probably a simple hack to the driver is good enough.  I can add it to my ath10k-ct driver, so it will be widely available in OpenWRT, at least.

NerdyProjects, 2020-02-16:
See #119