greearb/ath10k-ct #75: IBSS mesh and client mode do not work in parallel

CodeFetch, 2019-03-16:
According to https://github.com/freifunk-gluon/gluon/issues/1584 two VIFs one being an IBSS and the other an AP don't work in parallel.
I'm going to make some tests myself and try to provide logs. Maybe this is related to #68.
This is a rather critical issue for mesh networks. Please tell me if you have some ideas what I can do to help you resolve it.

greearb, 2019-03-18:
First, maybe see if this is only an issue with wave-1 ct firmware?

And second, try to create the simplest test case that uses little or no mesh logic.  That would give me a better chance of reproducing and fixing the problem locally.

Try disabling all encryption to see if that helps, see if any packets hit the air using a sniffer, etc.

For what it is worth, I expect that IBSS should coexist with station devices on all of the ct firmware.

greearb, 2019-03-25:
Did you have any luck testing this?

CodeFetch, 2019-03-25:
Unfortunately I didn't have time to test it, yet. QCA9886 is a Wave 1 chip, isn't it?  I've read on the OpenWrt mailing list, that you've just released a new version. I'm going to test it with that one. Yesterday someone gave me three UniFI AC Mesh. Their chip seems to be a QCA9888 clone, which is a Wave 2 chip. I'm going to test them, too.

BTW I have some stupid questions. Are you planning to support 802.11ax chips like IPQ8074? And another one is: As you've signed a NDA, does Qualcomm offer you to get insight into the topical code of the "official" ath10k firmware? I ask because Gluon uses your driver only for IBSS as the "official" firmware seems to be more stable for 802.11s.

greearb, 2019-03-25:
I do plan to support 11ax chips.
I'd be interested in debugging 11s issues in my wave-2 firmware.  I am not sure if the 'gluon' project had issues with wave-1 or wave-2 firmware.  I do not have plans to support .11s in wave-1 firmware.

maurerle, 2025-10-18:
No really sure what the right ticket would be.
I am trying to use 802.11s with qca9888 and ath10k-ct running in gluon and the device restarts after several firmware crashes and recoveries:

```
Sat Oct 18 22:28:59 2025 kern.info kernel: [ 1238.234880] ieee80211 phy0: Hardware restart was requested
Sat Oct 18 22:28:59 2025 kern.warn kernel: [ 1238.235114] ath10k_pci 0000:01:00.0: could not request stats (-108)
Sat Oct 18 22:29:03 2025 kern.info kernel: [ 1241.538980] ath10k_pci 0000:01:00.0: device successfully recovered
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.539188] ath10k_pci 0000:01:00.0: firmware crashed! (guid 963e26f3-0513-48f9-8905-9b223431dcb7)
Sat Oct 18 22:29:03 2025 kern.info kernel: [ 1241.544136] ath10k_pci 0000:01:00.0: qca9888 hw2.0 target 0x01000000 chip_id 0x00000000 sub 0000:0000
Sat Oct 18 22:29:03 2025 kern.info kernel: [ 1241.553080] ath10k_pci 0000:01:00.0: kconfig debug 1 debugfs 1 tracing 0 dfs 1 testmode 0
Sat Oct 18 22:29:03 2025 kern.info kernel: [ 1241.567021] ath10k_pci 0000:01:00.0: firmware ver 10.4b-ct-9888-fW-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl crc32 59e741e7
Sat Oct 18 22:29:03 2025 kern.info kernel: [ 1241.572004] ath10k_pci 0000:01:00.0: board_file api 2 bmi_id 0:23 crc32 5968d47d
Sat Oct 18 22:29:03 2025 kern.info kernel: [ 1241.582190] ath10k_pci 0000:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-nvmem max-sta 32 raw 0 hwcrypto 1
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.590072] ath10k_pci 0000:01:00.0: firmware register dump:
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.600700] ath10k_pci 0000:01:00.0: [00]: 0x0000000C 0x00000000 0x0098D153 0x00000000
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.605283] ath10k_pci 0000:01:00.0: [04]: 0x00000000 0x00060024 0x00000000 0x00000000
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.613067] ath10k_pci 0000:01:00.0: [08]: 0x00000000 0x00000000 0x00000000 0x00000000
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.620956] ath10k_pci 0000:01:00.0: [12]: 0x00000000 0x00000000 0x00000000 0x00000000
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.628832] ath10k_pci 0000:01:00.0: [16]: 0x0098208F 0x009606CA 0x00000000 0x0098D153
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.636772] ath10k_pci 0000:01:00.0: [20]: 0x00000000 0x00000000 0x00000000 0x00000000
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.644670] ath10k_pci 0000:01:00.0: [24]: 0x00000000 0x00000000 0x00000000 0x00000000
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.652552] ath10k_pci 0000:01:00.0: [28]: 0x00000000 0x00000000 0x00000000 0x00000000
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.660465] ath10k_pci 0000:01:00.0: [32]: 0x00000000 0x00000000 0x00000000 0x00000000
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.668303] ath10k_pci 0000:01:00.0: [36]: 0x00000000 0x00000000 0x00000000 0x00000000
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.676263] ath10k_pci 0000:01:00.0: [40]: 0x00000000 0x00000000 0x00000000 0x00000000
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.684152] ath10k_pci 0000:01:00.0: [44]: 0x00000000 0x00000000 0x00000000 0x00000000
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.692045] ath10k_pci 0000:01:00.0: [48]: 0x00000000 0x00000000 0x00000000 0x00000000
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.699930] ath10k_pci 0000:01:00.0: [52]: 0x00000000 0x00000000 0x00000000 0x00000000
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.707801] ath10k_pci 0000:01:00.0: [56]: 0x00000000 0x00000000 0x00000000 0x00000000
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.715740] ath10k_pci 0000:01:00.0: Copy Engine register dump:
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.724384] ath10k_pci 0000:01:00.0: [00]: 0x0004a000   4   4   3   3
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.729420] ath10k_pci 0000:01:00.0: [01]: 0x0004a400  16  16  86  87
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.736056] ath10k_pci 0000:01:00.0: [02]: 0x0004a800  16  16  15  16
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.742473] ath10k_pci 0000:01:00.0: [03]: 0x0004ac00  29  29  30  29
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.748873] ath10k_pci 0000:01:00.0: [04]: 0x0004b000  32  32  45  18
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.755314] ath10k_pci 0000:01:00.0: [05]: 0x0004b400  13  13  12  13
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.761740] ath10k_pci 0000:01:00.0: [06]: 0x0004b800   2   2   2   2
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.768134] ath10k_pci 0000:01:00.0: [07]: 0x0004bc00   1   1   1   1
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.774589] ath10k_pci 0000:01:00.0: [08]: 0x0004c000   0   0 127   0
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.781009] ath10k_pci 0000:01:00.0: [09]: 0x0004c400   1   1   1   1
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.787403] ath10k_pci 0000:01:00.0: [10]: 0x0004c800   0   0   0   0
Sat Oct 18 22:29:03 2025 kern.err kernel: [ 1241.793851] ath10k_pci 0000:01:00.0: [11]: 0x0004cc00   0   0   0   0
Sat Oct 18 22:29:03 2025 kern.info kernel: [ 1241.894552] ieee80211 phy0: Hardware restart was requested
Sat Oct 18 22:29:03 2025 kern.warn kernel: [ 1241.894884] ath10k_pci 0000:01:00.0: failed to send pdev bss chan info request
Sat Oct 18 22:29:03 2025 kern.warn kernel: [ 1241.899136] ath10k_pci 0000:01:00.0: failed to send pdev bss chan info request
Sat Oct 18 22:29:03 2025 kern.warn kernel: [ 1241.906896] ath10k_pci 0000:01:00.0: failed to set beacon mode for vdev 0: -108
[... cut ...]
Sat Oct 18 22:29:07 2025 kern.err kernel: [ 1245.567000] ath10k_pci 0000:01:00.0: [09]: 0x0004c400   0   0   0   0
Sat Oct 18 22:29:07 2025 kern.err kernel: [ 1245.573445] ath10k_pci 0000:01:00.0: [10]: 0x0004c800   0   0   0   0
Sat Oct 18 22:29:07 2025 kern.err kernel: [ 1245.579866] ath10k_pci 0000:01:00.0: [11]: 0x0004cc00   0   0   0   0
```

I don't quite know if this is worth looking into any further or if this is just the "ath10k-ct does not work well with mesh on 5ghz" behavior..?

Tested on a `linksys-whw03v2` running openwrt-24.10 with gluon adjustments