greearb/ath10k-ct #77: 3-way handshake failure with blink xt cameras

TheIvanSusanin, 2019-03-20:
Blink XT cameras that are based on TI CC3100  chip (and TI wifi/tcp stack I guess) at some point of their life will fail to complete 3way handshake. I.e. they will ignore SYN/ACK coming from server and will keep on sending SYNs.
I run R7800 (with openwrt).  Stock ath10k firmware/kernel drivers do not have this problem. When openwrt migrated to -ct firmware/driver half an year ago cameras weren't able to work at all. Over past half an year (i update once a month from snapshots) things got more stable but usually one of the cameras at any given time wont work.
After AP reboot usually things will work stable for 24h or so. After this camera failures with start. (nothing in dmesg on AP).
In case there was firmware/driver crash cameras wont work 100% of time till AP is rebooted (rest of wifi client will keep on working).

I guess there is something rather buggy on TI/Camera side of equation but it looks like something in firmware seems to trigger it.

Any interest in trying to figure it out ?

greearb, 2019-03-20:
I can build a series of firmware so you can bisect the problem?  If you want to do this, please let me know if you are using htt firmware or the normal wmi-mgt firmware.  Or, show me dmesg of the bootup and I can figure it out myself.

TheIvanSusanin, 2019-03-21:
I can try to bisect.
firmware ver: 10.4b-ct-9984-fW-012-38a5514c3 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT crc32 7de96fb7

Can dumping air traffic for working/non working sessions help to pinpoint the area ?

greearb, 2019-03-21:
Here is a set of binaries:  https://www.candelatech.com/downloads/ath10k-9984-10-4b/ath10k-fw-beta/all_builds-9984b-march-21-2019.tar.gz
See this page for how to bisect if you are unsure:  https://www.candelatech.com/ath10k-bugs.php

greearb, 2019-03-21:
Please see if this image works better for you:

[firmware-5-full-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2994339/firmware-5-full-community.bin.gz)

TheIvanSusanin, 2019-03-22:
> Please see if this image works better for you:
> 
> [firmware-5-full-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2994339/firmware-5-full-community.bin.gz)

Tried. Same random failed handshakes.

greearb, 2019-03-22:
Ok, it must be different from bug 78 then.  Please try to bisect to find the problem commit.  First check if the very early commits work, that should be very much like stock firmware.  If that doesn't work, then we need to think of some other way to track down the bug.

TheIvanSusanin, 2019-03-23:
will try to bissect.
got following message with firmware from above
[ 8924.812298] ------------[ cut here ]------------
[ 8924.812423] WARNING: CPU: 1 PID: 2033 at backports-4.19.23-1/net/wireless/util.c:1146 cfg80211_calculate_bitrate+0x1cc/0x254 [cfg80211]
[ 8924.816007] invalid rate bw=4, mcs=15, nss=4
[ 8924.828004] Modules linked in: pppoe ppp_async ath10k_pci ath10k_core ath pppox ppp_generic nf_conntrack_ipv6 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_of_simple ohci_platform ohci_hcd phy_qcom_dwc3 ahci ehci_platform
[ 8924.880801]  sd_mod ahci_platform libahci_platform libahci libata scsi_mod ehci_hcd gpio_button_hotplug
[ 8924.903027] CPU: 1 PID: 2033 Comm: hostapd Not tainted 4.14.105 #0
[ 8924.912165] Hardware name: Generic DT based system
[ 8924.918436] [<c030f2b4>] (unwind_backtrace) from [<c030b490>] (show_stack+0x14/0x20)
[ 8924.923200] [<c030b490>] (show_stack) from [<c07c01b8>] (dump_stack+0x88/0x9c)
[ 8924.931103] [<c07c01b8>] (dump_stack) from [<c0322be8>] (__warn+0xf0/0x11c)
[ 8924.938130] [<c0322be8>] (__warn) from [<c0322c48>] (warn_slowpath_fmt+0x34/0x4c)
[ 8924.945003] [<c0322c48>] (warn_slowpath_fmt) from [<bf282b40>] (cfg80211_calculate_bitrate+0x1cc/0x254 [cfg80211])
[ 8924.952675] [<bf282b40>] (cfg80211_calculate_bitrate [cfg80211]) from [<bf2ada8c>] (nl80211_put_sta_rate+0x48/0x2e8 [cfg80211])
[ 8924.962907] [<bf2ada8c>] (nl80211_put_sta_rate [cfg80211]) from [<bf295df0>] (nl80211_update_connect_params+0x72c/0xfdc [cfg80211])
[ 8924.974275] [<bf295df0>] (nl80211_update_connect_params [cfg80211]) from [<bf296760>] (nl80211_get_station+0xc0/0x124 [cfg80211])
[ 8924.986073] [<bf296760>] (nl80211_get_station [cfg80211]) from [<c06e323c>] (genl_rcv_msg+0x2ec/0x3a0)
[ 8924.997856] [<c06e323c>] (genl_rcv_msg) from [<c06e22a0>] (netlink_rcv_skb+0x94/0x110)
[ 8925.007052] [<c06e22a0>] (netlink_rcv_skb) from [<c06e2f34>] (genl_rcv+0x2c/0x48)
[ 8925.014950] [<c06e2f34>] (genl_rcv) from [<c06e1a04>] (netlink_unicast+0x164/0x224)
[ 8925.022502] [<c06e1a04>] (netlink_unicast) from [<c06e1ed4>] (netlink_sendmsg+0x334/0x390)
[ 8925.029979] [<c06e1ed4>] (netlink_sendmsg) from [<c068acc8>] (sock_sendmsg+0x18/0x34)
[ 8925.038303] [<c068acc8>] (sock_sendmsg) from [<c068b270>] (___sys_sendmsg+0x214/0x250)
[ 8925.046201] [<c068b270>] (___sys_sendmsg) from [<c068c1f0>] (__sys_sendmsg+0x48/0x78)
[ 8925.054015] [<c068c1f0>] (__sys_sendmsg) from [<c0307da0>] (ret_fast_syscall+0x0/0x54)
[ 8925.061982] ---[ end trace cfd74d73429fc58c ]---

greearb, 2019-05-01:
Please see if this firmware works any better.

[firmware-5-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/3135285/firmware-5-htt-mgt-community.bin.gz)

TheIvanSusanin, 2019-05-09:
Will try.
I did try to bisect firmware a few times but the problem that failures don't happen immediately but they may take a few days to appear.
The only thing that I managed to observe is that with original (or much older) firmwares video stream was starting quickly and with newer one it takes 5-10 seconds before it begins.

greearb, 2019-05-09:
A new firmware should hit OpenWRT shortly, it fixes a slow-to-start bug related to me decreasing retries.  To fix the issue, I start at a lower MCS when initializing rate-ctrl and that made DHCP work a lot better in my tests.

TheIvanSusanin, 2019-05-17:
I got a few crashes of firmware from above. Are you interested to see ?

greearb, 2019-05-17:
Yes, please post the dmesg output with the crash splat in it.  I found yet another way to tx-hang the firmware, though this took some pretty unfortunate events it seems.  I'll have new firmware that fixes this tx-hang bug I saw in our lab posted to OpenWRT shortly...

TheIvanSusanin, 2019-05-17:
attached,
[ath10.txt](https://github.com/greearb/ath10k-ct/files/3189930/ath10.txt)

greearb, 2019-05-17:
I added some extra debugging around the assert that hit in your crash, so if you see more crashes, let me know and maybe I can learn more about the issue.  Here is latest firmware with the extra debugging.
[firmware-5-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/3191470/firmware-5-htt-mgt-community.bin.gz)

TheIvanSusanin, 2019-06-02:
Ok. Will try this firmware.
But, previous firmware, despite the crashes seems like able to keep cameras working.
Everything before that - after crash cameras wont work anymore

TheIvanSusanin, 2019-07-11:
Well,
This firmware is best in -ct series. It doesn't mean that cameras work 100% of time now, but, there is no situation that I need to reboot AP in order to make them work again.
Sometime it does takes a few attempts to start video stream (and I guess some of the security alerts are missing).

Is there any other firmware to try or do you want me to try do do something ? (bissecting failed as it takes a few days for problem to appear).
Any idea if this firmware already hit openwrt ?

Thanks

greearb, 2019-07-11:
I think everything is upstream now....  Please try latest upstream OpenWRT when you get a chance, and if it fixes the issue, can close this.

greearb, 2019-10-29:
Closing this one, think it fixed now.

closed 2019-10-29 by greearb