greearb/ath10k-ct #78: wifi client Whatsapp problems

notgood, 2019-03-20:
Netgear R7800, latest master, ath10k-ct driver and 9984 firmware, VHT160

Wifi clients timeout on trying to download whatsapp messenger media files.
A bit of tcpdump on-router debugging reveals TCP SYN retransmissions to https://media.frix4-1.fna.whatsapp.net
Server ACK's are visible, but it seems aren't getting back to client.

Futher attempts to open this URL in browser on wifi clients
(android, laptops, etc) fails, problem persists on both 2.4GHz and 5GHz.
Upgrading to latest v12 beta = same failure
Downgrading firmware to v11 = same failure
Downgrading firmware to v10 = works (on 2.4Ghz, 5Ghz crashes, but guess thats another problem)
Switching to plain ath10k driver and vendor firmware = works.

Opening same link on ethernet connected PC or from router itself works OK, anytime.

greearb, 2019-03-20:
This looks a lot like bug 77.  Are you using htt or wmi mgt firmware, and if I build you a series of images, can you bisect to find the bad commit?

notgood, 2019-03-20:
I'm using non-htt firmware. Tried openwrt packaged htt one today as well, it was bugged as well.

I can try to test a set of binaries. Hope not too many commits, rebooting R7800 is getting boring quite fast.

greearb, 2019-03-20:
There are 1000+ commits, but you can bisect that fairly quick if you can reproduce the problem easily.  I'll start building...it will take a day or so to complete the series and I'll update the bug then.

greearb, 2019-03-21:
Here is a set of binaries:  https://www.candelatech.com/downloads/ath10k-9984-10-4b/ath10k-fw-beta/all_builds-9984b-march-21-2019.tar.gz
See this page for how to bisect if you are unsure:  https://www.candelatech.com/ath10k-bugs.php

notgood, 2019-03-21:
Bisect results:
firmware-5-full-community-commit-881-6727c5573.bin - OK
firmware-5-full-community-commit-882-3c429a2e0.bin - BAD

Bisecting on OpenWRT can be done without rebooting:
```
# on workstation
scp firmware-5-full-community-commit-NNN.bin root@myrouter:/lib/firmware/ath10k/QCA9984/hw1.0/firmware-5.bin

# on router
wifi down
sleep 3
rmmod ath10k_pci 
rmmod ath10k_core 
rmmod ath
rmmod mac80211
rmmod cfg80211
sleep 1
modprobe /lib/modules/4.14.106/ath10k_pci.ko
wifi up
ethtool -i wlan0

```

greearb, 2019-03-21:
Thanks...that big hairy patch could definitely cause this sort of problem.  I'll need to spend some time breaking it into smaller bits and give you some new binaries to test to narrow it down further.

greearb, 2019-03-21:
Please see if this one works, it has all of my normal changes except for the tricky parts of the bad commit you bisected to:  
[firmware-5-full-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2994334/firmware-5-full-community.bin.gz)

Assuming that works, then I can focus on trying to fix the problem part of the patch tomorrow.

notgood, 2019-03-22:
Yes, this one works.

greearb, 2019-03-22:
Please try this one, it is a re-work of the buggy patch.
[firmware-5-full-community.bin.gz](https://github.com/greearb/ath10k-ct/files/2998504/firmware-5-full-community.bin.gz)

notgood, 2019-03-23:
Thank you, re-worked version is OK as well

greearb, 2019-03-25:
I've uploaded new firmware and posted info for openwrt to pull this in.  Closing bug as fixed.  Thanks for the testing!

closed 2019-03-25 by greearb

gsustek, 2020-01-19:
Hi, after two days of searching looking into wireshark, i google it and  found out this section:_) now i fix this issue but this firmware is soo slow, 30MB/s. With this one (firmware-5-ct-non-commercial-full-htt-mgt-11.bin) i got 100MB/s. Is it possible to port fix to Ath10k CT 10.4 Release 11 or at least to Ath10k CT 10.4.3 Beta 12? 
Regards, 
Goran.

greearb, 2020-01-20:
Please try disabling MFP/PMF.  If that fixes things, then probably your device does not do block-ack encryption properly.

gsustek, 2020-01-21:
# ieee80211w: Whether management frame protection (MFP) is enabled
# 0 = disabled (default)
# 1 = optional
# 2 = required
#ieee80211w=0
# The most common configuration options for this based on the PMF (protected
# management frames) certification program are:
# PMF enabled: ieee80211w=1 and wpa_key_mgmt=WPA-EAP WPA-EAP-SHA256
# PMF required: ieee80211w=2 and wpa_key_mgmt=WPA-EAP-SHA256
# (and similarly for WPA-PSK and WPA-PSK-SHA256 if WPA2-Personal is used)
# WPA3-Personal-only mode: ieee80211w=2 and wpa_key_mgmt=SAE

I do not have this enabled and by default is 0(disabled).
"your device does not do block-ack encryption properly" i have this card..
https://compex.com.sg/shop/wifi-module/wle1216v5-20/
so it is a manufacturer fault or it can be fixed in firmware?
#####my config..######

driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=HT
ieee80211d=1
ieee80211h=1
hw_mode=a
beacon_int=100
channel=36

tx_queue_data2_burst=2.0
ieee80211n=1
#ht_coex=0
ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40]
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
ieee80211ac=1
vht_capab=[RXLDPC][SHORT-GI-80][SHORT-GI-160][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][VHT160-80PLUS80][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7]

interface=wlp6s0
bridge=br-lan
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
#chan_util_avg_period=600
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
#multi_ap=0
wpa_passphrase=XXXXXXXX
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=XXXX
#bridge=br-lan
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1

### hostapd event logger
logger_syslog=-1
logger_syslog_level=2
logger_stdout=-1
logger_stdout_level=2

#bssid

gsustek, 2020-01-24:
Hi, would be so kind to respond to my question. I have to be sure that it is not a hardware issue. Thank you so much.